cyring/CoreFreq #10: Issue with turning off CPU

netmonk, 2016-09-30:
I noticed that when putting a specific cpu offline, (echo 0 >/sys/devices/system/cpu/cpu1) The client stop refreshing data. 
Restarting daemon and cli doesn't bring back normal operation. 

This has to do with the kernel module which seems to stop providing data when the cpu is put offline. 

Worst even, is that when i tryed to rmmod the corefreq kernel module, i had a full system freeze which forces me to physically switch off, switch on the computer to restart the OS.

cyring, 2016-09-30:
Indeed no data is read from an offline cpu which was in BIOS or in OS.
It's an implementation choice that could be modified in future if you really need such functionality?
However I'm sorry to learn about your system freeze, I'll debug the rmmod case when some cores are offline
CoreFreq is in Beta stage, not ready for production yet and I do appreciate your return.

cyring, 2016-10-01:
Parts of the logarithm will be modified such as :
1- start a hrtimer for any enable cpu reported by BIOS (required if HTT is off)
2- don't take into account changes at the OS level
--> thus any OS disable cpu can still be monitored
3- during module shutdown, stop hrtimers created in step 1

netmonk, 2016-10-01:
I can kow some uses case where after booting and passing init stage, user
will put offline some cpu for forcing the kernel to move all tasks running
or pinned on it, and switch it back online to launch only what they need on
it.

If corefreq module is loaded at boot time, then it will be impacted by this
switch of status after being loaded in kernel.

That is why I was testing this scenario :)

cyring, 2016-10-01:
That's a nice feature I will carefully program and it will require several and strong stability testings. Any help appreciatied, I don't have such lab.
As a result, be aware that for each offline cpu you may get metrics with zero values and thus false averages.

netmonk, 2016-10-01:
Btw what is your email address ? I wanted to send unrelated pointer and was
unable to find your email ;)

cyring, 2016-10-02:
Hi netmonk,
I've reproduced your case.
The deamon log shows that its thread #1 has lost a cpu affinity. This can be recovered, but at the module side, the issue is trickier where the timer associated with cpu #1 crosses a state where its cpu has been put offline.

As a workaround, I suggest a shell script to manage disablement / enablement:
1- kill client with saved pid
2- kill deamon with saved pid
3- remove module
4- enable and / or disable some cpu
5- insert module
6- start deamon
    save its pid
7- start client
    save its pid

cyring, 2016-10-10:
Fixed

closed 2016-10-10 by cyring

cyring, 2020-10-29:
Salut l'ami.
J'espère que tout va bien pour toi.

Avec le ThreadRipper, 1ère génération, on a detecté une inconsitence sur le compteur des instructions résultant par des calculs d'IPC erronés.
Pourrais tu s.t.p. essayer sur ton 1950X, en idle, la dernière version de _CoreFreq_ et afficher cette vue ?
[image]

Merci,
Cyril

netmonk, 2020-11-01:
[image]

netmonk, 2020-11-01:
file:///home/plonky/Pictures/Screenshot%20from%202020-11-01%2011-52-29.png