cyring/CoreFreq #11: error: ‘MSR_NHM_TURBO_RATIO_LIMIT’ undeclared

JodiTheTigger, 2016-10-17:
Trying to install from Arch AUR and I get the build error:

```
aur-corefreq-git/src/CoreFreq/corefreqk.c:824:15: error: ‘MSR_NHM_TURBO_RATIO_LIMIT’ undeclared (first use in this function)
  RDMSR(Turbo, MSR_NHM_TURBO_RATIO_LIMIT);
               ^
```

```
==> Starting build()...
make: Entering directory 'aur-corefreq-git/src/CoreFreq'
cc -c corefreqd.c -o corefreqd.o
cc -lpthread -lrt -o corefreqd corefreqd.c
cc -c corefreq-cli.c -o corefreq-cli.o
cc -lrt -o corefreq-cli corefreq-cli.c
make -C /lib/modules/4.8.2-1-ARCH/build M=aur-corefreq-git/src/CoreFreq modules
make[1]: Entering directory '/usr/lib/modules/4.8.2-1-ARCH/build'
  CC [M]  aur-corefreq-git/src/CoreFreq/corefreqk.o
In file included from aur-corefreq-git/src/CoreFreq/corefreqk.c:23:0:
aur-corefreq-git/src/CoreFreq/corefreqk.c: In function ‘Nehalem_Platform_Info’:
aur-corefreq-git/src/CoreFreq/corefreqk.c:824:15: error: ‘MSR_NHM_TURBO_RATIO_LIMIT’ undeclared (first use in this function)
  RDMSR(Turbo, MSR_NHM_TURBO_RATIO_LIMIT);
               ^
aur-corefreq-git/src/CoreFreq/corefreqk.h:51:10: note: in definition of macro ‘RDMSR’
   : "c" (_reg)      \
          ^~~~
aur-corefreq-git/src/CoreFreq/corefreqk.c:824:15: note: each undeclared identifier is reported only once for each function it appears in
  RDMSR(Turbo, MSR_NHM_TURBO_RATIO_LIMIT);
               ^
aur-corefreq-git/src/CoreFreq/corefreqk.h:51:10: note: in definition of macro ‘RDMSR’
   : "c" (_reg)      \
          ^~~~
make[2]: *** [scripts/Makefile.build:296: aur-corefreq-git/src/CoreFreq/corefreqk.o] Error 1
make[1]: *** [Makefile:1471: _module_aur-corefreq-git/src/CoreFreq] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.8.2-1-ARCH/build'
make: *** [Makefile:6: all] Error 2
make: Leaving directory 'aur-corefreq-git/src/CoreFreq'
==> ERROR: A failure occurred in build().
    Aborting...
==> ERROR: Makepkg was unable to build corefreq-git.

```

```
cat /proc/cpuinfo 
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 60
model name      : Intel(R) Core(TM) i7-4790 CPU @ 3.60GHz
stepping        : 3
microcode       : 0x19
cpu MHz         : 3902.343
cache size      : 8192 KB
physical id     : 0
siblings        : 8
core id         : 0
cpu cores       : 4
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm epb tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm xsaveopt dtherm ida arat pln pts
bugs            :
bogomips        : 7198.35
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:

<snip for other 7 cores>
```

cyring, 2016-10-18:
Hello,
Do you have linux-headers package installed ?
Because MSR_NHM_TURBO_RATIO_LIMIT is defined into /usr/include/asm/msr-index.h which is a header file included in this package.

oleksandr-oksenenko, 2016-10-18:
Hey,
I have the same problem building the module on Arch (though not from AUR).
I do have `/usr/include/asm/msr-index.h` and i there's `MSR_NHM_TURBO_RATIO_LIMIT` in it.

oleksandr-oksenenko, 2016-10-18:
I figured it out, the problem is the custom kernel i use (`linux-pf`). For some reason, the module builds only  with stock Arch kernel.

oleksandr-oksenenko, 2016-10-18:
Hm, seems like the custom kernel is not the issue :)
I installed the Arch stock kernel of version `4.8.2` from `testing` and the build fails.
So i think the actual reason is the kernel version.

cyring, 2016-10-18:
Thanks for your return :100: 

I have also noticed that you are using a testing version 4.8 of the kernel (as JodiTheTigger's log) whereas I'm compiling with the current version 4.7.6-1 for ArchLinux.

Meanwhile I also found that 2 packages provide the msr definition:
1. linux-api-headers (/usr/include/asm/msr-index.h)
2. linux-headers /usr/lib/modules/4.7.6-1-ARCH/build/arch/x86/include/asm/msr-index.h)
(any diff however between these 2 files).

If you want to stick with the testing kernel, I suggest you to copy/write the few missing definitions into the CoreFreq header file 'intelmsr.h' , such as:

#define MSR_NHM_TURBO_RATIO_LIMIT	0x000001ad

oleksandr-oksenenko, 2016-10-18:
JFYI i found the following patch: https://patchwork.kernel.org/patch/9217367/ .
Seems like `MSR_TURBO_RATIO_LIMIT` should be used now, what do you think?

cyring, 2016-10-18:
Indeed, they are duplicated.
So, I have push an updated corefreqk.c to handle this change.
I also test the case (lines #889) that it will style build with previous kernel versions.

oleksandr-oksenenko, 2016-10-18:
Great, thanks :)

cyring, 2016-10-18:
Btw, I just few CPU to test CoreFreq with (Bloomfield, Core 2 duo T7250, AMD Turion)
I will appreciate if you can post in reply screenshots of corefreq-cli monitoring your processor(s).
That will be a great help.

oleksandr-oksenenko, 2016-10-18:
Sure.
I also loaded one core with CPU-bounded task.

[image]

cyring, 2016-10-18:
CoreFreq is running a Haswell, **Thank you** very much :+1: 

My 2 ways to reach the max turbo on 1 core:
sol 1. in BIOS, limit your settings to one core only activated
sol 2. disable all cores beside #0 using command
    echo 0 >  /sys/devices/system/cpu/cpu1/online
    echo 0 >  /sys/devices/system/cpu/cpu2/online
    ...
    echo 0 >  /sys/devices/system/cpu/cpu7/online

solution 2 is easier and CoreFreq is cpu hot-plug aware; thus you can disable/enable as wishes.

Prerequisites:
1- 'intel_idle' is your actual Linux cpudriver
--> you can check it by restarting corefreqk.ko module with the option IdleDriverQuery=1
--> next, in corefreq-cli the cpu driver will be printed beside the Linux version

2- Blacklist the module nmi_watchdog (which consumes permanently one fixed state counter)
--> In your kernel boot argument, you can add: nmi_watchdog=0

3- Decrease the sleep interval of corefreqk to its minimum of 500 ms
--> use the option SleepInterval=500 when starting module
This should provide an almost real time monitoring of the cores.

oleksandr-oksenenko, 2016-10-18:
I don't need max turbo on one core :)
I just ran it so there would be some activity on the screenshot other than just 0-1% jitter.
Anyway, thanks for the tips!

JodiTheTigger, 2016-10-19:
[image]

JodiTheTigger, 2016-10-19:
Ok, so I tried the AUR package again, and it all works! Thanks!

closed 2016-10-19 by JodiTheTigger

cyring, 2016-10-19:
@ JodiTheTigger: Thank you for your screenshot, but I can see some layout issues.
- red turbo bars going out of the view
- footer line is repeated 3 times
Do you confirm these layout issues ?

JodiTheTigger, 2016-10-19:
Running it in under `fish` using `konsole` under gnome3 on  arch linux.

yes, I consistently get both issues.