cyring/CoreFreq #12: Building issue

Dxhs, 2017-01-01:
Software looks great. However I'm unable to build it. Getting the following messages under make:

`[elx@elsxpc CoreFreq]$ make
cc -c corefreqd.c -o corefreqd.o
cc -lpthread -lrt -o corefreqd corefreqd.c
cc -c corefreq-cli.c -o corefreq-cli.o
cc -lm -lrt -o corefreq-cli corefreq-cli.c
make -C /lib/modules/4.4.39-1-MANJARO/build M=/home/elx/CoreFreq modules
make[1]: Entering directory '/usr/lib/modules/4.4.39-1-MANJARO/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/usr/lib/modules/4.4.39-1-MANJARO/build'
make: *** [Makefile:6: all] Error 2
`
What does this mean? Apologies I'm new to this.

cyring, 2017-01-01:
ArchLinux prerequisited packages to build the kernel module:
- Linux kernel headers
- fakeroot (Tool for simulating superuser privileges)
- make
- dkms (Dynamic Kernel Modules System)
- glibc
- gcc
- git

Most of them should be available if you switch your system to some developer profile.

cyring, 2017-01-01:
Manjaro kernel wiki: 
https://wiki.manjaro.org/index.php/Manjaro_Kernels

Dxhs, 2017-01-01:
Missed two packages and some headers. Thanks for the help!

closed 2017-01-01 by Dxhs

cyring, 2017-01-02:
May you please tell me the missing files ?
This could help for a Manjaro CoreFreq's documentation.
Regards