cyring/CoreFreq #14: Detect only 16Cores (2x xeon x5650 -> 12Cores -> 24Threads)

sl1pkn07, 2017-01-29:
Hi

my setup is a dual Xeon x5650 (Westmere), this cpu have 12 physical cores (24Threads) each one

Corefreqk shoy in dmesg:

`[102105.255176] CoreFreq: Processor [06_2C] Architecture [Westmere/EP] CPU [16/16]`

and the program

cyring, 2017-01-29:
Thanks for this report.
To be honest, it is the first time I see CoreFreq driving a dual Xeon. Indeed, I have programmed the software based on the Intel SSDM specs, and I read cpu count from cpuid(4).
However, Xeon needs probably another logic to get a true topology.

It could help if you dump all cpuid leaves. In corefreq-cli there is an argument option to print a (long) hexa dump. cpu #0 should be enough.

About UI, it has been conceived for the majors: xterm and the Linux console (b/c I have datacenters in mind)
Do you still have issues with those ?

Thanks for helping

sl1pkn07, 2017-01-29:
https://sl1pkn07.wtf/paste/view/45c3b088

cyring, 2017-01-29:
Btw, looking at the size of L3 cache means a multiple of 6 cores. Sounds good.

cyring, 2017-01-29:
Hello , I'm sorry not have be clear enough but _cpuid_ is not my program.

However ' corefreq-cli -h ' will display all the options.
Option ' dump ' will  print hexadecimal values of registers that I used to decode your processors.

sl1pkn07, 2017-01-29:
this?

https://sl1pkn07.wtf/paste/view/3383b648

corefreq-cli -u -> Print CPUID

cyring, 2017-01-29:
It's funny that Linux seems also lost with your dual xeon !
Look into the /proc/cpuinfo and read the physical id :
id 0 is given to the processor for cpus from 0 to 5 and processor id 1 for cpus from 6 to 23
So strange

sl1pkn07, 2017-01-29:
└───╼  cat /proc/cpuinfo |grep -e 'processor' -e "physical id"
processor       : 0
physical id     : 0
processor       : 1
physical id     : 0
processor       : 2
physical id     : 0
processor       : 3
physical id     : 0
processor       : 4
physical id     : 0
processor       : 5
physical id     : 0
processor       : 6
physical id     : 1
processor       : 7
physical id     : 1
processor       : 8
physical id     : 1
processor       : 9
physical id     : 1
processor       : 10
physical id     : 1
processor       : 11
physical id     : 1
processor       : 12
physical id     : 0
processor       : 13
physical id     : 0
processor       : 14
physical id     : 0
processor       : 15
physical id     : 0
processor       : 16
physical id     : 0
processor       : 17
physical id     : 0
processor       : 18
physical id     : 1
processor       : 19
physical id     : 1
processor       : 20
physical id     : 1
processor       : 21
physical id     : 1
processor       : 22
physical id     : 1
processor       : 23
physical id     : 1

cyring, 2017-01-29:
Perfect this the dump I need.

Please be patient for a corrected version; it will take time to support multiple procs.

I hope however the other features are OK with your setup; such as Turbo monitoring ?

sl1pkn07, 2017-01-29:
when try read the processor [p] inte the corefreq-cli (without arguments)

i get: https://sl1pkn07.wtf/paste/view/d3822bcf

i need open a new issue, right?

cyring, 2017-01-29:
My bad, forget for the physical id.

cyring, 2017-01-29:
Not needed ! It's a total refactoring of CoreFreq to support dual. It won't be the last crash you will face.
Be aware that I have programmed with a mono proc 8 threads core.

What I suggest is that we share a development release (tar file of source code) until we get stability then it will be published.

golem131, 2017-02-10:
On Xeon E5-2620 same behavior (topology 2x6x2).
```
# cat /proc/cpuinfo |grep -e 'processor' -e "physical id"
processor	: 0
physical id	: 0
processor	: 1
physical id	: 0
processor	: 2
physical id	: 0
processor	: 3
physical id	: 0
processor	: 4
physical id	: 0
processor	: 5
physical id	: 0
processor	: 6
physical id	: 1
processor	: 7
physical id	: 1
processor	: 8
physical id	: 1
processor	: 9
physical id	: 1
processor	: 10
physical id	: 1
processor	: 11
physical id	: 1
processor	: 12
physical id	: 0
processor	: 13
physical id	: 0
processor	: 14
physical id	: 0
processor	: 15
physical id	: 0
processor	: 16
physical id	: 0
processor	: 17
physical id	: 0
processor	: 18
physical id	: 1
processor	: 19
physical id	: 1
processor	: 20
physical id	: 1
processor	: 21
physical id	: 1
processor	: 22
physical id	: 1
processor	: 23
physical id	: 1
```

```
# dmesg | grep CoreF
[1297111.181375] CoreFreq: Processor [06_2D] Architecture [SandyBridge/eXtreme.EP] CPU [16/16]
```

sl1pkn07, 2017-02-10:
also, i get kernel panic (hard lockup) when reset the machine and the corefreck module is loaded

disable the module through grub can load the machine again

cyring, 2017-02-10:
golem131, sl1pkn07: as I mentioned above, CoreFreq is developed with a mainstream processor. A Core i7-920 Bloomfield Nehalem architecture.
bi-processors Xeon is untested.
I stated this bug as enhancement when I will have the chance to work with such hardware.
Sincerely.

cyring, 2017-02-18:
golem131, sl1pkn07: Hello, 

A special repository is created to focus on topology issues. 
(Fyi, this is a subset of the CoreFreq topology code.)

1- go to [topology](https://github.com/cyring/topology) 
2- build and test the driver (instructions supplied in README) 
3- dmesg will output the topology map 
4- post your results in the topology issues repository.

Thank you

sl1pkn07, 2017-02-18:
in that repository only exist the readme

cyring, 2017-02-18:
Sorry for my mess with the GitHub, I just fixed it

sl1pkn07, 2017-02-18:
```
[169280.032589] Topology: Intel(R) Xeon(R) CPU X5650 @ 2.67GHz
                Processor [06_2C] [GenuineIntel] CPU [16/16]
                Topology map [Extended]                         (w)rite-Back (i)nclusive
                 cpu#  ApicID  CoreID ThreadID L1-Inst Way  L1-Data Way     L2  Way      L3  Way
                  0*       0       0       0   32768  4..   32768  8..  262144  8..12582912 16.i
                  1        2       1       0   32768  4..   32768  8..  262144  8..12582912 16.i
                  2        4       2       0   32768  4..   32768  8..  262144  8..12582912 16.i
                  3       16       8       0   32768  4..   32768  8..  262144  8..12582912 16.i
                  4       18       9       0   32768  4..   32768  8..  262144  8..12582912 16.i
                  5       20      10       0   32768  4..   32768  8..  262144  8..12582912 16.i
                  6       32       0       0   32768  4..   32768  8..  262144  8..12582912 16.i
                  7       34       1       0   32768  4..   32768  8..  262144  8..12582912 16.i
                  8       36       2       0   32768  4..   32768  8..  262144  8..12582912 16.i
                  9       48  
```

sl1pkn07, 2017-02-18:
with other program:

```
└───╼  likwid-topology 
--------------------------------------------------------------------------------
CPU name:       Intel(R) Xeon(R) CPU           X5650  @ 2.67GHz
CPU type:       Intel Core Westmere processor
CPU stepping:   2
********************************************************************************
Hardware Thread Topology
********************************************************************************
Sockets:                2
Cores per socket:       6
Threads per core:       2
--------------------------------------------------------------------------------
HWThread        Thread          Core            Socket          Available
0               0               0               0               *
1               0               1               0               *
2               0               2               0               *
3               0               3               0               *
4               0               4               0               *
5               0               5               0               *
6               0               6               1               *
7               0               7               1               *
8               0               8               1               *
9               0               9               1               *
10              0               10              1               *
11              0               11              1               *
12              1               0               0               *
13              1               1               0               *
14              1               2               0               *
15              1               3               0               *
16              1               4               0               *
17              1               5               0               *
18              1               6               1               *
19              1               7               1               *
20              1               8               1               *
21              1               9               1               *
22              1               10              1               *
23              1               11              1               *
--------------------------------------------------------------------------------
Socket 0:               ( 0 12 1 13 2 14 3 15 4 16 5 17 )
Socket 1:               ( 6 18 7 19 8 20 9 21 10 22 11 23 )
--------------------------------------------------------------------------------
********************************************************************************
Cache Topology
********************************************************************************
Level:                  1
Size:                   32 kB
Cache groups:           ( 0 12 ) ( 1 13 ) ( 2 14 ) ( 3 15 ) ( 4 16 ) ( 5 17 ) ( 6 18 ) ( 7 19 ) ( 8 20 ) ( 9 21 ) ( 10 22 ) ( 11 23 )
--------------------------------------------------------------------------------
Level:                  2
Size:                   256 kB
Cache groups:           ( 0 12 ) ( 1 13 ) ( 2 14 ) ( 3 15 ) ( 4 16 ) ( 5 17 ) ( 6 18 ) ( 7 19 ) ( 8 20 ) ( 9 21 ) ( 10 22 ) ( 11 23 )
--------------------------------------------------------------------------------
Level:                  3
Size:                   12 MB
Cache groups:           ( 0 12 1 13 2 14 3 15 4 16 5 17 ) ( 6 18 7 19 8 20 9 21 10 22 11 23 )
--------------------------------------------------------------------------------
********************************************************************************
NUMA Topology
********************************************************************************
NUMA domains:           2
--------------------------------------------------------------------------------
Domain:                 0
Processors:             ( 0 12 1 13 2 14 3 15 4 16 5 17 )
Distances:              10 20
Free memory:            1885.52 MB
Total memory:           24139.3 MB
--------------------------------------------------------------------------------
Domain:                 1
Processors:             ( 6 18 7 19 8 20 9 21 10 22 11 23 )
Distances:              20 10
Free memory:            961.816 MB
Total memory:           24190.3 MB
--------------------------------------------------------------------------------
```

cyring, 2017-02-18:
sl1pkn07: thanks
My algorithm is lost with dual/multi socket. Needs improvements ...

cyring, 2017-08-15:
For testing purpose, added the Package ID into [topology](https://github.com/cyring/topology)

cyring, 2017-09-03:
Fixed with OS count.

closed 2017-09-23 by cyring