cyring/CoreFreq #15: Segmentation fault

bigfreak85, 2017-01-31:
see:
https://fb.naeherwohnen.de/index.php/mE69/

Sysspecs:
https://fb.naeherwohnen.de/index.php/j82c/

bigfreak85, 2017-01-31:
seems to be a problem when qemu domain is running (win10 vm). If no Guest is Running the App doesnt crash...

cyring, 2017-01-31:
Thanks to report this issue: I have not tested it w/ Skylake (blind coding upon specs)

What about the kernel driver module corefreqk.ko : does it start normally?
What about the daemon corefreqd ?

I would first to check both.

You can run daemon in debug mode with option '-d'. This will print the info of the detected processor, transmitted from driver to the daemon

cyring, 2017-01-31:
The general info of i7-6820HQ that CoreFreq should report the same:

http://www.cpu-world.com/CPUs/Core_i7/Intel-Core%20i7-6820HQ%20Mobile%20processor.html

cyring, 2017-01-31:
Does the client run without the UI, starting it with cstates option ?

corefreq-cli -c

bigfreak85, 2017-02-01:
Hi,

I will Answer one by one:

1. Yes the Kerneldriver starts normaly. 
2. The Daemon also starts normaly. When i start with -d flag set it give me the following output:
root@t460p /home/bigfreak > corefreqd -d
  SysGate(1)
CoreFreq Daemon.  Copyright (C) 2015-2017 CYRIL INGENIERIE

  Processor [Intel(R) Core(TM) i7-6820HQ CPU @ 2.70GHz]
  Architecture [Skylake/S] 8/8 CPU Online.
  SleepInterval(1000), SysGate(1)

    CPU #000 @ 2712.03 MHz
    CPU #001 @ 2712.57 MHz
    CPU #002 @ 2712.62 MHz
    CPU #003 @ 2712.57 MHz
    CPU #004 @ 2712.75 MHz
    CPU #005 @ 2712.61 MHz
    CPU #006 @ 2712.74 MHz
    CPU #007 @ 2712.76 MHz
    Thread [7fb87bf1e700] Init CPU 004
    Thread [7fb87c71f700] Init CPU 003
    Thread [7fb87cf20700] Init CPU 002
    Thread [7fb87a71b700] Init CPU 007
    Thread [7fb87af1c700] Init CPU 006
    Thread [7fb87b71d700] Init CPU 005
    Thread [7fb87df22700] Init CPU 000
    Thread [7fb87d721700] Init CPU 001

3.Yes the Client starts with -c Flag and doesnt crash (10 Minutes runtime)

4. No im running corefreq on the Local Machine not in the VM. 
no difference whether qemu runs or not (VM powerd on or off) Thats why im deleted the Comment.

bigfreak85, 2017-02-01:
After Playing a while with the Menu im getting this:

*** Error in `corefreq-cli': malloc(): memory corruption (fast): 0x0000000000cdc380 ***
======= Backtrace: =========
/usr/lib/libc.so.6(+0x70c4b)[0x7f7c03216c4b]
/usr/lib/libc.so.6(+0x76fe6)[0x7f7c0321cfe6]
/usr/lib/libc.so.6(+0x79411)[0x7f7c0321f411]
/usr/lib/libc.so.6(__libc_malloc+0x54)[0x7f7c03220d44]
corefreq-cli[0x4012e0]
corefreq-cli[0x403863]
corefreq-cli[0x414b76]
corefreq-cli[0x4170f8]
corefreq-cli[0x40999b]
corefreq-cli[0x42382d]
/usr/lib/libc.so.6(__libc_start_main+0xf1)[0x7f7c031c6291]
corefreq-cli[0x400f2a]
======= Memory map: ========
00400000-00427000 r-xp 00000000 08:02 51033243                           /usr/bin/corefreq-cli
00627000-00628000 r-xp 00027000 08:02 51033243                           /usr/bin/corefreq-cli
00628000-00629000 rwxp 00028000 08:02 51033243                           /usr/bin/corefreq-cli
00cbf000-00ce0000 rwxp 00000000 00:00 0                                  [heap]
7f7bfc000000-7f7bfc021000 rwxp 00000000 00:00 0 
7f7bfc021000-7f7c00000000 ---p 00000000 00:00 0 
7f7c02b6f000-7f7c02b85000 r-xp 00000000 08:02 50990620                   /usr/lib/libgcc_s.so.1
7f7c02b85000-7f7c02d84000 ---p 00016000 08:02 50990620                   /usr/lib/libgcc_s.so.1
7f7c02d84000-7f7c02d85000 r-xp 00015000 08:02 50990620                   /usr/lib/libgcc_s.so.1
7f7c02d85000-7f7c02d86000 rwxp 00016000 08:02 50990620                   /usr/lib/libgcc_s.so.1
7f7c02d86000-7f7c02f89000 rwxs 00000000 00:16 139777                     /dev/shm/corefreq-shm
7f7c02f89000-7f7c02fa1000 r-xp 00000000 08:02 50990227                   /usr/lib/libpthread-2.24.so
7f7c02fa1000-7f7c031a0000 ---p 00018000 08:02 50990227                   /usr/lib/libpthread-2.24.so
7f7c031a0000-7f7c031a1000 r-xp 00017000 08:02 50990227                   /usr/lib/libpthread-2.24.so
7f7c031a1000-7f7c031a2000 rwxp 00018000 08:02 50990227                   /usr/lib/libpthread-2.24.so
7f7c031a2000-7f7c031a6000 rwxp 00000000 00:00 0 
7f7c031a6000-7f7c0333b000 r-xp 00000000 08:02 50990246                   /usr/lib/libc-2.24.so
7f7c0333b000-7f7c0353a000 ---p 00195000 08:02 50990246                   /usr/lib/libc-2.24.so
7f7c0353a000-7f7c0353e000 r-xp 00194000 08:02 50990246                   /usr/lib/libc-2.24.so
7f7c0353e000-7f7c03540000 rwxp 00198000 08:02 50990246                   /usr/lib/libc-2.24.so
7f7c03540000-7f7c03544000 rwxp 00000000 00:00 0 
7f7c03544000-7f7c0354b000 r-xp 00000000 08:02 50990307                   /usr/lib/librt-2.24.so
7f7c0354b000-7f7c0374a000 ---p 00007000 08:02 50990307                   /usr/lib/librt-2.24.so
7f7c0374a000-7f7c0374b000 r-xp 00006000 08:02 50990307                   /usr/lib/librt-2.24.so
7f7c0374b000-7f7c0374c000 rwxp 00007000 08:02 50990307                   /usr/lib/librt-2.24.so
7f7c0374c000-7f7c0384f000 r-xp 00000000 08:02 50990304                   /usr/lib/libm-2.24.so
7f7c0384f000-7f7c03a4e000 ---p 00103000 08:02 50990304                   /usr/lib/libm-2.24.so
7f7c03a4e000-7f7c03a4f000 r-xp 00102000 08:02 50990304                   /usr/lib/libm-2.24.so
7f7c03a4f000-7f7c03a50000 rwxp 00103000 08:02 50990304                   /usr/lib/libm-2.24.so
7f7c03a50000-7f7c03a73000 r-xp 00000000 08:02 50990245                   /usr/lib/ld-2.24.so
7f7c03c33000-7f7c03c37000 rwxp 00000000 00:00 0 
7f7c03c71000-7f7c03c72000 rwxp 00000000 00:00 0 
7f7c03c72000-7f7c03c73000 r-xp 00022000 08:02 50990245                   /usr/lib/ld-2.24.so
7f7c03c73000-7f7c03c74000 rwxp 00023000 08:02 50990245                   /usr/lib/ld-2.24.so
7f7c03c74000-7f7c03c75000 rwxp 00000000 00:00 0 
7ffddf3dc000-7ffddf3fd000 rwxp 00000000 00:00 0                          [stack]
7ffddf5ce000-7ffddf5d0000 r--p 00000000 00:00 0                          [vvar]
7ffddf5d0000-7ffddf5d2000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Abgebrochen (Speicherabzug geschrieben)

bigfreak85, 2017-02-01:
playing a while with the Menu i get this:

*** Error in `corefreq-cli': malloc(): memory corruption (fast): 0x0000000000cdc380 ***
======= Backtrace: =========
/usr/lib/libc.so.6(+0x70c4b)[0x7f7c03216c4b]
/usr/lib/libc.so.6(+0x76fe6)[0x7f7c0321cfe6]
/usr/lib/libc.so.6(+0x79411)[0x7f7c0321f411]
/usr/lib/libc.so.6(__libc_malloc+0x54)[0x7f7c03220d44]
corefreq-cli[0x4012e0]
corefreq-cli[0x403863]
corefreq-cli[0x414b76]
corefreq-cli[0x4170f8]
corefreq-cli[0x40999b]
corefreq-cli[0x42382d]
/usr/lib/libc.so.6(__libc_start_main+0xf1)[0x7f7c031c6291]
corefreq-cli[0x400f2a]
======= Memory map: ========
00400000-00427000 r-xp 00000000 08:02 51033243                           /usr/bin/corefreq-cli
00627000-00628000 r-xp 00027000 08:02 51033243                           /usr/bin/corefreq-cli
00628000-00629000 rwxp 00028000 08:02 51033243                           /usr/bin/corefreq-cli
00cbf000-00ce0000 rwxp 00000000 00:00 0                                  [heap]
7f7bfc000000-7f7bfc021000 rwxp 00000000 00:00 0 
7f7bfc021000-7f7c00000000 ---p 00000000 00:00 0 
7f7c02b6f000-7f7c02b85000 r-xp 00000000 08:02 50990620                   /usr/lib/libgcc_s.so.1
7f7c02b85000-7f7c02d84000 ---p 00016000 08:02 50990620                   /usr/lib/libgcc_s.so.1
7f7c02d84000-7f7c02d85000 r-xp 00015000 08:02 50990620                   /usr/lib/libgcc_s.so.1
7f7c02d85000-7f7c02d86000 rwxp 00016000 08:02 50990620                   /usr/lib/libgcc_s.so.1
7f7c02d86000-7f7c02f89000 rwxs 00000000 00:16 139777                     /dev/shm/corefreq-shm
7f7c02f89000-7f7c02fa1000 r-xp 00000000 08:02 50990227                   /usr/lib/libpthread-2.24.so
7f7c02fa1000-7f7c031a0000 ---p 00018000 08:02 50990227                   /usr/lib/libpthread-2.24.so
7f7c031a0000-7f7c031a1000 r-xp 00017000 08:02 50990227                   /usr/lib/libpthread-2.24.so
7f7c031a1000-7f7c031a2000 rwxp 00018000 08:02 50990227                   /usr/lib/libpthread-2.24.so
7f7c031a2000-7f7c031a6000 rwxp 00000000 00:00 0 
7f7c031a6000-7f7c0333b000 r-xp 00000000 08:02 50990246                   /usr/lib/libc-2.24.so
7f7c0333b000-7f7c0353a000 ---p 00195000 08:02 50990246                   /usr/lib/libc-2.24.so
7f7c0353a000-7f7c0353e000 r-xp 00194000 08:02 50990246                   /usr/lib/libc-2.24.so
7f7c0353e000-7f7c03540000 rwxp 00198000 08:02 50990246                   /usr/lib/libc-2.24.so
7f7c03540000-7f7c03544000 rwxp 00000000 00:00 0 
7f7c03544000-7f7c0354b000 r-xp 00000000 08:02 50990307                   /usr/lib/librt-2.24.so
7f7c0354b000-7f7c0374a000 ---p 00007000 08:02 50990307                   /usr/lib/librt-2.24.so
7f7c0374a000-7f7c0374b000 r-xp 00006000 08:02 50990307                   /usr/lib/librt-2.24.so
7f7c0374b000-7f7c0374c000 rwxp 00007000 08:02 50990307                   /usr/lib/librt-2.24.so
7f7c0374c000-7f7c0384f000 r-xp 00000000 08:02 50990304                   /usr/lib/libm-2.24.so
7f7c0384f000-7f7c03a4e000 ---p 00103000 08:02 50990304                   /usr/lib/libm-2.24.so
7f7c03a4e000-7f7c03a4f000 r-xp 00102000 08:02 50990304                   /usr/lib/libm-2.24.so
7f7c03a4f000-7f7c03a50000 rwxp 00103000 08:02 50990304                   /usr/lib/libm-2.24.so
7f7c03a50000-7f7c03a73000 r-xp 00000000 08:02 50990245                   /usr/lib/ld-2.24.so
7f7c03c33000-7f7c03c37000 rwxp 00000000 00:00 0 
7f7c03c71000-7f7c03c72000 rwxp 00000000 00:00 0 
7f7c03c72000-7f7c03c73000 r-xp 00022000 08:02 50990245                   /usr/lib/ld-2.24.so
7f7c03c73000-7f7c03c74000 rwxp 00023000 08:02 50990245                   /usr/lib/ld-2.24.so
7f7c03c74000-7f7c03c75000 rwxp 00000000 00:00 0 
7ffddf3dc000-7ffddf3fd000 rwxp 00000000 00:00 0                          [stack]
7ffddf5ce000-7ffddf5d0000 r--p 00000000 00:00 0                          [vvar]
7ffddf5d0000-7ffddf5d2000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Abgebrochen (Speicherabzug geschrieben)

cyring, 2017-02-01:
Thanks for this info.

Can you please :
1- compile the UI w/ debug 
`cc -g -Wall corefreq-cli.c -o corefreq-cli -lm -lrt`
2- run it with gdb until crash
3- post the faulted source line of code

bigfreak85, 2017-02-01:
1 https://fb.naeherwohnen.de/index.php/ASZ/
2 done
3 https://fb.naeherwohnen.de/index.php/kodWq/

cyring, 2017-02-01:
Btw, in your kernel cmd line, have you blacklisted *nmi_watchdog* with 
'nmi_watchdog=0'

It is a requirement because nmi_watchdog use the same msr counters as CoreFreq read. 

Without blacklist, the values are unexpected and can lead the UI out of buffer.

bigfreak85, 2017-02-01:
Nope

root@t460p /boot/loader/entries > cat arch.conf 
title	Arch Linux
linux	/vmlinuz-linux
initrd	/initramfs-linux.img
options	root=PARTUUID=c8fae5ff-b11d-451d-8c10-58f8fde3b7f6 rw quiet splash

bigfreak85, 2017-02-01:
root@t460p /boot/loader/entries > cat /proc/sys/kernel/nmi_watchdog 
1

bigfreak85, 2017-02-01:
cat /proc/sys/kernel/nmi_watchdog 
0

after that it works :-D

closed 2017-02-01 by bigfreak85

cyring, 2017-02-01:
Cool. :)

Please show me some screenshots of your running UI w/ load on 1 core 8-) 

Check also you don't have system side effects (startup, suspend/resume, shutdown) while nmi_watchdog is blacklisted. 

If you like CoreFreq, just vote for it in AUR

Thks

(I let you close this issue)

bigfreak85, 2017-02-01:
here it comes :-D

https://fb.naeherwohnen.de/index.php/Ogq/

cyring, 2017-02-01:
Topology looks ok (Caches, Cores) in count & size
Ratios are good but it should top to x36
x8 is overwriting the header line: I need to improve this.
Architecture name sounds not correct ?
BCLK estimation is closed to factory value.