cyring/CoreFreq #16: On Gentoo with specific flag it aborted and by default menu is not showing

unixbhaskar, 2017-02-09:
One,

I need to press **q** to get the top menu. 

Second,

While running with **-s** flag , it aborted like below 

`
bhaskar@GentooLinux_21:33:10_Thu Feb 09:~/git-linux/CoreFreq>./corefreq-cli -s
Processor                             [Intel(R) Core(TM) i5-5200U CPU @ 2.20GHz]
|- Vendor ID                                                      [GenuineIntel]
|- Signature                                                             [06_3D]
|- Stepping                                                                [  4]
|- Architecture                                               [Broadwell/Mobile]
|- Online CPU                                                              [4/8]
|- Base Clock                                                              [ 99]
|- Ratio Boost:                                                                 
|                      Min Max  8C  7C  6C  5C  4C  3C  2C  1C                  
|                        5  22   -   -  25  25  25  25  25  27                  
Instruction set:                                                                
|- 3DNow!/Ext [N,N]           AES [Y]      AVX/AVX2 [Y/Y]        BMI1/BMI2 [Y/Y]
|- CLFSH        [Y]          CMOV [Y]       CMPXCH8   [Y]         CMPXCH16   [Y]
|- F16C         [Y]           FPU [Y]          FXSR   [Y]        LAHF/SAHF   [Y]
|- MMX/Ext    [Y/N]       MONITOR [Y]         MOVBE   [Y]         PCLMULDQ   [Y]
|- POPCNT       [Y]        RDRAND [Y]        RDTSCP   [Y]              SEP   [Y]
|- SSE          [Y]          SSE2 [Y]          SSE3   [Y]            SSSE3   [Y]
|- SSE4.1/4A  [Y/N]        SSE4.2 [Y]       SYSCALL   [Y]                       
                                                                                
Features:                                                                       
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Present]
|- CPL Qualified Debug Store                                  DS-CPL   [Present]
|- 64-Bit Debug Store                                         DTES64   [Present]
|- Fast-String Operation                                Fast-Strings   [Present]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
|- Pending Break Enable                                          PBE   [Present]
|- Process Context Identifiers                                  PCID   [Present]
|- Perfmon and Debug Capability                                 PDCM   [Present]
|- Page Global Enable                                            PGE   [Present]
|- Page Size Extension                                           PSE   [Present]
|- 36-bit Page Size Extension                                  PSE36   [Present]
|- Processor Serial Number                                       PSN   [Missing]
|- Restricted Transactional Memory                               RTM   [Missing]
|- Safer Mode Extensions                                         SMX   [Missing]
|- Self-Snoop                                                     SS   [Present]
|- Time Stamp Counter                                            TSC [  Variant]
|- Time Stamp Counter Deadline                          TSC-DEADLINE   [Present]
|- Virtual Mode Extension                                        VME   [Present]
|- Virtual Machine Extensions                                    VMX   [Present]
|- Extended xAPIC Support                                     x2APIC   [  xAPIC]                                                                                                                                    
*** Error in `./corefreq-cli': free(): invalid next size (fast): 0x0000000000cfe090 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x6f133)[0x7f0ce321b133]
/lib64/libc.so.6(+0x74796)[0x7f0ce3220796]
/lib64/libc.so.6(+0x74f9e)[0x7f0ce3220f9e]
./corefreq-cli[0x4014c0]
./corefreq-cli[0x4039f7]
./corefreq-cli[0x424238]
/lib64/libc.so.6(__libc_start_main+0xf0)[0x7f0ce31cc790]
./corefreq-cli[0x401029]
======= Memory map: ========
00400000-00428000 r-xp 00000000 fb:00 65345                              /home/bhaskar/git-linux/CoreFreq/corefreq-cli
00627000-00628000 r-xp 00027000 fb:00 65345                              /home/bhaskar/git-linux/CoreFreq/corefreq-cli
00628000-00629000 rwxp 00028000 fb:00 65345                              /home/bhaskar/git-linux/CoreFreq/corefreq-cli
00cfe000-00d1f000 rwxp 00000000 00:00 0                                  [heap]
7f0cdc000000-7f0cdc021000 rwxp 00000000 00:00 0 
7f0cdc021000-7f0ce0000000 ---p 00000000 00:00 0 
7f0ce2b76000-7f0ce2b8c000 r-xp 00000000 08:07 21666003                   /usr/lib64/gcc/x86_64-pc-linux-gnu/4.9.4/libgcc_s.so.1
7f0ce2b8c000-7f0ce2d8b000 ---p 00016000 08:07 21666003                   /usr/lib64/gcc/x86_64-pc-linux-gnu/4.9.4/libgcc_s.so.1
7f0ce2d8b000-7f0ce2d8c000 r-xp 00015000 08:07 21666003                   /usr/lib64/gcc/x86_64-pc-linux-gnu/4.9.4/libgcc_s.so.1
7f0ce2d8c000-7f0ce2d8d000 rwxp 00016000 08:07 21666003                   /usr/lib64/gcc/x86_64-pc-linux-gnu/4.9.4/libgcc_s.so.1
7f0ce2d8d000-7f0ce2f90000 rw-s 00000000 00:17 80559                      /dev/shm/corefreq-shm
7f0ce2f90000-7f0ce2fa7000 r-xp 00000000 08:07 54505121                   /lib64/libpthread-2.23.so
7f0ce2fa7000-7f0ce31a6000 ---p 00017000 08:07 54505121                   /lib64/libpthread-2.23.so
7f0ce31a6000-7f0ce31a7000 r-xp 00016000 08:07 54505121                   /lib64/libpthread-2.23.so
7f0ce31a7000-7f0ce31a8000 rwxp 00017000 08:07 54505121                   /lib64/libpthread-2.23.so
7f0ce31a8000-7f0ce31ac000 rwxp 00000000 00:00 0 
7f0ce31ac000-7f0ce333c000 r-xp 00000000 08:07 54505142                   /lib64/libc-2.23.so
7f0ce333c000-7f0ce353b000 ---p 00190000 08:07 54505142                   /lib64/libc-2.23.so
7f0ce353b000-7f0ce353f000 r-xp 0018f000 08:07 54505142                   /lib64/libc-2.23.so
7f0ce353f000-7f0ce3541000 rwxp 00193000 08:07 54505142                   /lib64/libc-2.23.so
7f0ce3541000-7f0ce3545000 rwxp 00000000 00:00 0 
7f0ce3545000-7f0ce354b000 r-xp 00000000 08:07 54505120                   /lib64/librt-2.23.so
7f0ce354b000-7f0ce374b000 ---p 00006000 08:07 54505120                   /lib64/librt-2.23.so
7f0ce374b000-7f0ce374c000 r-xp 00006000 08:07 54505120                   /lib64/librt-2.23.so
7f0ce374c000-7f0ce374d000 rwxp 00007000 08:07 54505120                   /lib64/librt-2.23.so
7f0ce374d000-7f0ce3851000 r-xp 00000000 08:07 54505111                   /lib64/libm-2.23.so
7f0ce3851000-7f0ce3a50000 ---p 00104000 08:07 54505111                   /lib64/libm-2.23.so
7f0ce3a50000-7f0ce3a51000 r-xp 00103000 08:07 54505111                   /lib64/libm-2.23.so
7f0ce3a51000-7f0ce3a52000 rwxp 00104000 08:07 54505111                   /lib64/libm-2.23.so
7f0ce3a52000-7f0ce3a75000 r-xp 00000000 08:07 54459364                   /lib64/ld-2.23.so
7f0ce3c54000-7f0ce3c58000 rwxp 00000000 00:00 0 
7f0ce3c73000-7f0ce3c75000 rwxp 00000000 00:00 0 
7f0ce3c75000-7f0ce3c76000 r-xp 00023000 08:07 54459364                   /lib64/ld-2.23.so
7f0ce3c76000-7f0ce3c77000 rwxp 00024000 08:07 54459364                   /lib64/ld-2.23.so
7f0ce3c77000-7f0ce3c78000 rwxp 00000000 00:00 0 
7ffe754b9000-7ffe754da000 rwxp 00000000 00:00 0                          [stack]
7ffe75503000-7ffe75505000 r--p 00000000 00:00 0                          [vvar]
7ffe75505000-7ffe75507000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted
`

unixbhaskar, 2017-02-09:
If you select **Features** under **window menu**, it aborted.

cyring, 2017-02-10:
This may be a buffer overflow :

just compile corefreq-cli.c in debug mode with a gcc -g ; then execute it with valgrind to locate the faulty line.

cyring, 2017-02-10:
Did you disable nmi_watchdog prior corefreqk.ko ?
Because it has to stopped to avoid msr registers conflicts

cyring, 2017-02-19:
Hello,
Does last version still crashes with a baseline kernel and nmi_watchdog disable ?

Btw, any non shortcut key will show the menu tab when area is cleared of any window.
This is a wanted behaviour.

cyring, 2017-03-05:
Hello,
Do you still confirm your issue  ?

unixbhaskar, 2017-03-05:
Oh I am so sorry Cyring ..that I didn't get a chance to re-apply your method yet...once I applied ..will certainly let you know. Sorry for the delay. Probably later in the day.

cyring, 2017-03-05:
Whilst a code review, I found lost codes that only x2APIC capable processor could reach.
```
	const char *x2APIC[] = {
		"Missing",
	    GoK	"  xAPIC" DoK,
	    GoK	" x2APIC" DoK
	};
```
Those "GoK", "DoK" ANSI strings have been removed.

Hope corefreq-cli will now run OK with your i5-5200U ?

regards,
CyrIng

unixbhaskar, 2017-03-06:
Well, I must be doing something wrong...check out 

`bhaskar@GentooLinux_07:46:59_Mon Mar 06:~/git-linux/CoreFreq>git pull
remote: Counting objects: 61, done.
remote: Compressing objects: 100% (61/61), done.
remote: Total 61 (delta 41), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (61/61), done.
From https://github.com/cyring/CoreFreq
   a191ecf9246d..9b4616ba19ed  master     -> origin/master
Updating a191ecf9246d..9b4616ba19ed
Fast-forward
 README.md      |  30 +++++++----
 bitasm.h       |  14 +++++
 corefreq-api.h |  68 ++++++++++++++++++-----
 corefreq-cli.c | 128 ++++++++++++++++++++++---------------------
 corefreq.h     |   1 +
 corefreqd.c    | 432 +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++------------------------
 corefreqk.c    | 320 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++-----------------
 coretypes.h    | 106 ++++++++++++++++++++++++++++++++++++
 intelmsr.h     | 195 ++++++++++++++++++++++++++++++++++++++++++++++++++++++------------
 9 files changed, 1057 insertions(+), 237 deletions(-)
bhaskar@GentooLinux_07:47:15_Mon Mar 06:~/git-linux/CoreFreq>ls
CoreFreq.svg    README.md  corefreq-api.h  corefreq-cli.o  corefreqd.c        corefreqk.c   corefreqk.mod.c     corefreqk.o  intelmsr.h
Makefile        amdmsr.h   corefreq-cli    corefreq.h      corefreqd.o        corefreqk.h   corefreqk.mod.gcno  coretypes.h  modules.order
Module.symvers  bitasm.h   corefreq-cli.c  corefreqd       corefreqd.service  corefreqk.ko  corefreqk.mod.o     dkms.conf
bhaskar@GentooLinux_07:47:24_Mon Mar 06:~/git-linux/CoreFreq>make
cc -Wall -pthread -c corefreqd.c -o corefreqd.o
corefreqd.c: In function 'SysGate_Update':
corefreqd.c:1286:6: warning: trampoline generated for nested function 'SortByRuntime' [-Wtrampolines]
  int SortByRuntime(const void *p1, const void *p2)
      ^
corefreqd.c:1294:6: warning: trampoline generated for nested function 'SortByUsertime' [-Wtrampolines]
  int SortByUsertime(const void *p1, const void *p2)
      ^
corefreqd.c:1302:6: warning: trampoline generated for nested function 'SortBySystime' [-Wtrampolines]
  int SortBySystime(const void *p1, const void *p2)
      ^
corefreqd.c:1310:6: warning: trampoline generated for nested function 'SortByState' [-Wtrampolines]
  int SortByState(const void *p1, const void *p2)
      ^
corefreqd.c:1318:6: warning: trampoline generated for nested function 'SortByPID' [-Wtrampolines]
  int SortByPID(const void *p1, const void *p2)
      ^
corefreqd.c:1326:6: warning: trampoline generated for nested function 'SortByCommand' [-Wtrampolines]
  int SortByCommand(const void *p1, const void *p2)
      ^
corefreqd.c:1345:6: warning: trampoline generated for nested function 'SortByTracker' [-Wtrampolines]
  int SortByTracker(const void *p1, const void *p2)
      ^
cc corefreqd.c -o corefreqd -lpthread -lrt
corefreqd.c: In function 'SysGate_Update':
corefreqd.c:1286:6: warning: trampoline generated for nested function 'SortByRuntime' [-Wtrampolines]
  int SortByRuntime(const void *p1, const void *p2)
      ^
corefreqd.c:1294:6: warning: trampoline generated for nested function 'SortByUsertime' [-Wtrampolines]
  int SortByUsertime(const void *p1, const void *p2)
      ^
corefreqd.c:1302:6: warning: trampoline generated for nested function 'SortBySystime' [-Wtrampolines]
  int SortBySystime(const void *p1, const void *p2)
      ^
corefreqd.c:1310:6: warning: trampoline generated for nested function 'SortByState' [-Wtrampolines]
  int SortByState(const void *p1, const void *p2)
      ^
corefreqd.c:1318:6: warning: trampoline generated for nested function 'SortByPID' [-Wtrampolines]
  int SortByPID(const void *p1, const void *p2)
      ^
corefreqd.c:1326:6: warning: trampoline generated for nested function 'SortByCommand' [-Wtrampolines]
  int SortByCommand(const void *p1, const void *p2)
      ^
corefreqd.c:1345:6: warning: trampoline generated for nested function 'SortByTracker' [-Wtrampolines]
[... 84 lines cut ...]
7fabf1697000-7fabf1896000 ---p 00190000 08:07 54505142                   /lib64/libc-2.23.so
7fabf1896000-7fabf189a000 r-xp 0018f000 08:07 54505142                   /lib64/libc-2.23.so
7fabf189a000-7fabf189c000 rwxp 00193000 08:07 54505142                   /lib64/libc-2.23.so
7fabf189c000-7fabf18a0000 rwxp 00000000 00:00 0 
7fabf18a0000-7fabf18a6000 r-xp 00000000 08:07 54505120                   /lib64/librt-2.23.so
7fabf18a6000-7fabf1aa6000 ---p 00006000 08:07 54505120                   /lib64/librt-2.23.so
7fabf1aa6000-7fabf1aa7000 r-xp 00006000 08:07 54505120                   /lib64/librt-2.23.so
7fabf1aa7000-7fabf1aa8000 rwxp 00007000 08:07 54505120                   /lib64/librt-2.23.so
7fabf1aa8000-7fabf1bac000 r-xp 00000000 08:07 54505111                   /lib64/libm-2.23.so
7fabf1bac000-7fabf1dab000 ---p 00104000 08:07 54505111                   /lib64/libm-2.23.so
7fabf1dab000-7fabf1dac000 r-xp 00103000 08:07 54505111                   /lib64/libm-2.23.so
7fabf1dac000-7fabf1dad000 rwxp 00104000 08:07 54505111                   /lib64/libm-2.23.so
7fabf1dad000-7fabf1dd0000 r-xp 00000000 08:07 54459364                   /lib64/ld-2.23.so
7fabf1fb1000-7fabf1fb5000 rwxp 00000000 00:00 0 
7fabf1fce000-7fabf1fd0000 rwxp 00000000 00:00 0 
7fabf1fd0000-7fabf1fd1000 r-xp 00023000 08:07 54459364                   /lib64/ld-2.23.so
7fabf1fd1000-7fabf1fd2000 rwxp 00024000 08:07 54459364                   /lib64/ld-2.23.so
7fabf1fd2000-7fabf1fd3000 rwxp 00000000 00:00 0 
7ffef2f70000-7ffef2f91000 rwxp 00000000 00:00 0                          [stack]
7ffef2fa5000-7ffef2fa7000 r--p 00000000 00:00 0                          [vvar]
7ffef2fa7000-7ffef2fa9000 r-xp 00000000 00:00 0                          [vdso]
ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
Aborted

`

cyring, 2017-03-06:
Although I don't have the Gentoo, I believe that all the warnings come from another compiler than gcc. Perhaps clang which does not support nested function.

I'm also uncertain about the x86 asm results of the driver with another compiler.

Can you change your compiler to gcc ?

Check also that nmi_watchdog=0 is set in your kernel boot loader.

cyring, 2017-03-10:
Can we close  ?

cyring, 2017-03-11:
Have you been able to run CoreFreq with its requirements ?

I will appreciate if you can give a status to the issue : Open or Close

closed 2017-03-15 by cyring