cyring/CoreFreq #17: Centos 7 , corefreq-cli segmentation fault

golem131, 2017-02-10:
- kernel - elrepo ml 4.8.7
- glibc version 2.17-157 rpm

Valgrind output:

```
=29746== Invalid write of size 8
==29746==    at 0x4C2DEF7: memset (vg_replace_strmem.c:1224)
==29746==    by 0x409EBC: Top (corefreq-cli.c:4410)
==29746==    by 0x423A18: main (corefreq-cli.c:4958)
==29746==  Address 0x5938c60 is 0 bytes after a block of size 9,504 alloc'd
==29746==    at 0x4C29975: calloc (vg_replace_malloc.c:711)
==29746==    by 0x406E4D: CreateLayer (corefreq-cli.c:1738)
==29746==    by 0x4166C6: AllocAll.7656 (corefreq-cli.c:3111)
==29746==    by 0x4098D5: Top (corefreq-cli.c:4345)
==29746==    by 0x423A18: main (corefreq-cli.c:4958)
==29746==

valgrind: m_mallocfree.c:303 (get_bszB_as_is): Assertion 'bszB_lo == bszB_hi' failed.
valgrind: Heap block lo/hi size mismatch: lo = 9568, hi = 9259542123273814144.
This is probably caused by your program erroneously writing past the
end of a heap block and corrupting heap metadata.  If you fix any
invalid writes reported by Memcheck, this assertion failure will
probably go away.  Please try that before reporting this as a bug.

host stacktrace:
==29746==    at 0x3803D208: ??? (in /usr/lib64/valgrind/memcheck-amd64-linux)
==29746==    by 0x3803D334: ??? (in /usr/lib64/valgrind/memcheck-amd64-linux)
==29746==    by 0x3803D4C1: ??? (in /usr/lib64/valgrind/memcheck-amd64-linux)
==29746==    by 0x3804ADA3: ??? (in /usr/lib64/valgrind/memcheck-amd64-linux)
==29746==    by 0x38036A6B: ??? (in /usr/lib64/valgrind/memcheck-amd64-linux)
==29746==    by 0x380352E3: ??? (in /usr/lib64/valgrind/memcheck-amd64-linux)
==29746==    by 0x380394AB: ??? (in /usr/lib64/valgrind/memcheck-amd64-linux)
==29746==    by 0x38034917: ??? (in /usr/lib64/valgrind/memcheck-amd64-linux)
==29746==    by 0x802D7F2A9: ???
==29746==    by 0x802CA9F2F: ???

sched status:
  running_tid=1

Thread 1: status = VgTs_Runnable (lwpid 29746)
==29746==    at 0x4C2DF10: memset (vg_replace_strmem.c:1224)
==29746==    by 0x409EBC: Top (corefreq-cli.c:4410)
==29746==    by 0x423A18: main (corefreq-cli.c:4958)

```

cyring, 2017-02-10:
Thanks for this dump.
Can you assure me that you have disable nmi_watchdog before running corefreqk.ko driver ?

golem131, 2017-02-13:
@cyring 
Yep.
```
# echo 0 > /proc/sys/kernel/nmi_watchdog
# cat /proc/sys/kernel/nmi_watchdog
0
```

cyring, 2017-02-19:
Hello,
Does last version still crashes (with a baseline kernel) ?

cyring, 2017-03-05:
Hello,
Can you instead disable nmi_watchdog from the kernel boot argument command line, and not _after_ the kernel is loaded.
Thank you for your return

cyring, 2017-03-10:
Can we close  ?

cyring, 2017-03-11:
Have you been able to run CoreFreq with its requirements ?

I will appreciate if you can give a  status to the issue : Open or Close

closed 2017-03-15 by cyring