cyring/CoreFreq #21: compile warning and induce losing efficancy  "pci.h:1206:2: warning: ignoring return value of ‘__pci_register_driver’"

hanqing0521, 2017-02-26:
#  compile warning 

```
cc -Wall -pthread -c corefreqd.c -o corefreqd.o
cc corefreqd.c -o corefreqd -lpthread -lrt
cc -Wall -c corefreq-cli.c -o corefreq-cli.o
cc corefreq-cli.c -o corefreq-cli -lm -lrt
make -C /lib/modules/4.9.10-200.fc25.x86_64/build M=/home/lixun/rpm/CoreFreq modules
make[1]: Entering directory '/usr/src/kernels/4.9.10-200.fc25.x86_64'
  CC [M]  /home/lixun/rpm/CoreFreq/corefreqk.o
In file included from /home/lixun/rpm/CoreFreq/corefreqk.c:9:0:
/home/lixun/rpm/CoreFreq/corefreqk.c: In function ‘CoreFreqK_init’:
./include/linux/pci.h:1206:2: warning: ignoring return value of ‘__pci_register_driver’, declared with attribute warn_unused_result [-Wunused-result]
  __pci_register_driver(driver, THIS_MODULE, KBUILD_MODNAME)
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/lixun/rpm/CoreFreq/corefreqk.c:3064:5: note: in expansion of macro ‘pci_register_driver’
     pci_register_driver(&CoreFreqK_pci_driver);
     ^~~~~~~~~~~~~~~~~~~
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/lixun/rpm/CoreFreq/corefreqk.mod.o
  LD [M]  /home/lixun/rpm/CoreFreq/corefreqk.ko
make[1]: Leaving directory '/usr/src/kernels/4.9.10-200.fc25.x86_64'
```

#  error of insmod 
```bash
insmod  corefreqk.ko 
insmod: ERROR: could not insert module corefreqk.ko: Required key not available
```

# computer info

## cpu info
```bash
cat /proc/cpuinfo 
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 61
model name      : Intel(R) Core(TM) M-5Y71 CPU @ 1.20GHz
stepping        : 4
microcode       : 0x24
cpu MHz         : 1262.512
cache size      : 4096 KB
physical id     : 0
siblings        : 4
core id         : 0
cpu cores       : 2
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 20
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch epb intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm rdseed adx smap xsaveopt dtherm ida arat pln pts
bugs            :
bogomips        : 2793.58
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:

processor       : 1
vendor_id       : GenuineIntel
cpu family      : 6
model           : 61
model name      : Intel(R) Core(TM) M-5Y71 CPU @ 1.20GHz
stepping        : 4
microcode       : 0x24
cpu MHz         : 1200.134
cache size      : 4096 KB
physical id     : 0
siblings        : 4
core id         : 1
cpu cores       : 2
apicid          : 2
initial apicid  : 2
fpu             : yes
fpu_exception   : yes
cpuid level     : 20
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch epb intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm rdseed adx smap xsaveopt dtherm ida arat pln pts
bugs            :
bogomips        : 2795.73
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:

processor       : 2
vendor_id       : GenuineIntel
cpu family      : 6
model           : 61
model name      : Intel(R) Core(TM) M-5Y71 CPU @ 1.20GHz
stepping        : 4
microcode       : 0x24
cpu MHz         : 1199.963
cache size      : 4096 KB
physical id     : 0
siblings        : 4
core id         : 0
cpu cores       : 2
apicid          : 1
initial apicid  : 1
fpu             : yes
fpu_exception   : yes
cpuid level     : 20
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch epb intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm rdseed adx smap xsaveopt dtherm ida arat pln pts
bugs            :
bogomips        : 2796.87
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:

processor       : 3
vendor_id       : GenuineIntel
cpu family      : 6
model           : 61
model name      : Intel(R) Core(TM) M-5Y71 CPU @ 1.20GHz
stepping        : 4
microcode       : 0x24
cpu MHz         : 1199.963
cache size      : 4096 KB
physical id     : 0
siblings        : 4
core id         : 1
cpu cores       : 2
apicid          : 3
initial apicid  : 3
fpu             : yes
fpu_exception   : yes
cpuid level     : 20
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf eagerfpu pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch epb intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm rdseed adx smap xsaveopt dtherm ida arat pln pts
bugs            :
bogomips        : 2795.84
clflush size    : 64
cache_alignment : 64
address sizes   : 39 bits physical, 48 bits virtual
power management:
```
## linux version
```bash
uname -r
4.9.10-200.fc25.x86_64
```

cyring, 2017-02-26:
Thanks a lot. These are valuable informations.
Based on your report (because I'm still using BIOS platforms) , third-party kernel modules with UEFI enable need to be signed on.
I would suggest you to follow the solution provided at http://askubuntu.com/questions/762254/why-do-i-get-required-key-not-available-when-install-3rd-party-kernel-modules

Please let me know if it works ?

CyrIng

cyring, 2017-02-27:
Compilation should now be fixed.
Can you try please ?

hanqing0521, 2017-02-28:
unfortunately, it still can not insmod Corefreqk,ko. 
*there are different problem. After i disable secure boot, run `sudo insmod Corefreqk,ko` my laptop crashed.*
I didn't find the crash journal

hanqing0521, 2017-02-28:
I clone your new commit, then there are new problem
**kernel info**
`4.9.11-200.fc25.x86_64`

# no compile warning 
```
make -C /lib/modules/4.9.11-200.fc25.x86_64/build M=/home/lixun/rpm/tmp/CoreFreq modules
make[1]: Entering directory '/usr/src/kernels/4.9.11-200.fc25.x86_64'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory '/usr/src/kernels/4.9.11-200.fc25.x86_64'
```
# *it's not a problem but a advise 
**SElinux will prevent your program**

# Still appear error info when insmod
```
sudo insmod corefreqk.ko                                                                           
insmod: ERROR: could not insert module corefreqk.ko: Invalid module format
```

cyring, 2017-02-28:
What is your Linux distribution ? 

Mine is ArchLinux.

I also do iterative compilation and execution testings with Debian, Ubuntu, Cento: all distros installed with default options.

Do you have any pointers to manage a driver with SElinux?

hanqing0521, 2017-02-28:
**My Linux distribution is fedora25**
I'm sorry, i don't know the method but maybe you can add some selinux policies. You know many distribution contain SElinux.  SElinux is not so important for me, considering just  using as desktop application, but it is very important for Servers.

cyring, 2017-02-28:
"By default, the Arch Kernel does not have the SELinux LSM enabled ..."
See https://wiki.archlinux.org/index.php/SELinux
I believe the distributions I have listed have abandoned SELinux since a while ?
I may consider SELinux in long terms delay; the servers roadmap is however to solve first topology issues for dual & multi-sockets then refactore code with indicators in consequence.

I will appreciate if you can post screenshots of your Core M-5Y71 monitored by CoreFreq

cyring, 2017-03-11:
Hello,
Have you been able to run CoreFreq in a basic way with the mainstream kernel ?

I will appreciate if you can give a  status to the issue : Open or Close

hanqing0521, 2017-03-12:
i clone your last version codes, but my latop crashed if i run  `sudo insmod corefreqk.ko`

cyring, 2017-03-12:
I've found the BDD58 errata for the [5th Generation](http://www.intel.com/content/www/us/en/processors/core/desktop-mobile-5th-gen-core-family-spec-update.html) that the counter register IA32_FIXED_CTR0 may not count consistently.

To eliminate this potential processor error, can you start CoreFreq with a generic algorithm :

`
insmod corefreqk.ko ArchID=1
`

closed 2017-04-22 by cyring