cyring/CoreFreq #22: Reading VID (vcore)

HebaNabil, 2017-02-27:
Thank you for this useful project.
I would like to ask you if there is a possibility to read the core voltage (vcore) using coreFreq?
 If it is currently not available feature. Do you know how can one do it ?

cyring, 2017-02-27:
Hello,

It's not available but planed on the CoreFreq roadmap.

So far, I find few information in table 35.18 for the only architecture Sandy Bridge of the [Intel Software Developer’s Manual](https://software.intel.com/en-us/articles/intel-sdm)

```
Register Addr = 198H (408)

Register Name = MSR_PERF_STATUS

Scope = Package

Bits = 47:32

Field = Core Voltage (R/O)

Description: P-state core voltage can be computed by MSR_PERF_STATUS[37:32] * (float) 1/(2^13).  
```

cyring, 2017-02-27:
A [topic](https://software.intel.com/en-us/forums/software-tuning-performance-optimization-platform-monitoring/topic/488785) to read

HebaNabil, 2017-02-27:
Thanks a lot for this useful information.

cyring, 2017-07-22:
Closing issue, lm_sensors is the answer for Linux

cyring, 2017-07-29:
VID for AMD K8 has been implemented.
One can change the current VID using a driver argument.
Enter " modinfo corefreqk.ko " to display all the options

CoreFreq is taking care of not exceeding limits.

Reminding work consists in translating VID to voltage and design a new UI layer.

Documentation shows that each architecture, probably each processors series , requires its own math function to convert to volt...

cyring, 2017-09-09:
Vcore is now showing in last commit
Run client with option '-V'

cyring, 2017-09-10:
Vcore is now showing in corefreq-cli

closed 2017-09-10 by cyring