cyring/CoreFreq #23: client coredumping a lot

netmonk, 2017-04-22:
-rw-r-----+ 1 root root 46225 22 avril 16:02 core.corefreq-cli.1000.e9f5528312614f4a889ff9e36bfd86c9.30313.1492869733000000000000.lz4
-rw-r-----+ 1 root root 45999 22 avril 16:03 core.corefreq-cli.1000.e9f5528312614f4a889ff9e36bfd86c9.30367.1492869798000000000000.lz4
-rw-r-----+ 1 root root 53244 22 avril 16:10 core.corefreq-cli.1000.e9f5528312614f4a889ff9e36bfd86c9.31486.1492870200000000000000.lz4
-rw-r-----+ 1 root root 47032 22 avril 16:11 core.corefreq-cli.1000.e9f5528312614f4a889ff9e36bfd86c9.31570.1492870286000000000000.lz4
-rw-r-----+ 1 root root 46208 22 avril 16:11 core.corefreq-cli.1000.e9f5528312614f4a889ff9e36bfd86c9.31746.1492870293000000000000.lz4
-rw-r-----+ 1 root root 46236 22 avril 16:11 core.corefreq-cli.1000.e9f5528312614f4a889ff9e36bfd86c9.31763.1492870301000000000000.lz4
-rw-r-----+ 1 root root 44063 22 avril 16:11 core.corefreq-cli.1000.e9f5528312614f4a889ff9e36bfd86c9.31789.1492870304000000000000.lz4
-rw-r-----+ 1 root root 46158 22 avril 16:11 core.corefreq-cli.1000.e9f5528312614f4a889ff9e36bfd86c9.31807.1492870308000000000000.lz4
-rw-r-----+ 1 root root 46200 22 avril 16:11 core.corefreq-cli.1000.e9f5528312614f4a889ff9e36bfd86c9.31826.1492870315000000000000.lz4

[plonky@netmonk CoreFreq]$  coredumpctl gdb corefreq-cli
           PID: 31826 (corefreq-cli)
           UID: 1000 (plonky)
           GID: 100 (users)
        Signal: 11 (SEGV)
     Timestamp: Sat 2017-04-22 16:11:55 CEST (2min 9s ago)
  Command Line: ./corefreq-cli
    Executable: /home/plonky/work/CoreFreq/corefreq-cli
 Control Group: /user.slice/user-1000.slice/session-c1.scope
          Unit: session-c1.scope
         Slice: user-1000.slice
       Session: c1
     Owner UID: 1000 (plonky)
       Boot ID: e9f5528312614f4a889ff9e36bfd86c9
    Machine ID: b6b4ab29a58b486a803dfb751f2caca6
      Hostname: netmonk
       Storage: /var/lib/systemd/coredump/core.corefreq-cli.1000.e9f5528312614f4a889ff9e36bfd86c9.31826.1492870315000000000000.lz4
       Message: Process 31826 (corefreq-cli) of user 1000 dumped core.
                
                Stack trace of thread 31826:
                #0  0x00007f8ca286aeb3 __memcpy_sse2_unaligned_erms (libc.so.6)
                #1  0x000000000040ac3e Top (corefreq-cli)
                #2  0x000000000042530a main (corefreq-cli)
                #3  0x00007f8ca2807291 __libc_start_main (libc.so.6)
                #4  0x0000000000400eea _start (corefreq-cli)

netmonk, 2017-04-22:
Ok sorry i was too fast, but the issue still occuring ! coredumping again and again !

cyring, 2017-04-22:
Hello netmonk

I need more details to help you solve the issue : 
1- I believe you have already disabled the nmi_watchdog ?
2- You are using recent and baseline kernel + gcc + libc ?
3- You don't have modified the compiler flags ? No optimization O1, O2 or O3 which could involve unexpected sse instructions ?

To investigate the issue, feel free to post details of your setup : Processor , Kernel distro and so on

Cyril

netmonk, 2017-04-22:
Hello cyril, 

Only the client is coredumping, do you think nmi_watchdog can have something to do with the client (kernel modul is ok and the daemon too). 

Im using archlinux with gcc 6.3.1 with kernel 4.9.11-1, standard make without any optims

cyring, 2017-04-22:
Yes it has a huge impact for a simple reason :

For drawing performance and cpu overhead, corefreq-cli does not check any buffer overflow. It's an implementation choice.

Programmatically speaking, it means no ' if ' condition is surrounding ascii buffers read/write because values have to be well defined by the driver, then the daemon.

When nmi_watchdog is enabled, it conflicts with CoreFreq driver whose values are corrupted and buffer overflows happen in client.

So what could be solutions :
1- Test any value  (thus cpu overhead)
2- Respect the prerequisites and disable nmi_watchdog (less cpu overhead)
3- others ?

So yes, please disable nmi_watchdog, and let me know if it runs better ?

Sincerely,
Cyril

netmonk, 2017-04-22:
Ok with this explanation I understand now :) I will test and give you
feedback

<https://github.com/cyring/CoreFreq/issues/23#issuecomment-296398045>, or mute
the thread

netmonk, 2017-04-23:
Disabling nmi_watchdog at cmdline seems to solve the issue ! 
No more coredump after reboot ! :) great

closed 2017-04-23 by netmonk