cyring/CoreFreq #24: Program received signal SIGSEGV, Segmentation fault.

papodaca, 2017-04-29:
Crashes when pushing cpu to 100%. (compiling nodejs with `make -j8`)

```
Program received signal SIGSEGV, Segmentation fault.
#0  0x00007ffff764a99f in __memset_avx2_unaligned_erms () from /usr/lib/libc.so.6
#1  0x000000000040ab37 in Top () at corefreq-cli.c:4473
#2  0x000000000042530a in main (argc=1, argv=0x7fffffffde78) at corefreq-cli.c:5028
```

enkore, 2017-04-29:
>    Q: The UI crashes frequently with segmentation faults ?
>    A: In the kernel boot command argument line, disable the NMI Watchdog
>
>     nmi_watchdog=0

?

cyring, 2017-04-29:
It means that you have to disable nmi_watchdog
Do this in your boot kernel command line.
Disabling nmi_watchdog will give to CoreFreq a full access to the UNC & URC counter registers, and avoid unexpected values.

FYI nmi_watchdog is frequently rewriting counters which leads corefreq-cli to randomized indexes and thus buffers overflowing.

So disable nmi_watchdog

netmonk, 2017-04-29:
It's not related to cpu charges. :) Disable NMI watchdog like @cyring is suggesting you and as it is written in the FAQ :)

papodaca, 2017-04-30:
Duly noted, feel free to close. Though would be nice if at run time this could be detected and a warning message printed or something?

cyring, 2017-04-30:
@papodaca : nice suggestion.

Another way could be to switch automatically to another free counter register taken from the general set ...

closed 2017-05-03 by cyring