cyring/CoreFreq #25: Intel i5-6300u bad cpu/thread count

lcaflc, 2017-05-03:
Hi,

I have an intel i5-6300u cpu: [ark.intel.com](http://ark.intel.com/products/88190/Intel-Core-i5-6300U-Processor-3M-Cache-up-to-3_00-GHz?q=i5-6300u)

corefreq badly detect the number of cpu and threads counting 4/8.
```
CoreFreq git:master > ./corefreq-cli -s                                                                                 ✭
Processor                             [Intel(R) Core(TM) i5-6300U CPU @ 2.40GHz]
|- Vendor ID                                                      [GenuineIntel]
|- Signature                                                             [06_4E]
|- Stepping                                                                [  3]
|- Architecture                                                     [Skylake/UY]
|- Online CPU                                                              [4/8]
|- Base Clock                                                              [ 99]
|- Ratio Boost:                                                                 
|                      Min Max  8C  7C  6C  5C  4C  3C  2C  1C                  
|                        4  25   -   -   -   -  29  29  29  30                  
Instruction set:                                                                
|- 3DNow!/Ext [N,N]           AES [Y]      AVX/AVX2 [Y/Y]        BMI1/BMI2 [Y/Y]
|- CLFSH        [Y]          CMOV [Y]       CMPXCH8   [Y]         CMPXCH16   [Y]
|- F16C         [Y]           FPU [Y]          FXSR   [Y]        LAHF/SAHF   [Y]
|- MMX/Ext    [Y/N]       MONITOR [Y]         MOVBE   [Y]         PCLMULDQ   [Y]
|- POPCNT       [Y]        RDRAND [Y]        RDTSCP   [Y]              SEP   [Y]
|- SSE          [Y]          SSE2 [Y]          SSE3   [Y]            SSSE3   [Y]
|- SSE4.1/4A  [Y/N]        SSE4.2 [Y]       SYSCALL   [Y]                       
                                                                                
Features:                                                                       
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Present]
|- CPL Qualified Debug Store                                  DS-CPL   [Present]
|- 64-Bit Debug Store                                         DTES64   [Present]
|- Fast-String Operation                                Fast-Strings   [Present]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Present]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
|- Pending Break Enable                                          PBE   [Present]
|- Process Context Identifiers                                  PCID   [Present]
|- Perfmon and Debug Capability                                 PDCM   [Present]
|- Page Global Enable                                            PGE   [Present]
|- Page Size Extension                                           PSE   [Present]
|- 36-bit Page Size Extension                                  PSE36   [Present]
|- Processor Serial Number                                       PSN   [Missing]
|- Restricted Transactional Memory                               RTM   [Present]
|- Safer Mode Extensions                                         SMX   [Present]
|- Self-Snoop                                                     SS   [Present]
|- Time Stamp Counter                                            TSC [Invariant]
|- Time Stamp Counter Deadline                          TSC-DEADLINE   [Present]
|- Virtual Mode Extension                                        VME   [Present]
|- Virtual Machine Extensions                                    VMX   [Present]
|- Extended xAPIC Support                                     x2APIC   [  xAPIC]
|- Execution Disable Bit Support                              XD-Bit   [Present]
|- XSAVE/XSTOR States                                          XSAVE   [Present]
|- xTPR Update Control                                          xTPR   [Present]
                                                                                
Technologies:                                                                   
|- Hyper-Threading                                               HTT       [ ON]
|- SpeedStep                                                    EIST       [ ON]
|- PowerNow!                                                PowerNow       [OFF]
|- Dynamic Acceleration                                          IDA       [ ON]
|- Turbo Boost                                             TURBO|CPB       [ ON]
                                                                                
Performance Monitoring:                                                         
|- Version                                                        PM       [  4]
|- Counters:          General                   Fixed                           
|                     4 x 48 bits             3 x 48 bits                       
|- Enhanced Halt State                                           C1E       [ ON]
|- C1 Auto Demotion                                              C1A       [ ON]
|- C3 Auto Demotion                                              C3A       [ ON]
|- C1 UnDemotion                                                 C1U       [ ON]
|- C3 UnDemotion                                                 C3U       [ ON]
|- Frequency ID control                                          FID       [OFF]
|- Voltage ID control                                            VID       [OFF]
|- P-State Hardware Coordination Feedback                MPERF/APERF       [ ON]
|- Hardware-Controlled Performance States                        HWP       [ ON]
|- Hardware Duty Cycling                                         HDC       [ ON]
|- MWAIT States:      C0      C1      C2      C3      C4                        
|                      0       2       1       2       4                        
|- Core Cycles                                                         [Present]
|- Instructions Retired                                                [Present]
|- Reference Cycles                                                    [Present]
|- Last Level Cache References                                         [Present]
|- Last Level Cache Misses                                             [Present]
|- Branch Instructions Retired                                         [Present]
|- Branch Mispredicts Retired                                          [Present]
                                                                                
Power & Thermal Monitoring:                                                     
|- Digital Thermal Sensor                                        DTS   [Present]
|- Power Limit Notification                                      PLN   [Present]
|- Package Thermal Management                                    PTM   [Present]
|- Thermal Monitor 1                                         TM1|TTP   [ Enable]
|- Thermal Monitor 2                                         TM2|HTC   [Present]
|- Clock Modulation                                             ODCM   [ 12.50%]
|- Energy Policy                                           Bias Hint   [      6]
```

If this can help fixing the cpu database of the software.

Thanks

cyring, 2017-05-03:
Thanks for your return.
Accordingly to the processor specs, values are indeed wrong.  CoreFreq should count 2 physical and 4 logical cores.
No cpu database to fix but the count algorithm, which is based on the cpuid instruction, has to be debug/improve.

MyPod, 2017-08-18:
Although in a different generation, I have a similar issue with a mismatched core count. My dual socket X5670 system only reports 16 cores and not the expected 24.

`|- Online CPU                                                            [16/16]`

I can provide more information/open a separate issue/test as needed.

cyring, 2017-08-18:
I'm working on a subset project to focus on topology @ http://github.com/cyring/topology
I suggest you switch to it for any issue.

Later when topology will be ok with DP, MP processors, I will back port code into CoreFreq

Regards

cyring, 2017-09-03:
Fixed with the OS cpu count.

closed 2017-09-23 by cyring

cyring, 2018-08-18:
Hello,

Could you please try the last changes ?
1. start the driver in Experimental mode
`insmod corefreqk.ko Experimental=1`
2. post back the results of:
`corefreq-cli -s` 
`corefreq-cli -m`
`corefreq-cli -M`
using low and high system load, 3 turns of `corefreq-cli -g` and `corefreq-cli -V` 

Thank you