cyring/CoreFreq #26: cannot compile with gcc 4.8.5

lwiechec, 2017-05-08:
Hi, I am trying to build CoreFreq but I am getting this:

```
⏚ [luke:~/fun/CoreFreq]↥ master ± make
cc -Wall -pthread -c corefreqd.c -o corefreqd.o
cc corefreqd.c -o corefreqd -lpthread -lrt
cc -Wall -c corefreq-cli.c -o corefreq-cli.o
cc corefreq-cli.c -o corefreq-cli -lm -lrt
make -C /lib/modules/4.4.8-040408-generic/build M=/home/luke/fun/CoreFreq modules
make[1]: Entering directory `/usr/src/linux-headers-4.4.8-040408-generic'
Makefile:660: Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler
  CC [M]  /home/luke/fun/CoreFreq/corefreqk.o
gcc: error: unrecognized command line option ‘-fstack-protector-strong’
make[2]: *** [/home/luke/fun/CoreFreq/corefreqk.o] Error 1
make[1]: *** [_module_/home/luke/fun/CoreFreq] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-4.4.8-040408-generic'
make: *** [all] Error 2
```

The full version of `gcc` is:

```
gcc (Ubuntu 4.8.5-2ubuntu1~14.04.1) 4.8.5
Copyright (C) 2015 Free Software Foundation, Inc
```

Do I need to upgrade my `gcc` ?...

cyring, 2017-05-08:
I believe you have to. Mine is currently 6.3.1
Any version 6.x should do the job

lwiechec, 2017-05-08:
hi again, thanks for response. I have installed `gcc` 6.2.0 but now the compilation fails:

```
⏚ [luke:~/fun/CoreFreq] master(+5/-4)* ± make CC=/usr/bin/gcc-6
/usr/bin/gcc-6 -Wall -pthread -c corefreqd.c -o corefreqd.o
/usr/bin/gcc-6 corefreqd.c -o corefreqd -lpthread -lrt
/usr/bin/gcc-6 -Wall -c corefreq-cli.c -o corefreq-cli.o
/usr/bin/gcc-6 corefreq-cli.c -o corefreq-cli -lm -lrt
make -C /lib/modules/4.11.0-041100-generic/build M=/home/luke/fun/CoreFreq modules
make[1]: Entering directory `/usr/src/linux-headers-4.11.0-041100-generic'
  CC [M]  /home/luke/fun/CoreFreq/corefreqk.o
In file included from /home/luke/fun/CoreFreq/corefreqk.c:26:0:
/home/luke/fun/CoreFreq/corefreqk.c: In function ‘PerCore_Nehalem_Query’:
/home/luke/fun/CoreFreq/corefreqk.c:1930:23: error: ‘MSR_NHM_SNB_PKG_CST_CFG_CTL’ undeclared (first use in this function)
   RDMSR(CStateConfig, MSR_NHM_SNB_PKG_CST_CFG_CTL);
                       ^
/home/luke/fun/CoreFreq/corefreqk.h:48:10: note: in definition of macro ‘RDMSR’
   : "c" (_reg)      \
          ^~~~
/home/luke/fun/CoreFreq/corefreqk.c:1930:23: note: each undeclared identifier is reported only once for each function it appears in
   RDMSR(CStateConfig, MSR_NHM_SNB_PKG_CST_CFG_CTL);
                       ^
/home/luke/fun/CoreFreq/corefreqk.h:48:10: note: in definition of macro ‘RDMSR’
   : "c" (_reg)      \
          ^~~~
/home/luke/fun/CoreFreq/corefreqk.c: In function ‘PerCore_SandyBridge_Query’:
/home/luke/fun/CoreFreq/corefreqk.c:1951:23: error: ‘MSR_NHM_SNB_PKG_CST_CFG_CTL’ undeclared (first use in this function)
   RDMSR(CStateConfig, MSR_NHM_SNB_PKG_CST_CFG_CTL);
                       ^
/home/luke/fun/CoreFreq/corefreqk.h:48:10: note: in definition of macro ‘RDMSR’
   : "c" (_reg)      \
          ^~~~
/home/luke/fun/CoreFreq/corefreqk.c: In function ‘Sys_DumpTask’:
/home/luke/fun/CoreFreq/corefreqk.c:2823:2: error: implicit declaration of function ‘for_each_process_thread’ [-Werror=implicit-function-declaration]
  for_each_process_thread(process, thread) {
  ^~~~~~~~~~~~~~~~~~~~~~~
/home/luke/fun/CoreFreq/corefreqk.c:2823:43: error: expected ‘;’ before ‘{’ token
  for_each_process_thread(process, thread) {
                                           ^
cc1: some warnings being treated as errors
make[2]: *** [/home/luke/fun/CoreFreq/corefreqk.o] Error 1
make[1]: *** [_module_/home/luke/fun/CoreFreq] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-4.11.0-041100-generic'
make: *** [all] Error 2
```

(I modified `Makefile` to be able to use alternative compiler).

Sorry for all (possibly stupid) questions, but I am a bit of n00b when it comes to C compilation...

cyring, 2017-05-08:
Not at all, I guess you put fingers on something which we'll give me a hard time !

MSR_NHM_SNB_PKG_CST_CFG_CTL is defined with value  0x000000e2 in  arch/x86/include/asm/msr-index.h in kernel source tree up to version 4.10.14

But the header file seems to have changed since versions 4.11

I suggest you to downgrade to 4.10.x or wait I can fix things ...

lwiechec, 2017-05-08:
ok, great - thanks for the analysis :) I will try to downgrade - or just wait... thanks for CoreFreq!

---- On Seg, 08 mai 2017 15:35:16 +0200 CYRIL INGENIERIE &lt;notifications@github.com&gt; wrote ----

Not at all, I guess you put fingers on something which we'll give me a hard time !

MSR_NHM_SNB_PKG_CST_CFG_CTL is defined with value 0x000000e2 in arch/x86/include/asm/msr-index.h in kernel source tree up to version 4.10.14

But the header file seems to have changed since versions 4.11

I suggest you to downgrade to 4.10.x or wait I can fix things ...

cyring, 2017-05-08:
Kernel version 11 is not released yet for Arch Linux, I will wait for it to measure impacts.

lwiechec, 2017-05-09:
Understood - thanks!

---- On Ter, 09 mai 2017 01:28:10 +0200 CYRIL INGENIERIE &lt;notifications@github.com&gt; wrote ----

Kernel version 11 is not released yet for Arch Linux, I will wait for it to measure impacts.

netmonk, 2017-05-11:
It reminds me the issue we had when we tryed to compile on production server with rhel 6.5 :) Upgrade gcc, seems to be the solution.

cyring, 2017-05-11:
Worst ! Upgrading to last version 11 kernel reveals 'define' changes in included headers.

cyring, 2017-05-13:
Here's the wall story why this msr has been renamed
http://lkml.iu.edu/hypermail/linux/kernel/1603.2/05317.html

cyring, 2017-05-25:
lwiechec : a commit is pushed to let CoreFreq be compatible with kernel version 4.11

Is the issue solved, please ?

cyring, 2017-05-28:
Is the fix ok for you now ?
Can you close this issue ?

cyring, 2017-06-02:
CoreFreq is now compiling with kernel 4.11

closed 2017-06-02 by cyring

netmonk, 2017-06-02:
Good news!!!