cyring/CoreFreq #28: Deepin 15.4 freezes out when installing CoreFreq on it

lothkang, 2017-06-04:
Downloading & compiling works fine, but `sudo insmod corefreqk.ko` freezes out Deepin. Tried with and without disabling NMI watchdog.
Kernel version: 4.9.8-5 x86_64
CPU: AMD A6 6310 APU

cyring, 2017-06-04:
Never heard about Deepin until your issue. However it appears Debian based.

Whatever, you are running CoreFreq with an untested processor.

Only series 0F have been implemented. Series above 0F, such as your AMD A6 6310, are not coded for the sole reason I don't have those ressources.

Spécifications available however within http://developer.amd.com/resources/developer-guides-manuals I just need some AMD processors to work with.

Cyri

lothkang, 2017-06-06:
Never heard about Deepin? Probably you don't visit DistroWatch that often (Deepin is the 6th on the list).
Anyway thanks for the fast answer. What about [Intel(R) Pentium(R) 4 CPU](http://paste.ubuntu.com/24794549/)? Is it supported or it's just too old?

cyring, 2017-06-07:
The target is Core generation and superior 64 bits architectures.

Pentium 4 may run using the ArchID argument when loading the corefreqk.ko module.

Try:
insmod corefreqk.ko ArchID=1

This will instruct to use generic instructions algorithm.
If ok, architecture name will report "Core/Yonah".

Many others IDs are listed at the bottom of corefreqk.h : any "_GenuineIntel" algorithms are the ones to target for unimplemented processors.

Warning: those are low level asm instructions run into the kernel space; thus save, close, sync all your files before inserting corefreqk.ko because it will probably crash processor.

Please, let me know about the result. If successful, I may add code for Pentium 4 64 bits.

Regards,
CyrIng

lothkang, 2017-06-07:
I tried to compile it, and it failed. And since x86 is not supported, i give up. Thanks again for the fast answer. I uploaded the output of make to [here](https://paste.ubuntu.com/24802128/).

cyring, 2017-06-08:
I have searched for long mode for your P4 ; it does not appear 64 bits capable.
Issue can be closed.

lothkang, 2017-06-08:
Of course it's not 64 bits capable. The computer has been made somewhere around 2002. Even the latest BIOS version for it is 11 years old.

closed 2017-06-08 by lothkang