cyring/CoreFreq #44: CLI segfaults on AMD EPYC when stressing system

Saruspete, 2018-01-03:
Hello Cyril,

When testing on an AMD 7551 EPYC 2x32 / Signature 8F_01, frequency monitoring is crashing when one of the CPU is under stress and monitoring frequency/cycles/cstate.... Works correctly under dashboard, and shows value of ~2492.

Processor Window shows:
  * Arch: 17h
  * base clock: 0.83
  * Frequency max: 1996.26MHz /  2400 ratio
  * Factory: 2400000 MHz / 2400 Ratio
  * 1C TurboBoost: 2079.44 / 2500 Ratio
  * 2C TurboBoost: 1996.26 / 2400 Ratio
 
Segfault comes from memcpy, from define "LayerFillAt()" called by Draw_Load:4625.
When commenting this LayerFillAt and adding stress, the bar goes beyond the end of line, thus back to newline, and displays freq of 2500 & Turbo 124%.

Seems because the max value of turbo is the 2400 advertised instead of the 2500 really used. 

Any idea ?

cyring, 2018-01-03:
Hello Adrien,

Thank you very much for this report.

Based on [processor features](http://www.amd.com/en/products/cpu/amd-epyc-7551) I have ideas.
It looks like P-States have been successfully queried but :
* multiplied by 100
* unordered ratios (non turbo & with turbo)
* BCLK seems odd

When those will be fixed in the driver, the UI should just run correctly.

cyring, 2018-02-04:
Hello Adrien,
Can you try last version  ?  it should fixe this SegFault. 
Thanks

cyring, 2018-05-11:
Delayed

closed 2018-05-11 by cyring