cyring/CoreFreq #51: i have same problem too

nikanonline, 2018-04-28:
hi i have same problem too what should i do?

[root@s129644 CoreFreq]# make
make -C /lib/modules/3.10.0-693.21.1.el7.x86_64/build M=/root/CoreFreq modules
make: *** /lib/modules/3.10.0-693.21.1.el7.x86_64/build: No such file or directory.  Stop.
make: *** [all] Error 2
[root@s129644 CoreFreq]#

nikanonline, 2018-05-03:
hi , it's not fixed
[root@s129644 ~]# yum group install 'Development Tools'
Loaded plugins: fastestmirror
Loading mirror speeds from cached hostfile
 * base: repos.forethought.net
 * epel: mirror.colorado.edu
 * extras: mirror.fdcservers.net
 * updates: mirror.den1.denvercolo.net
Maybe run: yum groups mark install (see man yum)
No packages in any requested group available to install or update
[root@s129644 ~]#

Already install the development package

[root@s129644 CoreFreq]# make make -C /lib/modules/3.10.0-693.21.1.el7.x86_64/build M=/root/CoreFreq modules make: *** /lib/modules/3.10.0-693.21.1.el7.x86_64/build: No such file or directory.  Stop. make: *** [all] Error 2 
[root@s129644 CoreFreq]#
--

centos 7 64 bit last version and updated.

cyring, 2018-05-03:
/lib/modules/3.10.0-693.21.1.el7.x86_64/build: No such file or directory.

Apparently kernel development package is missing in your FS

Please see DKMS for CentRO

nikanonline, 2018-05-04:
i was installed dkms with yum install -y dkms

please tell me step by step how i can install Corefreq?
i just installed original centos 7 x 64  from centos.org
i'm not professional like you ,so give me complete help to get Bitcoin reward as donation!
i want use corefreq to boost my server cpu for mining.
then if you don't put complete installation tutorial for users, the result is , they can not install corofreq and gone , maybe try an other modules.... so  all of your profit and retry will be lost. because they will be leave your Corefreq alone.

so put complete installation guide for both beginner and linux professional guys.

Have fun