cyring/CoreFreq #52: Errors

JanRadaa, 2018-05-02:
Hi guys 

Can you somone help mi with this ?

corefreq-cli.c:6430: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6430: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6430: warning: excess elements in union initializer
corefreq-cli.c:6430: warning: (near initialization for ‘hUncore.attr[8]’)
corefreq-cli.c:6430: error: unknown field ‘bf’ specified in initializer
corefreq-cli.c:6430: warning: excess elements in union initializer
corefreq-cli.c:6430: warning: (near initialization for ‘hUncore.attr[8]’)
corefreq-cli.c:6430: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6430: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6430: warning: excess elements in union initializer
corefreq-cli.c:6430: warning: (near initialization for ‘hUncore.attr[9]’)
corefreq-cli.c:6430: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6430: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6430: warning: excess elements in union initializer
corefreq-cli.c:6430: warning: (near initialization for ‘hUncore.attr[10]’)
corefreq-cli.c:6430: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6430: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6430: warning: excess elements in union initializer
corefreq-cli.c:6430: warning: (near initialization for ‘hUncore.attr[11]’)
corefreq-cli.c:6430: error: unknown field ‘bf’ specified in initializer
corefreq-cli.c:6430: warning: excess elements in union initializer
corefreq-cli.c:6430: warning: (near initialization for ‘hUncore.attr[11]’)
corefreq-cli.c: In function ‘Layout_Card_Bus’:
corefreq-cli.c:6451: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6451: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6451: warning: excess elements in union initializer
corefreq-cli.c:6451: warning: (near initialization for ‘hBus.attr[0]’)
corefreq-cli.c:6451: error: unknown field ‘bf’ specified in initializer
corefreq-cli.c:6451: warning: excess elements in union initializer
corefreq-cli.c:6451: warning: (near initialization for ‘hBus.attr[0]’)
corefreq-cli.c:6451: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6451: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6451: warning: excess elements in union initializer
corefreq-cli.c:6451: warning: (near initialization for ‘hBus.attr[1]’)
corefreq-cli.c:6451: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6451: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6451: warning: excess elements in union initializer
corefreq-cli.c:6451: warning: (near initialization for ‘hBus.attr[2]’)
corefreq-cli.c:6451: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6451: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6451: warning: excess elements in union initializer
corefreq-cli.c:6451: warning: (near initialization for ‘hBus.attr[3]’)
corefreq-cli.c:6451: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6451: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6451: warning: excess elements in union initializer
corefreq-cli.c:6451: warning: (near initialization for ‘hBus.attr[4]’)
corefreq-cli.c:6451: error: unknown field ‘bf’ specified in initializer
corefreq-cli.c:6451: warning: excess elements in union initializer
corefreq-cli.c:6451: warning: (near initialization for ‘hBus.attr[4]’)
corefreq-cli.c:6451: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6451: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6451: warning: excess elements in union initializer
corefreq-cli.c:6451: warning: (near initialization for ‘hBus.attr[5]’)
corefreq-cli.c:6451: error: unknown field ‘bf’ specified in initializer
corefreq-cli.c:6451: warning: excess elements in union initializer
corefreq-cli.c:6451: warning: (near initialization for ‘hBus.attr[5]’)
corefreq-cli.c:6451: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6451: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6451: warning: excess elements in union initializer
corefreq-cli.c:6451: warning: (near initialization for ‘hBus.attr[6]’)
corefreq-cli.c:6451: error: unknown field ‘bf’ specified in initializer
corefreq-cli.c:6451: warning: excess elements in union initializer
corefreq-cli.c:6451: warning: (near initialization for ‘hBus.attr[6]’)
corefreq-cli.c:6451: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6451: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6451: warning: excess elements in union initializer
[... 416 lines cut ...]
corefreq-cli.c:6665: warning: (near initialization for ‘_attr’)
corefreq-cli.c: In function ‘Draw_Card_Task’:
corefreq-cli.c:6789: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6789: error: unknown field ‘un’ specified in initializer
corefreq-cli.c:6789: warning: excess elements in union initializer
corefreq-cli.c:6789: warning: (near initialization for ‘_attr’)
corefreq-cli.c:6789: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6789: warning: excess elements in union initializer
corefreq-cli.c:6789: warning: (near initialization for ‘_attr’)
corefreq-cli.c:6789: error: unknown field ‘bf’ specified in initializer
corefreq-cli.c:6789: warning: excess elements in union initializer
corefreq-cli.c:6789: warning: (near initialization for ‘_attr’)
corefreq-cli.c: In function ‘Top’:
corefreq-cli.c:6990: error: unknown field ‘fg’ specified in initializer
corefreq-cli.c:6990: error: unknown field ‘un’ specified in initializer
corefreq-cli.c:6990: warning: excess elements in union initializer
corefreq-cli.c:6990: warning: (near initialization for ‘_attr’)
corefreq-cli.c:6990: error: unknown field ‘bg’ specified in initializer
corefreq-cli.c:6990: warning: excess elements in union initializer
corefreq-cli.c:6990: warning: (near initialization for ‘_attr’)
corefreq-cli.c:6990: error: unknown field ‘bf’ specified in initializer
corefreq-cli.c:6990: warning: excess elements in union initializer
corefreq-cli.c:6990: warning: (near initialization for ‘_attr’)

after i run " make " this is happend

cyring, 2018-05-02:
Warmly recommend to update your libc and gcc

Regards
CyrIng

cyring, 2018-05-08:
Hello,

Did you solve the problem after updating your system with an up to date development environment ?

such as:
```
Linux 4.16.7
dkms 2.5-1
make 4.2.1
gcc 8.1.0
as  2.30
glibc 2.27-3
```

closed 2018-05-11 by cyring