cyring/CoreFreq #54: No temperature report on Ryzen 2700X

adatum, 2018-05-12:
First I'd like to say I'm really impressed by this software. I've been looking for a lightweight program to monitor frequency (and hopefully temperature, vcore, etc) while I set up a new system, and CoreFreq works well and has a *ton* of features. Plus, I like that it runs in the terminal.

Now for the issue: there is currently no reporting of temperature, Vcore, or power on the Ryzen 2700X as can be seen in the following screenshots.

[image]

[image]

The 2700X is a new CPU so it is understandable. If you need info or want me to try something, just let me know. 

Even in the current state, CoreFreq is useful to me. Again, really impressed. It deserves more publicity!

cyring, 2018-05-12:
Hello,

I appreciate it.
Could you please return the results of command ```lspci -nn```
I'm expecting to read one PCI of device or function of value 0x60 to query thermal sensor.

adatum, 2018-05-12:
```
$ lspci -nn
00:00.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Root Complex [1022:1450]
00:00.2 IOMMU [0806]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) I/O Memory Management Unit [1022:1451]
00:01.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
00:01.3 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge [1022:1453]
00:02.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
00:03.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
00:03.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge [1022:1453]
00:04.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
00:07.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
00:07.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Internal PCIe GPP Bridge 0 to Bus B [1022:1454]
00:08.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge [1022:1452]
00:08.1 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Internal PCIe GPP Bridge 0 to Bus B [1022:1454]
00:14.0 SMBus [0c05]: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller [1022:790b] (rev 59)
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge [1022:790e] (rev 51)
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 0 [1022:1460]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 1 [1022:1461]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 2 [1022:1462]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 3 [1022:1463]
00:18.4 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 4 [1022:1464]
00:18.5 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 5 [1022:1465]
00:18.6 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 6 [1022:1466]
00:18.7 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 7 [1022:1467]
01:00.0 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] Device [1022:43d0] (rev 01)
01:00.1 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] Device [1022:43c8] (rev 01)
01:00.2 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43c6] (rev 01)
02:00.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43c7] (rev 01)
02:01.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43c7] (rev 01)
02:02.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43c7] (rev 01)
02:03.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43c7] (rev 01)
02:04.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43c7] (rev 01)
02:09.0 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD] Device [1022:43c7] (rev 01)
05:00.0 Ethernet controller [0200]: Intel Corporation I211 Gigabit Network Connection [8086:1539] (rev 03)
06:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. RTL8822BE 802.11a/b/g/n/ac WiFi adapter [10ec:b822]
08:00.0 USB controller [0c03]: ASMedia Technology Inc. Device [1b21:2142]
09:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP108 [GeForce GT 1030] [10de:1d01] (rev a1)
09:00.1 Audio device [0403]: NVIDIA Corporation GP108 High Definition Audio Controller [10de:0fb8] (rev a1)
0a:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Device [1022:145a]
0a:00.2 Encryption controller [1080]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Platform Security Processor [1022:1456]
0a:00.3 USB controller [0c03]: Advanced Micro Devices, Inc. [AMD] USB 3.0 Host controller [1022:145f]
0b:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Device [1022:1455]
0b:00.2 SATA controller [0106]: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] [1022:7901] (rev 51)
0b:00.3 Audio device [0403]: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) HD Audio Controller [1022:1457]

```

cyring, 2018-05-17:
Using last commit, you will need to add the Experimental argument when loading the driver.
```insmod corefreqk.ko Experimental=1```
Temperature is a raw value without decimal adjustment.
May crash, save you files.
CyrIng

cyring, 2018-05-19:
Thanks for these returns.

*/ p-state ratios enumeration differs from previous screenshot. Max was 41 and now 30 => reason of segfault.
Max ratio can be obtained during driver startup (insmod corefreqk.ko) if full load is applied simultaneously or governor set to performance.

*/ temp is so far a raw debug value. I need now to apply offsets.
Do you have any Celsius reference (BIOS or other value) I can compare with ? (sampled at the same period)

adatum, 2018-05-19:
You're welcome. Thanks for the quick development.

Does `insmod corefreqk.ko` have to be done at the time of max expected ratio? Is a fix or workaround possible/likely? Max ratio can go above 41, depending on the number of cores stressed, boosting algorithms, and potentially manual overclocking. 

There is a CPU temp readout from `sensors`/`psensor` (`lm-sensors` doesn't find any sensors):
```
$ sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

nouveau-pci-0900
Adapter: PCI adapter
temp1:        +44.0°C  (high = +95.0°C, hyst =  +3.0°C)
                       (crit = +105.0°C, hyst =  +5.0°C)
                       (emerg = +135.0°C, hyst =  +5.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +52.0°C  (high = +70.0°C)

```

It's certainly not perfect (no fan rpm) but it's the only way I have found so far to get a CPU temp reading in linux. At least kernel 4.16 fixed the offset (previously off by 49C or 59C).

cyring, 2018-05-19:
I commit a change of the sensor register.
Can you please post the value.

Remark: it is a debug value, not meaningful yet.

cyring, 2018-05-19:
Ok, let's try now with the index register commit

adatum, 2018-05-19:
Now the TMP value fluctuates. Max/Min are fixed. Sometimes the TMP value reads 0. 
The digits of the values have different colors depending on the column position, possibly for denoting decimals? Some shifting and truncation can change meanings between eg. 10(27), 102(77) in the images below:

[image]
[image]

cyring, 2018-05-20:
The purpose of this new change is to:
- bind temperature to service processor b/c according to specs, sensor is common to all cores
- adjust the current temperature with the formula: (sensor x 5 / 40) - 49

adatum, 2018-05-20:
There is a reading for only 1 core now. Max seems to behave as expected. Min is still 0. TMP  seems to be off by 10-11C compared to `psensor`. Is there anything you would like me to check specifically?

[image]
[image]

Maybe these links have useful info? 

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=1b59788979acd230b9627276c76f6e6ba2c4709c

https://www.phoronix.com/scan.php?page=news_item&px=Linux-4.16.6-Released

cyring, 2018-05-21:
Thermal adjustment added.

adatum, 2018-05-21:
Now TMP matches `sensors`/`psensor`and Min/Max behave as expected. Reporting is still for one core only.
[image]

I started using the [it87](https://github.com/groeck/it87) kernel driver for sensors and it looks like CoreFreq is matching temp1 rather than k10temp. Is there a way to check?

(Note: output below not taken at same time as screenshot above)
```
$ sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

nouveau-pci-0900
Adapter: PCI adapter
temp1:        +42.0°C  (high = +95.0°C, hyst =  +3.0°C)
                       (crit = +105.0°C, hyst =  +5.0°C)
                       (emerg = +135.0°C, hyst =  +5.0°C)

it8665-isa-0290
Adapter: ISA adapter
in0:          +1.49 V  (min =  +2.75 V, max =  +2.70 V)
in1:          +1.34 V  (min =  +1.92 V, max =  +0.97 V)
in2:          +2.39 V  (min =  +1.34 V, max =  +2.70 V)
in3:          +2.00 V  (min =  +0.68 V, max =  +0.87 V)
in4:          +1.12 V  (min =  +1.20 V, max =  +2.26 V)
in5:          +0.55 V  (min =  +2.23 V, max =  +2.26 V)
in6:          +0.91 V  (min =  +1.68 V, max =  +1.01 V)
3VSB:         +3.33 V  (min =  +5.49 V, max =  +5.47 V)
Vbat:         +3.25 V  
+3.3V:        +3.33 V  
fan1:         313 RPM  (min =   13 RPM)
fan5:           0 RPM  (min =   -1 RPM)  ALARM
temp1:        +44.0°C  (low  = +126.0°C, high = +119.0°C)
temp2:        +28.0°C  (low  = -12.0°C, high = +80.0°C)  sensor = thermistor
temp3:        +41.0°C  (low  = +48.0°C, high = -60.0°C)  sensor = thermistor
intrusion0:  ALARM

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +52.2°C  (high = +70.0°C)
```

cyring, 2018-05-21:
Nice to see we have the expected results.
CoreFreq employs the same register than k10temp.

2 sensors exist in this AMD PCI space : die and hardware temperatures.

As states previously, the sensor is unique for all cores. 
In your screenshot, the temperature is queried by the cpu #6 which is the current CoreFreq driver service cpu ;  I called it the service processor on which some driver handler, daemon threads and UI are bound to, for atomic synchronization reasons.
The service processor may change at startup or if the associated cpu is disable, but in any case the temperature query will migrate to the new service cpu number.

With Intel we found one sensor per physical core.

cyring, 2018-05-21:
As a roadmap, there are many other enhancements to do : 
- State of Turbo, C1E, Throttling and other features 
- Enable or disable Core performance boost 
- Add the XFR frequency ratio 
- Debug the SMT topology 
- Debug the cache level 3 size 
- Read the C-states counters 
- Enable, disable C-States (C6 seems feasible) 
- Compute the instructions IPC 
- Read the energy RAPL registers 
- Query the Uncore, Bus, DDR4 geometry

adatum, 2018-05-21:
I read somewhere that there are many CPU temperature sensors, but perhaps they are not exposed externally.

That roadmap is fantastic.

Would it be feasible to add:
- accurate idle ratio/frequency
- avoid segfault if max ratio changes
- bus frequency (including possibly asynchronous bus)
- voltages (Vcore, SOC, etc)
- memory frequency/timings

Of course we can make separate issues for them.

adatum, 2018-05-21:
It's easier to see the temperature difference between temp1 (dark green) and k10temp (orange) when stressing a single core:

[image]

I wanted to check which reading CoreFreq corresponds to, but it segfaults when trying to launch with single core stress tests. It launches fine when all cores are stressed.

cyring, 2018-05-21:
Seg faults happens in the UI due to a buffer overflow: the histogram is going far away from the max known ratio (41) --> XFR ratio is missing.

Can you run and post result of ```corefreq-cli -s```

You will notice that the highest ratio is 41 but this processor is able of 43 : the XFR frequency reached with a single core, I guess.

The remaing task is then to find the right register to query such ratio for all Ryzen series.

adatum, 2018-05-21:
Yes, on single core it can reach 4.35GHz (or more with watercooling). Though I'm not sure of the multiplier/ratio and base/bus clock combination for that. Some OC modes on some motherboards can change the bus clock from the default 100MHz to a bit more (or less), so that's another variable.

```
$ ./corefreq-cli -s
Processor                              [AMD Ryzen 7 2700X Eight-Core Processor ]
|- Architecture                                                     [Family 17h]
|- Vendor ID                                                      [AuthenticAMD]
|- Signature                                                            [ 8F_08]
|- Stepping                                                             [     2]
|- Microcode                                                            [     0]
|- Online CPU                                                           [ 16/16]
|- Base Clock                                                           [ 99.80]
|- Ratio Limited                                                        [  LOCK]
|- Frequency            (Mhz)                      Ratio                        
                 Min    399.21                    [   4 ]                       
                 Max   3692.71                    [  37 ]                       
|- Factory                                                                      
                       3700                       [  37 ]                       
|- Turbo Boost                                                                  
                  1C   2095.86                    [  21 ]                       
                  2C   3193.70                    [  32 ]                       
                  3C   2195.67                    [  22 ]                       
|- Uncore                                                                       
                                                                                
ISA Extensions:                                                                 
|- 3DNow!/Ext [N,N]        AES [Y]   AVX/AVX2 [Y/Y]   BMI1/BMI2 [Y/Y] 
|- CLFSH        [Y]       CMOV [Y]      CMPXCH8 [Y]      CMPXCH16 [Y] 
|- F16C         [Y]        FPU [Y]         FXSR [Y]     LAHF/SAHF [Y] 
|- MMX/Ext    [Y/Y]    MONITOR [Y]        MOVBE [Y]      PCLMULDQ [Y] 
|- POPCNT       [Y]     RDRAND [Y]       RDTSCP [Y]           SEP [Y] 
|- SGX          [N]        SSE [Y]         SSE2 [Y]          SSE3 [Y] 
|- SSSE3        [Y] SSE4.1/4A [Y/Y]      SSE4.2 [Y]       SYSCALL [Y] 
                                                                                
Features:                                                                       
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Present]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
|- Pending Break Enable                                          PBE   [Missing]
|- Process Context Identifiers                                  PCID   [Missing]
|- Perfmon and Debug Capability                                 PDCM   [Missing]
|- Page Global Enable                                            PGE   [Present]
|- Page Size Extension                                           PSE   [Present]
|- 36-bit Page Size Extension                                  PSE36   [Present]
|- Processor Serial Number                                       PSN   [Missing]
|- Restricted Transactional Memory                               RTM   [Missing]
|- Safer Mode Extensions                                         SMX   [Missing]
|- Self-Snoop                                                     SS   [Missing]
|- Time Stamp Counter                                            TSC [Invariant]
|- Time Stamp Counter Deadline                          TSC-DEADLINE   [Missing]
[... cut ...]
|- Package Thermal Management                                    PTM   [Missing]
|- Thermal Monitor 1                                         TM1|TTP   [Present]
|- Thermal Monitor 2                                         TM2|HTC   [Present]
|- Units                                                                        
   |- Power                                               watt   [  0.000000000]
   |- Energy                                             joule   [  0.000000000]
   |- Window                                            second   [  0.000000000]
```

cyring, 2018-05-21:
Looking at these ratios, I believe system was idleing when starting the driver.

Can load the kernel module ```corefreqk.ko``` in the following two cases and print ```corefreq-cli -s``` each time:
1- when all cores are stressed
2- when a single is core stressed

adatum, 2018-05-21:
Single core stressed (at least one core ~4.35GHz):
```
$ ./corefreq-cli -s
Processor                              [AMD Ryzen 7 2700X Eight-Core Processor ]
|- Architecture                                                     [Family 17h]
|- Vendor ID                                                      [AuthenticAMD]
|- Signature                                                            [ 8F_08]
|- Stepping                                                             [     2]
|- Microcode                                                            [     0]
|- Online CPU                                                           [ 16/16]
|- Base Clock                                                           [ 99.80]
|- Ratio Limited                                                        [  LOCK]
|- Frequency            (Mhz)                      Ratio                        
                 Min    399.21                    [   4 ]                       
                 Max   3692.71                    [  37 ]                       
|- Factory                                                                      
                       3700                       [  37 ]                       
|- Turbo Boost                                                                  
                  1C   2195.67                    [  22 ]                       
                  2C   3193.70                    [  32 ]                       
                  3C   2195.67                    [  22 ]                       
|- Uncore                                                                       
                                                                                
ISA Extensions:                                                                 
|- 3DNow!/Ext [N,N]        AES [Y]   AVX/AVX2 [Y/Y]   BMI1/BMI2 [Y/Y] 
|- CLFSH        [Y]       CMOV [Y]      CMPXCH8 [Y]      CMPXCH16 [Y] 
|- F16C         [Y]        FPU [Y]         FXSR [Y]     LAHF/SAHF [Y] 
|- MMX/Ext    [Y/Y]    MONITOR [Y]        MOVBE [Y]      PCLMULDQ [Y] 
|- POPCNT       [Y]     RDRAND [Y]       RDTSCP [Y]           SEP [Y] 
|- SGX          [N]        SSE [Y]         SSE2 [Y]          SSE3 [Y] 
|- SSSE3        [Y] SSE4.1/4A [Y/Y]      SSE4.2 [Y]       SYSCALL [Y] 
                                                                                
Features:                                                                       
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Present]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
|- Pending Break Enable                                          PBE   [Missing]
|- Process Context Identifiers                                  PCID   [Missing]
|- Perfmon and Debug Capability                                 PDCM   [Missing]
|- Page Global Enable                                            PGE   [Present]
|- Page Size Extension                                           PSE   [Present]
|- 36-bit Page Size Extension                                  PSE36   [Present]
|- Processor Serial Number                                       PSN   [Missing]
|- Restricted Transactional Memory                               RTM   [Missing]
|- Safer Mode Extensions                                         SMX   [Missing]
|- Self-Snoop                                                     SS   [Missing]
|- Time Stamp Counter                                            TSC [Invariant]
|- Time Stamp Counter Deadline                          TSC-DEADLINE   [Missing]
|- Virtual Mode Extension                                        VME   [Present]
|- Virtual Machine Extensions                                    VMX   [Missing]
|- Extended xAPIC Support                                     x2APIC   [Missing]
[... 161 lines cut ...]
[... cut ...]
|- Package Thermal Management                                    PTM   [Missing]
|- Thermal Monitor 1                                         TM1|TTP   [Present]
|- Thermal Monitor 2                                         TM2|HTC   [Present]
|- Units                                                                        
   |- Power                                               watt   [  0.000000000]
   |- Energy                                             joule   [  0.000000000]
   |- Window                                            second   [  0.000000000]
```

cyring, 2018-05-22:
Could you overclock the processor in the BIOS ?
And write down the selected coefficient of frequency.
Next run the two high load cases: single then multi-cores.

I would like to verify how ratios are reported by _CoreFreq_
It may help to discover which P-States registers are impacted.

adatum, 2018-05-22:
I believe manual OC disables XFR, Precision Boost2, etc. Is that still useful to you?

adatum, 2018-05-22:
Ok. Manual OC to 4.0GHz with 100MHz base clock and 40X ratio.

Single core stressed:
```
$ ./corefreq-cli -s
Processor                              [AMD Ryzen 7 2700X Eight-Core Processor ]
|- Architecture                                                     [Family 17h]
|- Vendor ID                                                      [AuthenticAMD]
|- Signature                                                            [ 8F_08]
|- Stepping                                                             [     2]
|- Microcode                                                            [     0]
|- Online CPU                                                           [ 16/16]
|- Base Clock                                                           [ 99.80]
|- Ratio Limited                                                        [  LOCK]
|- Frequency            (Mhz)                      Ratio                        
                 Min    399.20                    [   4 ]                       
                 Max   3991.96                    [  40 ]                       
|- Factory                                                                      
                       4000                       [  40 ]                       
|- Turbo Boost                                                                  
                  1C   2295.38                    [  23 ]                       
                  2C   2295.38                    [  23 ]                       
                  3C   2195.58                    [  22 ]                       
|- Uncore                                                                       
                                                                                
ISA Extensions:                                                                 
|- 3DNow!/Ext [N,N]        AES [Y]   AVX/AVX2 [Y/Y]   BMI1/BMI2 [Y/Y] 
|- CLFSH        [Y]       CMOV [Y]      CMPXCH8 [Y]      CMPXCH16 [Y] 
|- F16C         [Y]        FPU [Y]         FXSR [Y]     LAHF/SAHF [Y] 
|- MMX/Ext    [Y/Y]    MONITOR [Y]        MOVBE [Y]      PCLMULDQ [Y] 
|- POPCNT       [Y]     RDRAND [Y]       RDTSCP [Y]           SEP [Y] 
|- SGX          [N]        SSE [Y]         SSE2 [Y]          SSE3 [Y] 
|- SSSE3        [Y] SSE4.1/4A [Y/Y]      SSE4.2 [Y]       SYSCALL [Y] 
                                                                                
Features:                                                                       
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Present]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
|- Pending Break Enable                                          PBE   [Missing]
|- Process Context Identifiers                                  PCID   [Missing]
|- Perfmon and Debug Capability                                 PDCM   [Missing]
|- Page Global Enable                                            PGE   [Present]
|- Page Size Extension                                           PSE   [Present]
|- 36-bit Page Size Extension                                  PSE36   [Present]
|- Processor Serial Number                                       PSN   [Missing]
|- Restricted Transactional Memory                               RTM   [Missing]
|- Safer Mode Extensions                                         SMX   [Missing]
|- Self-Snoop                                                     SS   [Missing]
|- Time Stamp Counter                                            TSC [Invariant]
|- Time Stamp Counter Deadline                          TSC-DEADLINE   [Missing]
|- Virtual Mode Extension                                        VME   [Present]
[... 161 lines cut ...]
|                      1       1       0       0       0                        
|- Core Cycles                                                         [Missing]
[... cut ...]
|- Package Thermal Management                                    PTM   [Missing]
|- Thermal Monitor 1                                         TM1|TTP   [Present]
|- Thermal Monitor 2                                         TM2|HTC   [Present]
|- Units                                                                        
   |- Power                                               watt   [  0.000000000]
   |- Energy                                             joule   [  0.000000000]
   |- Window                                            second   [  0.000000000]
```

cyring, 2018-05-22:
Thank you. Very useful.

So, the max ratio means that the first msr register MSR_AMD_PSTATE_DEF_BASE has a significant frequency id which reflects the BIOS setting.

And when XFR is enabled, this register holds the max none turbo coefficient.

40 vs 37

An XFR status bit or equivalent has to be queried.

adatum, 2018-05-22:
Glad it's useful. 

The explanation is beyond my understanding, but let me know if there's anything else you need.

cyring, 2018-05-23:
A stress function will loop over the current frequency to reach the XFR or the max ratio.
One unit is added to the result.

Kernel scheduler may stall, save your files !

cyring, 2018-05-24:
Could you also check if TURBO is light on ?

adatum, 2018-05-24:
I get an error (not system crash) with `$ sudo insmod corefreqk.ko Experimental=1`: "watchdog: BUG: soft lockup - CPU#13 stuck for 22s!". (CPU # can vary) Loading the module takes several seconds and htop shows 100% cpu usage on only 1 core. Eventually the command exits and I'm able to run `corefreqd` and `corefreq-cli`.

CoreFreq hasn't crashed so far. It was started at idle. 

Turbo indicator is lit. Haven't tried with CPB/PE/PBO disabled in BIOS yet.

Screenshots below at idle, single core loaded, all cores loaded. Also output of `corefreq-cli -s`. Turbo Boost printout for 1C/2C/3C doesn't seem right.

[image]
[image]
[image]
```
$ ./corefreq-cli -s
Processor                              [AMD Ryzen 7 2700X Eight-Core Processor ]
|- Architecture                                                     [Family 17h]
|- Vendor ID                                                      [AuthenticAMD]
|- Signature                                                            [ 8F_08]
|- Stepping                                                             [     2]
|- Microcode                                                            [     0]
|- Online CPU                                                           [ 16/16]
|- Base Clock                                                           [100.59]
|- Ratio Limited                                                        [  LOCK]
|- Frequency            (Mhz)                      Ratio                        
                 Min    402.36                    [   4 ]                       
                 Max   3721.83                    [  37 ]                       
|- Factory                                                                      
                       3700                       [  37 ]                       
|- Turbo Boost                                                                  
                  1C   4425.96                    [  44 ]                       
                  2C   3218.88                    [  32 ]                       
                  3C   2212.98                    [  22 ]                       
|- Uncore                                                                       
                                                                                
ISA Extensions:                                                                 
|- 3DNow!/Ext [N,N]        AES [Y]   AVX/AVX2 [Y/Y]   BMI1/BMI2 [Y/Y] 
|- CLFSH        [Y]       CMOV [Y]      CMPXCH8 [Y]      CMPXCH16 [Y] 
|- F16C         [Y]        FPU [Y]         FXSR [Y]     LAHF/SAHF [Y] 
|- MMX/Ext    [Y/Y]    MONITOR [Y]        MOVBE [Y]      PCLMULDQ [Y] 
|- POPCNT       [Y]     RDRAND [Y]       RDTSCP [Y]           SEP [Y] 
|- SGX          [N]        SSE [Y]         SSE2 [Y]          SSE3 [Y] 
|- SSSE3        [Y] SSE4.1/4A [Y/Y]      SSE4.2 [Y]       SYSCALL [Y] 
                                                                                
Features:                                                                       
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Present]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
|- Pending Break Enable                                          PBE   [Missing]
|- Process Context Identifiers                                  PCID   [Missing]
|- Perfmon and Debug Capability                                 PDCM   [Missing]
|- Page Global Enable                                            PGE   [Present]
|- Page Size Extension                                           PSE   [Present]
[... 42 lines cut ...]
|                      1       1       0       0       0                        
|- Core Cycles                                                         [Missing]
|- Instructions Retired                                                [Missing]
[... cut ...]
|- Package Thermal Management                                    PTM   [Missing]
|- Thermal Monitor 1                                         TM1|TTP   [Present]
|- Thermal Monitor 2                                         TM2|HTC   [Present]
|- Units                                                                        
   |- Power                                               watt   [  0.000000000]
   |- Energy                                             joule   [  0.000000000]
   |- Window                                            second   [  0.000000000]
```

cyring, 2018-05-24:
Thanks
Those are the results I was expecting:
The XFR @ ~4.35GHz rounded to integer plus one
Thus max single core ratio frequency (1C) equals 44
However the algorithm is too aggressive, that why you get this bug.
Need to find another way to estimate or query the highest ratio...

adatum, 2018-05-24:
Thanks. Since it seems like the cores are being briefly benchmarked to establish max frequency/ratio, do you think it would be possible to rank the cores (best to worst)? 

Note that I accidentally had enabled "Ai Overclock Tuner" in the Asus BIOS which caused the 100.6MHz base clock in the screenshots above. This also allowed boosting to 4.38GHz which puzzled me initially.

cyring, 2018-05-24:
With BIOS combinations of CPB, XFR, Manual OC, can you report the state of TURBO (green or not) ?

I expect to read the TURBO as disable when OC is manual.

cyring, 2018-05-24:
I change for a table of Boost and XFR frequency ratios.

If "AMD Ryzen 7 2700X" is found and CPB is enable on any core then 

```1C = Max ratio + 6 (Boost) + 2 (XFR)``` 

else (case of manual OC)

```1C = Max ratio```

adatum, 2018-05-25:
Module loads quickly again, and `corefreq-cli` hasn't crashed so far starting from idle, stressing 1 to 16 cores, going back to idle. 

Some limitations:
- Hard coded 2700X identification: other Ryzen CPUs will have similar ratios
- Hard coded boosting limits: OCers with great cooling might exceed the `1C` limit (seems unlikely for this CPU). Manual OC isn't affected.

The good news is it works, and temperature (+min/max) match `sensors`

-----

There's a discrepancy in the idle ratios/frequencies compared to `watch -n1 "cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_cur_freq"`

[image]

cyring, 2018-05-25:
Give a look at source code for all other Ryzens:
https://github.com/cyring/CoreFreq/blob/b076c248e884b83132303d0b8ee0a2ab5122efba/corefreqk.c#L2559
Only additional beans are hard coded; base P0 ratio comes from the FID register (like the BIOS is doing, I guess).
However if CPB is off, because user is overclocking the processor, then this algorithm does not happen and the user's ratio becomes the limit.

Driver arguments can be listed with:
```modinfo corefreqk.ko```
Start with ```SleepInterval=100``` for the highest accuracy. This will have a CPU overhead b/c of a tight 100 ms loop, but it should match the values of scaling_cur_freq

adatum, 2018-05-25:
Nice work. I didn't know you included values for other CPUs as well. The 2200G[E]/2400G[E] are missing though.

`SleepInterval=100` gives the same values as before (refreshed much more quickly). I don't know what the lowest ratio is, but I haven't noticed much less than 18 maybe 16 in other tools, though I'm not fully confident of `scaling_cur_freq` either. Do you know what is the minimum frequency/ratio of the CPU?

cyring, 2018-05-25:
2200G[E]/2400G[E] but also EPYC series are missing into table: I need their exact cpuid brand string, with associated Core Coefficient and XFR ability: +50 or +100 or +200MHz or none
Any help to ding those informations are welcomed.

I believe scaling_cur_freq and _CoreFreq_ don't speak about the same subject: immediat vs relative frequency.

In the UI, look at lowest between 2C, 3C and so on

cyring, 2018-05-25:
Hello, there is something I can't see in your last screenshot: what is the highest ratio (computed by the driver) ?

adatum, 2018-05-25:
The highest ratio indicated in the GUI was 45 I believe.

Sorry I don't understand; what do you mean by immediate vs relative frequency? Seeing ~100MHz vs ~2GHz... they can't both be right or they are measuring different things. Since this is frequency and not a percentage I would think they should match.

cyring, 2018-05-26:
BIOS [P-states](http://www.overclockers.com/forums/showthread.php/781069-Crosshair-VI-P-State-modification-for-power-savings) guide.
If you have such options, it could be interested to check P1, P2 ... Pn against 2C, 3C, nC

cyring, 2018-05-26:
Here is the alpha code for the power energy and the voltage core. Remark: those are per Package readings.

In the UI, go to the view [Power & Voltage] to verify: 
1- the VID and the computed Vcore 
2- the Package and Cores Energy and Power consumed. (_no Uncore and Mermory_)

Open the window [Power & Thermal] to display the Units.

Could you also print the Processor topology
```corefreq-cli -m```

Thank you.

cyring, 2018-05-26:
Can you please run the ```rdmsr``` command for the followings hexa: 
```
0xc0010061
0xc0010062
0xc0010063
0xc0010071

0xc001029a
0xc001029b
0xc0010293

0x0005a000
```

adatum, 2018-05-26:
```
# rdmsr 0xc0010061
22

# rdmsr 0xc0010062
2

# rdmsr 0xc0010063
2

# rdmsr 0xc0010071
rdmsr: CPU 0 cannot read MSR 0xc0010071

# rdmsr 0xc001029a
41670f

# rdmsr 0xc001029b
13ad785

# rdmsr 0xc0010293
874c84

# rdmsr 0x0005a000
rdmsr: CPU 0 cannot read MSR 0x0005a000
```

adatum, 2018-05-26:
Also, I checked the P-states:

> P0: 3700, 1.21V
> P1: 3200, 0.99V
> P2: 2200, 0.81V
> P3: 400, 0V

Not defined for P4-P7.
The 0V for P3 caught my eye. But I really know nothing about P-states.

cyring, 2018-05-27:
I presume those are BIOS [P-States](https://www.thomas-krenn.com/en/wiki/Processor_P-states_and_C-states) frequency and voltage ?

adatum, 2018-05-27:
Yes.

cyring, 2018-05-28:
1- New VID reading delivered
2- About RAPL (running average power limit), I'm not sure if the Ryzen counters work like the Intel ones; incremental or instantaneous ?

adatum, 2018-05-28:
I try the new push later. Anything specific you'd like me to check?

For RAPL I just found it mentioned, but not defined, in an [AMD developer document ](https://developer.amd.com/wordpress/media/2013/12/User_Guide.pdf)you might have seen.

This [article](https://steemit.com/hardware/@tomasbrod/how-to-p-state-owerclock-ryzen-on-linux) mentions RAPL and this [hardware monitoring project](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/Hardware/CPU/AMD17CPU.cs) on github that maybe contains some hints?

adatum, 2018-05-29:
I see exactly two unique values: (VID, Vcore) = { (118, 0.8125), (54, 1.2125) }, where the first pair is at idle and the second pair under load.

cyring, 2018-05-29:
Do the RAPL hardware monitoring project match your BIOS ?

adatum, 2018-05-29:
The Vcore readings in BIOS and `sensors` (with it87 module) vary a lot more, so it does not match since it always shows one of only those two values.

The Vcore mostly remains at 0.8125 under low or single core load, occasionally jumping to 1.2125. Under full load it stays at 1.2125. I used mprime to stress (especially all cores).

I haven't tried the manual OC again.

Sorry I'm not familiar with RAPL. What would you like me to check?

cyring, 2018-05-29:
Thanks.

With a manual OC I would like to check if I stick with VID non boosted pstates or I have to switch to the VID of the Boosted register.
In the last case the algorithm will just test the CPB bit as we did with temperature.

For RAPL, we need a strong reference to compare with. Any AMD or motherboard tool for instance.

cyring, 2018-05-30:
Excellent report.

Because the _CoreFreq_ driver reads the P-States, we get only the VID programmed into these registers. That's why the computed Vcore is static; the VIDs set in BIOS are the values read later by the driver.
My understanding is that the Processor switches to another voltage plan by selecting another P-State. However a VID is a target voltage, not the effective voltage. Can be the same or closed to it

Same rule is applied with the FID, Frequency Identifier. This time however _CoreFreq_ computes and shows the relative frequency (based on performance counters rather than FID)

In the _CoreFreq_ UI, 48, the max ratio is too hight, the driver Ryzen table needs to be adjusted.

Do you get some values in the Power view ?

adatum, 2018-05-30:
Would it be possible to have effective voltage readings too?

For frequency, CoreFreq shows similar results to `turbostat` which I recently discovered. These frequencies are as low as ~0 MHz at idle. In contrast, scaling_cur_freq shows maybe a minimum of 1.8GHz at the same time. Clearly I'm not familiar with the distinction between these two measurements.

Still zero values for Power, Energy, exactly as in the previous screenshot.

cyring, 2018-05-30:
I don't have any specifications to read voltage.

0 MHz is not for real. It is a frequency tendency based on _usage_. (in fact, number of clock cycles elapsed when processor is working)
Off course, one can read the effective frequency, looping arround the  scaling_cur_freq; but with a  GHz processor, what will be the best sampling interval ?
Every second, you won't record data at the right time.
Every nano second, the measurements will itself put _load_ on processor !
=> performance counters have been created to solve this case.

RAPL: your previous msr readings show that we get values. Need to debug the scalling computation based on the retrieved power unit.

cyring, 2018-05-30:
Can you please try the last RAPL commit ?

adatum, 2018-05-30:
[image]

Non-zero energy and power values show up, though they are static. I could compare very roughly with readings from the UPS (total system + monitor power, not just CPU).

What is the significance of the usage based frequency and when might someone want to know it instead of effective frequency? 

I've been using `watch -n1 "cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_cur_freq"` but of course there's the issue of sampling rate. ( -n0 is crazy) 

A tool with min/max/avg/current values like [HWiNFO ](https://www.hwinfo.com/) seems lacking for linux so far; do you know what approach it takes on sampling and readings? It exists so it must be possible.

cyring, 2018-05-30:
Can't tell about HWInfo. Does it show, in the same time, the frequency increase/decrease ?

In first version of XFreq, I used to read the effective frequency without reaching the Turbo. See this thread at [Intel](https://software.intel.com/en-us/forums/software-tuning-performance-optimization-platform-monitoring/topic/499128).

adatum, 2018-05-30:
Not sure what you mean by frequency increase/decrease, but you can search for screenshots where it shows current/min/max/avg values for each core.

I would think that generally end users want to know effective values inclusive of Turbo and any other enhancements. Like in Task Manager, etc.

Very interesting thread. So much to learn from it.

cyring, 2018-05-30:
Definitely this is an idea to add on the roadmap.
I have programmed the fixed frequency and I suggest to combine then with the performance monitoring counters (pmc): in the UI, the histogram bars would be limited by the min, max and turbo frequency ratios; and with the help of the pmc, the bars could slide progressively between each limit.
This would be an innovation above the task mgr ?
I don't find motivation to clone other software -;)

We have however to complete the current _CoreFreq_ features for Ryzen ...

cyring, 2018-05-31:
Hello
Can you script 2 loops with a 1 second interval, arround respectively 0xc001029a 
0xc001029b 

I want see how these msr registers are moving.

[498 more replies]

closed 2020-03-18 by cyring