cyring/CoreFreq #57: RAPL for Skylake architecture

cyring, 2018-06-18:
https://github.com/cyring/CoreFreq/blob/a0f440a0dd043ea4b6621b6ad4269c7ee12028ea/corefreqk.c#L5216
https://github.com/cyring/CoreFreq/blob/a0f440a0dd043ea4b6621b6ad4269c7ee12028ea/corefreqk.c#L5320

Replace ```PWR_ACCU_SandyBridge``` with ```PWR_ACCU_Skylake```
```
#define PWR_ACCU_Skylake(Pkg, T)                                        \
({                                                                      \
        RDCOUNTER(Pkg->Counter[T].Power.ACCU[PWR_DOMAIN(PKG)],          \
                                                MSR_PKG_ENERGY_STATUS); \
                                                                        \
        RDCOUNTER(Pkg->Counter[T].Power.ACCU[PWR_DOMAIN(CORES)],        \
                                                MSR_PP0_ENERGY_STATUS); \
                                                                        \
        RDCOUNTER(Pkg->Counter[T].Power.ACCU[PWR_DOMAIN(UNCORE)],       \
                                                MSR_PP1_ENERGY_STATUS); \
                                                                        \
        RDCOUNTER(Pkg->Counter[T].Power.ACCU[PWR_DOMAIN(RAM)],          \
                                                MSR_DRAM_ENERGY_STATUS);\
})
```

cyring, 2018-06-28:
KBL RAPL and Voltage during high load ...
[image]
... during low load
[image]

cyring, 2018-07-30:
Measurements issue with i7-3770 (IvyBridge)
[image]
[image]
[image]

cyring, 2018-08-03:
Skylake: add missing code to measure DRAM power
https://github.com/cyring/CoreFreq/blob/6712f73d42313d07f85659eef70cc8dff95797b2/corefreqk.c#L6072
```
			Delta_PWR_ACCU(Proc, RAM);
```
https://github.com/cyring/CoreFreq/blob/6712f73d42313d07f85659eef70cc8dff95797b2/corefreqk.c#L6090
```
			Save_PWR_ACCU(Proc, RAM);
```
[image]
[image]
[image]

cyring, 2018-09-13:
Workaround for IVB [Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz]
https://github.com/cyring/CoreFreq/blob/68db124338be91b5a97a08cab7830cfbb24e472d/corefreqd.c#L443
```C
	Shm->Proc.Power.Unit.Watts = Proc->PowerThermal.Unit.PU > 0 ?
			1.0 / (double) (1 << Proc->PowerThermal.Unit.PU) : 0;

	Shm->Proc.Power.Unit.Watts /= (Proc->CPU.Count >> Proc->Features.HTT_Enable);
```
[image]
Accurate ?

cyring, 2018-12-01:
Issue closed until other hardware is available for testings.

cyring, 2019-07-21:
Architecture|Processor|ESU|Pkg load(J)|Cores load(J)|TDP(W)
----------------|-------------|------|----------------|-----------------|------------
Skylake/S|i5-6600K|0.000061035|84.761535645|75.259704590|91

cyring, 2019-07-21:
# Proposal for the `Power & Voltage` view

 `#`  | Freq MHz | VID | Vcore    |          Energy(J)               | Counter(Raw)
------|---------------|------|-------------|-------------------------------------|------------------
`#0` | `4015.73` |     .  |              . | `Package 131.852859497` | `9876543210`
`#1` | `4015.67` |     .  |              . | `Cores 122.611679077` | .
`#2` | `4015.74` | `54` | `1.2125` | `Uncore 10.000000000` | `9876543210`
`#3` | `4015.73` |     .  |              . | `Memory 5.000000000` | .

 `#`  | Freq MHz | VID | Vcore    |          Power(W)               | Core(W)
------|---------------|------|-------------|-------------------------------------|------------------
`#0` | `4015.73` |     .  |              . | `Package 131.852859497` | `13.000000000`
`#1` | `4015.67` |     .  |              . | `Cores 122.611679077` | .
`#2` | `4015.74` | `54` | `1.2125` | `Uncore 10.000000000` | `11.000000000`
`#3` | `4015.73` |     .  |              . | `Memory 5.000000000` | .

* One shortcut to toggle between the Energy(J) and Power(W) layouts

cyring, 2019-07-27:
# RAPL in the AMD Zen architecture
## Energy consumed
### Ryzen 3xxx
[image]

### Ryzen 2xxx
[image]

## Topology
### Ryzen 2xxx
```
CPU Pkg  Apic  Core Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID    ID     ID  L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
00: BSP     0     0      0      64  4        32  8       512  8     16384  8  
01:   0     1     0      1      64  4        32  8       512  8     16384  8  
02:   0     2     1      0      64  4        32  8       512  8     16384  8  
03:   0     3     1      1      64  4        32  8       512  8     16384  8  
04:   0     4     2      0      64  4        32  8       512  8     16384  8  
05:   0     5     2      1      64  4        32  8       512  8     16384  8  
06:   0     6     3      0      64  4        32  8       512  8     16384  8  
07:   0     7     3      1      64  4        32  8       512  8     16384  8  
08:   0     8     4      0      64  4        32  8       512  8     16384  8  
09:   0     9     4      1      64  4        32  8       512  8     16384  8  
10:   0    10     5      0      64  4        32  8       512  8     16384  8  
11:   0    11     5      1      64  4        32  8       512  8     16384  8  
12:   0    12     6      0      64  4        32  8       512  8     16384  8  
13:   0    13     6      1      64  4        32  8       512  8     16384  8  
14:   0    14     7      0      64  4        32  8       512  8     16384  8  
15:   0    15     7      1      64  4        32  8       512  8     16384  8 
```
### Ryzen 3xxx
```
CPU Pkg  Apic  Core Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID    ID     ID  L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
00: BSP     0     0      0      32  8        32  8       512  8     32768  9  
01:   0     2     1      0      32  8        32  8       512  8     32768  9  
02:   0     4     2      0      32  8        32  8       512  8     32768  9  
03:   0     6     3      0      32  8        32  8       512  8     32768  9  
04:   0     8     4      0      32  8        32  8       512  8     32768  9  
05:   0    10     5      0      32  8        32  8       512  8     32768  9  
06:   0    12     6      0      32  8        32  8       512  8     32768  9  
07:   0    14     7      0      32  8        32  8       512  8     32768  9  
08:   0     1     0      1      32  8        32  8       512  8     32768  9  
09:   0     3     1      1      32  8        32  8       512  8     32768  9  
10:   0     5     2      1      32  8        32  8       512  8     32768  9  
11:   0     7     3      1      32  8        32  8       512  8     32768  9  
12:   0     9     4      1      32  8        32  8       512  8     32768  9  
13:   0    11     5      1      32  8        32  8       512  8     32768  9  
14:   0    13     6      1      32  8        32  8       512  8     32768  9  
15:   0    15     7      1      32  8        32  8       512  8     32768  9
```
## Issue
### AMD specifications
```
MSRC001_029B [Package Energy Status] (Core::X86::Msr::PKG_ENERGY_STAT)
Read-only,Volatile. Reset: 0000_0000_0000_0000h. 
_lthree[1:0]; MSRC001_029B
Bits Description
63:32 Reserved. 
31:0 TotalEnergyConsumed.
```  
```
CCX: Core Complex where more than one core shares L3 resources.
```
where: 
`_lthree[1:0]` means CCX instance 0 and CCX instance 1

###  Improvements
* add the CCX identifier into the CPU topology
* read the Package energy counter per CCX
* sum the per CCX values into a Package value

### Questions
* what about the multi dies processors (TR, Naples, Rome)

cyring, 2019-07-27:
# Core Complex ID
## Ryzen 3xxx [SMT ON]

CPU#|Pkg ID|Apic Ext_ID|Core ID|Thread ID|Apic[3:0]
--------|---------|----------------|----------|-------------|------------
00|BSP|0|0|0|0000
01|0|2|1|0|0010
02|0|4|2|0|0100
03|0|6|3|0|0110
04|0|8|4|0|1000
05|0|10|5|0|1010
06|0|12|6|0|1100
07|0|14|7|0|1110
08|0|1|0|1|0001
09|0|3|1|1|0011
10|0|5|2|1|0101
11|0|7|3|1|0111
12|0|9|4|1|1001
13|0|11|5|1|1011
14|0|13|6|1|1101
15|0|15|7|1|1111

cyring, 2019-07-27:
After long hours of the 3700X debugging, it appears that the RAPL Package Energy Counter delta remains the same, whatever the CPU is used to read the msr
[image]

_Call for help on [Reddit](https://www.reddit.com/r/Amd/comments/cibwrv/ryzen_3000_fully_capable_of_advertised_boosts/ev7xqae?utm_source=share&utm_medium=web2x)_

cyring, 2019-08-04:
* To link with issue #60 for PL1 RAPL regusters

olejon, 2019-08-12:
Here are my screenshots as requested in #129 

* https://www.olejon.net/files/CF-X570-Ryzen-3600X/
* Also added the benchmarks from Blender Benchmark & Geekbench, because it shows Linux is already superior here :+1: 
* Both benchmarks run under same conditions (SMT on, Precision Boost Override on, System totally idle with no other programs open and Geekbench run in CLI-mode on Windows as well)

cyring, 2019-08-12:
To all reading this issue, look in the above screenshots for the Package Power measurements; Cores all stressed :

* X370 (Crosshair VI Hero) = 9.28W

* X570 (Primé X570 Pro) = 91.76W

Both cases, **same algorithm based on RAPL registers**; what can we conclude ?

olejon, 2019-08-12:
* That only 9.28W is definitely not right... Intel CPUs you have tested well and CoreFreq supports well, with more or less similar Cores & Threads, when they're all stressed, use way more Watts
* My Skylake 4-core i5-6600K [has a 91W TDP](https://ark.intel.com/content/www/us/en/ark/products/88191/intel-core-i5-6600k-processor-6m-cache-up-to-3-90-ghz.html) so the below makes sense
* When my Skylake 4-core i5-6600K used ~85W with All Cores Stressed using Conic Compute - UEFI "Sync All Cores" and multiplier 44, so 4.4 GHz which it holds fine with the cooler it has (45 multiplier and it gets too hot)
* [4.4 GHz is the max turbo speed for my 3600X](https://www.amd.com/en/products/cpu/amd-ryzen-5-3600x), and has a "Default TDP/TDPROM-06a" of **95W**, which makes sense for my result
* It can manage 4.4 GHz for 2 cores at a time. Seems to take advantage of that with regular use, offloading load from 2 active cores to 2 sleeping cores and so on, managing cooling very well
* Of course running a Stress Test is different. All Cores = ~4.0 GHz, as per the specs of the 3600X
* On Windows running *Blender Benchmark* - which is the bench that causes the most load (adding in more like CPU-Z or Geekbench does *not* stress it more) *Ryzen Master peaks at 70% of 128W "PPT (CPU)"
* **This is calculates to 89.6W, which is very close to the CoreFreq reading**
* CoreFreq showing a little more I think is because "Conic Compute" stresses the CPU *even more* than Blender Benchmark, since on Windows the CPU, **according to the most reliable tool, Ryzen Master* peaks at temp 79.xx-80.xx C (stays just a couple of seconds at 80.xx C and then goes below 80 C for a couple of seconds, then up again and so on, and above 80 C the color of the bar turns orange so easy to see)
* **Conclusion:** CoreFreq is right for my 3600X on an X570 MB
* Also CoreFreq manages to get it to at least more than 80 C using "Conic Compute" IIRC, even maybe 82-84 C.
* So 9.28W *can't be right*
* Also my benchs have shown Linux is faster, though didn't monitor temps (can do with Blender Benchmark)

**IMPORTANT NOTE:**
* As far as I've read and seen/hear on YouTube videos, **AMD straight out said they do NOT recommend Ryzen 3rd Gen on X370 MBs!**
* According to AMD, if using an older MB, the X470 ones can they be used just fine, **but that one does not benefit and may run into problems with a CPU with a TDP above 65W**, so not recommended for the X version of the 3600, BUT the 3700X has a TDP of 65W though. Problem is it costs more and according to benchs for gamers with a good enough GPU is just wasting money...
* AND one MUST remember to tell the seller/retailer to update to the latest BIOS (usually 2 updates, 1st to get it ready for the 2nd) *before shipping* as if NOT it will NOT RUN (3rd Gen)
* So one should stick to a CPU like the 3600 (without the X), which is 65W, AND is the CPU of choice recommended by the most respected YouTubers for gamers looking for a cheap update, *as good X470 MBs are very cheap now, and the 3600 is cheap*
* But with an X470 one can probably find a cheaper and better performing 2nd Gen Ryzen CPU, and you'll lose PCIe4 support if not using X570 anyways, so a 2nd Gen is
* As an example, with CPU-Z on Windows and its references, the 2700X outperforms the 3600X, like the 3700X it has 8c/16t compared to 6c/12t. Also the 1700X and 1800X are better (8c/16t) than the 3600X. For Ryzen I must go down to the 1600 (no X) 6c/12t to see my 3600X beating it clearly. On the Intel side my 3600X also clearly beats the i7-8700K (same 6c/12t), but of course not the i9-9900K since it has 8c/16t and costs a lot compared to a 8c/16t Ryzen like the 3700X, which I think the Intel i9 beats since it has Turbo to 5 GHz, but the 5800X (but *don't* buy that one!) *may beat it* with also 8c/16t like the 3700X, while Turbo is lower at 4.5 GHz, it has more cache and newer tech in general, well not much more Turbo compared to the 3700X's 4.4 GHz, but a TDP of 105W compared to the 3700X's 65W. The 3900X beats the i9 of course with 12c/24t, and clearly the 3950X when launching will beat it with 16c/32t.

olejon, 2019-08-12:
EDIT: Some additional info...

olejon, 2019-08-13:
What I wrote *very much confirmed*:

> CoreFreq manages to get it to at least more than this, using "Conic Compute" IIRC, even maybe 82-84 C IIRC

*Just running Conic Compute for ~1 minute and the CPU reached 85 C
* System otherwise totally idle. Not even logged in to desktop. Did this over SSH
* Much more than I manage on Windows with any tool/bench
* All fans are set to 100 % Duty Cycle above 75 C so not because of that
* Must be CPU using more Power. Since Limit according to Ryzen Master in Windows is 95 C, there's no reason it should stay around 80 C using popular tools that maxes out the CPU for a long time
* Basically Conic Compute in CoreFreq is better at stressing the CPU and hence is able to go to 91.5W. **Great job!**
* As said on Windows it calculates to 89.6W (see comment above how it was calculated)

cyring, 2019-08-13:
Is the B450 Chipset supported ? Because we also have the Package Power issue [above](https://user-images.githubusercontent.com/11563789/61993491-bcae3200-b06c-11e9-94e0-6ac42116845c.png)

cyring, 2019-08-13:
Processor|Chipset|Pkg Pwr
---|---|---
3700X|X370|KO
3700X|B450|KO
3600X|X570|OK

* I think we need results from the 3700X + X570 combinaison.

cyring, 2019-08-13:
Conic is an _old_ friend of the 486DX age
https://github.com/cyring/CoreFreq/blob/e21a6d5480fff6a095edec7b6ea01310ad25c0e9/corefreqm.c#L196

cyring, 2019-08-13:
# AMD64 Architecture-Programmer’s Manual
-Volume 2: 
System Programming
## Hardware Performance Monitoring and Control

```
Software Algorithm
The following algorithm should be used to calculate the average power consumed by a processor core 
during the measurement interval TM. To obtain a stable average power value, TM should be on the 
order of several milliseconds.
• Determine the value of the ratio of Tsample to the TSC period (CpuPwrSampleTimeRatio) by 
executing CPUID Fn8000_0007. Call this value N.
N = CPUID Fn8000_0007_ECX[31:0].
• Read the full range of the cumulative energy value from the register MaxCpuSwPwrAcc. 
Jmax = value returned from RDMSR MaxCpuSwPwrAcc.
• At time x, read CpuSwPwrAcc and the TSC
Jx = value returned by RDMSR CpuSwPwrAcc 
Tx = value returned by RDTSC
• At time y, read CpuSwPwrAcc and the TSC again
Jy = value returned by RDMSR CpuSwPwrAcc
Ty = value returned by RDTSC
Calculate the average power consumption for the processor core over the measurement interval TM = 
(Ty – Tx). 
• If (Jy < Jx), rollover has occurred; set Jdelta = (Jy + Jmax) – Jx
else Jdelta = Jy – Jx
• PwrCPUave = N * Jdelta / (Ty - Tx) 
Units of result is milliwatts.
```

olejon, 2019-08-13:
> I think we need results from the 3700X + X570 combinaison.

That will work just like 3600X.

cyring, 2019-08-13:
> That will work just like 3600X

I think the same, but the table won't be completed without such combination. 
Does the issue happen because of the chipset or the Processor model...

olejon, 2019-08-13:
I can't say anything before getting my 3950X, which is Ryzen 9, but if works like 3600X, why should Ryzen 7 don't?

I'll pre-order as soon as available here. If I'm quick maybe I'll get it in the end of September or beginning of October (depending on when AMD actually launches it, but says already on amd.com September), and if *that CPU* works exactly the same in CoreFreq, why wouldn't the 3700X?

IDK why they call all CPUs up to 3700X Ryzen 5, then 3700X up to 3900X Ryzen 7 and 3900X+ Ryzen 9. The Ryzen 5 is what makes no sense for me, for a 3600X CPU. Shouldn't it then be Ryzen 6? And 3800X be Ryzen 8?

Very few should buy the 3700X and no one the 3800X, after some reading. If for gaming? Buy the 3600 (without X), which doesn't bottleneck any modern GPU, at least not on X570, but I've seen similar results on X470 which is a cheaper option (and both that CPU and those older Gen MBs will lower in price).

At first considered the 3700X. AMD called it the competition to the latest i7-8700K I think, which is a darn good CPU so. But many said that they should call it more a direct competition to the latest i9, although the i9 does better in benchs, it's mostly because of the 5 GHz Turbo. But no real world usage difference, that probably goes in favor of AMD.

AMD said the 3900X was the direct competitor to the i9-9900K. Here the 3900X cost €60 more. Those prices will drop though - they were just released. *Nobody has it in stock*. It was but super demand.

And AMD's Turbo out of the box is SMT which equals to Sync All Cores *which even in Auto mode, meaning all Turbo to max stock Turbo and not above that, is a pretty serious OC on Intel*, letting all cores Turbo makes it MUCH hotter in my experience, while not noticable on AMD. Requires good cooling, *and I imagine especially since it has Hyperthreading, which my latest didn't*.

With a good cooler the 3700X/3800X can probably get a decent bump in Turbo GHz, maybe not 5 GHz but beat the i9 anyways.

At least maybe it can compete with the i9-9900 it (a CPU I never see mentioned in videos), but *both are Gen "Coffee Lake Refresh"*. When I checked my Norwegian online store, it seemed a better choice as cheaper *with a fan included* (doesn't say which, just that the product is "In Intel Box", but so does the K, but fan the K does never have, just an overly fancy box that people give it lower OR higher rating for, lower because really after opening it's just a waste of carton. Non-K being about just €2 cheaper though.

Differences:
* K has 3.6 GHz vs 3.1 GHz Base Clock Speed, *but both Turbo to 5 GHz
* K has 95W TDP vs 65W
* Non-K has "Intel Thermal Velocity Boost"

Similarities:
* 8c/16c
* "Coffee Lake Refresh"
* Exact same iGPU, pretty good, 1.2 GHz max, almost my RX 570's 1.28 GHz, and same Base Clock at 350 MHz as my RX 570

cyring, 2019-08-13:
EPYC Rome 64C/128T in a 2 sockets configuration is so far my dream setup.
But true, any Ryzen should be enough to study the architecture and those RAPL registers....

olejon, 2019-08-14:
You mean this?

> AMD EPYC 7742 64-Core 2.25GHz (3.4 GHz Max Boost) Socket SP3 225W 100-100000053WOF Server Processor

https://www.newegg.com/p/N82E16819113581

And that's just the price for the CPU :-O

[I count 6 Authorized Distributors & Online Retailers from amd.com :P](https://www.amd.com/en/where-to-buy/epyc-7002-series)

Try it in the cloud first at least just for a short time? :P Can run bare metal there I assume, and for much less.

olejon, 2019-08-14:
O-la-la

> A screenshot of popular benchmarking tool Geekbench revealed by regular Twitter leaker APISAK (@TUM_APISAK) supposedly shows AMD's new 16-core Ryzen 9 3950X beating Intel's current desktop flagship – the Core i9-9980XE. The latter has 18 cores and costs $2000 while AMD’s new king of the hill **costs just $749** and has two fewer cores.

[image]

cyring, 2019-08-14:
(just kidding) Should we report to them that we found some RAPL Package counter - chipset issues before they launch such a monster processor ?

cyring, 2019-08-14:
* Table updated with results in hands
  Same as 3600X, X570 + 3700X don't present RAPL issue

Processor|Chipset|Pkg Pwr
---|---|---
3700X|X370|KO
3700X|B450|KO
3600X|X570|OK
3700X|X570|OK

[image]

olejon, 2019-08-16:
Sorry I didn't understand this. Remember I'm not a CPU expert, and don't have enough RAM in my brain for more.

> Should we report to them that we found some RAPL Package counter - chipset issues before they launch such a monster processor ?

Yes, as said the 3700X would surely work on X570 since 3600X does. I wouldn't care about X370 as AMD doesn't recommend it. Though you should really try to get or loan an X470 and see if it works on that (after all BIOS updates applied). *That* is what budget users will go for, and likely the 3600 (without X) as IIRC AMD and YT videos recommend TDP = 65W and not more. It'll surely work with the 3700X as well, and if so I assume the 3600 shows correct data. I think the recommendation of TDP not higher than 65W is because of better OC capabilities.

Also X570 has a new standardized motherboard design all manufacturers follow. Like the PCH fan, VRAM cooling and these things work together for cooling the chips. May be the reason of the 65W recommendation as well on X470. I don't think AMD is lying here, since they *DO* want to sell their new CPUs, even if you use an X470. They have en entire program for compatibility after all (BIOS updates sent to all manufacturers, *before launch*, telling both them and sellers to update the MBs before selling to make sure a 3rd Gen Ryzen CPU actually works).

cyring, 2019-08-16:
I do not know the PC which hosts the B450. I just report  the strings the kernel has probe through its DMI implementation.  

Whatever I won't go further into this Ryzen RAPL issue, X570 is the way to go.  

(BTW plz let me know if you have a chance to test the modified code in the issue #140 ) 

Regards

olejon, 2019-08-16:
I think I see. MBs, most mATX or mITX. Looking at the "MSI B450I GAMING PLUS AC" and the "Gigabyte B450 I AORUS PRO WIFI" (latter having both Intel for Wi-Fi and Ethernet, former Realtek...) both mITX they're pretty impressive considering the price, both has 802.11ac Wi-Fi, and seems to have 2 antenna connectors, antennas don't seem to be included. People write positive things about all that have reviews. About the mITX that they're perfect for a PC in the bookshelf and such, and that Wi-Fi coverage is just "fine" (probably 2 antennas would make that much better).

The most expensive is ATX, "ASUS ROG STRIX B450-E GAMING" and comes with antennas. And being the most expensive it's still only €189 *in Norway*. And the first with Wi-Fi just €129.

They haven't updated the site that they support 3rd Gen Ryzen, but looked at the cheapest ASUS one (mATX) to see CPU support for it, and the mentioned ATX above, the one I would have bought since the sister I suppose to my PRIME-PRO, although it's not a PRO, and called "PRIME B450M-A". *Both say they support even the 3900X so*... Surely the 3950X when coming.

Of course lacking is PCIe4 or 3200 MHz RAM without (OC) behind it, but several has 2 x M.2 and says they support RAID 0. The MSI ones have most reviews.

MSI ones also brags about "2 x Turbo M.2 with you to 32 GB/s *per unit* using PCIe3 x4 and support for RAID 0" (so per unit, with RAID 0 it can give like 64 GB/s? Ehm fastest M.2 I can find is a Premium 2T actually packaged for a PCIe slot and has about 7 GB/s and only supports Windows Server *on the Windows front*, but also says "Linux 3.3 and newer and Freebase" haha. Surely other stores have faster ones but 32 GB/s? Marketing? What the heck does Turbo M.2 even mean? It's just PCIe3 on all and that'll be the limit. Also don't like how some B450's brag about supporting having an SSD and an HDD and making sure often used files are on the SSD and less used files on, yeah, you get it. Like the complete disaster Apple's Hybrid drives were. Speeds usually HDD-ish. And how's the setup? UEFI or probably require some Windows software.

Seems to be pretty RGB heavy built-in and connectors for more, all of them, kind of weird for a small MB that's not usually put for display? IDK, maybe they are. I've never built with anything else than ATX so.

Want space, and my Chassis is the [Graphite Series 760T Arctic White Full-Tower Windowed Case](https://www.corsair.com/us/en/Categories/Products/Cases/Graphite-Series%E2%84%A2-760T-Arctic-White-Full-Tower-Windowed-Case/p/CC-9011074-WW) (see extensive gallery). The "red" is really nice if you want all black, just you must change the 3 chassis fans it comes with then because they have yeah, red LEDs, while mine has white. And mix of "artic white" color with black and black tinted windows makes it fit perfectly my room. Chassis' fans aren't bad either.

With the Ryzen without custom OC can really set in UEFI "Allow fan stop" for the individual or all, and just spin them to max when reaching 75 C on the CPU, but they're always on. The CPU runs just fine with a sweetspot of lowest possible duty cycle 20% and 60% when 62 C and 100% when 72 C. This makes it silent even in games like CS:GO. More or less only benchs spins it up. 20% is barely noticable. Whereas the built-in "Silent" profile is more like 35-40%, and then the fan (well I always have TV/podcast/music on), but if silent it becomes like a little annoying, not as "white noise" as the bigger chassis fans.

At max the chassis' fans really makes the CPU fan not so audible. It all mixes together. Though people buying this chassis typically do serious overclocking, and say the chassis fans are a bit loud and replace them with Noctua ones (without LEDs but one can always add RGB strips or whatever if wanted, set to white always if wanted). AND/OR they primarily use water cooling which it has plenty of room for (radiator vertical or top, everything like the HDD/SSD mounts and 3 slots CD-ROM sized module can easily be removed, or one can stack the 2 x 3 HDD/SSD mounts on top of each other instead of side by side).

They're all awesome, no screws, and 2.5 disks is not a problem, just remove one snap pin and screw in the 4 SSD's screws from below. There's also room for 4 x 2.5 disks behind the back door, which are 2.5 only so snaps in, so in total 10 SATA disk slots, and 2 x M.2 disks which doesn't occupy SATA cables.... And of course in the 3 x CD-ROM module you can stack in a ton put they're not snap-ins. Haven't checked but may very well be screw holes for at least 3.5 disks. Otherwise some cheap stuff of eBay for a nice mount. Really just depends on SATA ports on MB and PSU SATA cables... I actually have an ASUS CD+RW ROM, not the cheapest but the cheapest *ASUS* black nice-looking one I found which was almost the cheapest. I really bought for my previous penguin chassis when buying Skylake (so it's pretty new and as fast as CD+RWs today) because the penguin chassis didn't come with a cover there, just a hole, or we put in one on a LAN decades ago for Linux server install stuff, maybe losing the cover, if so IDK where it is so if going to build in it again must buy a CD-ROM hah. Before it was a white one which we used a permanent marker to more or less align the colors.

Finishing touches at my LAN crew partner's house. State of the art camera then with diskette slot for easy transfer over image. Printed Bill Gates on crappy printer:

[image]

[image]

cyring, 2019-10-14:
1. To simulate _Per Core RAPL_ with this Xeon processor, code has been temporary modified for data injection.  
[image]  

2. The **sum** of each CPU measurement has to **be equaled** to the value of **`Cores`** (in footer),  
whatever the unit (Watt, Joule) and interval (sec) are.  
    This exemple: `6.1035 + 6.1035 + 12.2070 + ... + 6.1035 = 109.8630`  

3. Version 1.67.3 is releasing _Per Core RAPL_ for AMD Zen architecture.  
    Tests should fully stress all CPUs and check the Sum of Cores
    Other idle and mixed cases would be useful.

adatum, 2019-10-14:
Per core values do seem to sum to the summary Cores value, in all-core load, single-core load, and idle conditions.

Power and Energy both report the same values. (`[$]` Key)

[image]

cyring, 2019-10-14:
Looks great, thank you
I'm now waiting for Ryzen 3000 results to pursue with the cosmetics: masking zero values of temps and accus, single Vcore....

olejon, 2019-10-15:
Ryzen 3rd Gen + X570 seems fine. Under idle we're talking 0.00x Watts per core but seems about right:

[image]

**NUC/Intel however** is still showing 0, always:

[image]

olejon, 2019-10-15:
NOTE TO ABOVE: Maybe just show `TMP` on all cores. CPU will always report package anyway, not per core, unless you Sir find a loophole, keeping in mind not even Ryzen Master do this, I assume it isn't possible. Some monitoring tools on Windows might, like popular HWMonitor, *but last I saw last release was before Ryzen 3rd Gen release, so I don't trust those tools*, really only Ryzen Master, as I assume it would show temperatures per core if the CPU supported it.

olejon, 2019-10-15:
**NOTE AGAIN**: Strange that the NUC shows 0 on all Cores but below it's calculated to an actual total, which 0 * 4 of course doesn't calculate to.

olejon, 2019-10-15:
Sorry for all the comments, didn't see 2700X screenshot properly first. **Seems like screenshot from @adatum shows Cores Total *should* match Package Total?**

As seen below, *idle in this case but as seen above also under stress*, and written above, Cores Total don't match Package Total on 3600X/X570. If this is a bug or simply the different architecture/platform/CPU, IDK.

NUC calculating a total still strange when Cores show 0?

[image]

[image]

cyring, 2019-10-15:
@olejon : thanks a lot.
To clarify the difference between Intel and Amd RAPL registers : the first is per socket counter whereas the last is per physical core (thus no SMT values). However AMD is per socket for Package.
Intel specs are telling us that Package counter gathers socket and other RAM, Uncore, perhaps SOC, iGPU

The remaining programming task is to mask contextually the zero values... 

NUC is an issue : the issue may come from the Joule and Watt units also probed from registers. These adjustments are undocumented and it will several calibration tests. I believe most low power processors will be a painful journey to program.

cyring, 2019-10-16:
[Specification update](https://www.intel.com/content/dam/www/public/us/en/documents/specification-updates/4th-gen-core-family-mobile-specification-update.pdf) of Mobile 4th Gen Intel Core Processor Family:
See the Errata HSM82 about RAPL: is your NUC updated with the latest BIOS ?

olejon, 2019-10-16:
Did update it 1-2 months back. Can check again later. Intel is very good at supporting NUCs even from 2013 with improvements and security fixes, so *maybe a new one is out BUT I doubt it*. They own the entire shebang of chipset, MB and iGPU so they can do a lot if they find a bug or bottleneck.

NUCs should be one of the best documented Intel complete solutions out there, all kinds of weirdness one can do with them, at least some later models than mine (I jumped on board quickly), like their answer to Raspberry Pi. Their cheapest one now isn't very expensive and also supported by Ubuntu Core (2 of them are, cheapest out, and latest powerful and most expensive, basically).

They're supported by Clear Linux, which you probably should check out on an all-Intel PC, most are supported I think, at least consumer grade. *That distro kind of trashes all others in benchs on Phoronix for instance*, and has more and more packages, plenty for a complete desktop. But not for a daily driver. Still I wished I had tried it on my workstation. My NUC is a little busy and headless...

Just RAM and disk is your choice, and optional Wi-Fi/Bluetooth.

cyring, 2019-10-16:
> Did update it 1-2 months back. Can check again later. Intel is very good at supporting NUCs even from 2013 with improvements and security fixes, so _maybe a new one is out BUT I doubt it_. They own the entire shebang of chipset, MB and iGPU so they can do a lot if they find a bug or bottleneck.
> 
> NUCs should be one of the best documented Intel complete solutions out there, all kinds of weirdness one can do with them, at least some later models than mine (I jumped on board quickly), like their answer to Raspberry Pi. Their cheapest one now isn't very expensive and also supported by Ubuntu Core (2 of them are, cheapest out, and latest powerful and most expensive, basically).
> 
> They're of course supported by **Intel's own Clear Linux**, which you probably should check out on an all-Intel PC, most are supported I think, at least consumer grade. _That distro kind of trashes all others in benchs on Phoronix for instance_, and has more and more packages, plenty for a complete desktop. But not for a daily driver. Still I wished I had tried it on my workstation. My NUC is a little busy and headless...
> 
> Just RAM and disk is your choice, and optional Wi-Fi/Bluetooth.

It could be interesting to compare with the [PAPI](https://bitbucket.org/icl/papi/src/master/) measurements. You may have the PAPI package in your Linux distribution or clone/build. PAPI depends on the msr.ko driver and probably Perf.

cyring, 2019-10-16:
As a reminder: MSR `0x64d` for the SOC Power Domain added on the development roadmap.

olejon, 2019-10-16:
I'll see what I can do and check in to that when I have time, if it isn't to time expensive. Watching a documentary now (news & documentary slave here).

Just one of the cool things one can do with NUCs for instance; buy a cheap HDMI-CEC chip and connect it to the right pins on the (tiny) MB, and boom it appears as a HDMI-CEC adapter in Kodi (formerly XBMC), and such worked in harmony with my AV Receiver and TV when I used it as an HTPC. RPi has HDMI-CEC out of the box though. Last spec sheet I saw NUCs says HDMI-CEC supported (with mod). Also a USB to HDMI passthrough can be used for those less tech savvy.

Having a BIOS update in 2019, with this CPU from 2012 I think, shows that Intel cares about these things at least. And Canonical probably supported them officially from Ubuntu 12.04, if not 14.04 for sure. And then surely Red Hat. That means 2 major industry players supporting this thing for over 6 years now.

That's actually better than a Mac Mini... Which it's kind of a brilliant substitution for if you want that form factor as a desktop PC, for instance. No doubt it will be supported on Ubuntu 20.04 as well so 12+ years of **official Linux support** (Intel supported Linux on it from day 1, kind of why I bought it, comes with no OS, no "MS tax"). Probably best tech investment ever, considering price, different usage over the years, power consumption, just worked and works, as server or desktop. Has plenty of I/O ports... Went for the i3 model over the Celeron for future security, and SATA SSD version (which is now cheaper than the one with M.2), since there was no M.2 and the expansion disk card thing was kind of, well, bad reviews. M.2 NVMEs didn't exist basically. Fast but cheap enough SSD, didn't need 1 TB or anything, and kind of blew my mind, just like, very easy to upgrade RAM and disk, how nice, quiet and fast it was, even with "Ultra low TDP". And everything just worked, and still does. Only think lacking is UEFI support for BIOS updates through the Internet directly. Have to physically move it too, to get it to a monitor so. And cable and keyboard and USB with BIOS. Very easy and well made process, and as said still BIOS updates, just the hassle in *my case now as a server*.

cyring, 2019-10-16:
> Anyway would be really cool if you checked out Clear Linux (separate disk or whatever).

I would prefer you create another issue for this request but
_CoreFreq_ is dkms ready. dkms.conf released [here](https://github.com/cyring/CoreFreq/blob/master/dkms.conf)
One has to follow this Clear Linux [procedure](https://docs.01.org/clearlinux/latest/guides/kernel/kernel-modules-dkms.html#kernel-modules-dkms)  
Thanks to a nice contribution, `Makefile` has been enhanced with `install` and `module_install`  
Just type `make help`

cyring, 2019-10-16:
> As a reminder: MSR `0x64d` for the SOC Power Domain added on the development roadmap.

As mentioned In the SDM specifications, there is no guarantee this counter exists for the listed architecture families. Tested with a Skylake i7-6700, this MSR returns a zero value on all cores.

cyring, 2019-10-21:
* Bellow my tests with version 1.67.6
[image]
* Need to fix the Power Formula Scope at this line:
https://github.com/cyring/CoreFreq/blob/bb548ebd9bdd5df210a6cf51f30b932a59857f7c/coretypes.h#L342
```c
enum POWER_FORMULAS {
	POWER_FORMULA_NONE =						\
	(0b000000000000000000000000 << 8) | FORMULA_SCOPE_NONE,
	POWER_FORMULA_INTEL =						\
	(0b000000000000000000000001 << 8) | FORMULA_SCOPE_NONE,
	POWER_FORMULA_INTEL_ATOM =					\
	(0b000000000000000000000011 << 8) | FORMULA_SCOPE_NONE,
	POWER_FORMULA_AMD =						\
	(0b000000000001000000000000 << 8) | FORMULA_SCOPE_CORE,
	POWER_FORMULA_AMD_17h =						\
	(0b000100000001000000000000 << 8) | FORMULA_SCOPE_CORE
};
```
* Rebuild all and test again
[image]
* Results as expected for this architecture:
1. Voltage per Package
2. Temperature per Thread
3. No Power per CPU
4. Power for whole Package

* Please, let me know about yours -;)

adatum, 2019-10-21:
[image]

cyring, 2019-10-21:
@adatum : thank you
* Cores power looks OK
* Single temperature sensor: OK
* Voltage per Core or per SMT: I'm not sure what to display ?
   If a single CPU is stressed (first a CPU Core, next a CPU thread), do we read discret Vcore ?

adatum, 2019-10-21:
Looks like the voltages per core are independent, but the threads on the same core (threads 2 & 10 in screenshot) show the same voltage:

[image]

cyring, 2019-10-21:
> Looks like the voltages per core are independent, but the threads on the same core (threads 2 & 10 in screenshot) show the same voltage:

My understanding of the Zen SMT architecture is that CPU 10 is the logical peer of the core CPU 2, and they have the Vcore in common. 
The view is showing the relation between them.

olejon, 2019-10-22:
You can count 8 "down", because you have 16 threads, it seems, because for me it's counting 6 "down", since having 12 threads (3600X).

In my case, with 6c/12t, kind of makes sense stressing say core 0 you can always count down 6 steps to get to the other showing same voltage, so 000 (from CCX0) 006 from (from CCX1) shows the same voltage. Same goes if you stress something from CCX1 like 010, it'll be 004 (CCX0) that shows the same voltage. So they're spread as efficient as possible throughout to give the best performance. 

Anyway better to read; think it's shown and explained pretty ok in the article and picture in it below, but nothing shows it better than Ryzen Master, which also shows the best performing Cores or CCXs (one of them) with a yellow star, best silicon basically I suppose, those who'll probably respond best to live overclock:

https://www.tomshardware.com/reviews/amd-ccx-definition-cpu-core-explained,6338.html

The picture suggests this is good for heat spreading and hence performance, makes more cores easier then, as we see compared to Intel pricing, durability etc. Intel has nothing like this. They had this weird 2 CPU in one SKU architecture *many years back* (was on LTT YouTube channel for some retro "how well does it run"), with some bridge. Failed architecture.

adatum, 2019-10-22:
> CPU 10 is the logical peer of the core CPU 2, and they have the Vcore in common.

Yes, that's what I meant to highlight. It makes sense for the two virtual cores (what I meant by "threads") to have the same Vcore since it is the same physical core. I'm not sure if that necessarily has to be the case, but at least it makes sense.

olejon, 2019-10-22:
> Yes, that's what I meant to highlight. It makes sense for the two virtual cores (what I meant by "threads") to have the same Vcore since it is the same physical core. I'm not sure if that necessarily has to be the case, but at least it makes sense

Yeah. 2 and 10 = 8 between. For me it'll be 2 and 8 = 6 between.

cyring, 2019-10-22:
The Topology is the clue. Just checking at the screenshots inside the Wiki CPU Support and both 2700X and 3600X have a similar Topology, where the (CCX, CoreID, ThreadID) forms the cluster. 
Tests will also be interesting when facing the topology of the (dual EPYC) [https://gist.github.com/cyring/41f61cfa1db0c01c17303d47a46b3df0]

olejon, 2019-10-22:
Seems like the new Threadrippers will be called 3960X, 3970X and 3990X. At least AMD has a much more understandable *MODEL* naming than Intel (which is all over the place now), and for *generations* they are simply logical incremental numbers, with none only for laptops and skips desktops etc.

* Entry level desktop SKU: 3600 (that and all above has an X version simply meaning better clock speed and higher TDP than non-X versions)
* Laptop SKUs are reserved for lower numbers than 3600 (and AMD in their keynote had a lot more already committed laptop manufacturers on board for 3rd Gen than anyone had expected from AMD)
* Best regular consumer SKU: 3950X
* Top of the line SKUs: 3960X and above, bearing the name Threadripper too, and as prices leaked will reflect they're for "special use cases" like Xeon (which also has a disaster of model naming IMO) instead of (Intel) Core iX, which used to have an OK model naming until Coffee Lake, now caos
* I assume these 3960X+ will require another socket, and MB chipset differences, this and price will avoid people buying incompatible CPUs for their MB or another way around. *Hypothetically, if not, and one can be popped into a supported X570 card with a good enough PSU, it would be awesome if course*, but doubt that (my ASUS MB listed the 3950X already as supported before even having a confirmed launch date, but no 3960X etc).

[YouTube video worth a watch, just 7 minutes, about Intel's naming frustration referred to by many other YouTubers lately which cancels of 7 nm, only something for laptops and really just minor freq bumps for desktops in the end](https://youtu.be/1eiI4dxjnVU)

[AMD is planning three Threadripper 3000 SKUs: 3960X, 3970X and 3990X. The first two will be unveiled on November 5th, but the embargo on sales and reviews will be lifted on November 19th](https://videocardz.com/82293/exclusive-amd-ryzen-threadripper-3960x-3970x-and-3990x-launch-dates-leaked)

[New AMD Ryzen Threadripper shows off: hugely faster than the old 32-core](New AMD Ryzen Threadripper shows off: hugely faster than the old 32-core)

With the 3950X and X570 these are for extreme use cases, and enthusiasts like here with the $$$. 

It might be cheaper to buy a decent consumer 3rd Gen now, save a lot of money by doing that, even if it's "just" the 3900X and not the 3950X, and when 4th Gen is released you can simply pop in one of the best 4th Gen consumer SKUs and it may already on par with say at least the 3960X Threadripper of today... IDK but AMD could very well do some serious core count increase on the better 4th Gens, as they as well as Intel seem to have reached the frequency limit anyway. Heck Intel doesn't supply any stock cooler to their top of the line i9-9900K, but recommends "a good cooling solution". AMD has one even for the 3900X and surely will use the same for the 3950X.

AMD's game seems to be set on cores and smart design there. OC is very limited even using sub-zero cooling etc. The September BIOS that all serious MB manufacturers have shipped now should have fixed Precision Boost Override to do what it's supposed to (which many people have misunderstood), but you might want to wait, or at least search first, as some have experienced bricked system in certain MB + CPU configs.

I won't roll back BIOS, but it seems to be doing its thing now, at least it does something where before tests didn't really show any improvement on many configs. But to check I really have to boot Windows, check if a Ryzen Master update may be available, reset to default profile, then set to same "Auto Overclock" included safe profile and run GeekBench, Blender Benchmark (and CPU-Z I guess). Just glimpsing at CoreFreq it seems now that yeah, when a few cores (constantly changing to best available CCX) are used to the max and the others are hovering at some lower frequencies (typical for many tasks from games, booting to starting a program that does a little heavy work at first). Ryzen Master also shows when a Core or a whole CCX is completely Idle. The manual for Ryzen Master is just an HTML file which explains things OK enough, surely can be found on the web, but is included with the program and it's nice to see a single HTML document instead of say a PDF with embedded images etc.

Let's see if Linux still beats Windows by 4 minutes (!) with Blender Benchmark, the program that most stresses my CPU for longest. Many mistakes it for Blender, it's a separate program from Blender Foundation, but just for the 2 common benchmarks rendered in the background (BMW and Classroom). Just search it and make sure to download that because it's easy to end up on the main Blender website and download the entire Blender instead - filename says it all... still in Beta but works cross-platform just fine as it includes all it's libs and files so results shouldn't depend on distro or Windows version too much.

As in the Windows Insider Program, I got a significant update some weeks ago, that might have Ryzen improvements. One of those "enough" to show "Hi, we've got some news for you", and some setup stuff as if it was an almost clean install, mostly nagging to turn on more stuff to be sent to MS IIRC, and instead of a Skip button it was "Remind me later". Well that's part of the Insider Program. Fine, I chose it. Some Telemetry you can't turn off, or at least if you do Windows Update will temporarily stop Insider updates and notify which settings must be such and such. But Ubuntu LTS HWE stack has gotten a couple of kernel and graphics stack updates too, which made wonders for my say PCIe4 NVMEs' performance so... Some backports for Ryzen/X570 there too I guess. Well I'll check benchs later. Have previous for comparison. On Windows gotta check if new chipset drivers from ASUS too of course, ugh, although Windows Update installs everything you need, I want the AMD signed drivers, not the MS ones, and the version number is usually higher.

olejon, 2019-10-22:
Few more links and edits.

olejon, 2019-10-22:
Found the right YouTube videos. Even if not having that HW, pretty useful to know and informative.

And about Intel, shortages and what everybody feels about their messy naming now.

And the others regarding Threadripper, if you're not totally up to date on model naming an such. I guess so.

cyring, 2019-10-22:
@olejon : nice to read from you but do you mind to move your 2 last comments to our dedicated tchat thread.
I believe you'll understand that I want to focus on specific and tiny details such as this RAPL subject
See U soon on the tchat thread 
Regards 
Cyril

olejon, 2019-10-22:
Oh darn, true dat! Will do. Will delete this one afterwards. At least think links can be useful for people interested in Ryzen OC and news etc.

[14 more replies]

closed 2020-02-27 by cyring