cyring/CoreFreq #64: 2x Broadwell Xeon E5-2620 v4: Kernel panic on insmod in Corefreq livecd and centos 7.4

altmind, 2018-07-13:
Insmod'ing a corefreq kmod causes kernel panic, tested on both stock centos 7.4(tested both 3.10.0-514.2.2 and 3.10.0-862.6.3) and corefreq livecd.

The system is Supermicro X10DAL-i with TWO E5-2620 v4 @ 2.10GHz (fam: 06, model: 4f, stepping: 01)
I've atached some helpful info from centos crash(as corefreq livecd dont leave crash reports on disk):

[127.0.0.1-2018-07-12-20-26-38-dmesg.zip](https://github.com/cyring/CoreFreq/files/2193193/127.0.0.1-2018-07-12-20-26-38-dmesg.zip)
[127.0.0.1-2018-07-12-20-41-40-dmesg.zip](https://github.com/cyring/CoreFreq/files/2193194/127.0.0.1-2018-07-12-20-41-40-dmesg.zip)
[cpuinfo.zip](https://github.com/cyring/CoreFreq/files/2193195/cpuinfo.zip)
[dmidecode.zip](https://github.com/cyring/CoreFreq/files/2193196/dmidecode.zip)
[lshw.zip](https://github.com/cyring/CoreFreq/files/2193197/lshw.zip)

cyring, 2018-07-13:
Hello.
Thank you: Xeon has barely been tested.
What we can try is to start the driver with the most compatible architecture:
```
insmod corefreqk.ko ArchID=0
```
If no crash happens then I will specialized code for BDW-EX/EP and progressively chasse bugs.

In this header file are listed all the architectures with associated id, starting from zero for a 64 bits genuine Intel.
https://github.com/cyring/CoreFreq/blob/cf35ac1057fc973b2b706e96d94ce28d5d34fb4e/corefreqk.h#L998

When testing with the Live CD:
Do a CTRL+C to abort the _CoreFreq_ start during the 9 sec countdown. 
Next, change to directory /opt/CoreFreq
Build with `make`

insmod with ArchID=0 
then rmmod
Repeat with a upper and compatible architecture, probab

altmind, 2018-07-13:
Attaching some disasm from the start of Start_Haswell_ULT to crash site.

Seems like a crash inside the if statement somewhere around https://github.com/cyring/CoreFreq/blob/master/corefreqk.c#L5333

Corefreq commit cf35ac1057fc973b2b706e96d94ce28d5d34fb4e

[crash.zip](https://github.com/cyring/CoreFreq/files/2193699/crash.zip)

altmind, 2018-07-13:
Yes, ArchID=0 helped, no crash now. Checking if the reported values look sensible.

altmind, 2018-07-13:
JFYI, the same version of corefreq also cause kernel panic on dual xeon E5-2687W v3 @ 3.10GHz(HP  ProLiant DL360 Gen9).

Unfortunately, cannot produce much info about the crash as this is a remote server with crash dumps disabled.

altmind, 2018-07-13:
ArchID=27(SandyBridge_EP) also gives a kernel panic :(

cyring, 2018-07-13:
With RCX=0x630 the C8 Uncore counter msr is unavailable in Broadwell-EP and probably C9 and C10 .

Haswell-ULT is one of the few Xeon I have a chance to test _CoreFreq_ with.

What I suggest is to remove those msr readings, build and try again.

In `corefreqk.h`:
```
#define PKG_Counters_Haswell_ULT(Core, T)				\
({									\
	RDTSCP_COUNTERx7(Proc->Counter[T].PTSC,				\
			MSR_PKG_C2_RESIDENCY, Proc->Counter[T].PC02,	\
			MSR_PKG_C3_RESIDENCY, Proc->Counter[T].PC03,	\
			MSR_PKG_C6_RESIDENCY, Proc->Counter[T].PC06,	\
			MSR_PKG_C7_RESIDENCY, Proc->Counter[T].PC07,	\
})
```

altmind, 2018-07-13:
with only c2,c3,c6,c7, still getting a panic 
[crash2.txt](https://github.com/cyring/CoreFreq/files/2193995/crash2.txt)

cyring, 2018-07-13:
About the E5-2687W ([SandyBridge-EP](https://ark.intel.com/en/products/64582/Intel-Xeon-Processor-E5-2687W-20M-Cache-3_10-GHz-8_00-GTs-Intel-QPI)): can you start driver with `ArchID=0`
If ok, please run the daemon then post back the result of `corefreq-cli -s`

altmind, 2018-07-13:
re E5-2687W, will do in 3 hrs.

cyring, 2018-07-13:
The BDW-EP panic is due to MSR_PP1_ENERGY_STATUS: this msr(0x641) is unavailable.
In `corefreqk.c` : can you change the following macro to:
```
#define PWR_ACCU_SandyBridge(Pkg, T)					\
({									\
	RDCOUNTER(Pkg->Counter[T].Power.ACCU[PWR_DOMAIN(PKG)],		\
						MSR_PKG_ENERGY_STATUS); \
									\
	RDCOUNTER(Pkg->Counter[T].Power.ACCU[PWR_DOMAIN(CORES)],	\
						MSR_PP0_ENERGY_STATUS); \
									\
})
```
Build and test again.

(_I can wait for the other Xeon: Paris is falling to sleep_)

altmind, 2018-07-13:
dual xeon E5-2687W v3 @ 3.10GHz(HP ProLiant DL360 Gen9) with ArchID=0 loads, but gives incorrect freq(the server is running in loop on 6 cores, not on all of them)
[image]
[image]

altmind, 2018-07-13:
Some info from crashes on HP dual xeon E5-2687W v3(Sandy bridge):
[496724.467268] RIP: 0010:[<ffffffffc0692e5e>]  [<ffffffffc0692e5e>] Start_Uncore_SandyBridge+0x1e/0xa0 [corefreqk]
[496724.467436] RSP: 0018:ffff880466e4bae8  EFLAGS: 00010002
[496724.467519] RAX: 0000000000000001 RBX: ffff88090c21d000 RCX: 0000000000000391
[496724.467630] RDX: 004786cf0e5fd76a RSI: ffff88090c21e000 RDI: 0000000000000000
[496724.467740] RBP: ffff880466e4bae8 R08: 0000000000000001 R09: 000000000000003e

Dmesg lshw dmidecode and backtrace/disasm for vmcore:
[crash2.zip](https://github.com/cyring/CoreFreq/files/2194448/crash2.zip)
[dmesg2.zip](https://github.com/cyring/CoreFreq/files/2194449/dmesg2.zip)
[dmidecode2.zip](https://github.com/cyring/CoreFreq/files/2194450/dmidecode2.zip)
[lshw2.zip](https://github.com/cyring/CoreFreq/files/2194451/lshw2.zip)

cyring, 2018-07-14:
In this commit I'm fixing the CPUID signatures table for Broadwell Xeon D and Broadwell Xeon EP/EX

With the `Experimental=1` argument the driver will try to activate the Uncore Fixed Counter.

cyring, 2018-07-16:
Some fixes are pushed. 
Please let me know if _CoreFreq_ works fine with both processors  ?

altmind, 2018-07-17:
The newest version works on the E5-2620 v4(first reported), cannot test on the second box yet.

On the bright side, I can confirm corefreq works good on i7-8700k(kaby/coffee lake) :)

altmind, 2018-07-17:
Checked on E5-2687W v3, works good.
Thank you, I think this issue is resolved.

altmind, 2018-07-17:
Also works on E5-2689 v4.

cyring, 2018-07-18:
wahoo, that's a lot of tested architectures: thank you very much.
Did you keep some outputs  ? Especially the topology of cores on the multi-processors  ? 
Regards

closed 2018-07-18 by cyring

altmind, 2018-07-18:
I did not keep any info on systems(though I can get this info any time).
```
 CoreFreq]$ ./corefreq-cli -m
CPU Pkg  Apic  Core Thread  Caches      (w)rite-Back (i)nclusive
 #   ID   ID    ID     ID  L1-Inst Way  L1-Data Way      L2  Way      L3  Way
00: BSP     0     0      0   32768  8     32768  8    262144  8  20971520 20 i
01:   0     2     1      0   32768  8     32768  8    262144  8  20971520 20 i
02:   0     4     2      0   32768  8     32768  8    262144  8  20971520 20 i
03:   0     6     3      0   32768  8     32768  8    262144  8  20971520 20 i
04:   0     8     4      0   32768  8     32768  8    262144  8  20971520 20 i
05:   0    10     5      0   32768  8     32768  8    262144  8  20971520 20 i
06:   0    12     6      0   32768  8     32768  8    262144  8  20971520 20 i
07:   0    14     7      0   32768  8     32768  8    262144  8  20971520 20 i
08:   1    16     0      0   32768  8     32768  8    262144  8  20971520 20 i
09:   1    18     1      0   32768  8     32768  8    262144  8  20971520 20 i
10:   1    20     2      0   32768  8     32768  8    262144  8  20971520 20 i
11:   1    22     3      0   32768  8     32768  8    262144  8  20971520 20 i
12:   1    24     4      0   32768  8     32768  8    262144  8  20971520 20 i
13:   1    26     5      0   32768  8     32768  8    262144  8  20971520 20 i
14:   1    28     6      0   32768  8     32768  8    262144  8  20971520 20 i
15:   1    30     7      0   32768  8     32768  8    262144  8  20971520 20 i
16:   0     1     0      1   32768  8     32768  8    262144  8  20971520 20 i
17:   0     3     1      1   32768  8     32768  8    262144  8  20971520 20 i
18:   0     5     2      1   32768  8     32768  8    262144  8  20971520 20 i
19:   0     7     3      1   32768  8     32768  8    262144  8  20971520 20 i
20:   0     9     4      1   32768  8     32768  8    262144  8  20971520 20 i
21:   0    11     5      1   32768  8     32768  8    262144  8  20971520 20 i
22:   0    13     6      1   32768  8     32768  8    262144  8  20971520 20 i
23:   0    15     7      1   32768  8     32768  8    262144  8  20971520 20 i
24:   1    17     0      1   32768  8     32768  8    262144  8  20971520 20 i
25:   1    19     1      1   32768  8     32768  8    262144  8  20971520 20 i
26:   1    21     2      1   32768  8     32768  8    262144  8  20971520 20 i
27:   1    23     3      1   32768  8     32768  8    262144  8  20971520 20 i
28:   1    25     4      1   32768  8     32768  8    262144  8  20971520 20 i
29:   1    27     5      1   32768  8     32768  8    262144  8  20971520 20 i
30:   1    29     6      1   32768  8     32768  8    262144  8  20971520 20 i
31:   1    31     7      1   32768  8     32768  8    262144  8  20971520 20 i
 CoreFreq]$ ./corefreq-cli -s
Processor                            [Intel(R) Xeon(R) CPU E5-2620 v4 @ 2.10GHz]
|- Architecture                                                [Broadwell/EP/EX]
|- Vendor ID                                                      [GenuineIntel]
|- Signature                                                            [ 06_4F]
|- Stepping                                                             [     1]
|- Microcode                                                            [     0]
|- Online CPU                                                           [ 32/32]
|- Base Clock                                                           [100.00]
|- Ratio Limited                                                        [  LOCK]
|- Frequency            (Mhz)                      Ratio
                 Min   1200.04                    [  12 ]
                 Max   2100.07                    [  21 ]
|- Factory
                       2100                       [  21 ]
|- Turbo Boost
                  1C   3000.10                    [  30 ]
                  2C   3000.10                    [  30 ]
                  3C   2800.09                    [  28 ]
                  4C   2700.09                    [  27 ]
                  5C   2600.08                    [  26 ]
                  6C   2500.08                    [  25 ]
                  7C   2400.08                    [  24 ]
                  8C   2300.07                    [  23 ]
                  9C   2300.07                    [  23 ]
                 10C   2300.07                    [  23 ]
                 11C   2300.07                    [  23 ]
                 12C   2300.07                    [  23 ]
                 13C   2300.07                    [  23 ]
                 14C   2300.07                    [  23 ]
                 15C   2300.07                    [  23 ]
                 16C   2300.07                    [  23 ]
                 17C   2300.07                    [  23 ]
[... 92 lines cut ...]
|                      0       2       1       2       0
|- Core Cycles                                                         [Present]
|- Instructions Retired                                                [Present]
|- Reference Cycles                                                    [Present]
|- Last Level Cache References                                         [Present]
|- Last Level Cache Misses                                             [Present]
|- Branch Instructions Retired                                         [Present]
|- Branch Mispredicts Retired                                          [Present]

Power & Thermal Monitoring:
[... cut ...]
|- Power Limit Notification                                      PLN   [Present]
|- Package Thermal Management                                    PTM   [Present]
|- Thermal Monitor 1                                         TM1|TTP   [ Enable]
|- Thermal Monitor 2                                         TM2|HTC   [Present]
|- Units
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000061035]
   |- Window                                            second   [  0.000976562]
```