cyring/CoreFreq #68: Vcore always set to 0 in System Information json-encoded

leinardi, 2018-08-04:
It seems that the json-encoded System Information is always showing a value of 0 for the Vcore (`"Voltage":{"VID":9082,"Vcore":0}`), even if the dashboard view shows the correct value.

Another thing that I noticed is that only one CPU is showing VID and Vcore values, the others are all showing 0.

[image]

```
Processor                             [Intel(R) Core(TM) i7-4770K CPU @ 3.50GHz]
|- Architecture                                                [Haswell/Desktop]
|- Vendor ID                                                      [GenuineIntel]
|- Signature                                                            [ 06_3C]
|- Stepping                                                             [     3]
|- Microcode                                                            [     0]
|- Online CPU                                                           [  8/8 ]
|- Base Clock                                                           [ 97.23]
|- Frequency            (Mhz)                      Ratio                        
                 Min    777.84                    [   8 ]                       
                 Max   3403.03                    [  35 ]                       
|- Factory                                                              [100.00]
                       3500                       [  35 ]                       
|- Turbo Boost                                                          [UNLOCK]
                  1C   3791.95                    <  39 >                       
                  2C   3791.95                    <  39 >                       
                  3C   3694.72                    <  38 >                       
                  4C   3597.49                    <  37 >                       
|- Uncore                                                               [UNLOCK]
                 Min    777.84                    <   8 >                       
                 Max   3403.03                    <  35 >                       
                                                                                
ISA Extensions:                                                                 
|- 3DNow!/Ext [N,N]        AES [Y]   AVX/AVX2 [Y/Y]   BMI1/BMI2 [Y/Y] 
|- CLFSH        [Y]       CMOV [Y]      CMPXCH8 [Y]      CMPXCH16 [Y] 
|- F16C         [Y]        FPU [Y]         FXSR [Y]     LAHF/SAHF [Y] 
|- MMX/Ext    [Y/N]    MONITOR [Y]        MOVBE [Y]      PCLMULDQ [Y] 
|- POPCNT       [Y]     RDRAND [Y]       RDTSCP [Y]           SEP [Y] 
|- SGX          [N]        SSE [Y]         SSE2 [Y]          SSE3 [Y] 
|- SSSE3        [Y] SSE4.1/4A [Y/N]      SSE4.2 [Y]       SYSCALL [Y] 
                                                                                
Features:                                                                       
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Present]
|- CPL Qualified Debug Store                                  DS-CPL   [Present]
|- 64-Bit Debug Store                                         DTES64   [Present]
|- Fast-String Operation                                Fast-Strings   [Present]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
|- Pending Break Enable                                          PBE   [Present]
|- Process Context Identifiers                                  PCID   [Present]
|- Perfmon and Debug Capability                                 PDCM   [Present]
|- Page Global Enable                                            PGE   [Present]
|- Page Size Extension                                           PSE   [Present]
|- 36-bit Page Size Extension                                  PSE36   [Present]
|- Processor Serial Number                                       PSN   [Missing]
|- Restricted Transactional Memory                               RTM   [Missing]
|- Safer Mode Extensions                                         SMX   [Missing]
[... 41 lines cut ...]
|- Core Cycles                                                         [Present]
|- Instructions Retired                                                [Present]
|- Reference Cycles                                                    [Present]
|- Last Level Cache References                                         [Present]
|- Last Level Cache Misses                                             [Present]
|- Branch Instructions Retired                                         [Present]
|- Branch Mispredicts Retired                                          [Present]
                                                                                
Power & Thermal Monitoring:                                                     
|- Clock Modulation                                             ODCM   <Disable>
   |- DutyCycle                                                        <  6.25%>
|- Power Management                                         PWR MGMT   [   LOCK]
   |- Energy Policy                                        Bias Hint   [      0]
|- Junction Temperature                                        TjMax   [    100]
|- Digital Thermal Sensor                                        DTS   [Present]
|- Power Limit Notification                                      PLN   [Present]
|- Package Thermal Management                                    PTM   [Present]
|- Thermal Monitor 1                                         TM1|TTP   [ Enable]
|- Thermal Monitor 2                                         TM2|HTC   [Present]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000061035]
   |- Window                                            second   [  0.000976562]

```

cyring, 2018-08-04:
The register `IA32_PERF_STATUS MSR (0198H)` is specified as a Package scope register.
However this MSR is poorly documented and might provide VID per Core ?

Each next comments below will tell you what code to add in _CoreFreq_ files to test voltage per Core

1. add these two functions above line:
https://github.com/cyring/CoreFreq/blob/c5a96e24a80993b2dfb24763d3207ea9736e1e48/corefreqk.c#L5700
```
static enum hrtimer_restart Cycle_Haswell_DT(struct hrtimer *pTimer)
{
	PERF_STATUS PerfStatus = {.value = 0};
	unsigned int cpu = smp_processor_id();
	CORE *Core = (CORE *) KPublic->Core[cpu];

	if (BITVAL(KPrivate->Join[cpu]->TSM, MUSTFWD) == 1) {
		hrtimer_forward(pTimer,
				hrtimer_cb_get_time(pTimer),
				RearmTheTimer);

		SMT_Counters_SandyBridge(Core, 1);

		if (Core->Bind == Proc->Service.Core) {
			PKG_Counters_SandyBridge(Core, 1);

			PWR_ACCU_SandyBridge(Proc, 1);

			Delta_PC02(Proc);

			Delta_PC03(Proc);

			Delta_PC06(Proc);

			Delta_PC07(Proc);

			Delta_PTSC(Proc);

			Delta_UNCORE_FC0(Proc);

			Delta_PWR_ACCU(Proc, PKG);

			Delta_PWR_ACCU(Proc, CORES);

			Delta_PWR_ACCU(Proc, UNCORE);

			Save_PC02(Proc);

			Save_PC03(Proc);

			Save_PC06(Proc);

			Save_PC07(Proc);

			Save_PTSC(Proc);

			Save_UNCORE_FC0(Proc);

			Save_PWR_ACCU(Proc, PKG);

			Save_PWR_ACCU(Proc, CORES);

			Save_PWR_ACCU(Proc, UNCORE);

			Sys_Tick(Proc);
		}

		RDMSR(PerfStatus, MSR_IA32_PERF_STATUS);
		Core->Counter[1].VID = PerfStatus.SNB.CurrVID;

		Core_Intel_Temp(Core);

		RDCOUNTER(Core->Interrupt.SMI, MSR_SMI_COUNT);

		Delta_INST(Core);

		Delta_C0(Core);

		Delta_C3(Core);

		Delta_C6(Core);

		Delta_C7(Core);

		Delta_TSC(Core);

		Delta_C1(Core);

		Save_INST(Core);

		Save_TSC(Core);

		Save_C0(Core);

		Save_C3(Core);

		Save_C6(Core);

		Save_C7(Core);

		Save_C1(Core);

		BITSET(LOCKLESS, Core->Sync.V, 63);

		return(HRTIMER_RESTART);
	} else
		return(HRTIMER_NORESTART);
}

void InitTimer_Haswell_DT(unsigned int cpu)
{
	smp_call_function_single(cpu, InitTimer, Cycle_Haswell_DT, 1);
}
```

cyring, 2018-08-04:
2. add this code below this line:
https://github.com/cyring/CoreFreq/blob/c5a96e24a80993b2dfb24763d3207ea9736e1e48/corefreqk.h#L499
```
extern void InitTimer_Haswell_DT(unsigned int cpu) ;
```

cyring, 2018-08-04:
3. replace the timer at this line (Architecture number 30)
https://github.com/cyring/CoreFreq/blob/c5a96e24a80993b2dfb24763d3207ea9736e1e48/corefreqk.h#L1707
```
	.Timer = InitTimer_Haswell_DT,
```

cyring, 2018-08-04:
4. in the same structure (Arch number 30) replace also:
```
.voltageFormula = VOLTAGE_FORMULA_INTEL_SNB,
```
with:
```
.voltageFormula = VOLTAGE_FORMULA_INTEL_SKL_X,
```

cyring, 2018-08-04:
Now `make clean; make` and start _CoreFreq_
Please post a screenshot of Voltages with different load on Cores

leinardi, 2018-08-04:
Hey thanks for the quick reply!

Now the Vcore is correctly displayed for all the cores inside the Dasboard:
[image]

[image]

To make it work I had to do also a `sudo rmmod corefreqk.ko` and `sudo insod corefreqk.ko` after building.

But unfortunately in the json export is still set to 0:
[image]

This is the patch of the changes I made, just in case: 
[0001-WIP.patch.txt](https://github.com/cyring/CoreFreq/files/2259557/0001-WIP.patch.txt)

If you want me to try something else feel free to push a branch ore give me a patch file.

cyring, 2018-08-04:
For JSON please ask its [author](https://github.com/altmind/CoreFreq) to Pull Request me a fix

About Vcore: according to the screenshot, it's a **Core only** register.

1. in the function `Cycle_Haswell_DT`, replace these lines:
```
		RDMSR(PerfStatus, MSR_IA32_PERF_STATUS);
		Core->Counter[1].VID = PerfStatus.SNB.CurrVID;
```
with:
```
    if (Core->T.ThreadID == 0) {
		RDMSR(PerfStatus, MSR_IA32_PERF_STATUS);
		Core->Counter[1].VID = PerfStatus.SNB.CurrVID;
    }
```

2. rebuild and restart the whole thing.

3. apply load individually on each physical Core and check if Voltage is unique

Remark: in the UI, go to Window > Topology, each physical core has a thread id equal to zero

cyring, 2018-08-04:
Thanks a lot for the patch.

leinardi, 2018-08-04:
> For JSON please ask its author to Pull Request me a fix

Unfortunately the issue tracker is disabled on the fork you linked.

I will try asking here: @altmind could you please have a look at this issue? The Vcore is always 0 in the json export.

> About Vcore: according to the screenshot, it's a Core only register.

[image]

[image]

[image]

cyring, 2018-08-04:
The physical Cores are CPU numbers 0, 1, 2, and 3
Can you try to bind the load on one of them using `taskset` for example on CPU number 0 because I'm not sure the register is per Core ?

cyring, 2018-08-04:
Sorry I meant `taskset -c 0` to bind on CPU number 0

leinardi, 2018-08-04:
```
$ taskset -cp 0 27229
pid 27229's current affinity list: 0-7
pid 27229's new affinity list: 0
```

[image]

cyring, 2018-08-04:
VID (and thus Vcore) is the same on any Core with the load bound to only one CPU
The reading is per Package only.
Do you have any other tool that shows different Vcore ?

leinardi, 2018-08-04:
I am not aware of any other tool, if you some some tell me and I can try it.

cyring, 2018-08-04:
1. `sensors` which through the ACPI interface provides the following on my Xeon W3690
```
atk0110-acpi-0
Adapter: ACPI interface
3.3V Voltage:           +3.26 V  (min =  +2.97 V, max =  +3.63 V)
5V Voltage:             +5.02 V  (min =  +4.50 V, max =  +5.50 V)
12V Voltage:           +11.97 V  (min = +10.20 V, max = +13.80 V)
DRAM Bus Voltage:       +1.65 V  (min =  +1.40 V, max =  +1.90 V)
CPU Voltage:            +0.95 V  (min =  +0.80 V, max =  +1.80 V)
ICH Voltage:            +1.11 V  (min =  +0.90 V, max =  +1.35 V)
ICH PCIE Voltage:       +1.51 V  (min =  +1.20 V, max =  +1.80 V)
CPU PLL Voltage:        +1.82 V  (min =  +1.50 V, max =  +2.00 V)
IOH PCIE Voltage:       +1.51 V  (min =  +1.20 V, max =  +1.80 V)
IOH Voltage:            +1.12 V  (min =  +0.90 V, max =  +1.35 V)
QPI/DRAM Core Voltage:  +1.17 V  (min =  +0.80 V, max =  +1.50 V)
```
You will notice that only one Voltage CPU is retreived

2. Your system BIOS should show the Voltage Core: check if it is per Core or for the whole Package

leinardi, 2018-08-04:
```
$ sensors
nct6791-isa-0290
Adapter: ISA adapter
Vcore:                  +0.90 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +1.00 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
AVCC:                   +3.34 V  (min =  +2.98 V, max =  +3.63 V)
+3.3V:                  +3.34 V  (min =  +2.98 V, max =  +3.63 V)
in4:                    +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                    +1.99 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                    +0.43 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
3VSB:                   +3.42 V  (min =  +2.98 V, max =  +3.63 V)
Vbat:                   +3.36 V  (min =  +2.70 V, max =  +3.63 V)
in9:                    +1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in11:                   +0.85 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +0.50 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +0.57 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +0.58 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                   667 RPM  (min =    0 RPM)
fan2:                   694 RPM  (min =    0 RPM)
fan3:                   596 RPM  (min =    0 RPM)
fan4:                   609 RPM  (min =    0 RPM)
fan5:                   613 RPM  (min =    0 RPM)
fan6:                   631 RPM
SYSTIN:                 +38.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
CPUTIN:                 +44.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +81.0°C    sensor = thermistor
AUXTIN1:               -128.0°C    sensor = thermistor
AUXTIN2:                +34.0°C    sensor = thermistor
AUXTIN3:                +50.0°C    sensor = thermistor
PECI Agent 0:           +47.5°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled

acpitz-virtual-0
Adapter: Virtual device
temp1:        +27.8°C  (crit = +105.0°C)
temp2:        +29.8°C  (crit = +105.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +58.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +58.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:        +47.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:        +41.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:        +38.0°C  (high = +80.0°C, crit = +100.0°C)
```

I can't reboot now to check the BIOS but I remember there it was shown only one Vcore.

leinardi, 2018-08-04:
I am not familiar with these features, if you give me some step by step instructions on how to test I will be happy to do it (also, please, tell me also for each test how dangerous it is, I don't want to end up setting 2V as Vcore :D).

cyring, 2018-08-04:
Indeed a step by step instructions is missing, but I miss manpower to do so.

_CoreFreq_ don't touch voltage; the BIOS may however change the Vcore if set to AUTO. Nothing I can control.

Overclocking is made through the Turbo and the Uncore registers.
With my Xeon I can add +4 to the Core frequency (1C).
Above than that, the Processor crashes whenever I stress the CPU

The Experimental mode has been successfully tested on several architectures.
Because it is querying the PCI registers with some slight bits variations from one architecture to another one, _CoreFreq_ is safely starting without it.

What you can experiment through the UI:
1. Menu > Settings > Experimental : < Experimental operating mode >
2. Window > Memory Controller
3. Make screenshots of the DIMMs (timings, geometry, ...)

leinardi, 2018-08-04:
> What you can experiment through the UI:
> 
> Menu > Settings > Experimental : < Experimental operating mode >
> Window > Memory Controller
> Make screenshots of the DIMMs (timings, geometry, ...)

[image]

My RAM: https://www.gskill.com/en/product/f3-2400c10d-16gtx

When I reboot I can check the RAM settings from the BIOS if you want.

cyring, 2018-08-04:
Thanks a lot.

Yes please make BIOS screenshots of the current DRAM timings and speed.

Apparently Dual channel looks OK

Do you have one or two sticks ? Are they 8 or 16 GB each ?

cyring, 2018-08-04:
I just commit a change to only show the Vcore only bound to the measuring CPU.

leinardi, 2018-08-04:
[image]

cyring, 2018-08-04:
Thanks. Looks less confusing.

Can you tell if the measured power is good ?
The Processor [specifications](https://ark.intel.com/products/75123/Intel-Core-i7-4770K-Processor-8M-Cache-up-to-3_90-GHz) says a 84W TDP that should be compared against the measurement indicated by Package (W)

leinardi, 2018-08-04:
I am not sure and since I don't know how to measure it, but on full load the reported value is way higher than 84W:

[image]

cyring, 2018-08-04:
Yep sounds like an issue

cyring, 2018-08-05:
Hello,
I'm seeing a few glitches on the UI screenshots: do you have disable the nmi_watchdog ?

leinardi, 2018-08-05:
Hi, nope, I was lazy and I didn't changed it. Does it affect only the cli client or also the demon?

cyring, 2018-08-05:
It does affect a perf counter and consequently the accuracy of the indicators

leinardi, 2018-08-06:
Hey, I finally took a pic of the DDR settings:

[image]

[image]

[image]

cyring, 2018-08-06:
I find just a few matches with the previous _CoreFreq_ screenshot
RCD: 12
Cmd: 2N
RFC: 306

cyring, 2018-08-06:
Checking code, I notice that the Haswell DRAM timings are on my todo list !
I will complete soon.
Are the Dimms DDR3 ?

leinardi, 2018-08-06:
Yep, DDR3:
[image]

cyring, 2018-08-06:
Can you please print back the result of `lspci -nn`

leinardi, 2018-08-06:
Sure:
```
00:00.0 Host bridge [0600]: Intel Corporation 4th Gen Core Processor DRAM Controller [8086:0c00] (rev 06)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v3/4th Gen Core Processor PCI Express x16 Controller [8086:0c01] (rev 06)
00:16.0 Communication controller [0780]: Intel Corporation 8 Series/C220 Series Chipset Family MEI Controller #1 [8086:8c3a] (rev 04)
00:19.0 Ethernet controller [0200]: Intel Corporation Ethernet Connection I217-V [8086:153b] (rev 04)
00:1a.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #2 [8086:8c2d] (rev 04)
00:1c.0 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #1 [8086:8c10] (rev d4)
00:1c.1 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #2 [8086:8c12] (rev d4)
00:1c.3 PCI bridge [0604]: Intel Corporation 8 Series/C220 Series Chipset Family PCI Express Root Port #4 [8086:8c16] (rev d4)
00:1d.0 USB controller [0c03]: Intel Corporation 8 Series/C220 Series Chipset Family USB EHCI #1 [8086:8c26] (rev 04)
00:1f.0 ISA bridge [0601]: Intel Corporation Z87 Express LPC Controller [8086:8c44] (rev 04)
00:1f.2 SATA controller [0106]: Intel Corporation 8 Series/C220 Series Chipset Family 6-port SATA Controller 1 [AHCI mode] [8086:8c02] (rev 04)
00:1f.3 SMBus [0c05]: Intel Corporation 8 Series/C220 Series Chipset Family SMBus Controller [8086:8c22] (rev 04)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GP104 [GeForce GTX 1080] [10de:1b80] (rev a1)
01:00.1 Audio device [0403]: NVIDIA Corporation GP104 High Definition Audio Controller [10de:10f0] (rev a1)
03:00.0 SATA controller [0106]: ASMedia Technology Inc. ASM1062 Serial ATA Controller [1b21:0612] (rev 01)
04:00.0 PCI bridge [0604]: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch [10b5:8608] (rev ba)
05:01.0 PCI bridge [0604]: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch [10b5:8608] (rev ba)
05:04.0 PCI bridge [0604]: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch [10b5:8608] (rev ba)
05:05.0 PCI bridge [0604]: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch [10b5:8608] (rev ba)
05:06.0 PCI bridge [0604]: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch [10b5:8608] (rev ba)
05:08.0 PCI bridge [0604]: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch [10b5:8608] (rev ba)
05:09.0 PCI bridge [0604]: PLX Technology, Inc. PEX 8608 8-lane, 8-Port PCI Express Gen 2 (5.0 GT/s) Switch [10b5:8608] (rev ba)
06:00.0 PCI bridge [0604]: PLX Technology, Inc. PEX8112 x1 Lane PCI Express-to-PCI Bridge [10b5:8112] (rev aa)
07:04.0 Multimedia audio controller [0401]: C-Media Electronics Inc CMI8788 [Oxygen HD Audio] [13f6:8788]
08:00.0 Network controller [0280]: Qualcomm Atheros AR93xx Wireless Network Adapter [168c:0030] (rev 01)
09:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 11)
0c:00.0 SATA controller [0106]: ASMedia Technology Inc. ASM1062 Serial ATA Controller [1b21:0612] (rev 01)
```

May I ask you what are you looking for?

cyring, 2018-08-06:
Because here I have made a mistake: calling IVB rather than HSW architecture
https://github.com/cyring/CoreFreq/blob/3fba9c8258edb4ec70bdf42e364003c611bf68b0/corefreqk.h#L918
Can you change IVB_IMC with HSW_IMC
Rebuild and try again, please.

leinardi, 2018-08-06:
[image]

[image]

What differences should I see?

cyring, 2018-08-06:
The code seems following the same path as previously.

This PCI id: 0c00 from
```
00:00.0 Host bridge [0600]: Intel Corporation 4th Gen Core Processor DRAM Controller [8086:0c00] (rev 06)
```
is the one which has to be detected first in driver corefreqk.c , queried in HSW_IMC then timing data have to be computed by daemon corefreqd.c same function name.

cyring, 2018-08-07:
Last commit provides a partial fix to test the IMC.
So far I'm focusing on the major timings.

leinardi, 2018-08-08:
Hi @cyring, I just want to let you know that I have access to these CPUs and, if you want, I can do some test:

1. Intel(R) Core(TM) i7-6820HQ CPU @ 2.70GHz
2. Intel(R) Core(TM) i7-4800MQ CPU @ 2.70GHz
3. Intel(R) Xeon(R) CPU E3-1240 v6 @ 3.70GHz
4. Intel(R) Xeon(R) CPU E5-4650 v4 @ 2.20GHz

cyring, 2018-08-09:
Hello,
With great pleasure. Feel free to open an issue for each of them: it will easier to track problems and screeshots.
Btw, disable the nmi_watchdog before starting _CoreFreq_ (better in kernel cmdline)

If the VPS is virtualized, I not sure about results.

cyring, 2018-08-12:
About JSON: I have reviewed the code part. It's a bit tricky to use.

1. follow the key `Proc.Service.Service.Core` and mark down the value, for example: `this`

[image]

2. select `this` `Cpu` node

3. go to `FlipFlop` index `2`

4. read the voltage `VID` and `Vcore`
_like_ `Cpu[this].FlipFlop[2].Voltage.Vcore`

[image]

_remark_: if VID = 0 then Vcore is not meaningful

leinardi, 2018-08-13:
Yeah I was looking there but for me all the Vcore values were zero. I will try again this evening :+1:

leinardi, 2018-08-13:
Just tested and, unfortunately, on my I7-4770k the json file only shows 0 values for Vcore: [json.tar.gz](https://github.com/cyring/CoreFreq/files/2284178/json.tar.gz)

[image]

cyring, 2018-08-14:
JSON basic types don't comply with floating numbers but Vcore is `double` encoded. 

That's a big issue.

One needs to define a compatible exchange.
Strings ?
Like this:
```
"Voltage":{
    "VID":7125,
    "Vcore":"0.123456789"
}
```

leinardi, 2018-08-14:
Oh I see, the i7-4770k runs on an OS with Italian locale, and Italian uses commas as decimal separator, that can be the reason why it works on my other machine, where the OS is in English.

Is it possible to force the locale for the json serialization to US or to ROOT?

If is not possible I guess using a string would be a good enough workaround.

cyring, 2018-08-14:
I believe the locale won't help to fix the problem.

The bug is at this line:
https://github.com/cyring/CoreFreq/blob/b8b3d778bf535b3dac33f83e2ebc7d21b1341515/corefreq-cli-json.c#L1396
where the JSON output format is an unsigned integer whereas the data is encoded double.
Thus a voltage like `0.9370` will cast to `0`

What about Exponential numbers ?
Voltage could be exported to `937E+3`
Is this better than a string ?

_(my notes: using format "%uE%+d" rather than "%u" )_

leinardi, 2018-08-14:
Personally I would pick the string over the exponential notation.

cyring, 2018-08-15:
Fix of the decimal numbers done. Can you test please ?

leinardi, 2018-08-16:
Hey I can confirm that the fix for the 0 Vcore works! Thanks!

closed 2018-08-16 by leinardi

cyring, 2018-08-16:
YES !