cyring/CoreFreq #71: Intel(R) Core(TM) i7-6820HQ CPU @ 2.70GHz

leinardi, 2018-08-09:
As requested, some screenshots from this CPU:

NOMINAL MODE
[image]

[image]

EXPERIMENTAL MODE
[image]

[image]

Memory Controller doesn't open at all.

cyring, 2018-08-09:
There's a lot to fix ! Thank you.

But first, the thermal bit is on: HOT is red.
It can be a wrong register reading by _CoreFreq_ or the system has thermal issues ?

leinardi, 2018-08-09:
I am not aware of any thermal issues:

```
$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +46.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +46.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +46.0°C  (high = +100.0°C, crit = +100.0°C)

dell_smm-virtual-0
Adapter: Virtual device
Processor Fan: 3312 RPM
CPU:            +47.0°C  
Ambient:        +42.0°C  
SODIMM:         +45.0°C  
Other:          +40.0°C  

acpitz-virtual-0
Adapter: Virtual device
temp1:        +25.0°C  (crit = +107.0°C)

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +30.0°C  

pch_skylake-virtual-0
Adapter: Virtual device
temp1:        +57.5°C  

```

leinardi, 2018-08-09:
It's very possible that the CPU exceeded its threshold in the past, it's a laptop and I use it to build a big Java application (5+ min build).

It is also reported in the journal:
```
Aug 09 09:03:40 lpnb3132 kernel: CPU1: Core temperature above threshold, cpu clock throttled (total events = 1)
Aug 09 09:03:40 lpnb3132 kernel: CPU5: Core temperature above threshold, cpu clock throttled (total events = 1)
Aug 09 09:03:40 lpnb3132 kernel: CPU6: Package temperature above threshold, cpu clock throttled (total events = 1)
Aug 09 09:03:40 lpnb3132 kernel: CPU2: Package temperature above threshold, cpu clock throttled (total events = 1)
Aug 09 09:03:40 lpnb3132 kernel: CPU4: Package temperature above threshold, cpu clock throttled (total events = 1)
Aug 09 09:03:40 lpnb3132 kernel: CPU7: Package temperature above threshold, cpu clock throttled (total events = 1)
Aug 09 09:03:40 lpnb3132 kernel: CPU3: Package temperature above threshold, cpu clock throttled (total events = 1)
Aug 09 09:03:40 lpnb3132 kernel: CPU0: Package temperature above threshold, cpu clock throttled (total events = 1)
Aug 09 09:03:40 lpnb3132 kernel: CPU5: Package temperature above threshold, cpu clock throttled (total events = 1)
Aug 09 09:03:40 lpnb3132 kernel: CPU1: Package temperature above threshold, cpu clock throttled (total events = 1)
```

I don't know how to measure the temperature at the same time, this is the best I can do (1-2 sec difference):
[image]

cyring, 2018-08-09:
Measurements are pretty close. I will say that _CoreFreq_ temps are reliable.

In nominal mode, the "Power" view is showing some glitches. Do you have disable the nmi watchdog ?

In Experimental mode, the Cores' frequencies become zero:
* could you show the view "Core cycles"
* check if any driver error was reported in the journal

leinardi, 2018-08-09:
> In nominal mode, the "Power" view is showing some glitches. Do you have disable the nmi watchdog ?

I disabled it the first time I tried but not sure if was disabled for those screenshots. I have now disabled it and took new screenshots.

But the nominal mode had the core frequency to 0. I enabled the experimental mode and I got the values, than went back to nominal and the values are still there:
[image]

[image]

[image]

> could you show the view "Core cycles"

Nominal
[image]
Experimental
[image]

> check if any driver error was reported in the journal

These are the only errors I see in the journal:

```
Aug 09 17:15:30 lpnb3132 kernel: Uhhuh. NMI received for unknown reason 20 on CPU 4.
Aug 09 17:15:30 lpnb3132 kernel: Do you have a strange power saving mode enabled?
Aug 09 17:15:30 lpnb3132 kernel: Dazed and confused, but trying to continue
Aug 09 17:15:48 lpnb3132 kernel: Uhhuh. NMI received for unknown reason 20 on CPU 5.
Aug 09 17:15:48 lpnb3132 kernel: Do you have a strange power saving mode enabled?
Aug 09 17:15:48 lpnb3132 kernel: Dazed and confused, but trying to continue
Aug 09 17:15:48 lpnb3132 kernel: Uhhuh. NMI received for unknown reason 20 on CPU 3.
Aug 09 17:15:48 lpnb3132 kernel: Do you have a strange power saving mode enabled?
Aug 09 17:15:48 lpnb3132 kernel: Dazed and confused, but trying to continue
Aug 09 17:15:52 lpnb3132 kernel: Uhhuh. NMI received for unknown reason 20 on CPU 1.
Aug 09 17:15:52 lpnb3132 kernel: Do you have a strange power saving mode enabled?
Aug 09 17:15:52 lpnb3132 kernel: Dazed and confused, but trying to continue
Aug 09 17:15:57 lpnb3132 kernel: Uhhuh. NMI received for unknown reason 20 on CPU 0.
Aug 09 17:15:57 lpnb3132 kernel: Do you have a strange power saving mode enabled?
Aug 09 17:15:57 lpnb3132 kernel: Dazed and confused, but trying to continue
Aug 09 17:16:18 lpnb3132 kernel: Uhhuh. NMI received for unknown reason 20 on CPU 6.
Aug 09 17:16:18 lpnb3132 kernel: Do you have a strange power saving mode enabled?
Aug 09 17:16:18 lpnb3132 kernel: Dazed and confused, but trying to continue
Aug 09 17:16:56 lpnb3132 kernel: Uhhuh. NMI received for unknown reason 20 on CPU 7.
Aug 09 17:16:56 lpnb3132 kernel: Do you have a strange power saving mode enabled?
Aug 09 17:16:56 lpnb3132 kernel: Dazed and confused, but trying to continue
Aug 09 17:17:16 lpnb3132 kernel: Uhhuh. NMI received for unknown reason 20 on CPU 2.
Aug 09 17:17:16 lpnb3132 kernel: Do you have a strange power saving mode enabled?
Aug 09 17:17:16 lpnb3132 kernel: Dazed and confused, but trying to continue
```

cyring, 2018-08-09:
Thanks a lot. I don't see why frequencies, ratios and turbo are falling down to zero.
Could you also add to my previous request (see 'Edit'):
`corefreq-cli -c` (when frequencies are showing zero).
I want to check if it's an UI issue or it comes from the backend

leinardi, 2018-08-10:
I tried again today and I was not able to reproduce the 0 frequency bug.

I will let you know if it happens again.

```
$ ./corefreq-cli -s
Processor                            [Intel(R) Core(TM) i7-6820HQ CPU @ 2.70GHz]
|- Architecture                                                      [Skylake/S]
|- Vendor ID                                                      [GenuineIntel]
|- Signature                                                            [ 06_5E]
|- Stepping                                                             [     3]
|- Microcode                                                            [     0]
|- Online CPU                                                           [  8/8 ]
|- Base Clock                                                           [100.45]
|- Frequency            (Mhz)                      Ratio                        
                 Min    803.61                    [   8 ]                       
                 Max   2712.18                    [  27 ]                       
|- Factory                                                              [100.00]
                       2700                       [  27 ]                       
|- Turbo Boost                                                          [UNLOCK]
                  1C   3616.23                    <  36 >                       
                  2C   3415.33                    <  34 >                       
                  3C   3314.88                    <  33 >                       
                  4C   3214.43                    <  32 >                       
|- Uncore                                                               [  LOCK]
                 Max   25614.99                    [ 255 ]                      
|- TDP                                                            Level [  0:1 ]
   |- Programmable                                                      [UNLOCK]
   |- Configuration                                                     [UNLOCK]
   |- Turbo Activation                                                  [UNLOCK]
                                                                                
ISA Extensions:                                                                 
|- 3DNow!/Ext [N,N]        AES [Y]   AVX/AVX2 [Y/Y]   BMI1/BMI2 [Y/Y] 
|- CLFSH        [Y]       CMOV [Y]      CMPXCH8 [Y]      CMPXCH16 [Y] 
|- F16C         [Y]        FPU [Y]         FXSR [Y]     LAHF/SAHF [Y] 
|- MMX/Ext    [Y/N]    MONITOR [Y]        MOVBE [Y]      PCLMULDQ [Y] 
|- POPCNT       [Y]     RDRAND [Y]       RDTSCP [Y]           SEP [Y] 
|- SGX          [Y]        SSE [Y]         SSE2 [Y]          SSE3 [Y] 
|- SSSE3        [Y] SSE4.1/4A [Y/N]      SSE4.2 [Y]       SYSCALL [Y] 
                                                                                
Features:                                                                       
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Present]
|- CPL Qualified Debug Store                                  DS-CPL   [Present]
|- 64-Bit Debug Store                                         DTES64   [Present]
|- Fast-String Operation                                Fast-Strings   [Present]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Present]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
|- Pending Break Enable                                          PBE   [Present]
|- Process Context Identifiers                                  PCID   [Present]
|- Perfmon and Debug Capability                                 PDCM   [Present]
|- Page Global Enable                                            PGE   [Present]
|- Page Size Extension                                           PSE   [Present]
|- 36-bit Page Size Extension                                  PSE36   [Present]
|- Processor Serial Number                                       PSN   [Missing]
[... 72 lines cut ...]
00:02.0 VGA compatible controller [0300]: Intel Corporation HD Graphics 530 [8086:191b] (rev 06)
[... cut ...]
04:00.0 PCI bridge [0604]: Intel Corporation JHL6340 Thunderbolt 3 Bridge (C step) [Alpine Ridge 2C 2016] [8086:15da] (rev 02)
05:00.0 PCI bridge [0604]: Intel Corporation JHL6340 Thunderbolt 3 Bridge (C step) [Alpine Ridge 2C 2016] [8086:15da] (rev 02)
05:01.0 PCI bridge [0604]: Intel Corporation JHL6340 Thunderbolt 3 Bridge (C step) [Alpine Ridge 2C 2016] [8086:15da] (rev 02)
05:02.0 PCI bridge [0604]: Intel Corporation JHL6340 Thunderbolt 3 Bridge (C step) [Alpine Ridge 2C 2016] [8086:15da] (rev 02)
3c:00.0 USB controller [0c03]: Intel Corporation Device [8086:15db] (rev 02)
```

cyring, 2018-08-10:
Hello, Thanks for your replies.

The 0 frequency bug might be due to drivers conflict. In the [Wiki](https://github.com/cyring/CoreFreq/wiki/Software-incompatibility), I'm keeping track of any kernel codes and drivers which reprogram the fixed counters on behalf of _CoreFreq_
Not listed yet, the module `iTCO_wdt.ko` could also interfere and other HP divers I have crossed.

About the memory controller, the following PCI is the entry point to query the IMC registers
`00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Host Bridge/DRAM Registers [8086:1910] (rev 07)`

Code was already programmed but commented.
What I suggest is that you change the instructions as below. This will let the driver query the IMC based on my generic Skylake decoder.
Save your files, because you are the first to experiment it !
https://github.com/cyring/CoreFreq/blob/3fba9c8258edb4ec70bdf42e364003c611bf68b0/corefreqk.c#L2629
```
static PCI_CALLBACK SKL_SA(struct pci_dev *dev)
{
	SKL_SA_PLL_RATIOS PllRatios = {.value = 0};

	pci_read_config_dword(dev, 0x5918, &PllRatios.value);

	Proc->Uncore.Boost[UNCORE_BOOST(MAX)] = PllRatios.UCLK;
	Proc->Uncore.Boost[UNCORE_BOOST(MIN)] = 0;
/*ToDo:
	pci_read_config_dword(dev, 0xe4, &Proc->Uncore.Bus.SKL_Cap_A.value);
	pci_read_config_dword(dev, 0xe8, &Proc->Uncore.Bus.SKL_Cap_B.value);
*/
	return(Router(dev, 0x48, 64, 0x8000, Query_SKL_IMC));

	return(0);
}
```

leinardi, 2018-08-11:
I will try it on Monday, thanks :+1:

cyring, 2018-08-12:
Thanks.
Basically it decodes the IMC for Skylake Mobile and Desktop and superior architectures Kaby Lake, Coffee Lake. Feel free to try if you have such machines.

leinardi, 2018-08-13:
Hey good news: the Memory Controller is now showing :)

NOM
[image]

[image]

EXP
[image]

[image]

cyring, 2018-08-13:
Great ! Thank you for your answer.
Can you tell if the IMC values match those in BIOS (or other tools) ?

Unfortunately the zero frequency bug is back again.
Please take a screenshot of counters `C0:UCC` and `C0:URC` in the View > Core cycles

cyring, 2018-08-14:
Thanks 😉

* we now have the minimum and maximum Uncore ratios

The bad's
* TDP ratios are unlisted. May required the Experimental mode.
* Zero UCC: this counter is read from the register MSR_CORE_PERF_FIXED_CTR1.
The NMI watchdog can rewrite this register; do you confirm you have disable it from the kernel cmdline ?

leinardi, 2018-08-14:
> do you confirm you have disable it from the kernel cmdline ?

Not from kernel cmdline, I did it at runtime with `sudo sysctl kernel.nmi_watchdog=0`

cyring, 2018-08-15:
Hello,

There's a new feature to clear the temperature events.
[image]

Could you bring your processor to high temperature like it was when building the big Java application.

When the HOT indicator gets red , let the CPUs cool down.

Then press the [H] key (capital H) to open the "Clear Event" window and select "Thermal Sensor"

As a result, the HOT label should switch off.

Remark: if choosing "Thermal Sensor" does have no effect, please try the other items.

Regards

cyring, 2018-08-22:
Have you try with the nmi_watchdog fully disabled ?

Btw, last version brings the Package temperature (for architectures which provide the register). The sensor should display in the UI footer.

cyring, 2018-09-11:
No more issues with this i7-6820HQ ?