cyring/CoreFreq #72: CentOS 7.4.1708: insmod: ERROR: could not insert module corefreqk.ko: Invalid module format

leinardi, 2018-08-09:
I am trying to run CoreFreq on a CentOS 7.4.1708 installation but I am getting this error when I try to install the kernel mod:

```
# insmod corefreqk.ko
insmod: ERROR: could not insert module corefreqk.ko: Invalid module format
```

```
  Operating System: CentOS Linux release 7.4.1708 (Core) 
       CPE OS Name: cpe:/o:centos:centos:7
            Kernel: Linux 3.10.0-693.21.1.el7.x86_64
      Architecture: x86-64
```

cyring, 2018-08-09:
The minimum version I have successfully built the kernel module with is 3.14

Can you update the kernel to 4.x and btw the libc and gcc ?

leinardi, 2018-08-09:
I am sorry but I cannot do that on this machine.

leinardi, 2018-08-09:
> Is it the same machine on which CoreFreq has been compiled ?

Yes, I had to install the kernel headers to make it build.

> Don't you have any security software preventing from loading modules ?

I am not aware of any security software preventing from loading modules.

cyring, 2018-08-10:
I have virtualized CentOS w/ kernel 3.10.0-514.26.2.el7.x86_64
In kernel cmdline, I have added: 
```
nmi_watchdog=0 crashkernel=0 modprobe.blacklist=iTCO_wdt
```
No issue to insert corefreqk.ko

crashkernel might be the clue to the problem.

leinardi, 2018-08-11:
Hi @cyring, thanks for testing, I don't have access to the machine now, I will try again on Monday :+1:

leinardi, 2018-08-14:
Hey sorry, I did not notice that that test require a reboot, let me know if you want me to test something that doesn't require a reboot.

cyring, 2018-08-14:
For this CentOS issue, I don't see no other way to fix the problem.

cyring, 2018-08-22:
Hello,
Do you have a status of this issue ?

leinardi, 2018-08-23:
Hi @cyring, sorry but I cannot test any kernel cmd parameter. If you have a solution that doesn't require a reboot I am happy to try.

cyring, 2018-08-23:
Is this due to mismatches : custom kernel ? gcc version ?
Don't you see reject errors in `/var/log/messages` ?

Does the `vermagic:` of `modinfo corefreqk.ko` provide the same as the other loaded modules ?

leinardi, 2018-08-24:
I tried again everything from scratch, but I can't get the sources for the latest kernel:

```
# make
cc -Wall -pthread -c corefreqd.c -o corefreqd.o
cc -Wall -c corefreqm.c -o corefreqm.o
cc corefreqd.c corefreqm.c -o corefreqd -lpthread -lm -lrt
cc -Wall -c corefreq-cli.c -o corefreq-cli.o
cc -Wall -c corefreq-ui.c -o corefreq-ui.o
cc -Wall -c corefreq-cli-json.c -o corefreq-cli-json.o
cc -Wall -c corefreq-cli-extra.c -o corefreq-cli-extra.o
cc	corefreq-cli.c corefreq-ui.c \
	corefreq-cli-json.c corefreq-cli-extra.c \
	-o corefreq-cli -lm -lrt
make -C /lib/modules/3.10.0-693.21.1.el7.x86_64/build M=/root/temp/CoreFreq modules
make: *** /lib/modules/3.10.0-693.21.1.el7.x86_64/build: No such file or directory.  Stop.
make: *** [all] Error 2
```

```
ll /lib/modules/3.10.0-693.21.1.el7.x86_64/build
lrwxrwxrwx. 1 root root 43 13 mar 15.31 /lib/modules/3.10.0-693.21.1.el7.x86_64/build -> /usr/src/kernels/3.10.0-693.21.1.el7.x86_64
```

```
ll /usr/src/kernels/
total 4
drwxr-xr-x. 22 root root 4096 24 ago 09.57 3.10.0-862.11.6.el7.x86_64
```

```
# yum list kernel 
Loaded plugins: etckeeper, fastestmirror, product-id, subscription-manager
Loading mirror speeds from cached hostfile
 * base: artfiles.org
 * extras: artfiles.org
 * updates: centos.alpha-labs.net
Installed Packages
kernel.x86_64                                         3.10.0-693.17.1.el7                                          @updates                   
kernel.x86_64                                         3.10.0-693.21.1.el7                                          @updates 
```

```
# uname -r
3.10.0-693.21.1.el7.x86_64
```

```
# yum install kernel-devel kernel-headers 
Loaded plugins: etckeeper, fastestmirror, product-id, subscription-manager 
Loading mirror speeds from cached hostfile
[...]
 * base: artfiles.org
 * extras: artfiles.org
 * updates: centos.alpha-labs.net
Package kernel-devel-3.10.0-862.11.6.el7.x86_64 already installed and latest version
Package kernel-headers-3.10.0-862.11.6.el7.x86_64 already installed and latest version
Nothing to do
```

cyring, 2018-08-24:
Some ideas:
1. `yum install kernel-devel-$(uname -r) -y` @  [Finding Old kernel-devel Packages For CentOS](http://wiki.r1soft.com/display/kb3/Finding+Old+kernel-devel+Packages+For+CentOS)
1. `yum install "kernel-devel-uname-r == $(uname -r)"` from [stackexchange](https://unix.stackexchange.com/a/110685/209911)
1. install a Virtual Machine, same as the target one, including developer packages, but no updates

leinardi, 2018-08-24:
```
# yum install kernel-devel-$(uname -r) -y  
Loading mirror speeds from cached hostfile
 * base: artfiles.org
 * extras: artfiles.org
 * updates: centos.alpha-labs.net
No package kernel-devel-3.10.0-693.21.1.el7.x86_64 available.
Error: Nothing to do
```
```
# yum install "kernel-devel-uname-r == $(uname -r)"
Loaded plugins: etckeeper, fastestmirror, product-id, subscription-manager
Loading mirror speeds from cached hostfile
 * base: artfiles.org
 * extras: artfiles.org
 * updates: centos.alpha-labs.net
No package kernel-devel-uname-r == 3.10.0-693.21.1.el7.x86_64 available.
Error: Nothing to do
```
Thanks for trying but still no luck. I fear it depends on the mirror I am using: they are not the official one and I cannot change them.

cyring, 2018-08-24:
Your welcome.
It is an interesting experience for _CoreFreq_ which claims support among Distributions.
Meanwhile if you have chance to boot, a _CoreFreq_ [Live CD](https://github.com/cyring/CoreFreq/wiki/Live-CD) is ready to use.
Feel free to close this issue.
Regards,
CyrIng

closed 2018-09-01 by cyring