cyring/CoreFreq #81: Quad-Core Penryn read as Yorkfield

SairesArt, 2018-10-04:
This is no particularly CoreFreq specific, but I'm interested on input where to look for a solution.
My Thinkpad T500 with a QX9300 (Penryn) + [libreboot + no microcode updates] is being read in as a Q9300 (Yorkfield).
(See screenshot)
This is not exclusive to CoreFreq, as /proc/cpuinfo shows the same info, see attachments.
A coreboot dev suggested, that this may have been fixed by a microcode update.

[cpuinfo.txt](https://github.com/cyring/CoreFreq/files/2445966/cpuinfo.txt)
[image]

cyring, 2018-10-04:
Hello. Thanks for this.

Could you print back the results of `corefreq-cli -s` b/c I want to check the CPUID model signature

Could you also first
 ```
insmod corefreqk.ko AutoClock=0
``` 
b/c the TSC is variant and the Base Clock is wrongly estimated

SairesArt, 2018-10-04:
Sure, here.
The baseclock is correctly estimated... I guess...
The baseclock is 2.54 Ghz and boosts up to 2.8 ghz. This is correctly reflected in CoreFreq. Stress test on up to 2 Cores results in speeds slightly below 2.8 ghz and above 3 Cores it sticks to 2.54 Ghz

cyring, 2018-10-04:
1. In the following lines are made the link between the cpuid signature `06_17` and the architecture name `"Core2/Yorkfield"`

https://github.com/cyring/CoreFreq/blob/f3d0a2c6ed0291df7637fb12bedfa405fd337061/corefreqk.h#L796
https://github.com/cyring/CoreFreq/blob/f3d0a2c6ed0291df7637fb12bedfa405fd337061/corefreqk.h#L1391

I could just change the name to `"Core2/Yorkfield/Penryn"` . 
According to the Wiki, " The mobile version of Yorkfield is Penryn-QC "

Or we need to find another discriminant to distinguish [Penryn-QC](https://en.wikipedia.org/wiki/Penryn_(microprocessor)) [CPUID code | 1067x] from [Yorkfield](https://en.wikipedia.org/wiki/Yorkfield) [CPUID code 1067x]

------

2. This processor is IDA and SpeedStep technologies capable.  
It adds one bin to its frequency: 2540 + 266 = around 2.8 GHz

I need to fix the 1C Turbo Boost line.

------

cyring, 2018-10-05:
Hello

I would like to debug the One Core Turbo Boost (1C) ratio value.
Could you apply the following changes to the code ?

1. get the _CoreFreq_ source code from its last version (GitHub clone)

2. at the following line:  
https://github.com/cyring/CoreFreq/blob/f3d0a2c6ed0291df7637fb12bedfa405fd337061/corefreqk.c#L1451

3. replace these two functions `Intel_MaxBusRatio` and `Intel_Platform_Info` with the code below: 
```
int Intel_MaxBusRatio(PLATFORM_ID *PfID)
{
	struct SIGNATURE whiteList[] = {
		_Core_Conroe,		/* 06_0F */
		_Core_Yorkfield,	/* 06_17 */
		_Atom_Bonnell,		/* 06_1C */
		_Atom_Silvermont,	/* 06_26 */
		_Atom_Lincroft,		/* 06_27 */
		_Atom_Clovertrail,	/* 06_35 */
		_Atom_Saltwell,		/* 06_36 */
		_Silvermont_637,	/* 06_37 */
	};
	int id, ids = sizeof(whiteList) / sizeof(whiteList[0]);
	for (id = 0; id < ids; id++) {
		if((whiteList[id].ExtFamily == Proc->Features.Std.EAX.ExtFamily)
		 && (whiteList[id].Family == Proc->Features.Std.EAX.Family)
		 && (whiteList[id].ExtModel == Proc->Features.Std.EAX.ExtModel)
		 && (whiteList[id].Model == Proc->Features.Std.EAX.Model)) {

			RDMSR((*PfID), MSR_IA32_PLATFORM_ID);
			printk("corefreq:%s:%d:MSR_IA32_PLATFORM_ID[%llx]\n",
				__FUNCTION__,__LINE__,PfID->value);
			return(0);
		}
	}
	printk("corefreq:%s:%d:MSR_IA32_PLATFORM_ID[Blacklist]\n",
		__FUNCTION__,__LINE__);
	return(-1);
}

void Intel_Platform_Info(void)
{
	PLATFORM_ID PfID = {.value = 0};
	PLATFORM_INFO PfInfo = {.value = 0};
	PERF_STATUS PerfStatus = {.value = 0};
	unsigned int ratio0 = 10, ratio1 = 10, ratio2 = 10; // Arbitrary

	RDMSR(PfInfo, MSR_PLATFORM_INFO);
	printk("corefreq:%s:%d:MSR_PLATFORM_INFO[%llx]\n",
		__FUNCTION__,__LINE__,PfInfo.value);
	if (PfInfo.value != 0) {
		ratio0 = KMIN(PfInfo.MinimumRatio, PfInfo.MaxNonTurboRatio);
		ratio1 = KMAX(PfInfo.MinimumRatio, PfInfo.MaxNonTurboRatio);
		ratio2 = ratio1;
		printk("corefreq:%s:%d:MinimumRatio[%d] MaxNonTurboRatio[%d]\n",
			__FUNCTION__,__LINE__,
			PfInfo.MinimumRatio,PfInfo.MaxNonTurboRatio);
	}

	RDMSR(PerfStatus, MSR_IA32_PERF_STATUS);
	printk("corefreq:%s:%d:MSR_IA32_PERF_STATUS[%llx]\n",
		__FUNCTION__,__LINE__,PerfStatus.value);
	if (PerfStatus.value != 0) {				// §18.18.3.4
		if (PerfStatus.CORE.XE_Enable) {
			ratio2 = PerfStatus.CORE.MaxBusRatio;
			printk("corefreq:%s:%d:XE[ON] MaxBusRatio[%d]\n",
				__FUNCTION__,__LINE__,
				PerfStatus.CORE.MaxBusRatio);
		} else {
			if (Intel_MaxBusRatio(&PfID) == 0) {
				if (PfID.value != 0)
					ratio2 = PfID.MaxBusRatio;
				printk("corefreq:%s:%d:XE[OFF]" \
					" MaxBusRatio[%d]\n",
					__FUNCTION__,__LINE__,
					PfID.MaxBusRatio);
			}
		}
	} else {
			if (Intel_MaxBusRatio(&PfID) == 0) {
				if (PfID.value != 0)
					ratio2 = PfID.MaxBusRatio;
				printk("corefreq:%s:%d:Branch[alt]" \
					" MaxBusRatio[%d]\n",
					__FUNCTION__,__LINE__,
					PfID.MaxBusRatio);
			}
	}
	Proc->Boost[BOOST(MIN)] = ratio0;
	Proc->Boost[BOOST(MAX)] = KMIN(ratio1, ratio2);
	Proc->Boost[BOOST(1C)] = KMAX(ratio1, ratio2);
	printk("corefreq:%s:%d:BOOST:MIN[%d] MAX[%d] 1C[%d]\n",
		__FUNCTION__,__LINE__,
		Proc->Boost[BOOST(MIN)],
		Proc->Boost[BOOST(MAX)],
		Proc->Boost[BOOST(1C)]);

	Proc->Features.SpecTurboRatio = 1;
}
```
4. rebuild project `make clean; make`

5. insert module `insmod corefreqk.ko`

6. print all corefreq traces `dmesg -t | grep -i corefreq`

Result should look like this:
```
corefreq:Intel_Platform_Info:1490:MSR_PLATFORM_INFO[8080838f1012200]
corefreq:Intel_Platform_Info:1496:MinimumRatio[8] MaxNonTurboRatio[34]
corefreq:Intel_Platform_Info:1502:MSR_IA32_PERF_STATUS[231500002400]
corefreq:Intel_MaxBusRatio:1477:MSR_IA32_PLATFORM_ID[Blacklist]
corefreq:Intel_Platform_Info:1533:BOOST:MIN[8] MAX[34] 1C[34]
CoreFreq(3:-1): Processor [ 06_9E] Architecture [GenuineIntel] CPU [4/4]
```

In advance, thanks for any help

SairesArt, 2018-10-05:
@cyring 
Here it is:
```
corefreq:Intel_Platform_Info:1490:MSR_PLATFORM_INFO[801b4a294343061b]
corefreq:Intel_Platform_Info:1496:MinimumRatio[74] MaxNonTurboRatio[6]
corefreq:Intel_Platform_Info:1502:MSR_IA32_PERF_STATUS[61b492186324921]
corefreq:Intel_Platform_Info:1507:XE[ON] MaxBusRatio[9]
corefreq:Intel_Platform_Info:1533:BOOST:MIN[6] MAX[9] 1C[74]
CoreFreq(1:-1): Processor [ 06_17] Architecture [Core2/Yorkfield] CPU [4/4]
```

cyring, 2018-10-05:
Thanks a lot.

My notes to fix Core2's architectures code:
* MinimumRatio at MSR_PLATFORM_INFO[47:40] is not applicable, not specified
* MaxNonTurboRatio at MSR_PLATFORM_INFO[15:8] is not specified but can work

cyring, 2018-10-12:
In the Intel® Core™2 Extreme Quad-Core Specifications Update (ref. 320121-004), we can read a CPUID = 00001067Ah for QX9300 (SLB5J m-FCPGA)
So we may have a discriminant between Penryn and Yorkfield based on the Stepping number 10 vs 6

cyring, 2018-10-12:
Page 18, you will read:

"AZ2 CPUID Instruction Will Return Brand String With a Missing Letter"

Is it the reason why the "X" is missing in the returned CPUID name ?

cyring, 2018-10-15:
Hello,

Last version 1.36.1 you should get bug fixes for:
1. the architecture code name as `Core2/Penryn`
1. the boost ratio rounded to 11

Unfortunately there is nothing I can do for the missing letter X in the processor brand name.

SairesArt, 2018-10-19:
Awesome! many thanks.

closed 2018-10-19 by SairesArt

cyring, 2018-10-19:
Do you mind to post a screenshot, I would to see the ratios ?

cyring, 2018-10-30:
Thanks for the screenshot.

A **variant** TSC is a strong issue to compute the Base Clock with accuracy.
You can start the driver based on the Architecture specification `insmod corefreqk.ko AutoClock=0`
or apply a full stress load on the 4 Cores _while_ starting the driver.

Below are the definitions of the HDK color (_High Dark Black_)
https://github.com/cyring/CoreFreq/blob/61e8f51b081146c3efb20b3c3946693d47bb5ae3/corefreq-ui.h#L183
https://github.com/cyring/CoreFreq/blob/61e8f51b081146c3efb20b3c3946693d47bb5ae3/corefreq-ui.h#L61