cyring/CoreFreq #82: Feature request: No background color

SairesArt, 2018-10-04:
Minor wish:
I'd love to have no background color (currently the gray) in the cli client of CoreFreq. I have looked through the help command and read the doc, but have found no option to disable the background color.

cyring, 2018-10-04:
"No background color" is a difficult feature to program:
Space characters are part of cleaning drawing areas.
Removing color of spaces may lead to undefined results.
The CoreFreq UI is made with ASCII for Consoles.
A user theme with fixed background color is possible but it requires a long UI code refactoring.

cyring, 2018-10-04:
A quick workaround is to modify the color palette of the terminal.
Alter the RGB of the background and the Black colors.
[image]
[image]
[image]
[image]

SairesArt, 2018-10-04:
Gah, I wanted my Distro's sexy Glass terminal shining through CoreFreq. Too bad, the terminal doesn't support custom colors.
Thank's for the suggestion though
[image]
[image]

closed 2018-10-04 by SairesArt