cyring/CoreFreq #83: Clang

cyring, 2018-10-06:
To build _CoreFreq_ code with the Clang compiler.

All nested functions and shared variables have been moved in global space.

Get this source code at [blog.cyring.free.fr/live/CoreFreq.tar](http://blog.cyring.free.fr/live/CoreFreq.tar)

To build the project:
```
make CC=clang OPTIM_LVL=1
```
Issues:
* does not compil with level 0 => _invalid operand for inline asm constraint 'i'_
* can not insert module => _Invalid module format_

adatum, 2018-10-07:
There is a warning on compilation: 

```
In file included from ./arch/x86/include/asm/thread_info.h:53:
./arch/x86/include/asm/cpufeature.h:150:2: warning: "Compiler lacks ASM_GOTO support. Add -D __BPF_TRACING__ to your compiler arguments" [-W#warnings]
```

Aside from that, as mentioned in OP, inserting the kernel module fails:

```
$ sudo insmod corefreqk.ko Experimental=1
insmod: ERROR: could not insert module corefreqk.ko: Invalid module format
```

cyring, 2018-10-07:
@adatum thanks.
I've read that the ASM_GOTO was a issue when building Linux kernel w/ Clang
Unfortunately I've lost the article link...

adatum, 2018-10-07:
In the meantime, is there a way to test CoreFreq? Without inserting the kernel module, can't get far..

Also, given the frequent changes, perhaps it would be good to have unit tests, whether programmatic or manual.

cyring, 2018-11-10:
* Clang can build _CoreFreq_ with optimization levels 3 , 2 , 1  and undefine CC_HAVE_ASM_GOTO

```shell
$ make CC="clang -U CC_HAVE_ASM_GOTO" OPTIM_LVL=3
clang -U CC_HAVE_ASM_GOTO -O3 -Wall -pthread -c corefreqd.c \
	-D FEAT_DBG=1 -o corefreqd.o
clang -U CC_HAVE_ASM_GOTO -O3 -Wall -c corefreqm.c -o corefreqm.o
clang -U CC_HAVE_ASM_GOTO -O3 -Wall corefreqd.c corefreqm.c \
	-D FEAT_DBG=1 -o corefreqd -lpthread -lm -lrt
clang -U CC_HAVE_ASM_GOTO -O3 -Wall -c corefreq-cli.c -o corefreq-cli.o
clang -U CC_HAVE_ASM_GOTO -O3 -Wall -c corefreq-ui.c -o corefreq-ui.o
clang -U CC_HAVE_ASM_GOTO -O3 -Wall -c corefreq-cli-json.c \
	-o corefreq-cli-json.o
clang -U CC_HAVE_ASM_GOTO -O3 -Wall -c corefreq-cli-extra.c \
	-o corefreq-cli-extra.o
clang -U CC_HAVE_ASM_GOTO -O3 -Wall \
	corefreq-cli.c corefreq-ui.c \
	corefreq-cli-json.c corefreq-cli-extra.c \
	-o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/4.18.17-2-ck/build M=/workdir/CoreFreq modules
make[1]: Entering directory '/usr/lib/modules/4.18.17-2-ck/build'
  CC [M]  /workdir/CoreFreq/corefreqk.o
In file included from /workdir/CoreFreq/corefreqk.c:8:
In file included from ./include/linux/module.h:10:
In file included from ./include/linux/stat.h:19:
In file included from ./include/linux/time.h:6:
In file included from ./include/linux/seqlock.h:36:
In file included from ./include/linux/spinlock.h:51:
In file included from ./include/linux/preempt.h:81:
In file included from ./arch/x86/include/asm/preempt.h:7:
In file included from ./include/linux/thread_info.h:38:
In file included from ./arch/x86/include/asm/thread_info.h:53:
./arch/x86/include/asm/cpufeature.h:150:2: warning: "Compiler lacks ASM_GOTO
      support. Add -D __BPF_TRACING__ to your compiler arguments" [-W#warnings]
#warning "Compiler lacks ASM_GOTO support. Add -D __BPF_TRACING__ to you...
 ^
1 warning generated.
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /workdir/CoreFreq/corefreqk.mod.o
In file included from /workdir/CoreFreq/corefreqk.mod.c:1:
In file included from ./include/linux/module.h:10:
In file included from ./include/linux/stat.h:19:
In file included from ./include/linux/time.h:6:
In file included from ./include/linux/seqlock.h:36:
In file included from ./include/linux/spinlock.h:51:
In file included from ./include/linux/preempt.h:81:
In file included from ./arch/x86/include/asm/preempt.h:7:
In file included from ./include/linux/thread_info.h:38:
In file included from ./arch/x86/include/asm/thread_info.h:53:
./arch/x86/include/asm/cpufeature.h:150:2: warning: "Compiler lacks ASM_GOTO
      support. Add -D __BPF_TRACING__ to your compiler arguments" [-W#warnings]
#warning "Compiler lacks ASM_GOTO support. Add -D __BPF_TRACING__ to you...
 ^
1 warning generated.
  LD [M]  /workdir/CoreFreq/corefreqk.ko
make[1]: Leaving directory '/usr/lib/modules/4.18.17-2-ck/build'
```

* Using optimization level 0 compilation fails at pre-processing `__builtin_constant_p()`

cyring, 2018-11-10:
* Loading module fails. This subject is postponed until the driver port to BSDs ...
```
module: x86/modules: Skipping invalid relocation target, existing value is nonzero for type 1, loc (____addr____), val (____value____)
```

closed 2018-11-10 by cyring