cyring/CoreFreq #84: Ryzen C States Are Zero > C1

EUA, 2018-10-10:
Hi,
Thanks for great program.
I got Ryzen 1700 with B350 MB. I enabled deep sleep at bios etc...
I tried to do everything but don't have any success.

**corefreq only shows Turbo, C0 and C1 states.**
It shown at program that (CC6 & PC6 green) but C3/C6/C7 counters are always zero at frequency and Idle C-states view.
```
--- Freq(MHz) Ratio - Turbo --- C0 ---- C1 ---- C3 ---- C6 ---- C7 --Min TMP Max ----
#0   180.76 ( 2.29)   6.04%   6.87%  93.13%   0.00%   0.00%   0.00%  0  /  0/  0
#1   115.44 ( 1.16)   3.04%   3.92%  96.08%   0.00%   0.00%   0.00%  0  /  0/  0
#2   190.43 ( 1.91)   5.02%   7.40%  92.60%   0.00%   0.00%   0.00%  0  /  0/  0
#3   319.41 ( 3.20)   8.42%  11.72%  88.28%   0.00%   0.00%   0.00%  0  /  0/  0
...
```  
Why does C3/C6/C7 states not supported by CoreFreq?

PS: using cpupower monitor give this result.
```
    | Mperf              || Idle_Stats         
 CPU| C0   | Cx   | Freq  || POLL | C1   | C2    
   0|  7,33| 92,67|  1794||  0,00|  4,37| 88,32
   1|  3,91| 96,09|  1954||  0,00|  6,44| 89,58
   2|  3,52| 96,48|  1702||  0,00|  7,90| 88,86
...
```

cyring, 2018-11-10:
Based on [`cpuidle.c`](https://github.com/torvalds/linux/blob/master/tools/power/cpupower/lib/cpuidle.c) source code, can you print :
```
cat /sys/devices/system/cpu/cpu0/cpuidle/state?/name
```

EUA, 2018-11-12:
```
# cat /sys/devices/system/cpu/cpu0/cpuidle/state?/name
POLL
C1
C2
```
I think RyZen micro-code does not export all C states to Linux Kernel.

cyring, 2018-11-13:
Is this the same with Windows software ? 
Such as [ThrottleStop](https://www.techpowerup.com/download/techpowerup-throttlestop)

EUA, 2018-11-17:
Oopps. CPU is not supported by ThrottleStop

cyring, 2018-11-17:
Perhaps in the EFI/BIOS or the AMD software, but googling for Zen C-States doesn't show any monitoring of deep idle states.

EUA, 2018-11-23:
Okay. I dig further and found something. Will this help?
https://www.amd.com/system/files/TechDocs/49125_15h_Models_30h-3Fh_BKDG.pdf#M30.9.80656.Register.D24F4x120.Cstate.Policy.Control.Register

cyring, 2018-12-01:
I'm closing this issue but feel to try the experimental Zen overclocking @ issue #87

cyring, 2020-07-06:
Hello,
Among the last changes in issue #195 , an I/O ASM access to the FCH have been added: can you please check if the last _CoreFreq_ version is starting fine ?
Thank you

EUA, 2024-03-13:
Hello. Don't know why but I missed this notifications.

My older board is broken. Will try solder new mosfets next week...

Now I have 8600G CPU. I compiled from git-trunk and saw that;
Still only C0 and C1 available while `cpupower idle-info` shows C2 and C3 states for my 8600G
Also at windows, HWiNFO able to show C6 states of processors proper.
CoreFreq able to show C2/3/6 is enabled:

[image]

Don't know if "I/O ASM access to the FCH" improvements are enabled supported on my board/cpu. Might be, they work on old architectures. If any data required to you for implement your program, I can send any kind of dump.

When I able to repair my AM4 board, try to run cpufreqd on it to see if there are any improvement on C states with x1700 and 5700G CPUs.
Thanks.

cyring, 2024-03-13:
@EUA 
Still the same **missing** hardware counters specification since Zen `1`

Meanwhile I had implemented software kernel counters down to C6

Just start driver with these parameters:
```
insmod build/corefreqk.ko Register_ClockSource=1 Register_Governor=1 Register_CPU_Idle=1 Register_CPU_Freq=1

echo "corefreq_tsc" > /sys/devices/system/clocksource/clocksource0/current_clocksource
```

You can then alternate as will the idle limit from the UI window named `Kernel` 

You will have to unload or prevent any other driver to act as CPU-Idle, CPU-Freq and ClockSource. 
They have to be `Missing` prior registering _CoreFreq_

See [Wiki](https://github.com/cyring/CoreFreq/wiki)/`CoreFreq as the Clock Source, CPU Freq and CPU Idle driver`

When done, please provide me, using Mardown format, the output of the following CLI

```
build/corefreq-cli -s -n -m -n -B -n -M -n -k -n -C 1 -n -i 1 -n -V 1 -n -W 1 -n -c 1
```

EUA, 2024-03-14:
Hello there.
Thanks for quick reply.
I just loaded my built kernel module with parameters that you provided but...
Still same result. only C1 and C2 available. Why? Probably some other things like amd_pstate driver loaded already into kernel.
But I can't find amd_pstate module on my modules directory. Probably it's delivered in the kernel that I use (Arch Zen kernel). Not compiled as a module on my PC. Will try with standard kernel also. Do you know which other kernel modules could interfere with it?

At settings, CPU-IDLE driver and CPU-FREQ driver shows OFF while "CPU-IDLE route" shown as I/O

[image]

It's sad thing that learning AMD (still) do not exposes it's C states via MSRs, which is super easy to work with it.

Anyway, Here is the output of my CPU/System;

`build/corefreq-cli -s -n -m -n -B -n -M -n -k -n -C 1 -n -i 1 -n -V 1 -n -W 1 -n -c 1`

```
Processor                            [AMD Ryzen 5 8600G w/ Radeon 760M Graphics]
|- Architecture                                                [Zen4/Hawk Point]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a705203]
|- Signature                                                           [  AF_75]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 12/ 12]
|- Base Clock                                                          [101.168]
|- Frequency            (MHz)                      Ratio                        
                 Min   1618.69                    <  16 >                       
                 Max   4350.24                    <  43 >                       
|- Factory                                                             [100.000]
                       4300                       [  43 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   4350.24                    <  43 >                       
   |- CPPC                                                                      
                 Min   3844.40                    <  38 >                       
                 Max   4350.24                    <  43 >                       
                 TGT      AUTO                    <   0 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   5058.42                    [  50 ]                       
                 CPB   5058.42                    [  50 ]                       
                  1C   2225.70                    <  22 >                       
                  2C   1618.69                    <  16 >                       
|- Uncore                                                              [   LOCK]
                 CLK   2023.37                    [  20 ]                       
                 MEM   4046.73                    [  40 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
[... 355 lines cut ...]
010  659.88  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00
011 1062.34  000000000000000000    0.00   0.00   0.00    0.00   0.00   0.00

Energy(J)  Package[0]         Cores               Uncore              Memory
  29.45 38.12 74.32    9.40 27.62 57.78    0.00  0.00  0.00    0.00  0.00  0.00
Power(W)
  29.45 38.12 74.32    9.40 27.62 57.78    0.00  0.00  0.00    0.00  0.00  0.00

[... cut ...]
007 1017.24 (10.05)  23.38  22.50  77.50   0.00   0.00   0.00  0  /  0:0  /  0
008 1181.07 (11.67)  27.15  26.02  73.98   0.00   0.00   0.00  0  /  0:0  /  0
009 1155.23 (11.42)  26.56  25.42  74.58   0.00   0.00   0.00  0  /  0:0  /  0
010  500.60 ( 4.95)  11.51  11.45  88.55   0.00   0.00   0.00  0  /  0:0  /  0
011  825.14 ( 8.16)  18.97  18.27  81.73   0.00   0.00   0.00  0  /  0:0  /  0

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:    Pkg:
                     25.60  24.32  75.68   0.00   0.00   0.00       0 C     0 C
```

cyring, 2024-03-14:
Thank you for the output.
That Ryzen series is really filling the gap with Intel: all ISA and Features are available (beside oldish legacies)

`amd_pstate` might be compiled as __`static`__
Try the below boot parameters
```
modprobe.blacklist=acpi_cpufreq amd_pstate.shared_mem=0 amd_pstate=disable idle=halt
```

Temperature is not showing. That's an issue. I'm searching for thermal registers  and will come back with code for testings.

EUA, 2024-03-15:
With custom kernel boot parameters and again custom corefreqk kernel module parameters, there is an improvement.
[image]

Now CoreFreq reports something on C2-C3 column. But I believe, it should be need CC6 state indeed.

cyring, 2024-03-15:
> it should be need CC6

Using short key `k` or `Menu > Kernel` to open the Kernel window; next, select Idle limit to C6

cyring, 2024-03-15:
Btw can you disable any SPREAD SPECTRUM in your BIOS ?
The expected effect is that the Base Clock should estimate to `100,000,000 Hz`. At least, closer.

cyring, 2024-03-16:
@EUA 
> Now CoreFreq reports something on C2-C3 column. But I believe, it should be need CC6 state indeed.
>> Edit: I forgot to add a parameter to make limit permanent. I will add it soon. Meanwhile you can still use the UI to switch C-State limit as will.

----

I forgot I already made it available through the parameter **`Override_SubCstate`**

For example, looking at your `Sub C-State` (See "Performance Monitoring" window)

```console
|- MONITOR/MWAIT                                                                
   |- State index:    #0    #1    #2    #3    #4    #5    #6    #7              
   |- Sub C-State:     1     1     0     0     0     0     0     0              
```

If your want to make CC6 as a startup limit, you will set a `1` at each index up to the 6th and pad with `0` up to the last 8th

```console
insmod build/corefreqk.ko Register_ClockSource=1 Register_Governor=1 \
Register_CPU_Idle=1 Register_CPU_Freq=1 \
Override_SubCstate="1,1,1,1,1,1,0,0"
```

As a result, you will get such override

```console
|- MONITOR/MWAIT                                                                
   |- State index:    #0    #1    #2    #3    #4    #5    #6    #7              
   |- Sub C-State:     1     1     1     1     1     1     0     0              
```

And the kernel **Idle Limit** as the following

```console
Clock Source                                                  <    corefreq_tsc>
CPU-Freq driver                                               [  corefreqk-perf]
Governor                                                      [ corefreq-policy]
CPU-Idle driver                                               [  corefreqk-idle]
|- Idle Limit                                                 <              C6>
   |- State        POLL      C1      C2      C3      C4      C5      C6         
   |-           CPUIDLE  ZEN-C1  ZEN-C2  ZEN-C3  ZEN-C4  ZEN-C5  ZEN-C6         
   |- Power          -1       0       0       0       0       0       0         
   |- Latency         0       1      20      40      60      80     100         
   |- Residency       0       2      40      80     120     160     200         
```

Consequently Monitoring will start using CC6:

[image]

EUA, 2024-03-16:
Yup. Now it's looks better. Thanks.

[image]

cyring, 2024-03-17:
@EUA 

Can you compile my SMU tool [zencli](https://github.com/cyring/Tips/blob/master/C/zencli.c) ?

```sh
cc zencli.c -o zencli
```

As root, you will peek the base  [register](https://github.com/cyring/CoreFreq/blob/33fa721a97faaf542e0820f8f061a159a9e9cc3a/x86_64/amd_reg.h#L203) for thermal:

```sh
## since Zen gen1
zencli smu 0x59800
## per CCD
zencli smu 0x59954
## Family 19h APU
zencli smu 59B08
```

EUA, 2024-03-17:
Here is the registers you asking for:

```
[root@Ryzen CoreFreq]# ./zencli smu 0x59800
[0x00059800] READ(smu) = 0x822b0000 (2183856128)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1000 0010 0010 1011 0000 0000 0000 0000
[root@Ryzen CoreFreq]# ./zencli smu 0x59954
[0x00059954] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[root@Ryzen CoreFreq]# ./zencli smu 0x59B08
[0x00059b08] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[root@Ryzen CoreFreq]# 
```

cyring, 2024-03-17:
@EUA 

Can you please pull [`develop`](https://github.com/cyring/CoreFreq/tree/develop) branch and test temperature ?

EUA, 2024-03-17:
Output from develop branch ;

```
Processor                            [AMD Ryzen 5 8600G w/ Radeon 760M Graphics]
|- Architecture                                                [Zen4/Hawk Point]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [0x0a705203]
|- Signature                                                           [  AF_75]
|- Stepping                                                            [      2]
|- Online CPU                                                          [ 12/ 12]
|- Base Clock                                                          [101.171]
|- Frequency            (MHz)                      Ratio                        
                 Min   1618.74                    <  16 >                       
                 Max   4350.36                    <  43 >                       
|- Factory                                                             [100.000]
                       4300                       [  43 ]                       
|- Performance                                                                  
   |- P-State                                                                   
                 TGT   4350.36                    <  43 >                       
|- Turbo Boost                                                         [ UNLOCK]
                 XFR   5058.56                    [  50 ]                       
                 CPB   5058.56                    [  50 ]                       
                  1C   2225.77                    <  22 >                       
                  2C   1618.74                    <  16 >                       
|- Uncore                                                              [   LOCK]
                 CLK   2023.42                    [  20 ]                       
                 MEM   4046.85                    [  40 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N/N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX512-F     [Y]    AVX512-DQ [Y]  AVX512-IFMA [Y]   AVX512-PF [N] 
|- AVX512-ER    [N]    AVX512-CD [Y]    AVX512-BW [Y]   AVX512-VL [Y] 
|- AVX512-VBMI  [Y] AVX512-VBMI2 [Y]  AVX512-VNNI [Y]  AVX512-ALG [Y] 
|- AVX512-VPOP  [Y] AVX512-VNNIW [N] AVX512-FMAPS [N] AVX512-VP2I [N] 
|- AVX512-BF16  [Y] AVX-VNNI-VEX [N]    AVX-FP128 [N]   AVX-FP256 [Y] 
|- BMI1/BMI2  [Y/Y]         CLWB [Y]      CLFLUSH [Y] CLFLUSH-OPT [Y] 
|- CLAC-STAC    [Y]         CMOV [Y]    CMPXCHG8B [Y]  CMPXCHG16B [Y] 
|- F16C         [Y]          FPU [Y]         FXSR [Y]   LAHF-SAHF [Y] 
|- MMX/Ext    [Y/Y] MON/MWAITX [Y/Y]        MOVBE [Y]   PCLMULQDQ [Y] 
|- POPCNT       [Y]       RDRAND [Y]       RDSEED [Y]      RDTSCP [Y] 
|- SEP          [Y]          SHA [Y]          SSE [Y]        SSE2 [Y] 
|- SSE3         [Y]        SSSE3 [Y]  SSE4.1/4A [Y/Y]      SSE4.2 [Y] 
|- SERIALIZE    [N]      SYSCALL [Y]        RDPID [Y]        UMIP [Y] 
|- VAES         [Y]   VPCLMULQDQ [Y]   PREFETCH/W [Y]       LZCNT [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Capable]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Capable]
|- Advanced Programmable Interrupt Controller                   APIC   [Capable]
|- Advanced Virtual Interrupt Controller                        AVIC   [Capable]
|- APIC Timer Invariance                                        ARAT   [Capable]
|- LOCK prefix to read CR8                                    AltMov   [Capable]
|- Clear Zero Instruction                                     CLZERO   [Capable]
|- Core Multi-Processing                                  CMP Legacy   [Capable]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Collaborative Processor Performance Control                  CPPC   [Capable]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Capable]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast Short REP MOVSB                                         FSRM   [Capable]
|- Fast-String Operation                                        ERMS   [Capable]
|- Fused Multiply Add                                           FMA4   [Missing]
|- Fused Multiply Add                                            FMA   [Capable]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Hyper-Threading Technology                                    HTT   [Capable]
|- Hardware P-state control                                      HwP   [Capable]
|- Instruction Based Sampling                                    IBS   [Capable]
[... 270 lines cut ...]
  #2 15554  312  192  758   0    0    ON OFF  R0W0   0    6   1T    ON   0 
  #3 15554  312  192  758   0    0    ON OFF  R0W0   0    6   1T    ON   0 
[... cut ...]
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    32    1    131072      1024          32768    F5-8000J4048F24G
 DIMM Geometry for channel #3                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    32    1    131072      1024          32768    F5-8000J4048F24G
```

cyring, 2024-03-17:
As dump with
`zencli smu 0x59800`
I was expecting to read temperature.
But also 2 DRAM sticks.

Do you confirm building/running latest commits ?

https://github.com/cyring/CoreFreq/commits/develop/

EUA, 2024-03-18:
I did;
```
git branch develop
git pull
make
```
Only kernel module rebuilt. I am not git guru but, I think this is what should I need to do... Don't know how to verify it.

Than `insmod` as you write at top... Don't know if I am able to switch properly but... Something pulled from git.

I have 2 Ram slot on my MB and both of them used with 24GB Sticks.

cyring, 2024-03-18:
I think I have a digit mistake in register address.
Can you edit this header file at that line:
https://github.com/cyring/CoreFreq/blob/87344fae6395c7d3d5dd6b7b0876a6ee3aae413d/x86_64/amd_reg.h#L208

and replace code with this one:

```c

#define SMU_AMD_THM_TCTL_CCD_REGISTER_F19H_61H				\
	(SMU_AMD_THM_TCTL_REGISTER_F17H + 0x300)
```

Next please rebuild all
```
make clean
make
```
And reload all for temperature testing
(make sure to unload before loading driver again)

Thank you

EUA, 2024-03-18:
Replaced the lines as you indicated. Still no core temps appears on main screen.

```[root@Ryzen CoreFreq]# ## Prior Zen4
./zencli smu 0x50030
./zencli smu 0x50034
./zencli smu 0x50038
./zencli smu 0x5003C
## Since Zen4
./zencli smu 0x50040
./zencli smu 0x50044
./zencli smu 0x50048
./zencli smu 0x5004C
[0x00050030] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00050034] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00050038] READ(smu) = 0x00fffffe (16777214)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 1111 1111 1111 1111 1111 1110
[0x0005003c] READ(smu) = 0x00000000 (0)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000
[0x00050040] READ(smu) = 0x00150508 (1377544)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0101 0000 0101 0000 1000
[0x00050044] READ(smu) = 0x00150508 (1377544)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0101 0000 0101 0000 1000
[0x00050048] READ(smu) = 0x0025070c (2426636)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0010 0101 0000 0111 0000 1100
[0x0005004c] READ(smu) = 0x00150508 (1377544)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 0000 0000 0001 0101 0000 0101 0000 1000
```

I don't know what this SMU outputs means but... I definitely sure about that, at least, one of my sticks is placed at other slot and my Rams are 24GB. Looks like some electronic info at chips do not comply with reality. :smile: 

```
 DIMM Geometry for channel #0                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    32    1    131072      1024          32768    F5-8000J4048F24G
 DIMM Geometry for channel #1                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    32    1    131072      1024          32768    F5-8000J4048F24G
 DIMM Geometry for channel #2                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    32    1    131072      1024          32768    F5-8000J4048F24G
 DIMM Geometry for channel #3                                              
      Slot Bank Rank     Rows   Columns    Memory Size (MB)                
       #0                                                                  
       #1    32    1    131072      1024          32768    F5-8000J4048F24G

```

cyring, 2024-03-18:
@EUA 
> Replaced the lines as you indicated. Still no core temps appears on main screen.

So it appears that fix is working for EPYC/Genoa but fails on yours and 7950X

> I don't know what this SMU outputs means

I'm making a [page](https://gist.github.com/cyring/eb13992a3b9482eb4bde77a2cf62c605) to debug new DIMMs

cyring, 2024-03-18:
@EUA 

Can you load the `k10temp` module and show the temperature output from `sensors` command ?

EUA, 2024-03-18:
Of-course I am.

```k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +82.9°C
```
Unfortunately, **zenpower** does not work anymore.
With detailed info about Zen CPU's, might be you can fetch Voltage and Current info from SVI and revive the project as your fork...

cyring, 2024-03-18:
That's the odd thing:
Making use of the same SMU thermal address as the one mentioned in [k10temp source code](https://elixir.bootlin.com/linux/v6.8.1/source/drivers/hwmon/k10temp.c#L470) and not reading a temperature value !

`Tctl` Register has to be at `0x59800 + 0x308`:
```
zencli smu 0x59B08
```

cyring, 2024-03-19:
If `+82.9°C` was not your current CPU temperature, it means `sensors` returned a garbage value.

cyring, 2024-03-19:
> Unfortunately, **zenpower** does not work anymore. With detailed info about Zen CPU's, might be you can fetch Voltage and Current info from SVI and revive the project as your fork...

```
corefreq-cli -j
```
can output data in JSON, including voltage, power, temp

EUA, 2024-03-19:
And k10temp is actually reading from the die temp. So it's reading perfectly correct while I am mining crypto with my CPU right now :) nct6799-isa-0290 is also reporting it as :  `TSI0_TEMP:  +81.6°C  ` right now.

Unfortunately, 8000G series has ultra thick IHS, glued with low grade thermal glue. (really a big shame for AMD!) Just altering thermal paste with PTM7950 or a liquid metal reduces temps up to 25C degree (of-course with AIO setup). I am planning to delid it soon, when my PTM7950 thermal pad comes from AliExpress. So might be my fan could gonna holiday even if I am torturing the CPU with cryptos...

`zencli smu 0x59B08` returns all zeros, nothing more. So might be zencli not working OR, do you sure about if correct base address is really 0x59800 for my CPU?

While corefreq-cli -j can return those data, but AFAIK all linux users will definitely prefer to see it on lm-sensors directly. Not from another program. So we could move that value into Gnome shell, with freon extension. Think about it. I don't think if it's hard to do while you have already the working code... Probably req. just adding some numbers into header file and viola...

cyring, 2024-03-19:
@EUA 

> OR, do you sure about if correct base address is really 0x59800 for my CPU?

Brillant suggestion you have.

Looking closer into function `k10temp_get_ccd_support` which loops `8` times, starting from `0x59800 + 0x308`, adding index multiplied by `4` at each step, and attempting a temperature read if bit `11` is raised.

So can you please read this range of the `7` remaining addresses ?
```
zencli smu 0x59B0C
zencli smu 0x59B10
zencli smu 0x59B14
zencli smu 0x59B18
zencli smu 0x59B1C
zencli smu 0x59B20
zencli smu 0x59B24
```

cyring, 2024-03-20:
@EUA Hello. Not sure if you were notified about above `zencli` readings ?

EUA, 2024-03-20:
> @EUA
> 
> > OR, do you sure about if correct base address is really 0x59800 for my CPU?
> 
> Brillant suggestion you have.

Yup. Have some natural gift on beta testing. :) Busy with soldering mosfets to my old motherboard.

And about for those registers; all the range cleared with zero.

cyring, 2024-03-21:
> > @EUA
> > 
> > > OR, do you sure about if correct base address is really 0x59800 for my CPU?
> > 
> > Brillant suggestion you have.
> 
> Yup. Have some natural gift on beta testing. :) I have busy with soldering mosfets to my old motherboard last night.
> 
> And about for those registers; all the range cleared with zero.
> 
> 

If all cleared then no CCD.
And `TCTL` only at `read_tempreg_nb_zen()` which reads temperature at:

```
./zencli smu 0x00059800
```

Can you try, probably again, `0x00059800` ?

cyring, 2024-03-21:
Using your previous read of:
```sh
./zencli smu 0x00059800
```
```
[0x00059800] READ(smu) = 0x822b0000 (2183856128)
   60   56   52   48   44   40   36   32   28   24   20   16   12   08   04   00
 0000 0000 0000 0000 0000 0000 0000 0000 1000 0010 0010 1011 0000 0000 0000 0000
```
Now extracting bits and computing temperature:
```sh
echo $(( 0x822b0000 >> 21 ))
```
```
1041
```
```sh
echo $(( ((1041*5)/40)-49 ))
```
```
81
```

So the SMU thermal register address is definitely `0x00059800`

I just merged all development commits into `master` which you can pull and test for temperature.

EUA, 2024-03-21:
So what could be a problem? Some kernel options are not allowing to read them?
Temperature value is start to appear. But not for each core, yet...

cyring, 2024-03-22:
> So what could be a problem? Some kernel options are not allowing to read them? Temperature value is start to appear. But not for each core, yet...

Cam you show me screenshots ?

In contrary to `CCD`, `TCTL` has to be be a single sensor so the reading scope should be `Package` in "Settings" window.

EUA, 2024-03-23:
I got it. This from TCL, not CCD. At HWINFO (windows), we got detailed voltage and temperature info belongs to each core. Not one temp applied to each core. So, this is misleading. For now, unfil figure to read CCD from chip,it's better to not show SMT or Core but "Package" automatically at settings for 8000 Series APUs,...

[image]

cyring, 2024-03-23:
> I got it. This from TCL, not CCD. At HWINFO (windows), we got detailed voltage and temperature info belongs to each core. Not one temp applied to each core. So, this is misleading. For now, unfil [...]

* Sure it will be part of next commit.

  Phoenix v1 & v2 will also be set the same default.

  However I **need** Phoenix owners to help me confirm the SMU thermal register.

  My understanding is that Hawk Point is a single CCD: can you show the Core Topology ? 
```
corefreq-cli -m
```
  If true then `TCTL` is shared among all Cores: like any previous Ryzen generation

> Noticed that, core 010 minimum reported as 64... It's just a sw glitch I think. otherwise if you torture CPU 1, also CPU10 reports same value.

* Please show me as many screenshots and/or outputs as possible because it's difficult from here to figure what's the issue is.

* Also sensors screenshots of HWINFO when a single and only one CPU is stressed out.

EUA, 2024-03-25:
Indeed, at APU series, there are **no discrete CCD** as other Ryzen 7000 CPUs.
All is made from a single piece of silicon which includes the "CCD" complex.
This might be a reason about why your 7000 series code not work on APU.

Also different minimum report is not really important since it will be fade away with proper CCD thermal implementation...

Here is the info:

```
[root@Ryzen CoreFreq]# ./build/corefreq-cli -m
CPU Pkg  Apic  Core/Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID CCD CCX ID/ID L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
000:BSP    0   0  0   0  0      32  8        32  8      1024  8 i   16384 16w 
001:  0    2   0  0   1  0      32  8        32  8      1024  8 i   16384 16w 
002:  0    4   0  0   2  0      32  8        32  8      1024  8 i   16384 16w 
003:  0    6   0  0   3  0      32  8        32  8      1024  8 i   16384 16w 
004:  0    8   0  1   4  0      32  8        32  8      1024  8 i   16384 16w 
005:  0   10   0  1   5  0      32  8        32  8      1024  8 i   16384 16w 
006:  0    1   0  0   0  1      32  8        32  8      1024  8 i   16384 16w 
007:  0    3   0  0   1  1      32  8        32  8      1024  8 i   16384 16w 
008:  0    5   0  0   2  1      32  8        32  8      1024  8 i   16384 16w 
009:  0    7   0  0   3  1      32  8        32  8      1024  8 i   16384 16w 
010:  0    9   0  1   4  1      32  8        32  8      1024  8 i   16384 16w 
011:  0   11   0  1   5  1      32  8        32  8      1024  8 i   16384 16w 
```

Also my HWINFO

[image]

cyring, 2024-03-25:
Question: is there `CCX` in APU ?

cyring, 2024-03-25:
@EUA 

Last commit 3778ece916d702fe3811f387bc77b469b0552733 I'm removing Complex from the APU topology

Can you please test the change and post `corefreq-cli -m`

----

About HWINFO, I've no idea which those GFX temperature registers are ?

EUA, 2024-03-29:
```
[root@Ryzen CoreFreq]# ./build/corefreq-cli -m
CPU Pkg  Apic  Core/Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID    ID     ID  L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
000:BSP    0     0      0       32  8        32  8      1024  8 i   16384 16w 
001:  0    2     1      0       32  8        32  8      1024  8 i   16384 16w 
002:  0    4     2      0       32  8        32  8      1024  8 i   16384 16w 
003:  0    6     3      0       32  8        32  8      1024  8 i   16384 16w 
004:  0    8     4      0       32  8        32  8      1024  8 i   16384 16w 
005:  0   10     5      0       32  8        32  8      1024  8 i   16384 16w 
006:  0    1     0      1       32  8        32  8      1024  8 i   16384 16w 
007:  0    3     1      1       32  8        32  8      1024  8 i   16384 16w 
008:  0    5     2      1       32  8        32  8      1024  8 i   16384 16w 
009:  0    7     3      1       32  8        32  8      1024  8 i   16384 16w 
010:  0    9     4      1       32  8        32  8      1024  8 i   16384 16w 
011:  0   11     5      1       32  8        32  8      1024  8 i   16384 16w 

```

Since APU has GPU in it, CoreFreq might show the GPU clock also. Because soon gcc start to move some threads into GPU section at APU... So it counts as core too. Just brainstorming.

cyring, 2024-03-29:
The main issue is solved and the remaining actions are in list to do after this release candidate.
Thank you very much for your help. Feel free to close your issue.

haarp, 2025-09-01:
Hey, just discovered this tool. Running a Ryzen Pro 8700G and tested a few things.

With default settings the CPU goes down to C3 according to powertop, or C1 according to CoreFreq. Idle power at outlet: 12.2 watts.

After booting with `modprobe.blacklist=acpi_cpufreq amd_pstate.shared_mem=0 amd_pstate=disable idle=halt` and inserting the module with `insmod build/corefreqk.ko Register_ClockSource=1 Register_Governor=1 Register_CPU_Idle=1 Register_CPU_Freq=1 Override_SubCstate="1,1,1,1,1,1,0,0"` I can see C-states up to `C4:C6`. Idle power at outlet: 13.3 watts.

So the corefreq idle driver actually increases idle power consumption. I suspect that with the default driver, the CPU **does** go down to C6, but the driver doesn't expose that information. Whereas the corefreq driver does slightly worse.

haarp, 2025-09-03:
Yes, this is reproducible.

I see what you mean. This can also be seen at vcores, 0.7438 was the lowest on acpi_idle, 0.7313 on corefreq. This seems to indicate a difference in actual hardware C-states between the drivers. Would be interesting to investigate further.

Could this be an effect of cpufreq? Do you think it might be worthwhile to combine the `amd-pstate-epp` driver (which is supposedly pretty good as it leaves the p-states up to the hardware) with the corefreq idle driver?

cyring, 2025-09-03:
> Yes, this is reproducible.
>
> I see what you mean. This can also be seen at vcores, 0.7438 was the lowest on acpi_idle, 0.7313 on corefreq. This seems to indicate a difference in actual hardware [...]

It could be interesting to investigate AMD pstate module also I believe it deals more about CPPC, Collaborative Performance P-state

I just give a look into ACPI module and noticed thing like Cache management; I'm not doing prior MWAIT.

* CoreFreq MWAIT idle function 
https://github.com/cyring/CoreFreq/blob/12d2d2151a44c3bf258a76bca561e9c2f54e549d/x86_64/corefreqk.c#L21060

* ACPI Module MWAIT function 
https://codebrowser.dev/linux/linux/arch/x86/kernel/acpi/cstate.c.html#acpi_processor_ffh_play_dead

Both calls finish into `mwait_play_dead` which calls `wbinvd`, memory barriers, `clflush`

So adding cache invalidate and flushing the cache line

Those can be inserted prior the inline asm block in the CoreFreq function to see if they make any power consumption difference.

cyring, 2025-09-03:
> Also why the infinite loop `while (1)` and how the driver is breaking out from it ?

The `while (1)` is a scheduler construct, not a hardware requirement for the asm _idle_  instructions: `hlt mwait mwaitx i/o`

cyring, 2025-09-04:
> I have an additional data point. There's an Optane SSD attached to CPU's PCIe lanes, which won't do ASPM. This is a known cause preventing the CPU from reaching deeper sleep states. Yet corefreqk-idle claims it reaches C6. Is it possible it's reading bogus values?

When _CoreFreq_ shows C3 C6; those are _software_ C-states that I computing myself.
Those counters won't assure that the processor had indeed fall into hardware deep sleep.
You are indeed reading the sleep level (C1, C3, C6) in the UI because the kernel is _forced_ to call my hook each time the scheduler has no task to execute, and the kernel is passing the desired sleep level to my function. Thus I'm able to dispatch in my _statistics_ the cycles for a desired level.

In short, why you see is not necessary what you get.
And you made the demonstration with your power outlet measurements that no power is saved on Zen.

How I can fix this mess ? 
MONITOR/MWAIT works perfectly with Intel processors, following their SDM specification but with AMD, only God has the answer. AMD specs are exactly the same explanations that Intel but a few bits are not mentioned ...

cyring, 2025-09-05:
* `corefreqk-idle`

[image]

cyring, 2025-09-05:
Like the `MWAIT` [kernel implementation](https://codebrowser.dev/linux/linux/arch/x86/kernel/smpboot.c.html#mwait_play_dead), I'm adding all kinds of memory barrier (`mfence`) and cache operations (`wbinvd`, `clflush`) which don't improve the power consumption when `MWAIT` is the `CPU-IDLE route`

```c
	/*			MWAIT Idle methods			*/
static int CoreFreqK_MWAIT_Handler(struct cpuidle_device *pIdleDevice,
				struct cpuidle_driver *pIdleDriver, int index)
{/*	Source: /drivers/cpuidle/cpuidle.c				*/
	unsigned long MWAIT=(CoreFreqK.IdleDriver.states[index].flags>>24)&0xff;
	UNUSED(pIdleDevice);
	UNUSED(pIdleDriver);

	__asm__ volatile
	(
		"xorq	%%rcx	,	%%rcx"	"\n\t"
		"xorq	%%rdx	,	%%rdx"	"\n\t"
		"leaq	%[addr] ,	%%rax"	"\n\t"
		"wbinvd"			"\n\t"
		"mfence"			"\n\t"
		"clflush %[addr]"		"\n\t"
		"mfence"			"\n\t"
		"monitor"			"\n\t"
		"mfence"			"\n\t"
		"# Bit 0 of ECX must be set"	"\n\t"
		"movq	$0x1	,	%%rcx"	"\n\t"
		"movq	%[hint] ,	%%rax"	"\n\t"
		"mwait"
		:
		: [addr] "m" (current_thread_info()->flags),
		  [hint] "ir" (MWAIT)
		: "%rax", "%rcx", "%rdx", "memory"
	);

	return index;
}

static int CoreFreqK_MWAIT_AMD_Handler(struct cpuidle_device *pIdleDevice,
				struct cpuidle_driver *pIdleDriver, int index)
{
	return CoreFreqK_MWAIT_Handler(pIdleDevice, pIdleDriver, index);
}

#if ((LINUX_VERSION_CODE < KERNEL_VERSION(5, 9, 0)) && (RHEL_MAJOR == 0)) \
 || ((RHEL_MAJOR == 8) && (RHEL_MINOR < 4))
static void CoreFreqK_S2_MWAIT_Handler(struct cpuidle_device *pIdleDevice,
				struct cpuidle_driver *pIdleDriver, int index)
#else
static int CoreFreqK_S2_MWAIT_Handler(struct cpuidle_device *pIdleDevice,
				struct cpuidle_driver *pIdleDriver, int index)
#endif /* 5.9.0 */
{
	unsigned long MWAIT=(CoreFreqK.IdleDriver.states[index].flags>>24)&0xff;
	UNUSED(pIdleDevice);
	UNUSED(pIdleDriver);

	__asm__ volatile
	(
		"xorq	%%rcx	,	%%rcx"	"\n\t"
		"xorq	%%rdx	,	%%rdx"	"\n\t"
		"leaq	%[addr] ,	%%rax"	"\n\t"
		"wbinvd"			"\n\t"
		"mfence"			"\n\t"
		"clflush %[addr]"		"\n\t"
		"mfence"			"\n\t"
		"monitor"			"\n\t"
		"mfence"			"\n\t"
		"# Bit 0 of ECX must be set"	"\n\t"
		"movq	$0x1	,	%%rcx"	"\n\t"
		"movq	%[hint] ,	%%rax"	"\n\t"
		"mwait"
		:
		: [addr] "m" (current_thread_info()->flags),
		  [hint] "ir" (MWAIT)
		: "%rax", "%rcx", "%rdx", "memory"
	);

#if ((LINUX_VERSION_CODE >= KERNEL_VERSION(5, 9, 0)) || (RHEL_MINOR >= 4))
	return index;
#endif /* 5.9.0 */
}
```

[image]

cyring, 2025-09-05:
> HWinfo on Windows seems to be able to, or claims to be showing C6 residency. But not C2-5

I think it comes from the _reversed_ SMU mailbox protocol. If I remember well, one of the commands will retrieve in percent the C6. You should observe the same with Ryzen Master. Unfortunately those are not cycles.