cyring/CoreFreq #85: Add support SCHED_MUQSS kernel

fedan, 2018-11-09:
Hi, I have a kernel with SCHED_MUQSS (http://ck.kolivas.org/patches/muqss). I created a patch to build a module with muqss support:
```diff
diff --git a/corefreqk.c b/corefreqk.c
index 68cec48..96be6ef 100644
--- a/corefreqk.c
+++ b/corefreqk.c
@@ -4390,8 +4390,11 @@ void Sys_DumpTask(SYSGATE *SysGate)
 	rcu_read_lock();
 	for_each_process_thread(process, thread) {
 		task_lock(thread);
-
+#ifdef CONFIG_SCHED_MUQSS
+		SysGate->taskList[cnt].runtime  = tsk_seruntime(thread);
+#else
 		SysGate->taskList[cnt].runtime  = thread->se.sum_exec_runtime;
+#endif
 		SysGate->taskList[cnt].usertime = thread->utime;
 		SysGate->taskList[cnt].systime  = thread->stime;
 		SysGate->taskList[cnt].pid      = thread->pid;

```

cyring, 2018-11-09:
Seems fine.
Is ```tsk_seruntime()``` safe in interrupt context ?
How can I quickly test it before with ArchLinux or any live ISO ?

fedan, 2018-11-09:
> Is `tsk_seruntime()` safe in interrupt context ?

This is a define in include/linux/sched.h:
```c
+#ifdef CONFIG_SCHED_MUQSS
+#define tsk_seruntime(t)		((t)->sched_time)
+#define tsk_rttimeout(t)		((t)->rt_timeout)
...
+#else /* CFS */
+#define tsk_seruntime(t)	((t)->se.sum_exec_runtime)
+#define tsk_rttimeout(t)	((t)->rt.timeout)
...
+#endif /* CONFIG_SCHED_MUQSS */
```

cyring, 2018-11-09:
Works nicely. Thanks a lot.
[image]

closed 2018-11-09 by fedan