cyring/CoreFreq #86: Executing make does not enter kernel directory

wordlessly, 2018-11-19:
Also it does not create corefreqk.ko. See make output below:

> make
cc  -Wall -pthread -c corefreqd.c \
	-D FEAT_DBG=1 -o corefreqd.o
cc  -Wall -c corefreqm.c -o corefreqm.o
cc  -Wall corefreqd.c corefreqm.c \
	-D FEAT_DBG=1 -o corefreqd -lpthread -lm -lrt
cc  -Wall -c corefreq-cli.c -o corefreq-cli.o
cc  -Wall -c corefreq-ui.c -o corefreq-ui.o
cc  -Wall -c corefreq-cli-json.c \
	-o corefreq-cli-json.o
cc  -Wall -c corefreq-cli-extra.c \
	-o corefreq-cli-extra.o
cc  -Wall \
	corefreq-cli.c corefreq-ui.c \
	corefreq-cli-json.c corefreq-cli-extra.c \
	-o corefreq-cli -lm -lrt

cyring, 2018-11-19:
Do you confirm you have installed the kernel headers/source ? 

Depending of your Linux distribution, if you install the dkms package, this may drag all the prerequisites for kernel module development.

wordlessly, 2018-11-19:
I installed headers/source manually. Maybe you want to extend the description on how to make CoreFreq on Ubuntu.

Also add-apt-repository ppa:cyring/corefreq is empty. Maybe you want to remove the advice to add this repo to install CoreFreq.

Cheers!

wordlessly, 2018-11-20:
Sorry, but I am not sure how to build the driver step by step. Can I tell make to do this?

Maybe I wasn't clear in my last message. I installed the headers/source for my kernel. After that the build did work flawlessly.

cyring, 2018-11-20:
Ok I will soon release a new Makefile. Meanwhile the instructions I post above should build the missing `corefreqk.ko`

cyring, 2018-11-24:
Please let me know if you can successfully build _CoreFreq_ ?

cyring, 2018-12-01:
I'm closing the issue : a simple `Makefile` is released.

closed 2018-12-01 by cyring