cyring/CoreFreq #87: AMD Zen: experimental overclocking

cyring, 2018-11-24:
To overclock the Zen family processors from the UI:
1. Disable Boost in [Technologies]
2. Activate the Experimental mode in [Settings]
3. In [Processor] select the 1C ratio
4. Change _slightly_ the clock

It might be advised to also disable CC6 and PC6 before overclocking

adatum, 2018-11-24:
Changing 1C ratio isn't actually changing the frequency ratio as seen in the max ratio value and the frequency reading during a burn test:

[image]

Also, upon restoring Boost, the cpu is still stuck to a maximum of 3.7 GHz.

cyring, 2018-11-25:
Don't you reach a ratio of 39 when burning one and only core ?

It may work like Intel when set for a one core ratio, you need to stress one, and only one CPU 

The same with 2C, you reach the defined ratio when 2 CPU are only loaded (plus other processor limits: TDP) 

You can use the integrated stress tool [F3] and choose 'Turbo Random CPU' or 'Turbo Round Robin' to burn 1C

Also try to disable CC6 and PC6

It may overclock with the 2C P-State ? 

Boost restoring is a remaining task but unlike Intel,  none AMD register is giving me the factory ratio values.

As current code, I can compute 1C with the max non turbo (factory 37) then add XFR but what about 2C or 3C...

cyring, 2018-11-25:
In AMD specs, I'm reading the `MSRC001_0061 [P-state Current Limit]`  

|Bits|Description|
|-----|---------------|
|<p align="right">`63:7`</p>| Reserved.|
|<p align="right">`6:4`</p>| PstateMaxVal: P-state maximum value... </br>Specifies the lowest-performance non-boosted P-state (highest non-boosted value) allowed.|
|<p align="right">`3`</p>| Reserved. Read-only. Reset: Fixed,0.|
|<p align="right">`2:0`</p>| CurPstateLimit: current P-state limit... </br>Specifies the highest-performance P-state (lowest value) allowed.|

These could be reliable values to restore ratio when restoring Boost.

Can you please `rdmsr 0xC0010061` before then after changing ratio. Also when restoring boost ?

adatum, 2018-11-25:
AMD's boost is different than Intel's and kind of complicated with PB2, XFR2, and PBO algorithmically managing multi-core boosting depending on various safety limits. The trouble is that those technologies are named differently in the BIOS, depending on the vendor. Some info about the boosting [here](https://community.amd.com/community/gaming/blog/2017/11/27/), [here](https://community.amd.com/community/gaming/blog/2018/08/13/understanding-precision-boost-overdrive-in-three-easy-steps), [here](https://www.hardocp.com/article/2018/04/30/amd_precision_boost_2_wraith_prism_deep_dive/1), and [here](https://www.anandtech.com/show/13124/the-amd-threadripper-2990wx-and-2950x-review/3).

Normally the all-core frequency I see is at least 3.8GHz up to 4.1GHz depending on the type of load.

But I don't think the ratio actually changed because I tried higher (39), and also lower (34) ratios, but the stress test always resulted in the same 3.7GHz all-core frequency. I also tried with CC6 and PC6 disabled.

Also, my understanding is that when manually overclocking, defined by changing the ratio from the stock 37, it is an all-core overclock. 

Without changing ratio `rdmsr 0xC0010061` reads 20. The trouble with testing the ratio right now is that the only way I have found to get the normal boosting back is to reboot.

cyring, 2018-11-25:
Sorry for the reboot inconvenience. I hope we could find a way to restore values online.
Also about MSR 0xC0010061 we have already worked on it https://github.com/cyring/CoreFreq/issues/54#issuecomment-392280424 . Those registers don't supply ratios but just P-States. I could compute the core coefficient from them but how to be sure they have not been altered since reboot.

As shown in some BIOS screens, perhaps the OC works when applied to P-State `P0`

Could you replace this function with the following code:
https://github.com/cyring/CoreFreq/blob/677adf07ac05ab310ccf5a945cd36f3fac00913f/corefreqk.c#L3132
```C
long TurboClock_AMD_Zen(CLOCK_ARG *pClockMod)
{
	long rc = 0;
    if (Proc->Registration.Experimental)
    {
	HWCR HwCfgRegister = {.value = 0};
	/* Make sure the Core Performance Boost is disabled. */
	RDMSR(HwCfgRegister, MSR_K7_HWCR);

	if (HwCfgRegister.Family_17h.CpbDis && (pClockMod != NULL)) {
		unsigned long long PstateAddr = MSR_AMD_PSTATE_DEF_BASE;
		PSTATEDEF PstateDef = {.value = 0};

/*	    switch (pClockMod->Ratio) {
	    case 1 ... 7:
		PstateAddr += pClockMod->Ratio;		*/
		RDMSR(PstateDef, PstateAddr);
		/* Apply if and only if the P-State is enabled */
		if (PstateDef.Family_17h.PstateEn) {
/*			unsigned int index = BOOST(SIZE) - pClockMod->Ratio;*/
			unsigned int index = BOOST(MAX);
			/* Compute the P-State Frequency ID from the ratio */
			unsigned int FID = 0, COF = 0;
			FID = AMD_Zen_CoreFID(	Proc->Boost[index]
						+ pClockMod->Offset,
						PstateDef.Family_17h.CpuDfsId);
			/* Write then Read the P-State MSR */
			PstateDef.Family_17h.CpuFid = FID;
			WRMSR(PstateDef, PstateAddr);
			RDMSR(PstateDef, PstateAddr);
			/* Re-compute the Core coefficient */
			COF = AMD_Zen_CoreCOF(	PstateDef.Family_17h.CpuFid,
						PstateDef.Family_17h.CpuDfsId);
			/* Update the Boost ratio with COF */
			Proc->Boost[index] = COF;
			rc = 2;
		}
/*		break;
	    }	*/
	}
    }
	return(rc);
}
```

This code will add any selected offset (last column) to the Max Frequency (aka P0 State)
[image]
For example, you select `+2` in the [Turbo Clock 1C] window then the Max Frequency should be OCed to `37 + 2 = 39`

adatum, 2018-11-25:
> For the test, I recommend to start the driver with the argument AutoClock=0. You will get 100MHz
>
> 1. Increase the ratio by one unit then apply load to check the result
> 2. Next, decrease by one unit to check if frequency is back to nominal.

This worked with changing Max Frequency ratio, but increasing +1 (38) resulted in a 200 MHz jump (3900 MHz) when load testing single CPU in round robin. Decreasing by -1 back to 37 also worked and load testing saw 3700 MHz. 

Also, the Atomic Burn test, after moving the ratio +1 and also -1, made CoreFreq unresponsive for a while and practically froze the entire computer (not normal). This testing feels rather risky, much more so than what we've previously done.

After testing and re-enabling CBP, the actual max frequency is still capped at 3700 MHz.

cyring, 2018-11-25:
So it went from 3.7 to 4.0 GHz despite you have selected 3.8 GHz. I suspect a rounded math issue. Zen can do decimal frequencies (.25 ; .5 ; .75)

It might also be necessary to delay after modifying the FID in the P-State register. But I don't read such thing in specs.

Is it also unresponsive when testing with an external stress tool ?

cyring, 2018-12-01:
I have brought code changes to overclock per Core. Can you please try the last version ?

cyring, 2018-12-01:
... and also print `corefreq-cli -s`

adatum, 2018-12-01:
By Boost you mean CPB, yes? FYI, PBO is also enabled.

So far CPU `#0` and `#7` give the same `rdmsr` readings. 

|When | PStateDef `0xc0010064`| HWCR `0xc0010015` | Unknown PState `0xc0010293`|
|---|---|---|---|
|BIOS Boost Enabled [nominal] | 80000000494d8894 | 49000011 | Varies: 8550ae, 97cc80, 9ccc84, 978c80, 974c80, 97cc84, 5d0880, 8510ae, 980c80, 9d0c80, 83cc80, 840c80, 8390ae, etc |
|BIOS Overclock @ 3800 MHz | TBA | TBA | TBA | 
|BIOS Overclock @ 3900 MHz | TBA | TBA | TBA | 

---
Did you want `corefreq-cli -s` for each BIOS state?

`corefreq-cli -s` with BIOS CPB Enabled:

```
$ ./corefreq-cli -s
Processor                              [AMD Ryzen 7 2700X Eight-Core Processor ]
|- Architecture                                            [Zen+ Pinnacle Ridge]
|- Vendor ID                                                      [AuthenticAMD]
|- Signature                                                            [ 8F_08]
|- Stepping                                                             [     2]
|- Microcode                                                            [     0]
|- Online CPU                                                           [ 16/16]
|- Base Clock                                                           [100.00]
|- Frequency            (Mhz)                      Ratio                        
                 Min    400.00                    [   4 ]                       
                 Max   3700.00                    <  37 >                       
|- Factory                                                              [100.00]
                       3700                       [  37 ]                       
|- Turbo Boost                                                          [UNLOCK]
                  1C   4400.00                    <  44 >                       
                  2C   3200.00                    <  32 >                       
                  3C   2200.00                    <  22 >                       
|- Uncore                                                               [  LOCK]
                                                                                
ISA Extensions:                                                                 
|- 3DNow!/Ext [N,N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [N]  BMI1/BMI2 [Y/Y]        CLFSH [Y]        CMOV [Y] 
|- CMPXCH8      [Y]     CMPXCH16 [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/Y]     MONITOR [Y] 
|- MOVBE        [Y]          MPX [N]     PCLMULDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [Y]       RDTSCP [Y]         SEP [Y] 
|- SGX          [N]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/Y]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features:                                                                       
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- 100 MHz multiplier Control                            100MHzSteps   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Present]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Missing]
|- CPL Qualified Debug Store                                  DS-CPL   [Missing]
|- 64-Bit Debug Store                                         DTES64   [Missing]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- No-Execute Page Protection                                     NX   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
[... 45 lines cut ...]
|                      1       1       0       0       0                        
|- Core Cycles                                                         [Present]
|- Instructions Retired                                                [Present]
|- Reference Cycles                                                    [Present]
|- Last Level Cache References                                         [Missing]
|- Last Level Cache Misses                                             [Missing]
|- Branch Instructions Retired                                         [Missing]
|- Branch Mispredicts Retired                                          [Missing]
[... cut ...]
|- Package Thermal Management                                    PTM   [Missing]
|- Thermal Monitor 1                                         TM1|TTP   [Present]
|- Thermal Monitor 2                                         TM2|HTC   [Present]
|- Units                                                                        
   |- Power                                               watt   [  0.015625000]
   |- Energy                                             joule   [  0.000001907]
   |- Window                                            second   [  0.000976562]
```

cyring, 2018-12-01:
Yes I meant CPB.
Meanwhile I have pushed code which hopefully can overclock the Ryzen. You may have to clone again
What's new is that I now modify the frequency on all Zen cores (thread scope)  whereas Intel processors use common registers (package scope)

adatum, 2018-12-01:
All done with the latest code.

cyring, 2018-12-02:
Thanks. I will analyze those bits... 
Can you try now to overclock from CoreFreq UI b/c the algorithm has changed.

adatum, 2018-12-02:
Do you want me to change the Max Frequency ratio or the Turbo Boost 1C ratio?

cyring, 2018-12-02:
Beside the `CpbDis` and `IRPerfEn` flags, the `HWCR` register does not show major differences when activating overclocking in BIOS.

I don't have a factory ratio to restore CPB with. When overclock is enabled in BIOS, the nominal ratio is overrided (in the P0 P-State). You will have to do all manually and write down the max frequency like bellow.

A new version is available for testing.

1. Boot with BIOS CPB = ON
From _CoreFreq_ :
2. In the Processor view: write down the max frequency ratio (which should be 37)
3. In the Technology view: disable Turbo
4. In the Processor view: select a new max frequency ratio
5. Stress and check this new overclock frequency
6. In the Processor view: restore the nominal max frequency ratio (select  37)
7. In the Technology view: enable Turbo
8. Stress and check if Boost and XFR are in effect.

(please provide screenshots, this help me to chase bugs)

adatum, 2018-12-02:
It works fine until step 7-8. When restoring CPB, 1C, 2C don't change back to what they were before. Manually trying to restore them to the original values doesn't seem to work.

Pre-OC:
[image]

OC:
[image]

Post-OC:
[image]

Got 1C to 44 but then 2C dropped to 22. Changing 3C to 20 made it vanish. The Turbo ratios are behaving strangely.

[image]

cyring, 2018-12-02:
Nice progress😎

It's hard to tell the magic inside P-States without further specifications from manufacturer

Probably, Turbo has to be disabled prior changing one of the Turbo ratio. 

How is the BCLK estimation behaving with this version ?
Can you test the max frequency overclocking without the driver argument `AutoClock`

cyring, 2018-12-02:
Btw I don't alter the voltage VID. This requires to work directly with the hardware and to notice any slight change. 

Thus overclocking is limited to the default P-State voltage. 
In your case the VID provided during boot time and BIOS settings.

adatum, 2018-12-02:
BCLK was reading 99.81 MHz without the `AutoClock` argument. While changing ratios for a bit, I did not notice the BCLK change much if at all.

Disabling/re-enabling CPB does seem to make a difference with the Turbo ratios. eg. when 3C disappears, disabling and re-enabling CPB brings it back. 

But changing the Turbo ratios are like a game of whack-a-mole. Changing one ratio sometimes changes other ratios. And re-enabling CPB sometimes results in weird setups like having 2C with a higher ratio than 1C.

Also my system has totally locked up a couple of times while changing ratios.

cyring, 2018-12-02:
Thanks for your reply.
Restoring the CPB will be a long road. I wonder how the BIOS is managing that; beside rebooting to safe values.

There's an "enablement" bit in each of the 8 P-State registers, from P0 to P7
My algorithm don't take into account a disabled P-State
Consequently a disabled P-State won't show into the frequency view (as 1C, 2C,... 7C)
When starting up `corefreqk.ko` finds 4 enabled P-States in your system which translates into :
Max freq. = P0
1C = P1
2C = P2
3C = P3

Min freq. being a hard coded value, based on datasheet.

Like the VID, the enablement bit is left unchanged and I don't know yet why the 2C had vanished. If it is a bug or a processor rule.

cyring, 2018-12-03:
As a reminder, the next change is to leave 1C with the P1 value whatever the CPB state is.
In case CPB is enabled I will introduce 2 artificial Boost such as 4C and 5C, respectively :
1C = P1 = 37
4C = P1 + Boost = 37 + 5
5C = P1 + Boost + XFR = 37 + 5 + 2

cyring, 2018-12-03:
Can you  please replace the following function with the code below and try OC and CPB restore.
https://github.com/cyring/CoreFreq/blob/ff031e1f3b9d0fdfd0608747f44f1561b78eb7be/corefreqk.c#L3088
```C
void Compute_AMD_Zen_Boost(void)
{
	PROCESSOR_SPECIFIC *pSpecific = NULL;
	unsigned int COF = 0, pstate, sort[8] = { /* P[0..7]-States */
		BOOST(MAX), BOOST(1C), BOOST(2C), BOOST(3C),
		BOOST(4C) , BOOST(5C), BOOST(6C), BOOST(7C)
	};
	HWCR HwCfgRegister = {.value = 0};
	PSTATEDEF PstateDef = {.value = 0};

	for (pstate = BOOST(MIN); pstate < BOOST(SIZE); pstate++)
		Proc->Boost[pstate] = 0;

	/*Core & L3 frequencies < 400MHz are not supported by the architecture*/
	Proc->Boost[BOOST(MIN)] = 4;
	/* Loop over all frequency ids. */
	for (pstate = 0, pstate <= 7; pstate++) {
		RDMSR(PstateDef, (MSR_AMD_PSTATE_DEF_BASE + pstate));
		/* Handle only valid P-States. */
		if (PstateDef.Family_17h.PstateEn) {
			COF = AMD_Zen_CoreCOF(	PstateDef.Family_17h.CpuFid,
						PstateDef.Family_17h.CpuDfsId);

			Proc->Boost[sort[pstate]] = COF;
		}
	}
	Proc->Features.SpecTurboRatio = pstate;

	/* If CPB is enabled then add Boost + XFR to the P0 ratio. */
	RDMSR(HwCfgRegister, MSR_K7_HWCR);
	if (!HwCfgRegister.Family_17h.CpbDis)
	{
		Proc->Boost[BOOST(8C)] = Proc->Boost[BOOST(MAX)];
		Proc->Boost[BOOST(9C)] = Proc->Boost[BOOST(MAX)];

	    if ((pSpecific = LookupProcessor()) != NULL) {
		/* Save thermal parameters to compute per Core temperature */
		Arch[Proc->ArchID].Specific[0].Param = pSpecific->Param;

		Proc->Boost[BOOST(8C)] += pSpecific->Boost[0];

		Proc->Boost[BOOST(9C)] += pSpecific->Boost[0];
		Proc->Boost[BOOST(9C)] += pSpecific->Boost[1]); /* XFR */

		OverrideCodeNameString(pSpecific);
		OverrideUnlockCapability(pSpecific);
	    } else {
		Arch[Proc->ArchID].Specific[0].Param.Target = 0;
	    }
		Proc->Features.SpecTurboRatio += 2;
	}
}
```

adatum, 2018-12-04:
With the above code, max load frequency is not stuck at 3700 MHz after disabling/re-enabling CPB, but I think it's because there's something strange going on with BCLK again. 

Also, I'm not sure I understand what you are trying to do, but I don't think AMD intends boost and XFR as fixed ratio increases.

CPB off, OC=38, AutoClock default (note 108.89 MHz which doesn't make sense):
[image]

CPB on, no OC (=37), AutoClock default:
[image]

AutoClock off (99.81 MHz):
[image]

cyring, 2018-12-04:
Hello,

> AMD intends boost and XFR as fixed ratio increases

Intel specifies MSR registers to query most Processor ratios [Min, Max, Turbo, Activation, Uncore, ...]; with AMD I'm just getting a few and have to refer to market values.  _Same misery we had with temperature offset_.  
</br>

About BCLK, I could have a TSC delta computing bug here:
https://github.com/cyring/CoreFreq/blob/ff031e1f3b9d0fdfd0608747f44f1561b78eb7be/corefreqk.c#L7016

Could you replace with this code and check BLCK during OC and CPB 

```C
static enum hrtimer_restart Cycle_AMD_Family_17h(struct hrtimer *pTimer)
{
	PSTATESTAT PstateStat;
	PSTATEDEF PstateDef;
	CORE *Core;
	unsigned int pstate;
	unsigned int cpu;

	cpu = smp_processor_id();
	Core = (CORE *) KPublic->Core[cpu];

	Mark_OVH(Core);

	if (BITVAL(KPrivate->Join[cpu]->TSM, MUSTFWD) == 1) {
		hrtimer_forward(pTimer,
				hrtimer_cb_get_time(pTimer),
				RearmTheTimer);

		SMT_Counters_AMD_Family_17h(Core, 1);
		/*ToDo:	Compute Core Performance Boost */

		if (Core->Bind == Proc->Service.Core) {
			PKG_Counters_Generic(Core, 1);

			Core_AMD_Family_17h_Temp(Core);

			PWR_ACCU_AMD_Family_17h(Proc, 1);

			Delta_PTSC(Proc);

			Delta_PWR_ACCU(Proc, PKG);

			Delta_PWR_ACCU(Proc, CORES);

			Save_PWR_ACCU(Proc, PKG);

			Save_PWR_ACCU(Proc, CORES);

			if (AutoClock & 0b10)
				Pkg_OVH(Proc, Core);

			Save_PTSC(Proc);

			Sys_Tick(Proc);
		}
		/* Read the current P-State number. */
		RDMSR(PstateStat, MSR_AMD_PERF_STATUS);
		/* Offset the P-State base register. */
		pstate = MSR_AMD_PSTATE_DEF_BASE + PstateStat.Current;
		/* Read the voltage ID at the offset */
		RDMSR(PstateDef, pstate);
		Core->PowerThermal.VID = PstateDef.Family_17h.CpuVid;

		Delta_INST(Core);

		Delta_C0(Core);

		Delta_TSC(Core);

		Delta_C1(Core);

		Save_INST(Core);

		Save_C0(Core);

		Save_C1(Core);

		if (AutoClock & 0b10) {
			Core_OVH(Core);

			REL_BCLK(Core->Clock,
				Proc->Boost[BOOST(MAX)],
				Core->Delta.TSC,
				Proc->SleepInterval);
		}
		Save_TSC(Core);

		BITSET(LOCKLESS, Core->Sync.V, 63);

		return(HRTIMER_RESTART);
	} else
		return(HRTIMER_NORESTART);
}
```

cyring, 2018-12-04:
AMD Ryzen Master Over-clocking User’s Guide
Pub. # 55931 Rev. 1.0  March 2, 2017:
> The following changes take effect when the values are re-programmed and the processor enters 
Overclocking Mode: All enabled CPU cores operate at the newly user-programmed voltage and P0 frequency 
value. Adjustment of the CPU clock is in 25MHz steps.

cyring, 2018-12-04:
However the Turbo bit is here:
https://github.com/cyring/CoreFreq/blob/ff031e1f3b9d0fdfd0608747f44f1561b78eb7be/amdmsr.h#L108
I just reprogram the 25th bit of the msr register HWCR on all Cores to disable (value 1) or enable (1) the CPB technology

cyring, 2018-12-04:
In the function `Query_AMD_Zen`  
https://github.com/cyring/CoreFreq/blob/ff031e1f3b9d0fdfd0608747f44f1561b78eb7be/corefreqk.c#L3366
Can you replace these instructions:
```C
	case COREFREQ_TOGGLE_OFF:
	case COREFREQ_TOGGLE_ON:
		HwCfgRegister.Family_17h.CpbDis = !TurboBoost_Enable;
		WRMSR(HwCfgRegister, MSR_K7_HWCR)
```
With these ones
```C
	case COREFREQ_TOGGLE_OFF:
	case COREFREQ_TOGGLE_ON:
		HwCfgRegister.Family_17h.CpbDis = !TurboBoost_Enable;
		HwCfgRegister.Family_17h.LockTscToCurrP0 = !HwCfgRegister.Family_17h.CpbDis;
		WRMSR(HwCfgRegister, MSR_K7_HWCR)

```
Then check if the BCLK reading remains stable during OC and CPB

adatum, 2018-12-04:
Which set of changes would you like me to test? I've already done `git checkout HEAD corefreqk.c` to erase previous code changes we tested. 

What do you mean by "stable" for BCLK readings?

cyring, 2018-12-05:
Just keep all code changes and please apply this last change which concerns `LockTscToCurrP0`
Then the purpose of the test is to do OC and CPB restoration. 

What I try to do is to experiment the `LockTscToCurrP0` bit and how it improves the BCLK

cyring, 2018-12-05:
Sorry, apply this instruction:
```C
HwCfgRegister.Family_17h.LockTscToCurrP0 = HwCfgRegister.Family_17h.CpbDis;
```
(Fyi, it removes the not)

adatum, 2018-12-05:
BCLK shows ~99.81 MHz with the 37 ratio, but changing ratio also changes the BCLK reading. Reverting to 37 shows 99.81 MHz again.

OC 39, BCLK 94.69 MHz:
[image]

OC 38, BCLK 97.18 MHz :
Note that the MHz (eg. 3693) shown does not correspond to the ratio (eg. 38) nor to the max frequency under load (eg. 3793).
[image]

Restoring ratio to 37 restores the BCLK to 99.81 MHz:
The ratio and MHz correspond again as well.
Restoring CPB does not restore the boosted max frequencies; it's still capped at ~3700 MHz.
[image]

cyring, 2018-12-05:
Thanks a lot. Probably that TSC-P0 bit has had a impact on OC. However the BCLK estimation makes things difficult to read, especially with the UI.
I suggest to pursue with argument `AutoClock=0`

I have to gather all code changes made so far, and to avoid confusion, I will use the factory base clock (yours: 100MHz) for static informations like the frequencies selector.

adatum, 2018-12-05:
> I suggest to pursue with argument `AutoClock=0`

To be clear, does `0` correspond to ON or OFF?

> to avoid confusion, I will use the factory base clock (yours: 100MHz) for static informations like the frequencies selector.

Which would be fine unless the base clock is changed, which is possible (in BIOS).

cyring, 2018-12-05:
AutoClock | BCLK 
------------------|-----------
 0 | Factory 
 1 | Once 
 2 | Dynamic

cyring, 2018-12-11:
**Make use of P-States arguments to overclock when the driver is starting up** (_draft notes_)

1. Family 0Fh: run the below function only when `PState_FID` or `PState_VID` are valued
https://github.com/cyring/CoreFreq/blob/f66676a8c45221a98088a35a560106714a9ef17c/corefreqk.c#L4098

2. Family 17h:  overclock the max non boost frequency P0 when `PState_FID` is valued
https://github.com/cyring/CoreFreq/blob/f66676a8c45221a98088a35a560106714a9ef17c/corefreqk.c#L4486
```C
if ( (PState_FID >= 0) && (PState_FID <= MAXCLOCK_TO_RATIO(Core->Clock.Hz)) )
{
	CLOCK_ARG clockMod = {
		.NC = CLOCK_MOD_MAX,
		.Offset = PState_FID - Proc->Boost[BOOST(MAX)]
	};
	CLOCK_ZEN_ARG clockZen = {	/* P0:Max non-boosted P-State */
		.pClockMod  = &clockMod,
		.PstateAddr = MSR_AMD_PSTATE_DEF_BASE,
		.BoostIndex = BOOST(MAX)
	};

	TurboBoost_Enable = COREFREQ_TOGGLE_OFF;
	CC6_Enable = COREFREQ_TOGGLE_OFF;
	PC6_Enable = COREFREQ_TOGGLE_OFF;

	Query_AMD_Zen(Core);

	TurboBoost_Enable = -1;
	CC6_Enable = -1;
	PC6_Enable = -1;

	TurboClock_AMD_Zen_PerCore((void *) &clockZen);
}
```

cyring, 2019-04-05:
Closing the issue.

For next versions, I'm thinking about UI cells refreshment, such as the BCLK in Processor window; frequencies scale in OC; IMC estimation; and so on

closed 2019-04-05 by cyring