cyring/CoreFreq #90: Unreasonably high frequencies sometimes reported on one or more of my cores.

tstandley, 2019-01-09:
CoreFreq often momentarily lists cores at unreasonably high frequencies.

Issue seen on a i7 9800x and i9 9900k in Ubuntu 18.04.

In the screenshot below, my thread #2 is listed as having 37.6 ghz for a split second.

[image]

cyring, 2019-01-09:
Hello,

You will have to disable the NMI watchdog during kernel boot to let _CoreFreq_ employ the fixed perf counters. (MSR_CORE_PERF_FIXED_CTR1 and MSR_CORE_PERF_FIXED_CTR2)

-> Just append `nmi_watchdog=0` to the command line.
</br>

If does not get better, rebuild _CoreFreq_ to use other performance registers (APERF/MPERF):
```
make MSR_CORE_PERF_UCC=MSR_IA32_APERF MSR_CORE_PERF_URC=MSR_IA32_MPERF
```

At any time you can get check the building config:
```
make info
```
returns:
```
NMI [0]
HYPERVISOR [1]
MSR_CORE_PERF_UCC [MSR_IA32_APERF]
MSR_CORE_PERF_URC [MSR_IA32_MPERF]
```

If you still encounter the issue, please check the software incompatibility list in the [Wiki](https://github.com/cyring/CoreFreq/wiki/Software-incompatibility).

cyring, 2019-01-13:
Hello,
Can you report if you still encounter the issue with the prerequisites applied ?

tstandley, 2019-01-13:
Hi. I didn't read your message very carefully, and I didn't see the `nmi_watchdog=0 ` solution, so I just recompiled with `MSR_CORE_PERF_UCC=MSR_IA32_APERF MSR_CORE_PERF_URC=MSR_IA32_MPERF`
This solved my problem, but a better solution might be to use `nmi_watchdog=0 `

cyring, 2019-01-13:
Using the fixed perf counters, it may improve the accuracy of the turbo frequencies.

At your will, feel free to close this issue; and open any other ones based on these Processor features.

Regards 
CyrIng

closed 2019-01-13 by tstandley

tstandley, 2019-01-13:
Update: sending nmi_watchdog=0 into the cli also works to fix the problem:
`./corefreq-cli nmi_watchdog=0`

cyring, 2019-01-13:
I'm not sure to understand : "sending nmi_watchdog". 
There is no such option into corefreq-cli

nmi_watchdog is a Kernel option. 

So if you don't reproduce the issue, it is because corefreqk.ko has been built with the APERF/MPERF counters.

But if you rebuild with the default fixed perf counters, you will encounter the issue again. *Except* if you boot the Kernel with nmi_watchdog off