cyring/CoreFreq #91: No temperature report on Opteron Bulldozer

LexJackson, 2019-01-10:
First, this is fantastic software and exactly what I was looking for! Thank you for writing this! 

I read through the thread about no temperature report from Ryzen and suspect I have a similar issue.

With an insmod /var/lib/dkms/corefreqk/1.39/4.20.0-arch1-1-ARCH/x86_64/module/corefreqk.ko Experimental=1

Output of corefreq-cli -s is:

[image]
[image]
[image]

I see no thermal data of course.

[image]

The output of sensors does show the CPU Temps here:

[image]

At your leisure, let me know what I should try. THANK YOU!

LexJackson, 2019-01-10:
Thank you for the rapid reply!  

Here the output requested. 

[image]

Base clock i'm unsure about but the CPU's tend to clock pretty low when idle. Here's an output of cat /proc/cpuinfo | grep MHz.  They get pretty sleepy.

[image]

LexJackson, 2019-01-12:
Also adding, looks like detailed temp per core can be obtained. I'm using Turion Power Control to get this info. (tpc -temp). I believe it's using the cpuid kernel module for that info. 

[image]

cyring, 2019-01-13:
Hello,

For your testings, new code is available to read the processor temperature.

Remark: I presume there is only one sensor per package.

LexJackson, 2019-01-13:
Thanks for the work! For some reason I'm not seeing any temp data now. 

I loaded corefreq with:
insmod /var/lib/dkms/corefreqk/1.39/4.20.0-arch1-1-ARCH/x86_64/module/corefreqk.ko Experimental=1
systemctl start corefreqd

Here's what I'm seeing (currently running an x265 encode btw)

[image]

cyring, 2019-01-13:
Can you try new code ?

You should also read `Bulldozer/Piledriver` as the architecture name.

LexJackson, 2019-01-13:
Sorry, I thought I did, did a git clone on the repository and built it. I may have done something wrong. So sorry.

cyring, 2019-01-13:
In the working directory, just `git pull` to get last source code then `make clean all` to rebuild

LexJackson, 2019-01-13:
Will building from this PKGBUILD not pull the latest code?

https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=corefreq-git

That's how I've been installing it.

cyring, 2019-01-14:
The PKGBUILD may pull the latest and rebuild with`makepkg -sif`; however for our testing session, which implies many small changes, I recommend to clone the source code directly from the github then pull and rebuild whenever I notify a new push.

1. Uninstall, remove package dir
2. Clone from the GitHub and change to the cloned dir
3.  Force rebuild `make clean all`
4. Fully restart _CoreFreq_ , especially its driver corefreqk.ko

cyring, 2019-01-14:
Using the latest code, please post screenshots of temperature b/c I need to know if the SMU queries are working? 
Thks

LexJackson, 2019-01-14:
OK done. Still not seeing temperature data for some reason. Happy to keep testing, please let me know what I can do to help. THANK YOU!

[image]

LexJackson, 2019-01-15:
Here's the confirmation on the Experimental=1 flag and result. 

[image]

[image]

I am compiling the new code now. Thanks!

LexJackson, 2019-01-15:
[dmesgCoreFreqCapture.txt](https://github.com/cyring/CoreFreq/files/2761168/dmesgCoreFreqCapture.txt)

cyring, 2019-01-16:
Can you replace and try with the following functions:

* corefreqk.c
```C
void Core_AMD_Family_15h_Temp(CORE *Core)
{
	TCTL_REGISTER TctlSensor = {0};

	RDPCI(TctlSensor, PCI_CONFIG_ADDRESS(0, 0x18, 0x3, 0xa4));
	Core->PowerThermal.Sensor = TctlSensor.CurTmp;

	if (Proc->Features.AdvPower.EDX.TTP == 1) {
		THERMTRIP_STATUS ThermTrip = {0};

		WRPCI(	SMU_AMD_THM_TRIP_REGISTER_F15H,
			SMU_AMD_INDEX_REGISTER_F15H);
		RDPCI(ThermTrip, SMU_AMD_DATA_REGISTER_F15H);

		Core->PowerThermal.Events = ThermTrip.SensorTrip << 0;
	}
}
```

* coretypes.h
```C
#define COMPUTE_THERMAL_AMD_15h(Temp, Param, Sensor)			\
	(Temp = (Sensor * 5 / 40) - 49)
```

LexJackson, 2019-01-16:
Looks like it's showing something.. Not sure what. ;-)

[image]

dmesg | grep Core 
[    0.969195] ACPI: Core revision 20181003
[    4.976436] systemd[1]: Listening on Process Core Dump Socket.
[ 9920.687007] CoreFreq(31:-1): Processor [ 6F_02] Architecture [Bulldozer/Piledriver] CPU [32/32]

LexJackson, 2019-01-16:
Looks like that worked! What do I need to tweak to get all CPU's reporting? ;-) Thanks for your help!

[image]

cyring, 2019-01-16:
According to the datasheet, it is a sensor per socket.
To my understanding, no temperature per Core. 
But is your proc dual sockets, thus we could read two sensors.

Can you print the topology discovered by _CoreFreq_

`corefreq-cli -m`

LexJackson, 2019-01-16:
I'm only seeing temp data for one of the 32 CPU's in that case shouldn't at least two temps be displayed since there are 2 CPU's?

[image]

cyring, 2019-01-16:
So far only one sensor is queried (on the service thread; which is driven by the CPU wirh an enlightened number in the UI)

The remaining work consists in improving the AMD topology to make the difference between the two sockets: Node ID, Pkg, Module, Core and so on.
Then, set a CPU affinity of the service thread to the each socket. 
Finally collect the PCI sensor from the determined CPU.

LexJackson, 2019-01-16:
Perfect, thank you very much for your help!

cyring, 2019-01-16:
Btw do you know the size of the caches?
The L3 looks wrong. 
(due to a unit change between Zen and previous architectures)

LexJackson, 2019-01-18:
Thanks for working on it!! Looking good so far. 

[image]

I'm able to see temps pretty well with tpc for now. Oh, and I have 4 CPU's now (as of yesterday)

[image]

cyring, 2019-01-18:
Nice setup. 
64 CPU is so far the _CoreFreq_ limit.

What to understand from the Turion Power screenshot : the temperature granularity is at least per Node ?

cyring, 2019-01-18:
Can you show the full topology with the 4 processors

LexJackson, 2019-01-18:
Yes with the Turion screenshot I was simply showing that it's possible to see per node temp. How would you like me to show the topology? Happy to do it.

cyring, 2019-01-18:
> Yes with the Turion screenshot I was simply showing that it's possible to see per node temp. How would you like me to show the topology? Happy to do it.

Just copy/past the output `corefreq-cli -m` with the Markdown code format

LexJackson, 2019-01-18:
[corefreq-topology.log](https://github.com/cyring/CoreFreq/files/2774004/corefreq-topology.log)

cyring, 2019-01-19:
I was expecting to read 16384 KB of L3 cache.
Can you modify the source as bellow at these lines: 
https://github.com/cyring/CoreFreq/blob/1554e89f4b77fe80bd7a50a3df51e97c4208fd2e/corefreqk.c#L1194

```C
	case AMD_Family_15h:
	  if ((Proc->Features.Std.EAX.ExtModel == 0x0)
	   && (Proc->Features.Std.EAX.Model >= 0x0)
	   && (Proc->Features.Std.EAX.Model <= 0xf))
	  {
		PROBE_FILTER_CTRL PF;
		RDPCI(PF, PCI_AMD_PROBE_FILTER_CTRL);
/*	    if (PF.Mode != 0b00) {*/
		/* Add to L3 the Sub Caches in 512 KB unit size.	*/
		Core->T.Cache[3].Size = Core->T.Cache[3].Size
		+ PF.SubCache0En ? (1 << (1 + (PF.SubCacheSize0 & 0b01))) : 0
		+ PF.SubCache1En ? (1 << (1 + (PF.SubCacheSize1 & 0b01))) : 0
		+ PF.SubCache2En ? (1 << (1 + (PF.SubCacheSize2 & 0b01))) : 0
		+ PF.SubCache3En ? (1 << (1 + (PF.SubCacheSize3 & 0b01))) : 0;
	    }
/*	  }*/
```

LexJackson, 2019-01-23:
Thanks for the bump, so sorry, work and life! Looks like I did something wrong, I get build errors on the make clean all. 

Is this pasted in correctly?

`case AMD_Family_10h:
        case AMD_Family_11h:
        case AMD_Family_12h:
        case AMD_Family_14h:
        case AMD_Family_15h:
  if ((Proc->Features.Std.EAX.ExtModel == 0x0)
           && (Proc->Features.Std.EAX.Model >= 0x0)
           && (Proc->Features.Std.EAX.Model <= 0xf))
          {
                PROBE_FILTER_CTRL PF;
                RDPCI(PF, PCI_AMD_PROBE_FILTER_CTRL);
/*          if (PF.Mode != 0b00) {*/
                /* Add to L3 the Sub Caches in 512 KB unit size.        */
                Core->T.Cache[3].Size = Core->T.Cache[3].Size
                + PF.SubCache0En ? (1 << (1 + (PF.SubCacheSize0 & 0b01))) : 0
                + PF.SubCache1En ? (1 << (1 + (PF.SubCacheSize1 & 0b01))) : 0
                + PF.SubCache2En ? (1 << (1 + (PF.SubCacheSize2 & 0b01))) : 0
                + PF.SubCache3En ? (1 << (1 + (PF.SubCacheSize3 & 0b01))) : 0;
/*          }*/
          }
        case AMD_Family_16h:
                Core->T.ApicID    = leaf1_ebx.Init_APIC_ID;
                Core->T.CoreID    = leaf1_ebx.Init_APIC_ID;
                Core->T.PackageID = leaf1_ebx.Init_APIC_ID
                                  >> leaf80000008.ECX.ApicIdCoreIdSize;
                break;
        case AMD_Family_17h:`

cyring, 2019-01-23:
New version pushed.
Please refresh source code then post:
* the topology `corefreq-cli -m` 
* the CPUID dump `corefreq-cli -u`

LexJackson, 2019-01-24:
[CPUID-Dump.txt](https://github.com/cyring/CoreFreq/files/2793777/CPUID-Dump.txt)
[CoreFreq-m.txt](https://github.com/cyring/CoreFreq/files/2793779/CoreFreq-m.txt)

cyring, 2019-01-26:
Hello,
This version 1.39.15 is focusing on the L3 Cache size. 
If your Piledriver processor is detected then any L3 Sub-Cache  has to be summed up to L3

cyring, 2019-01-26:
The size of the L3 Cache in the Topology, please.

LexJackson, 2019-01-29:
Hope this helps! Thanks!

[corefreq-cli-u.txt](https://github.com/cyring/CoreFreq/files/2805888/corefreq-cli-u.txt)
[corefreq-cli-m.txt](https://github.com/cyring/CoreFreq/files/2805889/corefreq-cli-m.txt)

cyring, 2019-01-29:
Thanks for your return.

The Core ID topology looks better, but the L3 cache size remains at 12MB despite the PCI queries of the Sub-Caching data.

The next coding steps will consist in implementing the temperature sensor reading for each cores identified with an ID of # 0 . Thus four sensors (1 per package processor), if available in this architecture ?
```
CPU Pkg  Apic  Core Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID    ID     ID  L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
00: BSP     0     0     -1      64  2        16  4      2048 16     12288 14  
01:   0     1     1     -1      64  2        16  4      2048 16     12288 14  
02:   0     2     2     -1      64  2        16  4      2048 16     12288 14  
...
15:   0    15    15     -1      64  2        16  4      2048 16     12288 14  
16:   1    32     0     -1      64  2        16  4      2048 16     12288 14  
17:   1    33     1     -1      64  2        16  4      2048 16     12288 14  
...
63:   3   111    15     -1      64  2        16  4      2048 16     12288 14  
```

cyring, 2019-01-30:
Hello,
In the last code, you will get the temperature for each Core ID 0 (per Package) and the Voltage ID for any CPU.

I don't find a precise formula how to convert the VID to voltage, in Specs and this [AMD FX tuning guide](https://www.amd.com/Documents/AMD_FX_Performance_Tuning_Guide.pdf)

The same formula of the Zen architecture is copy-past to this function `COMPUTE_VOLTAGE_AMD_15h`
https://github.com/cyring/CoreFreq/blob/24bbcfe1dce4c580d8637b2bb7ef157154164788/coretypes.h#L284
You will get wrong voltage results but the VID should be correct. Feel free to modify the formula.

In the UI, please returns screenshots of the view "Power & Voltage".

The VID is specified to be per P-State, but you will have to experiment and determine if the VID is per Package, Node, or Core ?
For example, stress individually or group of Cores to observe how the VID differs.

cyring, 2019-02-04:
Hello,
Any result from above request ?
Regards
CyrIng

cyring, 2019-02-04:
I have done successful non-regression builds on several Linux (Arch, Ubuntu, Suse, CentOS)
You have to `git clone` from scratch again.

cyring, 2019-02-04:
No code to edit. Everything has been back-ported; in last version 1.39.18, you should read the temperature and voltage ID for each processor.

cyring, 2019-02-05:
Do you still have issues to build the latest version ?

LexJackson, 2019-02-08:
Latest builds good. Thanks, here's the screen shot. Thanks!

[image]

cyring, 2019-02-08:
Per CPU does not work. What is the specs voltage Vcore ? (BIOS)

About temperature, do you read distinct one per package processor ?

cyring, 2019-02-09:
Hey, version 1.40 brings a new feature to stress a chosen CPU. See issue #97 
Based on the topology, you can stress a processor and check how its voltage VID differs from the others. 
Please let me know about the testing results b/c I don't if I'm on the good coding path with Opteron

cyring, 2019-02-13:
Hello,
Just bumping my requests above 
Thank you

cyring, 2019-02-20:
Any results of the voltage ?

closed 2019-03-11 by cyring