cyring/CoreFreq #93: RCU CPU stall warning

cyring, 2019-01-23:
Kernel crashes with a RCU stall warning w/ Xen, less often when booting bare-metal.
```
rcu: INFO: rcu_preempt detected stalls on CPUs/tasks:
rcu:    (detected by 3, t=18002 jiffies, g=206749, q=28439)
rcu: All QSes seen, last rcu_preempt kthread activity 17996 (4295547332-4295529336), jiffies_till_next_fqs=2, root ->qsmask 0x0
swapper/3       R  running task        0     0      1 0x80000000
Call Trace:
 <IRQ>
 sched_show_task.cold.14+0xa6/0xd3
 rcu_check_callbacks.cold.35+0x416/0x43c
 ? tick_sched_handle.isra.6+0x60/0x60
 update_process_times+0x28/0x60
 tick_sched_handle.isra.6+0x1f/0x60
 tick_sched_timer+0x3b/0x80
 __hrtimer_run_queues+0x100/0x2d0
 hrtimer_interrupt+0x10a/0x220
 xen_timer_interrupt+0x29/0x30
 __handle_irq_event_percpu+0x43/0x1c0
 handle_irq_event_percpu+0x30/0x80
 handle_percpu_irq+0x55/0x70
 generic_handle_irq+0x27/0x30
 __evtchn_fifo_handle_events+0x17d/0x190
 __xen_evtchn_do_upcall+0x66/0xd0
 xen_evtchn_do_upcall+0x27/0x40
 xen_do_hypervisor_callback+0x29/0x40
 </IRQ>
RIP: e030:xen_hypercall_sched_op+0xa/0x20
...
```
This could from [a periodic interrupt whose handler takes longer than the time interval between successive pairs of interrupts](https://www.kernel.org/doc/Documentation/RCU/stallwarn.txt).

Until a solution is found (ie. kthread), disable the SysGate when starting the daemon:
```
corefreqd -goff
```

cyring, 2019-01-24:
The Tasks RCU query is moved out of the timer handler function (bound to the Service thread)
The side effects of querying the RCU from an IOCTL are:
* increase SysCalls
* push the requester PID to the top of the list

cyring, 2019-01-25:
Fix stable 24H

cyring, 2019-01-27:
The tasks RCU can be read from Timer when building _CoreFreq_ with `make FEAT_DBG=2`

A few kernel build stress tests has revealed a stability of the fix.

Remark: reading goes through IOCTL with default make options.

cyring, 2019-01-29:
Uptime `12:15:05 up 1 day`: the Timer based implementation is now stable.
Will go back to this solution in next versions.

closed 2019-01-31 by cyring

rogerioahf, 2023-04-29:
didnt got how do you guys solved it... i'm trying to install linux ubuntu budgie 22.04 but it gets stucked in this point, dont know what to do... need a step by step solution