cyring/CoreFreq #96: Maybe Add versions?

pmav99, 2019-02-07:
I think that adopting a versioning scheme (e.g. semver.org ) + adding versions as git tags + adding a `--version` cli flags would be a useful addition

cyring, 2019-02-07:
Lots of good things into this scheme. Indeed I need to improve versionning and packaging.

Adding a `-v` will definitely be part of next release.

`--version` impacts code, and need a refactoring of the arguments parsing.

Btw, do you mind to post some screenshots, I would like to check how your processor is supported by _CoreFreq_

pmav99, 2019-02-07:
I added a wiki page: https://github.com/cyring/CoreFreq/wiki/CPU-support

cyring, 2019-02-07:
Thanks for the Wiki initiative : there are many Processor informations to check for accuracy.

I believe you CPU is ultra low C-State capable: do you see any value against the C10 counter ?

Are the memory timings, geometry and frequency accurately equaled to the Dimm specs (ie BIOS) ?

pmav99, 2019-02-07:
I added some formatting to the wiki. Please review. If you do need feedback, it would not be a bad idea to ask on the arch forum for data. I think it would also be helpful, if you specified which CLI flags  you actually need.

WRT your question, I don't know! I need to check. Since it is a laptop though, chances are that the BIOS will be severely limited.

cyring, 2019-02-07:
The addition in the wiki looks very good and well organized. I will definitely add a reference to the "CPU-support" page in the README.md

With a laptop, you may find details using [memtest86](https://github.com/Distrotech/memtest86) or CPU-Z Windows like software.

cyring, 2019-02-09:
Should it print the version only ?
```
$ corefreq-cli -v
1.39.18
```
That way, the version could be embedded into user script for any purpose.

pmav99, 2019-02-09:
I think it is just fine.

cyring, 2019-02-09:
Version released.

pmav99, 2019-02-09:
Works just fine. Closing the issue

cyring, 2019-02-09:
Hello,

In the README.md, under the question " The CPU freeze or the System crash " is now added a reference to the CPU support Wiki page.

Thanks again for your contribution.

CyrIng

JohnAZoidberg, 2019-06-05:
Please also add a tag to git and maybe a GitHub release.
Also a mention in the README about which version we're at would be nice :)

I'll add the outputs of my system to the wiki as soon as I've packaged it.

cyring, 2019-06-08:
To my taste there are already too much version references in the project: 
* `coretypes.h`: the only one **needed** by the source code
* `dkms.conf` : for a DKMS version consistency
* PKGBUILD in the [`corefreq-git`](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=corefreq-git) ArchLinux AUR package
* The GitHub Wiki, see the [footer](https://github.com/cyring/CoreFreq/wiki/Live-CD).

Although I believe adding the version in the README.md is a must, I prefer to free myself from this maintenance and focus on developments.
I therefore chose the ArchLinux package as the only support; because the deb and rpm are time consuming.

I am open to any suggestions that lead indeed to more clarity in reading the current version.

pmav99, 2019-06-11:
@cyring Not sure if you are interested in adding such dev dependencies, but I have been using [bump2version](https://github.com/c4urself/bump2version) and it works quite nicely

cyring, 2019-08-27:
Closing the issue.

Feel free to come back for any suggestion

closed 2019-08-27 by cyring