cyring/CoreFreq #97: Tools: stress a selected CPU

cyring, 2019-02-09:


cyring, 2019-02-09:
This is the implementation of a feature requested by a user: trigger the Turbo frequency of a chosen CPU.

[image]
[image]

Please comment.

StephanieSunshine, 2019-02-09:
Works great!  Is there a way to toggle it on and off from the command line?

cyring, 2019-02-09:
Unfortunately no in the current code state.
The client passes those commands to the daemon through a ring protocol.

cyring, 2019-02-09:
About the stress time duration: it is not defined in time units but in a number of loops passed as an argument in the function below.
https://github.com/cyring/CoreFreq/blob/fa57a81c160dd121f672618cf1a24146696e1e01/corefreqm.c#L123
However, the loops have been calibrated on my Xeon processor and I'm now in the reflexion of how to precisely convert a user specified duration in relation to the current processor base clock. Second step will consist in offering a list to choose securely among duration values.

cyring, 2019-02-10:
Hello,

Version 1.41 : in the ` Slice counters [T] ` view is added the `Error` column which shows the CPUs calculus errors detected from the `Tools` algorithms.  
[image]
_Remark: the above screenshot presents fake errors for the testing purpose_  

[image]

_Remarks:_
_2. in nominal mode, no value is printed if no error is found_ 
_3. the error checking applies to Atomic operations, CRC result, and Conic calculus_ 

The CPU selector window has also be enhanced to present the topology information:
[image]

If you can not read the PMC0 counter and thus the Instructions elapsed, enter the following command as root
```
echo 2 > /sys/devices/cpu/rdpmc
```

Feel free to comment.
CyrIng

cyring, 2019-02-11:
Display issue may happen in the below function when the ID equals -1
https://github.com/cyring/CoreFreq/blob/5dbc44a087f0a57511425014c46c6b12f8fb844d/corefreq-cli.c#L3478

Such as the Topology() function, fix with this format:
```C
		sprintf((char*) item, 	"   %02u  %4d%6d%6d   ",
					cpu,
					Shm->Cpu[cpu].Topology.PackageID,
					Shm->Cpu[cpu].Topology.CoreID,
					Shm->Cpu[cpu].Topology.ThreadID);
```

cyring, 2019-03-11:
Closing the issue. 
In the future, I may add other cli commands.

cyring, 2019-07-30:
Stress single Core with Ryzen 3700X does not reach Turbo.
Perhaps the same with Zen first generation and Zen+

Reopening this issue to collect users tests.

adatum, 2019-07-30:
Turbo Select CPU for single cores is working fine for my Ryzen 2700X, reaching ~4.3-4.35 GHz. Tested on several cores.

cyring, 2019-07-30:
> Turbo Select CPU for single cores is working fine for my Ryzen 2700X, reaching ~4.3-4.35 GHz. Tested on several cores.

Thank you.
I have remotely stressed a 3700X but could not trigger Turbo. Despite [here](https://github.com/cyring/CoreFreq/issues/133#issuecomment-511626249), it shows I can. Pretty confusing.

cyring, 2019-08-15:
Closing this issue

closed 2019-08-15 by cyring

olejon, 2019-08-19:
> Stress single Core with Ryzen 3700X does not reach Turbo

Works fine on X570 + 3600X. Goes to almost 4.4 GHz as expected, also using Round Robin. Assume same results with 3rd Gen on X570. We'll see when I get my 3950X though.