cyring/CoreFreq #99: Ultra low C-States

cyring, 2019-02-18:
* PC10
[image]
[image]
[image]

cyring, 2019-02-18:
* PC09
[image]

cyring, 2019-02-18:
* PC08
[image]

cyring, 2019-02-23:
Nothing shows up in the above screenshots.

Either the i7-6700 doesn't have C8, C9 and C10 Package C-States, or I need to set some additional registers up to measure these msr:

Value | MSR
--------|------------
0x00000632 | PC10
0x00000631 | PC09
0x00000630 | PC08

Q: If you have a processor of the Skylake, Kaby Lake families, capable with ultra low States (those generally found in Laptops) please copy-past the code below and let me know about results ?

Starts with PC10, next substitutes with PC09, then PC08 , in the following functions:

PKG_Counters_Skylake() | Cycle_Skylake()
-----------------------------------|---------------------
`MSR_PKG_C10_RESIDENCY,Proc->Counter[T].PC10` | `Delta_PC10(Proc);` `Save_PC10(Proc);`
`MSR_PKG_C9_RESIDENCY,Proc->Counter[T].PC09` | `Delta_PC09(Proc);` `Save_PC09(Proc);`
`MSR_PKG_C8_RESIDENCY,Proc->Counter[T].PC08` | `Delta_PC08(Proc);` `Save_PC08(Proc);`

https://github.com/cyring/CoreFreq/blob/d6f9be9a5efe9be3e44c0a7ac6afb029f6b12cea/corefreqk.c#L5165

```C
#define PKG_Counters_Skylake(Core, T)					\
({									\
	RDTSCP_COUNTERx6(Proc->Counter[T].PTSC,				\
			MSR_PKG_C2_RESIDENCY, Proc->Counter[T].PC02,	\
			MSR_PKG_C3_RESIDENCY, Proc->Counter[T].PC03,	\
			MSR_PKG_C6_RESIDENCY, Proc->Counter[T].PC06,	\
			MSR_PKG_C7_RESIDENCY, Proc->Counter[T].PC07,	\
			MSR_PKG_C10_RESIDENCY,Proc->Counter[T].PC10,	\
	MSR_SKL_UNCORE_PERF_FIXED_CTR0, Proc->Counter[T].Uncore.FC0);	\
})
```

https://github.com/cyring/CoreFreq/blob/d6f9be9a5efe9be3e44c0a7ac6afb029f6b12cea/corefreqk.c#L6629

```C
static enum hrtimer_restart Cycle_Skylake(struct hrtimer *pTimer)
{
	PERF_STATUS PerfStatus = {.value = 0};
	CORE *Core;
	unsigned int cpu;

	cpu = smp_processor_id();
	Core = (CORE *) KPublic->Core[cpu];

	Mark_OVH(Core);

	if (BITVAL(KPrivate->Join[cpu]->TSM, MUSTFWD) == 1) {
		hrtimer_forward(pTimer,
				hrtimer_cb_get_time(pTimer),
				RearmTheTimer);

		SMT_Counters_SandyBridge(Core, 1);

		if (Core->Bind == Proc->Service.Core) {
			PKG_Counters_Skylake(Core, 1);

			RDMSR(PerfStatus, MSR_IA32_PERF_STATUS);
			Core->PowerThermal.VID = PerfStatus.SNB.CurrVID;

			PWR_ACCU_Skylake(Proc, 1);

			Delta_PC02(Proc);

			Delta_PC03(Proc);

			Delta_PC06(Proc);

			Delta_PC07(Proc);

			Delta_PC10(Proc);

			Delta_PTSC_OVH(Proc, Core);

			Delta_UNCORE_FC0(Proc);

			Delta_PWR_ACCU(Proc, PKG);

			Delta_PWR_ACCU(Proc, CORES);

			Delta_PWR_ACCU(Proc, UNCORE);

			Delta_PWR_ACCU(Proc, RAM);

			Save_PC02(Proc);

			Save_PC03(Proc);

			Save_PC06(Proc);

			Save_PC07(Proc);

			Save_PC10(Proc);

			Save_PTSC(Proc);

			Save_UNCORE_FC0(Proc);

			Save_PWR_ACCU(Proc, PKG);

			Save_PWR_ACCU(Proc, CORES);

			Save_PWR_ACCU(Proc, UNCORE);

			Save_PWR_ACCU(Proc, RAM);

			Sys_Tick(Proc);
		}

		Core_Intel_Temp(Core);

		RDCOUNTER(Core->Interrupt.SMI, MSR_SMI_COUNT);

		Delta_INST(Core);

		Delta_C0(Core);

		Delta_C3(Core);

		Delta_C6(Core);

		Delta_C7(Core);

		Delta_TSC_OVH(Core);

		Delta_C1(Core);

		Save_INST(Core);

		Save_TSC(Core);

		Save_C0(Core);

		Save_C3(Core);

		Save_C6(Core);

		Save_C7(Core);

		Save_C1(Core);

		BITSET(LOCKLESS, Core->Sync.V, 63);

		return(HRTIMER_RESTART);
	} else
		return(HRTIMER_NORESTART);
}
```

cyring, 2019-03-08:
To anyone reading this, if your processor is ultra low C-states capable, please post some results : screenshots and/or cli outputs
Let me know if CoreFreq matches the BIOS settings (if available) 

Thanks for your help 😣

metebalci, 2019-03-14:
I have this, will it help ? I am not sure how many C/PC states it has.

```
Processor                             [Intel(R) Core(TM) i7-7567U CPU @ 3.50GHz]
|- Architecture                                                   [Kaby Lake/UY]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [       154]
|- Signature                                                            [ 06_8E]
|- Stepping                                                             [     9]
|- Online CPU                                                           [  4/4 ]

|- Package C-State
   |- Configuration Control                                   CONFIG   [   LOCK]
   |- Lowest C-State                                           LIMIT   [      0]
   |- I/O MWAIT Redirection                                  IOMWAIT   [Disable]
   |- Max C-State Inclusion                                    RANGE   [      0]
|- MWAIT States:    C0    C1    C2    C3    C4    C5    C6    C7
|                    0     2     1     2     4     1     1     1
```

cyring, 2019-03-15:
MWAIT States : the c-states are counted from the cpuid instruction. Check the idle view for values coming from the free running counter msr registers

Pkg C-states are enumerated from msr decoding in function `Intel_CStatesConfiguration`: can you print the hexa value of `MSR_PKG_CST_CONFIG_CONTROL` (using `rdmsr` command)

cyring, 2019-04-30:
Using a SandyBridge, _CoreFreq_ can read the Package C-States down to PC07
[image]

Unfortunately I don't have other hardware to check PC-States down to PC10  
Topic closed.

closed 2019-04-30 by cyring