cyring/CoreFreq #100: Intel Xeon E3-1200 v5 series processor

metebalci, 2019-02-27:
PCs are reported as 0.

```
$ ./corefreq-cli -s
Processor                            [Intel(R) Xeon(R) CPU E3-1245 v5 @ 3.50GHz]
|- Architecture                                                      [Skylake/S]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [       198]
|- Signature                                                            [ 06_5E]
|- Stepping                                                             [     3]
```

```
$ ./corefreq-cli -k
Linux:
|- Release                                                   [4.15.0-45-generic]
|- Version                         [#48-Ubuntu SMP Tue Jan 29 16:28:13 UTC 2019]
|- Machine                                                              [x86_64]
```

```
$ ./corefreq-cli -g
		Cycles		State(%)
PC02	                 0	   0.00
PC03	                 0	   0.00
PC06	                 0	   0.00
PC07	                 0	   0.00
PC08	                 0	   0.00
PC09	                 0	   0.00
PC10	                 0	   0.00
PTSC	        3504154098
UNCORE	          25160034
```

cyring, 2019-02-27:
I have an idea where it comes from:
Both, an `i7-6700HQ` and a `Xeon E3-1200 v5` appear to share the same cpuid signature `06_5E` revision `3`
Thus they are both classified in the micro-architecture Skylake/S
However both don't share all of the same MSR registers, probably Package C-state counters. I have to digg the Skylake Xeon datasheet to find this... 
Fortunately I had implemented into CoreFreq a Processor specific identification algorithm to sub-class architectures
Next, I may to implement a specific timer loop to query your Xeon registers. 

Fyi, below the i7-6700HQ:
```
Processor                            [Intel(R) Core(TM) i7-6700HQ CPU @ 2.60GHz]
|- Architecture                                                      [Skylake/S]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [       198]
|- Signature                                                            [ 06_5E]
|- Stepping                                                             [     3]
|- Online CPU                                                           [  8/8 ]
|- Base Clock                                                           [ 99.69]
|- Frequency            (MHz)                      Ratio                        
                 Min    797.53                    [   8 ]                       
                 Max   2591.97                    [  26 ]                       
|- Factory                                                              [100.00]
                       2600                       [  26 ]                       
|- Turbo Boost                                                          [UNLOCK]
                  1C   3489.20                    <  35 >                       
                  2C   3289.81                    <  33 >                       
                  3C   3190.12                    <  32 >                       
                  4C   3090.43                    <  31 >                       
|- Uncore                                                               [UNLOCK]
                 Min    797.53                    <   8 >                       
                 Max   3489.20                    <  35 >                       
|- TDP                                                            Level [  0:3 ]
   |- Programmable                                                      [UNLOCK]
   |- Configuration                                                     [UNLOCK]
   |- Turbo Activation                                                  [UNLOCK]
             Nominal   2591.97                    [  26 ]                       
              Level1   2093.52                    [  21 ]                       
               Turbo   2492.28                    [  25 ]                       
```

cyring, 2019-02-27:
Remark: in issue #99 the i7-6700, a desktop class processor, also has the same cpuid !
PC02 is however showing up.

metebalci, 2019-02-27:
I think I also found sth. From corefreq -s output:

```
Package C-State
   |- Configuration Control                                   CONFIG   [   LOCK]
   |- Lowest C-State                                           LIMIT   [      0]
```

I guess this is the reason, since the lowest reported state is 0, there is nothing above C0. I do not know why it is like this, I may have changed something in the BIOS or in the kernel that also modifies it. I will check.

For CoreFreq, maybe it can be useful to somehow relate this to -g output, as a warning or information why the values could be zero.

metebalci, 2019-02-27:
More info:
  
I played with two BIOS settings:

"Runtime power management" controls SpeedStep, Dynamic Acceleration and Turbo Boost.

"Enhanced idle power states" control C1A, C3A, C1U and C3U. Also, when enabled, it changes the Package C-State's Max C-State Inclusion to 0 from 3.

I had both of them disabled before. Now, with both settings enabled (which is also the default), if I execute corefreq-cli -g, I see these:

```
    Cycles    State(%)
PC02          1155128590    27.75
PC03          1428193620    35.60
PC06                   0     0.00
PC07                   0     0.00
PC08                   0     0.00
PC09                   0     0.00
PC10                   0     0.00
PTSC          3504012216
UNCORE              297874
```

and corefreq-cli -s reports:

```
|- Package C-State
   |- Configuration Control                                   CONFIG   [   LOCK]
   |- Lowest C-State                                           LIMIT   [      0]
   |- I/O MWAIT Redirection                                  IOMWAIT   [Disable]
   |- Max C-State Inclusion                                    RANGE   [      0]
```

I did not look into the code in detail but I am not sure why it reports these as such because:

```
$ sudo rdmsr 0xe2
7e008006
```

I think Package C-State Limit is 6=0110, so C8, but above it reports 0.

```
$ sudo rdmsr 0xe4
31814
```

I think C-State Range is 3=011, but above it reports 0.

I checked the PC states with powertop utility also, and it reports same right now. So I guess there is no problem with PCxy output.

cyring, 2019-02-27:
Great news, the Package states are showing up. 
* At least we are getting PC02 which seems common among the SKL architectures.
* For the ultra low states, the issue #99 supplies the source code to modify the driver loop and test PC08, PC09, PC10

To explain the source call flow:
The C-states range is decoded in the below function:
https://github.com/cyring/CoreFreq/blob/b2b11c4a1e8cabaca69456e1f320bdf5a243c8d7/corefreqk.c#L3909
I have struggled with the bit encoding of those msr which differs among 3 uarch (according to the SDM specs). 
I have then classified those " i7 generations " in 3 classes: 
0x061A , 0x062A , and 0x0645

Here, Skylake is queried like SandyBridge: 
https://github.com/cyring/CoreFreq/blob/b2b11c4a1e8cabaca69456e1f320bdf5a243c8d7/corefreqk.h#L768
which calls the decoding here:
https://github.com/cyring/CoreFreq/blob/b2b11c4a1e8cabaca69456e1f320bdf5a243c8d7/corefreqk.c#L4479

Perhaps the SKL claims a new msr encoding

cyring, 2019-02-27:
Indeed, 011 value read from MSR_PMG_IO_CAPTURE_BASE (0x000000e4) has to be added to the switch cases:
https://github.com/cyring/CoreFreq/blob/b2b11c4a1e8cabaca69456e1f320bdf5a243c8d7/corefreqk.c#L4179
And also when set by user:
https://github.com/cyring/CoreFreq/blob/b2b11c4a1e8cabaca69456e1f320bdf5a243c8d7/corefreqk.c#L4160

Don't you think 011 is range C8 ?

metebalci, 2019-02-27:
> Don't you think 011 is range C8 ?

I actually tried to find it in Intel docs, but I couldnt. Probably C8.

cyring, 2019-02-28:
Apparently an undocumented value. I also don't find much more.

Back from my travel, I will refactor the function `Intel_CStatesConfiguration` to handle your `limit` and `range` values

Btw, do you get any value from in the "Memory Controller" view ?

metebalci, 2019-02-28:
Only this:

```
$ ./corefreq-cli -M
                          GenuineIntel   
```

and same output with Experimental=1 also.

metebalci, 2019-02-28:
I have a question about the voltage view. You said in the comment in my post that it is a package scoped value so that is why it shows only on one core. I have read in Intel docs that the Freq can be adjusted per package or per core (I do not know if this is software configurable), and also I was thinking P-states are per core. So if Freq is adjusted per core, I was assuming the voltage is adjusted as well. If it is not the case, that sounds like a limitation on P-state selection. What am I missing ?

metebalci, 2019-02-28:
> Back from my travel, I will refactor the function `Intel_CStatesConfiguration` to handle your `limit` and `range` values.

Looking a bit more to the docs, this is what I understand for MSR_PKG_CST_CONFIG_CONTROL:

- My processor (06_5E Skylake) and other Skylakes (06_4EH and 06_5EH, -see below for 06_55-) and also Kaby Lake and Coffee Lake (06_8EH and 06_9EH) and Cannon Lake (06_66H) all uses the Table 2-35.
- Broadwell (06_3DH and 06_47H) uses Table 2-35 (4-bits C-limit).
- Broadwell (06_56H, 06_4FH) uses Table 2-36 (3-bits C-limit).
- Skylake (06_55H) is confusing, first it says it uses Table 2-35 like other Skylakes, but then it says it uses Table 2-45 (3-bits C-limit).
- Xeon Phi (06_57H and 06_85H) uses Table 2-46 (3-bits C-limit).

I think you need to implement 2 different PerCore_Broadwell_Query rather than using the SandyBridge code and use one of these (implementing Table 2-35) for Skylake and future generations other than Skylake 06_55 which requires a different implementation, and also another implementation for Xeon Phi. I did not check the values in the tables, maybe some contains the same info.

cyring, 2019-02-28:
The Vcore is read from MSR_IA32_PERF_STATUS
Looking into the SDM, the scope differs between uarch (starting from the old Core Solo, Duo) : thus testing and classification is one way to do. 
https://github.com/cyring/CoreFreq/blob/b2b11c4a1e8cabaca69456e1f320bdf5a243c8d7/corefreqk.c#L6650

Strangely I have found P-States msr easier to understand in AMD specs where FID and VID are encoded in same register (one per state level)
With Intel, we have ratios here, turbo, there, VID in other msr (sometimes through pci or **NDA**); any architecture test helps me to understand the specifics (and the undocumented)

I'm not sure to clearly answer to your question but any R&D contribution is welcomed.

metebalci, 2019-02-28:
I will investigate it.

cyring, 2019-03-05:
Just test code on an i7-6700, and the `110` binary value seems to be the C8 level.

[image]

https://github.com/cyring/CoreFreq/blob/b2b11c4a1e8cabaca69456e1f320bdf5a243c8d7/corefreqk.c#L4093
```
		switch (CStateConfig.Pkg_CStateLimit & 0x7) {
		case 0b110:
			Core->Query.CStateLimit = 8;
			break;
		case 0b101:
		case 0b100:
			Core->Query.CStateLimit = 7;
			break;
		case 0b011:
			Core->Query.CStateLimit = 6;
			break;
		case 0b010:
			Core->Query.CStateLimit = 3;
			break;
		case 0b001:
			Core->Query.CStateLimit = 2;
			break;
		case 0b000:
		default:
			Core->Query.CStateLimit = 0;
			break;
		}
```

https://github.com/cyring/CoreFreq/blob/b2b11c4a1e8cabaca69456e1f320bdf5a243c8d7/corefreqk.c#L4179
```
	switch (CState_IO_MWAIT.CStateRange) {
	case 0b011:
		Core->Query.CStateInclude = 8;
		break;
	case 0b010:
		Core->Query.CStateInclude = 7;
		break;
	case 0b001:
		Core->Query.CStateInclude = 6;
		break;
	case 0b000:
		Core->Query.CStateInclude = 3;
		break;
	default:
		Core->Query.CStateInclude = 0;
		break;
	}
```

cyring, 2019-03-06:
Fixed in vesion 1.41.4  
Please try to change with various Package C-States limits
[image]

metebalci, 2019-03-13:
I have this now, so I think it is fixed.

```
|- Package C-State                                                              
   |- Configuration Control                                   CONFIG   [   LOCK]
   |- Lowest C-State                                           LIMIT   [      8]
   |- I/O MWAIT Redirection                                  IOMWAIT   [Disable]
   |- Max C-State Inclusion                                    RANGE   [      8]
```

cyring, 2019-03-13:
Great news. Thank you.

About the IMC, I'm reading the 2 datasheet volumes of the Xeon Scalable but there is less PCI registers information to decode the Dimm geometry, channels data, etc.

What is strange is that your Xeon has the same CPUID signature than the i7-6700 I have implemented the memory controller with.

Probably yours works the same and it is just a matter of PCI device IDs : can you list them plz ?

Regards

metebalci, 2019-03-14:
Do you mean the lspci output related to CPU and chipset ?

cyring, 2019-03-14:
Yes `lspci -nn` will be just fine.

metebalci, 2019-03-18:
Here it is:

```
00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Host Bridge/DRAM Registers [8086:1918] (rev 07)
00:01.0 PCI bridge [0604]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) [8086:1901] (rev 07)
00:14.0 USB controller [0c03]: Intel Corporation 100 Series/C230 Series Chipset Family USB 3.0 xHCI Controller [8086:a12f] (rev 31)
00:14.2 Signal processing controller [1180]: Intel Corporation 100 Series/C230 Series Chipset Family Thermal Subsystem [8086:a131] (rev 31)
00:16.0 Communication controller [0780]: Intel Corporation 100 Series/C230 Series Chipset Family MEI Controller #1 [8086:a13a] (rev 31)
00:16.3 Serial controller [0700]: Intel Corporation 100 Series/C230 Series Chipset Family KT Redirection [8086:a13d] (rev 31)
00:17.0 RAID bus controller [0104]: Intel Corporation SATA Controller [RAID mode] [8086:2822] (rev 31)
00:1b.0 PCI bridge [0604]: Intel Corporation 100 Series/C230 Series Chipset Family PCI Express Root Port #17 [8086:a167] (rev f1)
00:1f.0 ISA bridge [0601]: Intel Corporation C236 Chipset LPC/eSPI Controller [8086:a149] (rev 31)
00:1f.2 Memory controller [0580]: Intel Corporation 100 Series/C230 Series Chipset Family Power Management Controller [8086:a121] (rev 31)
00:1f.3 Audio device [0403]: Intel Corporation 100 Series/C230 Series Chipset Family HD Audio Controller [8086:a170] (rev 31)
00:1f.4 SMBus [0c05]: Intel Corporation 100 Series/C230 Series Chipset Family SMBus [8086:a123] (rev 31)
00:1f.6 Ethernet controller [0200]: Intel Corporation Ethernet Connection (2) I219-LM [8086:15b7] (rev 31)
01:00.0 VGA compatible controller [0300]: NVIDIA Corporation GM107GLM [Quadro M620 Mobile] [10de:13b4] (rev a2)
01:00.1 Audio device [0403]: NVIDIA Corporation Device [10de:0fbc] (rev a1)
02:00.0 Non-Volatile memory controller [0108]: Samsung Electronics Co Ltd NVMe SSD Controller SM951/PM951 [144d:a802] (rev 01)
```

cyring, 2019-03-18:
Thanks,
`[8086:1918]` is the DID not implemented yet:
https://github.com/cyring/CoreFreq/blob/941b5ea90ca92fc365e17b324b783f29e8d90ff2/coretypes.h#L1204
Now looking for its datasheet ...

cyring, 2019-03-24:
Hello,
IMC decoding implemented for Skylake/DT.
It may crash, **save, sync** your files before start
If it goes OK, please post the UI "Memory Controller" view or using option `-M`
Regards
CyrIng

metebalci, 2019-03-28:
It looks ok I think. I experienced no crash.

```
                           Sunrise Point                              
Controller #0                                           Dual Channel  
 Bus Rate  8000 MT/s      Bus Speed 8009 MT/s     DRAM Speed 2133 MHz 
                                                                      
 Cha    CL  RCD   RP  RAS  RRD  RFC   WR RTPr WTPr  FAW  B2B  CWL Rate
  #0    15   15   15   35    0  374    0    8   34    0    0   13   2N
  #1    15   15   15   35    0  374    0    8   34    0    0   13   2N
      ddWR drWR srWR ddRW drRW srRW ddRR drRR srRR ddWW drWW srWW  ECC
  #0     0    0    0    0    0    0    0    0    0    0    0    0   3 
  #1     0    0    0    0    0    0    0    0    0    0    0    0   3 
                                                                      
 DIMM Geometry for channel #0                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0    16    2     65536      1024          16384               
 DIMM Geometry for channel #1                                         
      Slot Bank Rank     Rows   Columns    Memory Size (MB)           
       #0    16    2     65536      1024          16384               
```

cyring, 2019-03-28:
Thanks a lot for your return.

cyring, 2019-04-05:
Package C-states and `8086:1918` are implemented.
Let's close the issue. 
Regards,
CyrIng

closed 2019-04-05 by cyring