cyring/CoreFreq #104: Problem TURBO BOOST. TURBO BOOST is on but processor not support this

vincenzogianfelice, 2019-04-12:
Hi. when I type "./corefreq-cli -s" I see that the turbo boost feature is active ... but the processor doesn't support turbo boost. I have an intel core i3-6100 @ 3.70GHZ.
Output commans "./corefreq-cli -s | grep -i turbo" is:

`|- Turbo Boost                                                          [UNLOCK]
   |- Turbo Activation                                                  [UNLOCK]
|- Turbo Boost                                                 TURBO       < ON>
`

cyring, 2019-04-13:
Hi. Thanks for your report.
Indeed, I have also noticed that some architecture reports the wrong Turbo capability.
For example, my Xeon W3690 has a locked turbo but in fact it is the opposite.
My workaround will be to override the Turbo capability of your  `Core i3-6100` based on its CPU name.

1. can you print its full brand name, such as:
```
Intel(R) Xeon(R) CPU W3690 @ 3.47GHz
```
2. can you stress only one Core to show the frequency limit.
* press `[F3]` next choose `Turbo Select CPU...` then select CPU # 0
[image]

cyring, 2019-04-14:
This fix will query the TDP registers when Skylake is detected.
Remark: this a common code with Broadwell

1. Using the last version of _CoreFreq_, in the `corefreqk.c` file, replace the function `Query_Broadwell` with the code bellow:
https://github.com/cyring/CoreFreq/blob/68b559a28a4a1d4f9001afb768ac31bdd22bbcff/corefreqk.c#L2945
```C
void Query_Broadwell(void)
{
	Proc->Features.Uncore_Unlock = 1;
	Nehalem_Platform_Info();
	HyperThreading_Technology();
	Haswell_Uncore_Ratio(NULL);
	Intel_Turbo_TDP_Config();
	SandyBridge_PowerInterface();
}
```

2. Then rebuild `make clean all` and (re)start _CoreFreq_

3. Please post back the full output of `corefreq-cli -s`

CyrIng

cyring, 2019-04-14:
My notes (from the Intel Software Developer’s Manual)
```
6th generation: Skylake microarchitecture
CPUID signatures: 06_4EH [Skylake_UY], 06_5EH [Skylake_S], and 06_55H [Skylake_X].

7th generation: Kaby Lake microarchitecture and 8th generation Coffee Lake microarchitecture
CPUID signatures: 06_8EH [Kabylake_UY], [Whiskey Lake] and 06_9EH [Kabylake].

Future Intel® Core™ processors: Cannon Lake microarchitecture
CPUID signature: 06_66H [Cannon Lake].

These processors support the MSR interfaces listed in :
Table 2-20
Table 2-21
Table 2-25
Table 2-29 : MSR_TURBO_ACTIVATION_RATIO, MSR_CONFIG_TDP_CONTROL, MSR_CONFIG_TDP_LEVEL2, MSR_CONFIG_TDP_LEVEL1, MSR_CONFIG_TDP_NOMINAL
Table 2-35
Table 2-39
Table 2-40
```

cyring, 2019-04-14:
Fix released.
Can you please test and post back the full output of `corefreq-cli`

vincenzogianfelice, 2019-04-14:
I updated...But core_freq still reports that the turbo boost is there:

`|- Turbo Boost                                                          [UNLOCK]
   |- Turbo Activation                                                  [UNLOCK]
|- Turbo Boost                                                 TURBO       < ON>`

[image]

cyring, 2019-04-14:
Can you post the output of

1. `corefreq-cli -s`

2. screenshot of a single stressed CPU. For this, press F3, then Turbo and select CPU number 0.
Screenshots when the frequency is at its most value.

cyring, 2019-04-14:
Fix released.
In the UI `IDA,TURBO` have to be OFF
Same in the `Technologies` window

vincenzogianfelice, 2019-04-15:
Thank you. Now everything works correctly. The data is displayed correctly. For completeness, I post my output:

`|- Turbo Boost                                                          [UNLOCK]
   |- Turbo Activation                                                  [UNLOCK]
|- Turbo Boost                                                 TURBO       <OFF>`

Here is also the screenshot:

[image]

...even if I didn't understand what "UNLOCK" means. What does it mean?

cyring, 2019-04-15:
Great news that it works.  

`UNLOCK` is a bit status that the associated registers can be altered.
With your processor, `Turbo Activation` should have no effect.  

Verifying will be easier for me if you can post the full output of `corefreq-cli -s`

vincenzogianfelice, 2019-04-15:
Sure, sorry:

`Processor                              [Intel(R) Core(TM) i3-6100 CPU @ 3.70GHz]
|- Architecture                                                      [Skylake/S]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [       198]
|- Signature                                                            [ 06_5E]
|- Stepping                                                             [     3]
|- Online CPU                                                           [  4/4 ]
|- Base Clock                                                           [ 99.90]
|- Frequency            (MHz)                      Ratio                        
                 Min    799.17                    [   8 ]                       
                 Max   3696.18                    [  37 ]                       
|- Factory                                                              [100.00]
                       3700                       [  37 ]                       
|- Turbo Boost                                                          [UNLOCK]
                  1C   3696.18                    <  37 >                       
                  2C   3696.18                    <  37 >                       
                  3C   3696.18                    <  37 >                       
                  4C   3696.18                    <  37 >                       
|- Uncore                                                               [UNLOCK]
                 Min    799.17                    <   8 >                       
                 Max   3696.18                    <  37 >                       
|- TDP                                                            Level [  0:3 ]
   |- Programmable                                                      [UNLOCK]
   |- Configuration                                                     [  LOCK]
   |- Turbo Activation                                                  [UNLOCK]
             Nominal   3696.18                    [  37 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N,N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [N]  BMI1/BMI2 [Y/Y]        CLFSH [Y]        CMOV [Y] 
|- CMPXCH8      [Y]     CMPXCH16 [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/N]     MONITOR [Y] 
|- MOVBE        [Y]          MPX [Y]     PCLMULDQ [Y]      POPCNT [Y] 
|- RDRAND       [Y]       RDSEED [Y]       RDTSCP [Y]         SEP [Y] 
|- SGX          [Y]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/N]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features                                                                        
|- 1 GB Pages Support                                      1GB-PAGES   [Present]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Present]
|- CPL Qualified Debug Store                                  DS-CPL   [Present]
|- 64-Bit Debug Store                                         DTES64   [Present]
|- Fast-String Operation                                Fast-Strings   [Present]
|- Fused Multiply Add                                       FMA|FMA4   [Present]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
|- Pending Break Enable                                          PBE   [Present]
|- Process Context Identifiers                                  PCID   [Present]
|- Perfmon and Debug Capability                                 PDCM   [Present]
|- Page Global Enable                                            PGE   [Present]
|- Page Size Extension                                           PSE   [Present]
|- 36-bit Page Size Extension                                  PSE36   [Present]
|- Processor Serial Number                                       PSN   [Missing]
|- Restricted Transactional Memory                               RTM   [Missing]
|- Safer Mode Extensions                                         SMX   [Missing]
[... 39 lines cut ...]
[... cut ...]
|- Package Thermal Management                                    PTM   [Present]
|- Thermal Monitor 1                                         TM1|TTP   [ Enable]
|- Thermal Monitor 2                                         TM2|HTC   [Present]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000061035]
   |- Window                                            second   [  0.000976562]`

cyring, 2019-04-15:
Thanks for this output. 

Surprisingly, the Turbo Activation Nominal ratio of [  37 ] is provided by the register altb we know this processor does have any Turbo feature.

cyring, 2019-04-15:
Last commit, you will get a fix for the Uncore lock state.
This should look this way (_depending on the current BCLK_):
```
|- Uncore                                                               [  LOCK]
                 Min    800.00                    [   8 ]                       
                 Max   3700.00                    [  37 ]
```

cyring, 2019-04-17:
The remaining work is to determine which processors have a Uncore coefficient unlocked or not.

vincenzogianfelice, 2019-04-17:
Good job.
Now I'd like to ask you a question. But what changes, as regards temperature control, between this tool and lm-sensors? I read around that lm-sensors can be dangerous ... so how does corefreq see temperatures? Is the operation like that of the lm-sensor tool? And how reliable is this measurement?

Thank you.

cyring, 2019-04-17:
Thanks.
I **never** have an issue with lm-sensors. 
_CoreFreq_ is also using low level accesses to read the Processor digital sensors registers.
lm-sensors is however able to give much more information, especially chipsets voltage, fanspeed, etc. 
This of-course implies that the user has a good knowledge of her/his hardware.

_CoreFreq_ try to be as accurate as lm-sensors.

The true difficulties come from the lack of specifications supplied by Manufacturers to the open source projects.

cyring, 2019-04-19:
I have rollback the Uncore Limit change which will now be applied only for IDA capable processors; starting from the Haswell architectures.

Please let me know if you still get the same result ?

vincenzogianfelice, 2019-04-19:
Yes.
I saved the output of 'corefreq-cli -s' and saw the differences. Apparently nothing has changed. Also place a screenshots
[image]

cyring, 2019-04-19:
Great, Thank you.
I believe we can close this topic.

closed 2019-04-19 by vincenzogianfelice