cyring/CoreFreq #105: CALL TRACE Error kill corefreqd in background

vincenzogianfelice, 2019-04-13:
Hi. Day, i have see new problem...kill daemon corefreqd.
This is problem:

`CoreFreq: Unload
[ 3174.822898] BUG: unable to handle kernel paging request at ffffffffc0850388
[ 3174.822915] IP: [<ffffffff96e07f4f>] filp_close+0x1f/0x70
[ 3174.822928] PGD b800d067 
[ 3174.822934] PUD b800f067 
[ 3174.822940] PMD 232c57067 
[ 3174.822943] PTE 0

[ 3174.822951] Oops: 0000 [#1] SMP
[ 3174.822958] Modules linked in: xt_nat xt_mac ipt_REJECT nf_reject_ipv4 xt_connlimit xt_tcpudp xt_conntrack ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack arptable_filter arp_tables iptable_filter snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic fuse intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul hid_generic evdev i915 ghash_clmulni_intel intel_cstate drm_kms_helper snd_hda_intel drm intel_uncore intel_rapl_perf snd_hda_codec serio_raw snd_hda_core snd_hwdep usbhid pcspkr i2c_algo_bit snd_pcm hid snd_timer iTCO_wdt iTCO_vendor_support snd mei_me soundcore mei shpchp sg wmi video acpi_pad button bonding parport_pc ppdev lp parport ip_tables x_tables autofs4
[ 3174.823178]  ext4 crc16 jbd2 crc32c_generic fscrypto ecb mbcache sd_mod uas usb_storage crc32c_intel aesni_intel aes_x86_64 glue_helper lrw gf128mul ablk_helper cryptd psmouse ahci libahci i2c_i801 i2c_smbus libata xhci_pci scsi_mod xhci_hcd r8169 mii usbcore usb_common fan thermal [last unloaded: corefreqk]
[ 3174.823295] CPU: 2 PID: 3635 Comm: corefreqd-cmgr Tainted: G  R        O    4.9.0-8-amd64 #1 Debian 4.9.144-3.1
[ 3174.823338] Hardware name: Gigabyte Technology Co., Ltd. H110M-S2H/H110M-S2H-CF, BIOS F22 06/12/2017
[ 3174.823380] task: ffff97f4f1561040 task.stack: ffffb5ed80d68000
[ 3174.823403] RIP: 0010:[<ffffffff96e07f4f>]  [<ffffffff96e07f4f>] filp_close+0x1f/0x70
[ 3174.823444] RSP: 0018:ffffb5ed80d6bf18  EFLAGS: 00010246
[ 3174.823466] RAX: ffffffffc0850320 RBX: ffff97f4f2826a00 RCX: ffff97f4f2826a00
[ 3174.823492] RDX: ffff97f4f12c1ad8 RSI: ffff97f4f12c1a40 RDI: ffff97f4f2826a00
[ 3174.823517] RBP: ffff97f4f12c1a40 R08: 000056436ac9d030 R09: 00007f9b468e6900
[ 3174.823543] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000000
[ 3174.823569] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[ 3174.823595] FS:  00007f9b475c1b40(0000) GS:ffff97f4fed00000(0000) knlGS:0000000000000000
[ 3174.823634] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 3174.823657] CR2: ffffffffc0850388 CR3: 0000000231e78000 CR4: 00000000003606f0
[ 3174.823683] Stack:
[ 3174.823699]  ffffb5ed80d6bf58 0000000000000000 0000000000000000 ffffffff96e07fbf
[ 3174.823740]  ffffffff96c03b7d 0000000000000000 0000000000000000 ffffffff9721924e
[ 3174.823781]  0000000000000000 0000000000000000 00007fff9de97bc0 0000000000000000
[ 3174.823821] Call Trace:
[ 3174.823840]  [<ffffffff96e07fbf>] ? SyS_close+0x1f/0x50
[ 3174.823863]  [<ffffffff96c03b7d>] ? do_syscall_64+0x8d/0xf0
[ 3174.823887]  [<ffffffff9721924e>] ? entry_SYSCALL_64_after_swapgs+0x58/0xc6
[ 3174.823912] Code: 00 48 83 c7 10 e9 f2 cf 01 00 66 90 0f 1f 44 00 00 41 54 55 53 48 8b 47 38 48 85 c0 74 4a 48 8b 47 28 45 31 e4 48 89 fb 48 89 f5 <48> 8b 40 68 48 85 c0 74 08 e8 f3 71 41 00 41 89 c4 f6 43 45 40 
[ 3174.824024] RIP  [<ffffffff96e07f4f>] filp_close+0x1f/0x70
[ 3174.824048]  RSP <ffffb5ed80d6bf18>
[ 3174.824067] CR2: ffffffffc0850388
[ 3174.824857] ---[ end trace 2f202f513d8c7c92 ]---
[ 3174.825077] BUG: unable to handle kernel paging request at ffffffffc0850388
[ 3174.825367] IP: [<ffffffff96e07f4f>] filp_close+0x1f/0x70
[ 3174.825567] PGD b800d067 
[ 3174.825660] PUD b800f067 
[ 3174.825854] PMD 232c57067 
[ 3174.825945] PTE 0

[ 3174.826242] Oops: 0000 [#2] SMP
[ 3174.826348] Modules linked in: xt_nat xt_mac ipt_REJECT nf_reject_ipv4 xt_connlimit xt_tcpudp xt_conntrack ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack arptable_filter arp_tables iptable_filter snd_hda_codec_hdmi snd_hda_codec_realtek snd_hda_codec_generic fuse intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul hid_generic evdev i915 ghash_clmulni_intel intel_cstate drm_kms_helper snd_hda_intel drm intel_uncore intel_rapl_perf snd_hda_codec serio_raw snd_hda_core snd_hwdep usbhid pcspkr i2c_algo_bit snd_pcm hid snd_timer iTCO_wdt iTCO_vendor_support snd mei_me soundcore mei shpchp sg wmi video acpi_pad button bonding parport_pc ppdev lp parport ip_tables x_tables autofs4
[ 3174.832975]  ext4 crc16 jbd2 crc32c_generic fscrypto ecb mbcache sd_mod uas usb_storage crc32c_intel aesni_intel aes_x86_64 glue_helper lrw gf128mul ablk_helper cryptd psmouse ahci libahci i2c_i801 i2c_smbus libata xhci_pci scsi_mod xhci_hcd r8169 mii usbcore usb_common fan thermal [last unloaded: corefreqk]
[ 3174.836171] CPU: 2 PID: 3634 Comm: corefreqd-pmgr Tainted: G  R   D    O    4.9.0-8-amd64 #1 Debian 4.9.144-3.1
[ 3174.836301] Hardware name: Gigabyte Technology Co., Ltd. H110M-S2H/H110M-S2H-CF, BIOS F22 06/12/2017
[ 3174.836430] task: ffff97f4efd39040 task.stack: ffffb5ed82498000
[ 3174.836541] RIP: 0010:[<ffffffff96e07f4f>]  [<ffffffff96e07f4f>] filp_close+0x1f/0x70
[ 3174.836757] RSP: 0018:ffffb5ed8249bf18  EFLAGS: 00010246
[ 3174.836867] RAX: ffffffffc0850320 RBX: ffff97f4f2826a00 RCX: ffff97f4f2826a00
[ 3174.836980] RDX: ffff97f4f1031958 RSI: ffff97f4f10318c0 RDI: ffff97f4f2826a00
[ 3174.837093] RBP: ffff97f4f10318c0 R08: 000056436ac9d030 R09: 00007f9b460e5900
[ 3174.837206] R10: 0000000000000000 R11: 0000000000000000 R12: 0000000000000000
[ 3174.837319] R13: 0000000000000000 R14: 0000000000000000 R15: 0000000000000000
[ 3174.837433] FS:  00007f9b475c1b40(0000) GS:ffff97f4fed00000(0000) knlGS:0000000000000000
[ 3174.837560] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 3174.837670] CR2: ffffffffc0850388 CR3: 0000000233a38000 CR4: 00000000003606f0
[ 3174.837783] Stack:
[ 3174.837887]  ffffb5ed8249bf58 0000000000000000 0000000000000000 ffffffff96e07fbf
[ 3174.838367]  ffffffff96c03b7d 0000000000000000 0000000000000000 ffffffff9721924e
[ 3174.838847]  0000000000000000 0000000000000000 00007fff9de97bc0 0000000000000000
[ 3174.839327] Call Trace:
[ 3174.839433]  [<ffffffff96e07fbf>] ? SyS_close+0x1f/0x50
[ 3174.839544]  [<ffffffff96c03b7d>] ? do_syscall_64+0x8d/0xf0
[ 3174.839655]  [<ffffffff9721924e>] ? entry_SYSCALL_64_after_swapgs+0x58/0xc6
[ 3174.839767] Code: 00 48 83 c7 10 e9 f2 cf 01 00 66 90 0f 1f 44 00 00 41 54 55 53 48 8b 47 38 48 85 c0 74 4a 48 8b 47 28 45 31 e4 48 89 fb 48 89 f5 <48> 8b 40 68 48 85 c0 74 08 e8 f3 71 41 00 41 89 c4 f6 43 45 40 
[ 3174.845700] RIP  [<ffffffff96e07f4f>] filp_close+0x1f/0x70
[ 3174.845899]  RSP <ffffb5ed8249bf18>
[ 3174.846006] CR2: ffffffffc0850388
[ 3174.846112] ---[ end trace 2f202f513d8c7c93 ]---`

cyring, 2019-04-13:
Although I'm trying to make _CoreFreq_ as robust as possible but how did you shutdown the programs ?

From traces, I understand the following sequence: 
* Firstly: kernel has been removed
  - `CoreFreq: Unload`
  - Was it forcely removed ?
* Secondly: the daemon has been killed
   - Which signal did you send to  the daemon ?
   - Did you kill `corefreqd-pmgr` or `corefreqd-cmgr` or one of the ring handlers `corefreqd-ring0 corefreqd-ring1` or any other threads `corefreqd/0 corefreqd#0` ?

Remarks:
1- the **corefreqd.service** lists the daemon shutdown procedure
2- Cli will automatically exit when the daemon is down
3- then, the driver `corefreqk.ko` can be safely removed from kernel

CyrIng

vincenzogianfelice, 2019-04-14:
I solved. My "corefreqd.service" contained an error. Thanks for the "clarifications" and sorry for my English. ;)

cyring, 2019-04-14:
Your welcome.
Feel free to share your `corefreqd.service`, it may help me to improve the released script.

vincenzogianfelice, 2019-04-15:
This code of cpufreqd file:

`#!/bin/sh

case "$1" in

start)
      /sbin/insmod /opt/CoreFreq/corefreqk.ko
      /opt/CoreFreq/corefreqd -id &
      ;;
stop)
      killall -g /opt/CoreFreq/corefreqd
      sleep 1
      rmmod corefreqk
      ;;
esac
`

cyring, 2019-04-15:
OK I see.
Options `-i` `-d` `-q` arz

closed 2019-04-15 by cyring

vincenzogianfelice, 2019-04-17:
I solved this way, since 'wait' ended correctly as soon as the kill command had finished killing the process, but when I tried to remove the module it was always busy:

`stop)
                kill $(pgrep corefreqd | head -n1)
                while [ "$status" != "0" ]; do sleep 0.5; rmmod corefreqk; status="$?"; done
                ;;`

cyring, 2019-04-17:
Hello,
Check this part of code and you will notice that the shutdown procedure is engaged by the signal QUIT 
https://github.com/cyring/CoreFreq/blob/49e96094c93c9957b5a4d804af46aae92ac84b28/corefreqd.c#L3370

`grep` won't guarantee to return the main pid and you may get the child process pid, which must not receive any signal.
I will suggest to : 
  * store in one script variable the pid of the launched `corefreqd` job
or
  * use `pidof corefreqd` as the `kill` argument 

The _CoreFreq_ default timer is one second. The script should at least sleep this delay after trying to remove the module

vincenzogianfelice, 2019-04-19:
The problem is just that ... variables are lost after the script is closed. So at the next invocation, that variable no longer contains anything. Unless you save the pid in a file. Furthermore, even with the 'export' command it does not work, as the script at launch is in a subshell, so the variable is saved in the subshell which "kills" the environment variable at closure.

cyring, 2019-04-19:
I think `pidof` will reliably returned the main pid