cyring/CoreFreq #107: CFlop->Relative.Freq showing a frequency that is greater than the factory core frequency/Turbo boost frequency. How is this calculated?

NRevathi, 2019-04-19:
My Intel Laptop(SandyBridge) is 2.50 GHz but relative freq (MHz) is showing 23650/24554 sometimes.

cyring, 2019-04-19:
Hello,

Regarding the build, when you run `make info` what are the counters in used ?  

  * Fixed perf. counters
```
MSR_CORE_PERF_UCC [MSR_CORE_PERF_FIXED_CTR1]
MSR_CORE_PERF_URC [MSR_CORE_PERF_FIXED_CTR2]
```
  * Or: HCF counters
```
MSR_CORE_PERF_UCC [MSR_IA32_APERF]
MSR_CORE_PERF_URC [MSR_IA32_MPERF]
```

If the fixed performance counters are in use; is the [NMI_Watchdog blacklisted](https://github.com/cyring/CoreFreq/wiki/HowTo-solve-crashes), since the kernel boot ?

NRevathi, 2019-04-20:
$ make info
NMI [0]
HYPERVISOR [0]
MSR_CORE_PERF_UCC [MSR_CORE_PERF_FIXED_CTR1]
MSR_CORE_PERF_URC [MSR_CORE_PERF_FIXED_CTR2]
make: Nothing to be done for 'info'.
$ cat /proc/sys/kernel/nmi_watchdog
1
$

cyring, 2019-04-20:
So the NMI Watchdog is running and CoreFreq uses the fixed performance counters : there is a conflict.

The NMI watchdog has to be fully disabled, straight from the boot kernel command line:
In your boot loader (SysLinux or Grub) add : nmi_watchdog=0

See mine bellow.
```
LABEL arch
        MENU LABEL Arch ^Linux
        LINUX ../vmlinuz-linux
        APPEND root=/dev/disk/by-label/ROOT rw nmi_watchdog=0 modprobe.blacklist=pcspkr,iTCO_wdt cpu0_hotplug
        INITRD ../intel-ucode.img,../initramfs-linux.img
```

NRevathi, 2019-04-21:
Hello,
I have disabled NMI_watchdog in GRUB /etc/default/grub and rebooted. But still I'm getting relative freq (MHz) as 52238.84, 47790.89, 416481.91 in some of the records. Also, the Instructions cycles shows the following values when I ran NSB3.0 bt.C benchmark program :

1.        **7395187774.13503** IPS/s, **19941625126.805** IPC/c, 0 CPI/i,  **1.84467440736681E+019** INST for relative freq 1283.28 MHz.
2.        0.650029 IPS/s, 0.665204 IPC/c, 1.503297 CPI/i,, 1621391182 INST, 2521.37 MHz.

These values are retrieved through the fields: CFlop->State.IPS, CFlop->State.IPC,  CFlop->State.CPI, CFlop->Delta.INST. The total number of instructions executed doesn't match with the perf instructions. 

Lot of differences are there in total number of instructions executed. Please clarify. 

Thanks in advance.
N Revathi

--

--

--

--

--

cyring, 2019-04-21:
Hello,
I have collected software which conflicts with _CoreFreq_ in this [list](https://github.com/cyring/CoreFreq/wiki/Software-incompatibility).  

Every interval, the _CoreFreq_ driver reads the TSC and these 3 fixed PMC registers to sample the following counters:  

Counter | MSR | Addr 
------------|--------|-------- 
INST | MSR_PERF_FIXED_CTR0 | 0x309
UCC | MSR_PERF_FIXED_CTR1 | 0x30A
URC | MSR_PERF_FIXED_CTR2 | 0x30B

The above registers are freely read/write by Softwares. For example, the Watchdog will reset the value and thus corrupt the _CoreFreq_ measurements.  
I'm not using the Linux perf framework but I understand it also employs the same registers. [Here](https://perf.wiki.kernel.org/index.php/Tutorial#multiplexing_and_scaling_events)

Alternatively the _CoreFreq_ driver can be built with the APERF/MPERF registers:
```
make MSR_CORE_PERF_UCC=MSR_IA32_APERF MSR_CORE_PERF_URC=MSR_IA32_MPERF
```
This built may solve the UCC and URC conflicts but not the INST counter which remains on the FIXED_CTR0

Can you please post the output of:
1. `corefreq-cli -c`
2. `corefreq-cli -i`

cyring, 2019-04-21:
In addition to above, where is the NSB source code ?
Does it use the PMC registers, directly or through other frameworks ?
Is NSB a virtualized environment ?

NRevathi, 2019-04-21:
I ran NASA parallel benchmark openmp C program "NPB3.0-omp-C" separately.
It's a c program and doesnt use any PMC registers. I ran to stress the CPU
and study temp variations and instructions retired. It has nothing to do
with virtualization.

Output:
$./corefereq-cli -c
CPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS
Max
#00 2451.89 (24.57)  98.30  99.44   0.56   0.00   0.00   0.00  62 / 81:19 /
83
#01 1180.94 (11.84)  47.34  99.62   0.21   0.00   0.00   0.17  62 / 83:17 /
85
#02 2468.58 (24.74)  98.97  99.88   0.12   0.00   0.00   0.00  62 / 81:19 /
83
#03 5833.45 (58.47) 233.87  98.56   1.27   0.00   0.00   0.17  62 / 83:17 /
85

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:
Pkg:
                    119.62  99.38   0.54   0.00   0.00   0.08     100 C
83 C
CPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS
Max
#00 1997.50 (20.02)  80.08  99.87   0.13   0.00   0.00   0.00  62 / 82:18 /
83
#01 5466.92 (54.79) 219.17  99.80   0.20   0.00   0.00   0.00  62 / 83:17 /
85
#02 2005.37 (20.10)  80.40  99.85   0.15   0.00   0.00   0.00  62 / 82:18 /
83
#03 1664.81 (16.69)  66.74  99.76   0.24   0.00   0.00   0.00  62 / 83:17 /
85

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:
Pkg:
                    111.60  99.82   0.18   0.00   0.00   0.00     100 C
83 C
CPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS
Max
#00 1896.69 (19.01)  76.04  99.77   0.23   0.00   0.00   0.00  62 / 90:10 /
91
#01 3576.08 (35.84) 143.37  99.73   0.27   0.00   0.00   0.00  62 / 90:10 /
93
#02 1936.59 (19.41)  77.64  99.73   0.27   0.00   0.00   0.00  62 / 90:10 /
91
#03  299.53 ( 3.00)  12.01  99.77   0.23   0.00   0.00   0.00  62 / 90:10 /
93

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:
Pkg:
                     77.26  99.75   0.25   0.00   0.00   0.00     100 C
90 C
CPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS
Max
#00 1945.74 (19.50)  78.01  97.24   0.47   0.00   0.00   2.29  62 / 93:7  /
94
#01 6521.91 (65.37) 261.47  96.81   0.89   0.00   0.00   2.30  62 / 94:6  /
95
#02 1754.03 (17.58)  70.32  97.30   0.41   0.00   0.00   2.29  62 / 93:7  /
94
#03 2521.16 (25.27) 101.08  97.03   0.67   0.00   0.00   2.30  62 / 94:6  /
94

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:
Pkg:
                    127.72  97.09   0.61   0.00   0.00   2.29     100 C
94 C

$./corefreq-cli -i
CPU     IPS            IPC            CPI
#00 7395747638.079905/s 14825968327.488356/c     0.000000/i
#01     0.145059/s     0.291262/c     3.433333/i
#02     0.391889/s     0.784100/c     1.275348/i
#03     0.147659/s     0.299186/c     3.342406/i

CPU     IPS            IPC            CPI
#00     0.080224/s     0.164589/c     6.075740/i
#01     0.297429/s     0.604885/c     1.653206/i
#02     0.136694/s     0.279467/c     3.578240/i
#03     0.021247/s     0.043217/c    23.138784/i

CPU     IPS            IPC            CPI
#00     0.551701/s     1.118151/c     0.894334/i
#01     0.194079/s     0.401173/c     2.492692/i
#02 7395247999.218527/s 15368568174.839579/c     0.000000/i
#03     0.244990/s     0.508714/c     1.965741/i

CPU     IPS            IPC            CPI
#00     0.402021/s     0.801495/c     1.247669/i
#01     0.175478/s     0.349610/c     2.860333/i
#02 7395245784.565212/s 14666123572.040688/c     0.000000/i
#03     0.175293/s     0.350184/c     2.855644/i

CPU     IPS            IPC            CPI
#00 7395441203.971069/s 14584444745.454603/c     0.000000/i
#01     0.090407/s     0.177835/c     5.623174/i
#02 7395440943.215435/s 14608870379.821009/c     0.000000/i
#03     0.312060/s     0.618291/c     1.617362/i

CPU     IPS            IPC            CPI
#00     0.295338/s     0.602497/c     1.659760/i
#01 7395662776.940892/s 14874043157.448429/c     0.000000/i
#02     0.183577/s     0.377290/c     2.650483/i
#03     0.296519/s     0.598992/c     1.669471/i

NRevathi, 2019-04-21:
I executed my C program as given below:
$sudo perf stat -e cpu-cycles,instructions ./NPB3.0-omp-C/bin/lu.C

Does it conflict with corefreq PMC counter reg.?

Now I ran without perf:
$./NPB3.0-omp-C/bin/lu.C
but now I'm getting only NULL values......

./corefreq-cli -i
CPU     IPS            IPC            CPI
#00     0.000000/s     0.000000/c     0.000000/i
#01     0.000000/s     0.000000/c     0.000000/i
#02     0.000000/s     0.000000/c     0.000000/i
#03     0.000000/s     0.000000/c     0.000000/i

./corefreq-cli -c
CPU Freq(MHz) Ratio  Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)  Min TMP:TS
Max
#00    0.00 ( 0.00)   0.00  98.20   0.43   0.00   0.00   1.37  62 / 92:8  /
96
#01    0.00 ( 0.00)   0.00  99.59   0.41   0.00   0.00   0.00  62 / 93:7  /
97
#02    0.00 ( 0.00)   0.00  98.14   0.49   0.00   0.00   1.37  62 / 92:8  /
96
#03    0.00 ( 0.00)   0.00  99.52   0.48   0.00   0.00   0.00  62 / 93:7  /
97

    Averages:        Turbo  C0(%)  C1(%)  C3(%)  C6(%)  C7(%)    TjMax:
Pkg:
                      0.00  98.86   0.45   0.00   0.00   0.68     100 C
93 C

I've attached the code for lu.c

Thank you.

NRevathi, 2019-04-21:
$ sudo ./corefreq-cli -s
Processor                             [Intel(R) Core(TM) i5-2450M CPU @ 2.50GHz]
|- Architecture                                                    [SandyBridge]
|- Vendor ID                                                      [GenuineIntel]
|- Microcode                                                        [        46]
|- Signature                                                            [ 06_2A]
|- Stepping                                                             [     7]
|- Online CPU                                                           [  4/4 ]
|- Base Clock                                                           [ 99.77]
|- Frequency            (MHz)                      Ratio                        
                 Min    798.19                    [   8 ]                       
                 Max   2494.34                    [  25 ]                       
|- Factory                                                              [100.00]
                       2500                       [  25 ]                       
|- Turbo Boost                                                          [  LOCK]
                  1C   3092.98                    [  31 ]                       
                  2C   2893.43                    [  29 ]                       
                  3C   2893.43                    [  29 ]                       
                  4C   2893.43                    [  29 ]                       
|- Uncore                                                               [  LOCK]
                 Min    798.19                    [   8 ]                       
                 Max   2494.34                    [  25 ]                       
                                                                                
ISA Extensions:                                                                 
|- 3DNow!/Ext [N,N]          ADX [N]          AES [Y]  AVX/AVX2 [Y/N] 
|- AVX-512      [N]  BMI1/BMI2 [N/N]        CLFSH [Y]        CMOV [Y] 
|- CMPXCH8      [Y]     CMPXCH16 [Y]         F16C [N]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/N]     MONITOR [Y] 
|- MOVBE        [N]          MPX [N]     PCLMULDQ [Y]      POPCNT [Y] 
|- RDRAND       [N]       RDSEED [N]       RDTSCP [Y]         SEP [Y] 
|- SGX          [N]          SSE [Y]         SSE2 [Y]        SSE3 [Y] 
|- SSSE3        [Y]  SSE4.1/4A [Y/N]       SSE4.2 [Y]     SYSCALL [Y] 
                                                                                
Features:                                                                       
|- 1 GB Pages Support                                      1GB-PAGES   [Missing]
|- Advanced Configuration & Power Interface                     ACPI   [Present]
|- Advanced Programmable Interrupt Controller                   APIC   [Present]
|- Core Multi-Processing                                  CMP Legacy   [Missing]
|- L1 Data Cache Context ID                                  CNXT-ID   [Missing]
|- Direct Cache Access                                           DCA   [Missing]
|- Debugging Extension                                            DE   [Present]
|- Debug Store & Precise Event Based Sampling               DS, PEBS   [Present]
|- CPL Qualified Debug Store                                  DS-CPL   [Present]
|- 64-Bit Debug Store                                         DTES64   [Present]
|- Fast-String Operation                                Fast-Strings   [Missing]
|- Fused Multiply Add                                       FMA|FMA4   [Missing]
|- Hardware Lock Elision                                         HLE   [Missing]
|- Long Mode 64 bits                                         IA64|LM   [Present]
|- LightWeight Profiling                                         LWP   [Missing]
|- Machine-Check Architecture                                    MCA   [Present]
|- Model Specific Registers                                      MSR   [Present]
|- Memory Type Range Registers                                  MTRR   [Present]
|- OS-Enabled Ext. State Management                          OSXSAVE   [Present]
|- Physical Address Extension                                    PAE   [Present]
|- Page Attribute Table                                          PAT   [Present]
|- Pending Break Enable                                          PBE   [Present]
|- Process Context Identifiers                                  PCID   [Present]
|- Perfmon and Debug Capability                                 PDCM   [Present]
|- Page Global Enable                                            PGE   [Present]
|- Page Size Extension                                           PSE   [Present]
|- 36-bit Page Size Extension                                  PSE36   [Present]
|- Processor Serial Number                                       PSN   [Missing]
|- Restricted Transactional Memory                               RTM   [Missing]
|- Safer Mode Extensions                                         SMX   [Missing]
|- Self-Snoop                                                     SS   [Present]
|- Time Stamp Counter                                            TSC [Invariant]
|- Time Stamp Counter Deadline                          TSC-DEADLINE   [Present]
|- Virtual Mode Extension                                        VME   [Present]
|- Virtual Machine Extensions                                    VMX   [Present]
[... cut ...]
|- Package Thermal Management                                    PTM   [Present]
|- Thermal Monitor 1                                         TM1|TTP   [ Enable]
|- Thermal Monitor 2                                         TM2|HTC   [Present]
|- Units                                                                        
   |- Power                                               watt   [  0.125000000]
   |- Energy                                             joule   [  0.000015259]
   |- Window                                            second   [  0.000976562]
$

cyring, 2019-04-21:
The thing with the Processor PMU is that any software which wants to use the PMC counters has to start them, and **stop** therm when over. 

Although _CoreFreq_ save the current PMC enablement configuration before use, and restore it when closing; I can't tell about the other softwares. (But it happens).

I will suggest that you first boot the _CoreFreq_ [live ISO image](https://github.com/cyring/CoreFreq/wiki/Live-CD) to tell me if your processor is well supported.

Regards
CyrIng

NRevathi, 2019-04-21:
Thank you. 

It works. I rebooted and executed benchmark without perf tool and read PMC thro' CoreFreq...I'm getting relative freq within the limit and Instruction is also reasonable...

You are right. There was a conflict bcoz I used both tools at the same time.

cyring, 2019-04-21:
Glad to read that it runs ok.
I believe that the _CoreFreq_'s documentation needs improvements.

closed 2019-04-23 by cyring