cyring/CoreFreq #110: *** stack smashing detected ***: <unknown> terminated zsh: abort (core dumped)  corefreq-cli

johnpiers, 2019-04-23:
I hope that this is in the right place, apologies if not.

```
:: Proceed with installation? [Y/n] Y
(1/1) checking keys in keyring                                           [########################################] 100%
(1/1) checking package integrity                                         [########################################] 100%
(1/1) loading package files                                              [########################################] 100%
(1/1) checking for file conflicts                                        [########################################] 100%
(1/1) checking available disk space                                      [########################################] 100%
:: Running pre-transaction hooks...
(1/1) Remove DKMS modules
==> dkms remove corefreqk/1.48 -k 5.0.9-arch1-1-ARCH
:: Processing package changes...
(1/1) reinstalling corefreq-git                                          [########################################] 100%

[ CoreFreq Starting Instructions ]
 
1: load the kernel module
  # modprobe corefreqk
 
2: start the daemon
  # systemctl start corefreqd
 
3: run the client
  $ corefreq-cli

:: Running post-transaction hooks...
(1/3) Install DKMS modules
==> dkms install corefreqk/1.48 -k 5.0.9-arch1-1-ARCH
(2/3) Reloading system manager configuration...
(3/3) Still FuckallToSeeHere...

┌─[~]  47s
└─> sudo modprobe corefreqk

┌─[~]  
└─> sudo systemctl start corefreqd

┌─[~]  
└─> corefreq-cli
*** stack smashing detected ***: <unknown> terminated
zsh: abort (core dumped)  corefreq-cli

```
So the above is what I get. I've had this thing working for about a year (or more) and only recently with newer ver of the program and kernel updates have I started to have problems. Stopped working altogether, then with the last kernel update and the latest version from the ARCH repository it installed OK and the DKMS module installed correctly (==> dkms install corefreqk/1.48 -k 5.0.9-arch1-1-ARCH) I am a noob when it comes to these things and I am at a loss as regards what to look for or how to proceed from here. Thank you.

cyring, 2019-04-23:
Sorry for this bug.  
It comes from my yesterday commit that I have not been able to test with `HWP` capable processors.
Work is in progress in the issue #109 and I will provide a fix ASAP

With the current version, try the workaround below (which forces the driver to compute the Target (TGT) ratio)  
```
insmod corefreqk.ko TurboBoost_Enable=1
```
Next starts normally the Daemon and the Client.

johnpiers, 2019-04-23:
Thanks Cyril, that worked just fine. Thank you.

Have a good one!
Piers.

cyring, 2019-04-23:
Workaround released.

cyring, 2019-04-28:
Hello,  
Feel free to close the problem if you are satisfied with the latest version.  
Regards

johnpiers, 2019-04-29:
Thanks again Cyril.

closed 2019-04-29 by johnpiers