cyring/CoreFreq #111: How to get PMC that gives the number of Instructions executed by a process at run time?

NRevathi, 2019-04-28:


cyring, 2019-04-28:
At run time ?
But, if you are programming in C, your program will open the SHM and read the [structure](https://github.com/cyring/CoreFreq/blob/1e8a92ad7780fa6039d920bbb5d67c8cb49331e6/corefreq.h#L85) of the Instruction counter at the point of execution. 

Like this loop:  
https://github.com/cyring/CoreFreq/blob/1e8a92ad7780fa6039d920bbb5d67c8cb49331e6/corefreq-cli.c#L2243

```C
{
	struct FLIP_FLOP *CFlop = &Shm->Cpu[cpu].FlipFlop[!Shm->Cpu[cpu].Toggle];

	if (!BITVAL(Shm->Cpu[cpu].OffLine, OS))
		printf("#%02u %12.6f/s %12.6f/c %12.6f/i %18llu\n",
			cpu,
			CFlop->State.IPS,
			CFlop->State.IPC,
			CFlop->State.CPI,
			CFlop->Delta.INST);
}
```

Because **only one connection to the daemon** is possible, your program will take the place of the Cli.  

For other langages, you can invoke the JSON export with the `j` option of the Cli.
However, I believe it will have more latency.  
```
corefreq-cli -j
```

Going further, you could also mod the Daemon code and make use of the PMC counters in user space:  
https://github.com/cyring/CoreFreq/blob/1e8a92ad7780fa6039d920bbb5d67c8cb49331e6/corefreqm.c#L37

Hope this helps.

NRevathi, 2019-04-29:
I wanted  to trace the C program(openMP) during execution and get the 

> number of instructions executed by that process/task alone across all the cores dynamically at runtime...

I think using SHM, we can get the number of instructions executed by the individual CPUs...and not individual process.

Please help me. Using other tools to read PMC also conflict with corefreq.

cyring, 2019-04-29:
Do you have any examples of the C openMP case ?

NRevathi, 2019-04-29:
For example, when I run the just_omp.c, i wanted to get the number of instructions executed dynamically by the process a.out identified by its PID.
```
$ cat just_omp.c
#include <omp.h>
#include <stdio.h>
#include <stdlib.h>

int main (int argc, char *argv[]) 
{ int nthreads, tid;

/* Fork a team of threads giving them their own copies of variables */
  #pragma omp parallel private(nthreads, tid)
  {
  /* Obtain thread number */
  tid = omp_get_thread_num();
  printf("Hello World from thread = %d\n", tid);

  /* Only master thread does this */
  if (tid == 0) 
    {
    nthreads = omp_get_num_threads();
    printf("Number of threads = %d\n", nthreads);
    }
  }  /* All threads join master thread and disband */
}
```

> $ gcc -fopenmp just_omp.c 
> $ ./a.out
> Hello World from thread = 0
> Number of threads = 4
> Hello World from thread = 2
> Hello World from thread = 1
> Hello World from thread = 3

> $ objdump -dw a.out | wc -l
> 216
> $ 

Here, objdump will give the static count of instructions after linking...it doesn't take into account the problem size or branch instructions...

cyring, 2019-04-29:
Have you check the [opmc](https://github.com/opcm/pcm) project on this ?

cyring, 2019-05-04:
Hey did you check the project above ?

Unfortunately, my code is made per CPU.

To track a PID which can migrate, you will have to measure at the scheduler level. I believe than Linux Perf can help you.

closed 2019-05-04 by cyring

cyring, 2019-05-29:
Try [NanoBench](https://github.com/andreas-abel/nanoBench)