cyring/CoreFreq #112: Tempature not shown correctly on Threadripper 2950x

samsungrocks10, 2019-05-02:
Hi, I've got a TR 2950x and have used it for checking overclock freq, but the temperature readout seems to be totally wrong, it shows 80C when lm-sensors shows 29C, on my threadripper the temp is called Tdie with an offset version called Tctl. Here's a screenshot of sensors and corefreq-ctl if that helps, thanks.
[image] 
[image]

cyring, 2019-05-02:
Hello,

In this structure for Threadripper 2950 brand 
https://github.com/cyring/CoreFreq/blob/3ae40b633267b1262f46f89e602709bfc0d0adf8/corefreqk.h#L2131
Can you replace the offsets with these values: 
```
	.Param.Offset = {10,49},
```

Next, rebuild and test again  please.

samsungrocks10, 2019-05-03:
It seems to be lower but still not really near what sensors shows 
[image]

cyring, 2019-05-03:
This was the offset calibrated for the 2700X 

We need another reference such as BIOS values to compare with. 

The formula is here:
https://github.com/cyring/CoreFreq/blob/3ae40b633267b1262f46f89e602709bfc0d0adf8/coretypes.h#L232

```
	(Temp = ((Sensor * 5 / 40) - Param.Offset[1]) - Param.Offset[0])
```

Can you simplify it to a simple sensor value, like:

```
	(Temp = Sensor)
```

Rebuild and monitor in plain text with with this argument `corefreq-cli -c`

cyring, 2019-05-03:
Also when loading driver, can you bind to the bootstrap processor.

```
insmod corefreqk.ko ServiceProcessor=0
```

cyring, 2019-05-03:
Looking at k10temp, the _CoreFreq_ offsets would be:  
```
	.Param.Offset = {27,49},
```

samsungrocks10, 2019-05-04:
Sorry for the taking a bit, i changed those few things and now it seems to match sensors, thanks! Also, i am overclocked, currently its at 4.15GHz. Here's the output of these commands

```
samsungrocks10@debian-main:~/CoreFreq$ ./corefreq-cli -m
CPU Pkg  Apic  Core Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID    ID     ID  L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
00: BSP     0     0      0      64  4        32  8       512  8     32768 10  
01:   0     2     1      0      64  4        32  8       512  8     32768 10  
02:   0     4     2      0      64  4        32  8       512  8     32768 10  
03:   0     6     3      0      64  4        32  8       512  8     32768 10  
04:   0     8     4      0      64  4        32  8       512  8     32768 10  
05:   0    10     5      0      64  4        32  8       512  8     32768 10  
06:   0    12     6      0      64  4        32  8       512  8     32768 10  
07:   0    14     7      0      64  4        32  8       512  8     32768 10  
08:   1    16     0      0      64  4        32  8       512  8     32768 10  
09:   1    18     1      0      64  4        32  8       512  8     32768 10  
10:   1    20     2      0      64  4        32  8       512  8     32768 10  
11:   1    22     3      0      64  4        32  8       512  8     32768 10  
12:   1    24     4      0      64  4        32  8       512  8     32768 10  
13:   1    26     5      0      64  4        32  8       512  8     32768 10  
14:   1    28     6      0      64  4        32  8       512  8     32768 10  
15:   1    30     7      0      64  4        32  8       512  8     32768 10  
16:   0     1     0      1      64  4        32  8       512  8     32768 10  
17:   0     3     1      1      64  4        32  8       512  8     32768 10  
18:   0     5     2      1      64  4        32  8       512  8     32768 10  
19:   0     7     3      1      64  4        32  8       512  8     32768 10  
20:   0     9     4      1      64  4        32  8       512  8     32768 10  
21:   0    11     5      1      64  4        32  8       512  8     32768 10  
22:   0    13     6      1      64  4        32  8       512  8     32768 10  
23:   0    15     7      1      64  4        32  8       512  8     32768 10  
24:   1    17     0      1      64  4        32  8       512  8     32768 10  
25:   1    19     1      1      64  4        32  8       512  8     32768 10  
26:   1    21     2      1      64  4        32  8       512  8     32768 10  
27:   1    23     3      1      64  4        32  8       512  8     32768 10  
28:   1    25     4      1      64  4        32  8       512  8     32768 10  
29:   1    27     5      1      64  4        32  8       512  8     32768 10  
30:   1    29     6      1      64  4        32  8       512  8     32768 10  
31:   1    31     7      1      64  4        32  8       512  8     32768 10  
samsungrocks10@debian-main:~/CoreFreq$ ./corefreq-cli -s
Processor                      [AMD Ryzen Threadripper 2950X 16-Core Processor ]
|- Architecture                                                    [Zen+ Colfax]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [ 134251019]
|- Signature                                                            [ 8F_08]
|- Stepping                                                             [     2]
|- Online CPU                                                           [ 32/32]
|- Base Clock                                                           [101.03]
|- Frequency            (MHz)                      Ratio                        
                 Min    404.11                    <   4 >                       
                 Max   4142.15                    <  41 >                       
|- Factory                                                              [100.00]
                       4100                       [  41 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   2222.62                    [  22 ]                       
|- Turbo Boost                                                          [UNLOCK]
                 XFR   5051.40                    [  50 ]                       
                 CPB   5051.40                    [  50 ]                       
                  1C   4142.15                    <  41 >                       
                  2C   2222.62                    <  22 >                       
|- Uncore                                                               [  LOCK]
                 Min       nan                    [   0 ]                       
                 Max       nan                    [   0 ]                       
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N,N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [N]  BMI1/BMI2 [Y/Y]        CLFSH [Y]        CMOV [Y] 
|- CMPXCH8      [Y]     CMPXCH16 [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/Y]     MONITOR [Y] 
|- MOVBE        [Y]          MPX [N]     PCLMULDQ [Y]      POPCNT [Y] 
[... 74 lines cut ...]
[... cut ...]
|- Package Thermal Management                                    PTM   [Missing]
|- Thermal Monitor 1                                         TM1|TTP   [Present]
|- Thermal Monitor 2                                         TM2|HTC   [Present]
|- Units                                                                        
   |- Power                                               watt   [  0.007812500]
   |- Energy                                             joule   [  0.000000954]
   |- Window                                            second   [  0.000976562]
```

cyring, 2019-05-04:
Temperature offsets code committed.  

You can now pull the latest version of _CoreFreq_  

I'm in need of many of other Zen & EPYC architectures results in case if you have such hardware.

Cyril

cyring, 2019-05-04:
# Additional request
In BIOS or other motherboard tools, using the stock frequency, can you tell me what are
* the base frequency
* the turbo boost frequency
* the XFR frequency

I need those values to adjust the Boost ratios in the source code line below:
https://github.com/cyring/CoreFreq/blob/a8f6b7c266804f01970bd9d470334e7144093f6a/corefreqk.h#L2130
where the first parameter is the # of CPB ratios added to stock ratio  
and the second parameter is the # of XFR added to the CPB

FYI, this is currently the value 50 you see in the UI ratios ruler and the XFR & CPB of `corefreq-cli -s`; you can adjust these 2 parameters and rebuild until it matches the hardware.

samsungrocks10, 2019-05-05:
Thanks. I've got 2 other linux machines running ryzen 5 2600's so i probably could help with info on those if needed. I'll probably put my TR back to stock tomorrow(not sure what time zone your in, tomorrow for me is the 5th) to get that info.

cyring, 2019-05-05:
Nice. Can't wait to see results 😉
Same day here in Paris, France, summer time: UTC + 2

samsungrocks10, 2019-05-06:
Ok, I'm back down to stock, and now for some reason I'm getting segmentation fault on -s and -m. 
```
samsungrocks10@debian-main:~$ corefreq-cli -m
CPU Pkg  Apic  Core Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID    ID     ID  L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
00:   0     0     0      0       0  0         0  0         0  0         0 13  
01:   0     0     0      0       -  -         -  -         -  -         -  -  
02:   0     0     0      0       -  -         -  -         -  -         -  -  
03:   0     0     0      0       -  -         -  -         -  -         -  -  
04:   0     0     0      0       -  -         -  -         -  -         -  -  
05:   0     0     0      0       -  -         -  -         -  -         -  -  
06:   0     0     0      0       -  -         -  -         -  -         -  -  
07:   0     0     0      0       -  -         -  -         -  -         -  -  
08:   0     0     0      0       -  -         -  -         -  -         -  -  
09:   0     0     0      0       -  -         -  -         -  -         -  -  
10:   0     0     0      0       -  -         -  -         -  -         -  -  
11:   0     0     0      0       -  -         -  -         -  -         -  -  
12:   0     0     0      0       -  -         -  -         -  -         -  -  
13:   0     0     0      0       -  -         -  -         -  -         -  -  
14:   0     0     0      0       -  -         -  -         -  -         -  -  
15:   0     0     0      0       -  -         -  -         -  -         -  -  
16:   0     0     0      0       -  -         -  -         -  -         -  -  
17:   0     0     0      0       -  -         -  -         -  -         -  -  
18:   0     0     0      0       -  -         -  -         -  -         -  -  
19:   0     0     0      0       -  -         -  -         -  -         -  -  
20:   0     0     0      0       -  -         -  -         -  -         -  -  
21:   0     0     0      0       -  -         -  -         -  -         -  -  
22:   0     0     0      0       -  -         -  -         -  -         -  -  
23:   0     0     0      0       -  -         -  -         -  -         -  -  
24:   0     0     0      0       -  -         -  -         -  -         -  -  
25:   0     0     0      0       -  -         -  -         -  -         -  -  
26:   0     0     0      0       -  -         -  -         -  -         -  -  
27:   0     0     0      0       -  -         -  -         -  -         -  -  
28:   0     0     0      0       -  -         -  -         -  -         -  -  
29:   0     0     0      0       -  -         -  -         -  -         -  -  
30:   0     0     0      0       -  -         -  -         -  -         -  -  
31:   0     0     0      0       -  -         -  -         -  -         -  -  
32:   0     0     0      0       -  -         -  -         -  -         -  -  
33:   0     0     0      0       0  0         0  0         0  0         0  0  
34: BSP   128     0    273       0 18         0  0    786450  0         0 17  
35:   0     7     8      0       0  0         0  0         0  0         0  0  
36: BSP-1256135691238464265 505697421       -  -         -  -         -  -         -  -  
37: BSP-2062614528   827 5409022161207959552  2  10326691843456w    36888617  237030605018687w 
38: BSP-336567480-108443642 1220970824120821574553761w 136787406830473 i230351901736168w 28752384516  
39: BSP-1991755582-1958211509 -1052184502       -  -         -  -         -  -         -  -  
40: BSP1207959566-29325941 -19581501458266139862376 i25591910317875w 170976500135121  4670861918626w 
41: BSP264275455-1905276 376831390909132935657wi1681191365517wi230337766865516wi18304204833675  
42: BSP-200449861616777214 -402653184     148436  224049891935144w 426459687532525  429496693665534wi
43: BSP522652061963065344 -1786033966       -  -         -  -         -  -         -  -  
44: BSP-13585013-1645674497 1224736765       -  -         -  -         -  -         -  -  
45: BSP 65683-1149024256    296143517579665416wi     68050052  282719264855045  427927273934063 i
46: BSP1166756076-2060088560 120796007323367516162048  230273843333807 i  13971636169 i11372364835144w 
47: BSP137536     2 29590344       -  -         -  -         -  -         -  -  
48: BSP2335653214202255 606374733       -  -         -  -         -  -         -  -  
49: BSP10966374321096630620 -101716752211498332163904w 1365946311419 i77759759  0  389233846834120  
50: BSP53925184922036456 6095203841224736766  1         118687w  14892179308  230293504152105w 
51: BSP-596767349-914030591 860828312886675616733  778486623  0  25185484816895wi122473676653893wi
52: BSP134523136-2062614528    438       -  -         -  -         -  -         -  -  
53: BSP1426063362-668695413 1275068416121111457228648w 58574605751505w 1294210187  0wi23367516169284  
54: BSP12247367662095701385 268435454 420225523232wi14186918965332wi39258685437951  4278190084  0  
55: BSP637043464-1958155659 -40771       -  -         -  -         -  -         -  -  
56: BSP-52828434932015107 251787395998965248  4  2277982665  0  117440511928821wi429496710465512 i
57: BSP10908828881207959712 166266761       -  -         -  -         -  -         -  -  
58: BSP-106496000160195855 -1958215680       -  -         -  -         -  -         -  -  
59: BSP-385928722-53408 1082493768       -  -         -  -         -  -         -  -  
[... cut ...]
|- Junction Temperature                                        TjMax   [26740:30017]                                                                                                                                                                                                                                                                                                                                
|- Digital Thermal Sensor                                        DTS   [Present]
|- Power Limit Notification                                      PLN   [Missing]
|- Package Thermal Management                                    PTM   [Missing]
Segmentation fault
```

cyring, 2019-05-06:
1. Make sure you **unload** the driver before starting _CoreFreq_ b/c the _data structure has been changed_ since the last version.

*  Download a fresh copy of the software and rebuild. Once again, make sure to `rmmod corefreqk.ko`

2. If you still encounter those issues, please boot the [live image](https://github.com/cyring/CoreFreq/wiki/Live-CD).

* Once booted, If _CoreFreq_ starts correctly, exit [F4] and get the latest version with the command
```
corefreq_online_update.sh
```

* Build and start again with:

```
corefreq_build_run.sh
```

cyring, 2019-05-08:
Up

samsungrocks10, 2019-05-10:
Sorry for the delay, I was a bit busy and not on my PC much. All i did was rmmod, git pull and then reinstall/start the insmod and daemon, and it didn't do the segmentation fault again, not sure what was up there, i think thats what i did before too but i guess i must have missed something. here's the non errored corefreq-cli -s and -m.

```
samsungrocks10@debian-main:~/CoreFreq$ ./corefreq-cli -m
CPU Pkg  Apic  Core Thread  Caches      (w)rite-Back (i)nclusive              
 #   ID   ID    ID     ID  L1-Inst Way  L1-Data Way      L2  Way      L3  Way 
00: BSP     0     0      0      64  4        32  8       512  8     32768 10  
01:   0     2     1      0      64  4        32  8       512  8     32768 10  
02:   0     4     2      0      64  4        32  8       512  8     32768 10  
03:   0     6     3      0      64  4        32  8       512  8     32768 10  
04:   0     8     4      0      64  4        32  8       512  8     32768 10  
05:   0    10     5      0      64  4        32  8       512  8     32768 10  
06:   0    12     6      0      64  4        32  8       512  8     32768 10  
07:   0    14     7      0      64  4        32  8       512  8     32768 10  
08:   1    16     0      0      64  4        32  8       512  8     32768 10  
09:   1    18     1      0      64  4        32  8       512  8     32768 10  
10:   1    20     2      0      64  4        32  8       512  8     32768 10  
11:   1    22     3      0      64  4        32  8       512  8     32768 10  
12:   1    24     4      0      64  4        32  8       512  8     32768 10  
13:   1    26     5      0      64  4        32  8       512  8     32768 10  
14:   1    28     6      0      64  4        32  8       512  8     32768 10  
15:   1    30     7      0      64  4        32  8       512  8     32768 10  
16:   0     1     0      1      64  4        32  8       512  8     32768 10  
17:   0     3     1      1      64  4        32  8       512  8     32768 10  
18:   0     5     2      1      64  4        32  8       512  8     32768 10  
19:   0     7     3      1      64  4        32  8       512  8     32768 10  
20:   0     9     4      1      64  4        32  8       512  8     32768 10  
21:   0    11     5      1      64  4        32  8       512  8     32768 10  
22:   0    13     6      1      64  4        32  8       512  8     32768 10  
23:   0    15     7      1      64  4        32  8       512  8     32768 10  
24:   1    17     0      1      64  4        32  8       512  8     32768 10  
25:   1    19     1      1      64  4        32  8       512  8     32768 10  
26:   1    21     2      1      64  4        32  8       512  8     32768 10  
27:   1    23     3      1      64  4        32  8       512  8     32768 10  
28:   1    25     4      1      64  4        32  8       512  8     32768 10  
29:   1    27     5      1      64  4        32  8       512  8     32768 10  
30:   1    29     6      1      64  4        32  8       512  8     32768 10  
31:   1    31     7      1      64  4        32  8       512  8     32768 10  

samsungrocks10@debian-main:~/CoreFreq$ ./corefreq-cli -s
Processor                      [AMD Ryzen Threadripper 2950X 16-Core Processor ]
|- Architecture                                                    [Zen+ Colfax]
|- Vendor ID                                                      [AuthenticAMD]
|- Microcode                                                        [ 134251019]
|- Signature                                                            [ 8F_08]
|- Stepping                                                             [     2]
|- Online CPU                                                           [ 32/32]
|- Base Clock                                                           [ 99.81]
|- Frequency            (MHz)                      Ratio                        
                 Min    399.25                    <   4 >                       
                 Max   3493.41                    <  35 >                       
|- Factory                                                              [100.00]
                       3500                       [  35 ]                       
|- Performance                                                                  
   |- OSPM                                                                      
                 TGT   2794.73                    [  28 ]                       
|- Turbo Boost                                                          [UNLOCK]
                 XFR   4391.72                    [  44 ]                       
                 CPB   4391.72                    [  44 ]                       
                  1C   2794.73                    <  28 >                       
                  2C   2195.86                    <  22 >                       
|- Uncore                                                               [  LOCK]
                                                                                
Instruction Set Extensions                                                      
|- 3DNow!/Ext [N,N]          ADX [Y]          AES [Y]  AVX/AVX2 [Y/Y] 
|- AVX-512      [N]  BMI1/BMI2 [Y/Y]        CLFSH [Y]        CMOV [Y] 
|- CMPXCH8      [Y]     CMPXCH16 [Y]         F16C [Y]         FPU [Y] 
|- FXSR         [Y]    LAHF/SAHF [Y]    MMX/Ext [Y/Y]     MONITOR [Y] 
|- MOVBE        [Y]          MPX [N]     PCLMULDQ [Y]      POPCNT [Y] 
[... cut ...]
|- Thermal Monitor 1                                         TM1|TTP   [Present]
|- Thermal Monitor 2                                         TM2|HTC   [Present]
|- Units                                                                        
   |- Power                                               watt   [  0.007812500]
   |- Energy                                             joule   [  0.000000954]
   |- Window                                            second   [  0.000976562]
samsungrocks10@debian-main:~/CoreFreq$
```

cyring, 2019-05-10:
I'm glad to read it works better, thank you.

The frequency ratios Max and CPB match the 2950X specifications with respectively the values of 35 and 44

What is missing is the real XFR ratio (which is not documented by AMD). 
To discover it, you will stress one and only one CPU. 
For example, press F3, stress the CPU number 0, and watch its frequency until it goes beyond the CPB. 
Then post back the highest ratio you have read.

I'm still interested by 2600' results

Regards 
Cyril

cyring, 2019-05-10:
Remark: googling with "2950X XFR highest frequency" and I don't find better than a frequency ratio of 44 

In your BIOS, do you have options to disable / enable :
* XFR
* Precision Boost 
* Core Performance Boost

_whatever the names defined by the BIOS manufacturer_

And using the single Core stress method, posted above, discover the respective frequency ratios.

Note: I'm aware it's tedious way to reveal the undocumented boosted P-States, but we have learned things using the 2700X

samsungrocks10, 2019-05-11:
So far it only got to around 4377 max on one core stress. In my bios it looks to be called Precision Boost Overdrive, It's set to auto, should it be on for this? There's also Performance Enhancer set to auto, other options are Level 1 though 3. My mobo is a ASRock Fatal1ty x399 if that helps at all

cyring, 2019-05-11:
Thank you. 

43.75 (44.00 rounded in _CoreFreq_) seems to be the highest COF (coefficient of frequency)

I don't find the difference between CPB and XFR frequencies for this Processor architecture, I will leave the 2950X table as it is. 

Please let me know how _CoreFreq_ works with the 2600's

cyring, 2019-06-17:
Hello,
Latest commit brings a change in the temperature computation based on a sensor register bit.
Can you please make some non-regression tests ?

cyring, 2019-07-14:
Hello, if you are still on board with _CoreFreq_ -;)

Can you test the Power measurements of the latest version ?
1. Press `V`
2. For each case { idle; single Core load; full load }  (press `F3` to apply load)
    * Monitor the `Cores` power against your 2950X specs
    * Check if the `Package` power has no regression
3. screenshots welcomed

Thank U

closed 2019-07-23 by cyring

cyring, 2020-07-06:
Hello,
Among the last changes in issue #195 , an I/O ASM access to the FCH have been added: can you please check if the last _CoreFreq_ version is starting fine ?
Thank you