cyring/CoreFreq #113: CorFreq is not compatible with BMQ

RomanHargrave, 2019-05-10:
In kernels with the BMQ scheduler patch, `struct task_struct` does not have `wake_cpu`. There isn't a lot of good documentation (that I could find, presently) on the purpose of `wake_cpu` but the name and related patches suggest that it is the currently active CPU for a given task. I believe this isn't available due to the way BMQ reasons about task scheduling.

While I cannot make any good suggestions as to alternatives, `struct task_struct.on_cpu` is still available when BMQ is enabled (though it may be less correct, at least when BMQ isn't in use).

RomanHargrave, 2019-05-10:
I have had success in replacing `wake_cpu` with `on_cpu`, though I'm not certain what effect it will have on the accuracy of measurements (if any).

It also appears that when BMQ is enabled, `struct task_struct.se` is not present (as with MUQSS). I updated the MUQSS config test macro to test `CONFIG_SCHED_BMQ` as well, which appears to work.

cyring, 2019-05-10:
Thanks for this.

For any low latency user reason, I've made the Tasks Monitoring optional in _CoreFreq_ 

Just start the daemon with the argument `-goff`

About building the driver in BMQ Kernel configuration , the BMQ macro directive could indeed leave the missing struct se fields with a zero value.

cyring, 2019-05-10:
I'm using ArchLinux: is this the BMQ patch package [linux-ck](https://aur.archlinux.org/packages/linux-ck) ?

RomanHargrave, 2019-05-10:
It's actually pf-kernel, so there may be some additional stuff in the way, but the `CONFIG_SCHED_MUQSS` test is also applicable to `CONFIG_SCHED_BMQ` (see [my hack](https://github.com/RomanHargrave/CoreFreq/blob/bmq/corefreqk.c#L5130) - though I think that there is a different, more applicable config flag that should be tested for this rather than `CONFIG_SCHED_*`).

cyring, 2019-05-10:
Fyi, I have pulled the last linux-ck version and successfully build & run _CoreFreq_
```
$ make
cc  -Wall -pthread -c corefreqd.c \
        -D FEAT_DBG=1 -o corefreqd.o
cc  -Wall -c corefreqm.c -o corefreqm.o
cc  -Wall corefreqd.c corefreqm.c \
        -D FEAT_DBG=1 -o corefreqd -lpthread -lm -lrt
cc  -Wall -c corefreq-cli.c -o corefreq-cli.o
cc  -Wall -c corefreq-ui.c -o corefreq-ui.o
cc  -Wall -c corefreq-cli-rsc.c \
        -o corefreq-cli-rsc.o
cc  -Wall -c corefreq-cli-json.c \
        -o corefreq-cli-json.o
cc  -Wall -c corefreq-cli-extra.c \
        -o corefreq-cli-extra.o
cc  -Wall \
        corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
        corefreq-cli-json.c corefreq-cli-extra.c \
        -o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/5.0.15-1-ck/build M=CoreFreq modules
make[1]: Entering directory '/usr/lib/modules/5.0.15-1-ck/build'
  CC [M]  CoreFreq/corefreqk.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC     CoreFreq/corefreqk.mod.o
  LD [M] CoreFreq/corefreqk.ko
make[1]: Leaving directory '/usr/lib/modules/5.0.15-1-ck/build'
```
[image]

cyring, 2019-05-10:
OK, so only this change is enough to be compatible with your kernel config ?
```C
...
#if defined(CONFIG_SCHED_MUQSS) || defined(CONFIG_SCHED_BMQ)
		SysGate->taskList[cnt].runtime  = tsk_seruntime(thread);
#else
...
```

cyring, 2019-05-11:
Hello,
Is the latest doing the job ?
Feel free to close the issue if ok.
Regards 
Cyril

RomanHargrave, 2019-05-12:
@cyring sorry for the delay. no, unfortunately that's just part of the required changes. the other issue is that of `wake_cpu` not being part of `struct task_struct`.

cyring, 2019-05-12:
But how is defined your [task_struct](https://elixir.bootlin.com/linux/latest/source/include/linux/sched.h#L586) ?
The only condition of `wake_cpu` is `CONFIG_SMP`
Can you explain why this field is not part of the structure?

RomanHargrave, 2019-05-12:
In linux-pf, there are a few other patches on top of BMQ that could relate to this, but `wake_cpu`'s existence is determined by `#ifndef CONFIG_SCHED_BMQ`.

```c
struct task_struct {
#ifdef CONFIG_THREAD_INFO_IN_TASK
	/*
	 * For reasons of header soup (see current_thread_info()), this
	 * must be the first element of task_struct.
	 */
	struct thread_info		thread_info;
#endif
	/* -1 unrunnable, 0 runnable, >0 stopped: */
	volatile long			state;

	/*
	 * This begins the randomizable portion of task_struct. Only
	 * scheduling-critical items should be added above here.
	 */
	randomized_struct_fields_start

	void				*stack;
	atomic_t			usage;
	/* Per task flags (PF_*), defined further below: */
	unsigned int			flags;
	unsigned int			ptrace;

#if defined(CONFIG_SMP) && !defined(CONFIG_SCHED_BMQ)
	struct llist_node		wake_entry;
#endif
#if defined(CONFIG_SMP) || defined(CONFIG_SCHED_BMQ)
	int				on_cpu;
#endif
#ifdef CONFIG_SMP
#ifdef CONFIG_THREAD_INFO_IN_TASK
	/* Current CPU: */
	unsigned int			cpu;
#endif
#ifndef CONFIG_SCHED_BMQ
	unsigned int			wakee_flips;
	unsigned long			wakee_flip_decay_ts;
	struct task_struct		*last_wakee;

	/*
	 * recent_used_cpu is initially set as the last CPU used by a task
	 * that wakes affine another task. Waker/wakee relationships can
	 * push tasks around a CPU where each wakeup moves to the next one.
	 * Tracking a recently used CPU allows a quick search for a recently
	 * used CPU that may be idle.
	 */
	int				recent_used_cpu;
	int				wake_cpu;
#endif /* !CONFIG_SCHED_BMQ */
#endif
	int				on_rq;

	int				prio;
	int				static_prio;
	int				normal_prio;
	unsigned int			rt_priority;

#ifdef CONFIG_SCHED_BMQ
	u64				last_ran;
	s64				time_slice;
	int				boost_prio;
	int				bmq_idx;
	struct list_head		bmq_node;
	/* sched_clock time spent running */
	u64				sched_time;
#else /* !CONFIG_SCHED_BMQ */
	const struct sched_class	*sched_class;
[... 554 lines cut ...]
	void				*security;
#endif

#ifdef CONFIG_GCC_PLUGIN_STACKLEAK
	unsigned long			lowest_stack;
	unsigned long			prev_lowest_stack;
#endif

	/*
	 * New fields for task_struct should be added above here, so that
	 * they are included in the randomized portion of task_struct.
	 */
	randomized_struct_fields_end

	/* CPU-specific state of this task: */
	struct thread_struct		thread;

	/*
	 * WARNING: on x86, 'thread_struct' contains a variable-sized
	 * structure.  It *MUST* be at the end of 'task_struct'.
	 *
	 * Do not put anything below here!
	 */
};
```

RomanHargrave, 2019-05-12:
To clarify, I did modify CoreFreq to use `on_cpu`, but as stated previously I'm unsure of the effect that will have vs `wake_cpu` primarily due to a (apparent) lack of good documentation for the members of `task_struct`.

cyring, 2019-05-12:
How does the `top` command run w/ this patch ? 
If it correctly shows the last cpu the task was on, then it means thar somewhere, between the task structure and `top`, some code is handling the case.

The same, If `on_cpu` is coherent with the results of any scheduling tracer, such as `/proc/sched_debug`

RomanHargrave, 2019-05-13:
I'll have a look at the top source code.

RomanHargrave, 2019-05-13:
OK. Just briefly looking at it, it appears to be that `on_cpu` is usually `0` (although i have seen tasks appear on other CPUs as well).

Here is running `zpaq` on a large file with seven threads. You can see that some tasks are shown on CPU1 but almost all are on CPU0, you can see that the threads are distributed (or are most active) across CPU0-3

[image]

Here is `zpaq` running with a single thread and 100% affinity for CPU7. You can see that CPU7 is busy but that the tasks are still reported on the incorrect CPU.

[image]

Based on this, `on_cpu` is _not_ an accurate value. I'll have a read through BMQ scheduler code and see if I can glean any understanding of how (and where) it tracks CPU migration.

Also note - BMQ is a continuation of (though still different from) PDS, if you ever had workarounds for PDS.

RomanHargrave, 2019-05-13:
After a brief inspection, it appears that BMQ tracks the CPU for a given task in `struct_task.cpu`, unless `CONFIG_THREAD_INFO_IN_TASK` in which case it is `task_thread_info(struct_task)->cpu` (which is `(((struct thread_info *)(task)->stack))->cpu` on every arch except ia64)

cyring, 2019-05-13:
Congratulations.

Do you have the change available in your _CoreFreq_ fork ?

I'm reading above:
```
/*
 * Wrappers for p->thread_info->cpu access. No-op on UP.
 */
```
Does that mean the CPU will be undefined if Kernel is booted with a single Core ?

PDS ?

cyring, 2019-05-13:
Also notice those limits:

* Max number of tasks (stored in shared memory) 

https://github.com/cyring/CoreFreq/blob/124d9e54ab2b5b0385855745d416d93b57346cfa/coretypes.h#L1435

  You can gently increase the order if needed. 

* Max PID
`wake_cpu` field type in SHM is a `short int`  

https://github.com/cyring/CoreFreq/blob/124d9e54ab2b5b0385855745d416d93b57346cfa/coretypes.h#L1416

RomanHargrave, 2019-05-14:
@cyring I may have pushed it to my fork. Though I want to do some cleanup before submitting a PR.

> Does that mean the CPU will be undefined if Kernel is booted with single Core ?

I believe it refers to what happens if `!defined(CONFIG_SMP)`.

cyring, 2019-06-23:
Hello,
Can we backport the adapted code ?

RomanHargrave, 2019-06-24:
@cyring Hey, sorry about the silence. I've been very busy. Yes - we can backport it. I'll make a point of preparing a PR tonight.

Apologies.

closed 2019-07-23 by cyring