cyring/CoreFreq #116: CoreFreq and /proc/cpuinfo output different frequency information

BowenWang, 2019-05-18:
Hi,
I am using Intel i9 9900k in overclocking mode, after I install CoreFreq, I found that the frequency is different it outputs is different from /proc/cpuinfo.
The screenshots on my machine(these three pictures are shot at the same time):
[image]
[image]
[image]

Is there any problem? Which software should I trust?

BowenWang, 2019-05-18:
I also found that the difference only occurs when there is no process running, if I try to run some stress test, then the frequency output will be the same(both are 4.7 GHz)

BowenWang, 2019-05-18:
Screenshots of i7z and CoreFreq at the same time.
[image]
[image]

cyring, 2019-05-18:
Indeed, they are the same but _CoreFreq_ shows the relative frequency which is based on the Processor performance counters sampling. In other terms, the frequency based on the usage.

Fyi, at a gigahertz speed, it's almost impossible to query the frequency in real time with an algorithm that will overflow the CPU. That's why manufacturers are providing the PMU to sample counters on a large scale.

At any time, with an idle processor, you can press the [p] key to display the Processor view in which you will find frequencies from the Minimum up to the Turbo.

Btw, your screenshot shows artifacts beside temperature: you have to disable the NMI watchdog

BowenWang, 2019-05-19:
Ok, but what do you mean by 
> your screenshot shows artifacts beside temperature
?

cyring, 2019-05-19:
The values in gray, at the right side of the Max temperature values. 
Looks like they are sliding out of their place.  
The same in the Averages, this 23% is unexpected.  
See this reference screen for example:  
[image]

BowenWang, 2019-05-19:
Ok, this definitely seems weird, could it because of the NMI watchdog?

cyring, 2019-05-19:
Yes if you don't have disable them (straight from the kernel boot command line)

cyring, 2019-05-28:
Hello, 
If things are OK, and before closing the issue, [here](https://github.com/cyring/CoreFreq/wiki/CPU-support) is maintained a Hall of Fame: feel free to send me a Gist link with various CoreFreq Cli screenshots which I'll add under your Processor name.

cyring, 2019-06-25:
Your 9900K is Thermal Velocity Boost capable, are you OK to test some code which will query the state of this technology ?

cyring, 2019-08-27:
Closing the issue.
However let me know if your want to contribute to Thermal Velocity Boost at issue #121

closed 2019-08-27 by cyring