cyring/CoreFreq #117: build issue on CentOS7 [cpuidle_poll_state_init] [enter_s2idle]

bobjiang82, 2019-05-22:
hi, I run into the following issue when doing "make". I went through the issue list and don't find a similar one. Could you help to give me a guide? Thanks in advance.

make -j1 -C /lib/modules/3.10.0-514.el7.x86_64/build M=/home/jiangbo/sspdat/CoreFreq modules
make[1]: Entering directory `/usr/src/kernels/3.10.0-514.el7.x86_64'
  CC [M]  /home/jiangbo/sspdat/CoreFreq/corefreqk.o
/home/jiangbo/sspdat/CoreFreq/corefreqk.c: In function ‘CoreFreqK_IdleDriver_Init’:
/home/jiangbo/sspdat/CoreFreq/corefreqk.c:7971:3: error: implicit declaration of function ‘cpuidle_poll_state_init’ [-Werror=implicit-function-declaration]
   cpuidle_poll_state_init(&CoreFreqK.IdleDriver);
   ^
/home/jiangbo/sspdat/CoreFreq/corefreqk.c:8002:4: error: ‘struct cpuidle_state’ has no member named ‘enter_s2idle’
   ].enter_s2idle = CoreFreqK_S2IdleHandler;
    ^
cc1: some warnings being treated as errors
make[2]: *** [/home/jiangbo/sspdat/CoreFreq/corefreqk.o] Error 1
make[1]: *** [_module_/home/jiangbo/sspdat/CoreFreq] Error 2
make[1]: Leaving directory `/usr/src/kernels/3.10.0-514.el7.x86_64'
make: *** [all] Error 2

Some additional info on my environment:
$ uname -a
Linux cattle-node4 3.10.0-514.el7.x86_64 #1 SMP Tue Nov 22 16:42:41 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
I have run the following cmds successfully
$ sudo yum install kernel-devel
$ sudo yum groups install "Development Tools"
gcc version 4.8.5 20150623 (Red Hat 4.8.5-36) (GCC)

bobjiang82, 2019-05-22:
It looks the cpuidle.h is updated on Aug. 2017, which has Rename ->enter_freeze to ->enter_s2idle.
https://github.com/torvalds/linux/commit/28ba086ed30fb3fb714598aa029b894c3754fa7b#diff-33a183593d2c4cda9b451b3e3e32cc53
Then what's the minimal kernel version for CoreFreq? :)

cyring, 2019-05-23:
My fault: I have not make non regression tests with 3.1 kernel
Will fix asap: find alternative API or surround those functions with ifdef version macros

cyring, 2019-05-23:
Fix released.  
Build tested with Kernel  3.10.0

bobjiang82, 2019-05-24:
It works perfectly on my CentOS 7 server now. Thanks for the quick response. :)

closed 2019-05-24 by bobjiang82