cyring/CoreFreq #120: compilation failed.

kprasadvnsi, 2019-05-28:
kp@vivo:~/CoreFreq$ make
cc  -Wall corefreqd.c corefreqm.c \
	-D FEAT_DBG=1 -o corefreqd -lpthread -lm -lrt
cc  -Wall \
	corefreq-cli.c corefreq-ui.c corefreq-cli-rsc.c \
	corefreq-cli-json.c corefreq-cli-extra.c \
	-o corefreq-cli -lm -lrt
make -j1 -C /lib/modules/5.1.3/build M=/home/kp/CoreFreq modules
make[1]: *** /lib/modules/5.1.3/build: No such file or directory.  Stop.
Makefile:68: recipe for target 'all' failed
make: *** [all] Error 2

cyring, 2019-05-28:
Hello,

Do you confirm the kernel headers prerequisite for kernel module compilation is installed ?
* [Ubuntu, Debian](https://github.com/cyring/CoreFreq#debian-ubuntu)
* [Red Hat, CentOS](https://github.com/cyring/CoreFreq#red-hat-centos)
* [ArchLinux](https://aur.archlinux.org/packages/corefreq-git/)

kprasadvnsi, 2019-05-28:
Yes, they are installed. using the following command:
sudo apt-get install libpthread-stubs0-dev

kprasadvnsi, 2019-05-28:
ok, I have updated the kernel to 5.1.5 and restart. don't know which one solves the problem but now it compiles and runs.

kprasadvnsi, 2019-05-28:
thanks, @cyring for the prompt reply. I really appreciate the help I got here.

closed 2019-05-28 by kprasadvnsi

cyring, 2019-05-28:
Your welcome! 
Hope you will appreciate _CoreFreq_ and it fulfills your needs.
[Here](https://github.com/cyring/CoreFreq/wiki/CPU-support) is maintained a hall of fame, feel free to send me a Gist link with your Processor screenshots.