cyring/CoreFreq #121: Thermal Velocity Boost

cyring, 2019-05-29:
* [Intel® Turbo Boost Max Technology 3.0](https://www.intel.com/content/www/us/en/architecture-and-technology/turbo-boost/turbo-boost-max-technology.html)
`CPUID(0x6) > EAX[15:14]`

* [Thermal Velocity Boost](https://en.wikichip.org/wiki/intel/thermal_velocity_boost)
Maximus X/XI & Strix Motherboards (Z370/Z390) > [MCE explanations and others](https://rog.asus.com/forum/showthread.php?106375-MCE-explanations-and-others)
> MSR 0x150 ... set bit 63 to 1 and [39:32] to 18h
> Bit 3 shows you if TVB is enabled or disabled (0=disabled).

_Originally posted by @cyring in https://github.com/cyring/CoreFreq/issues/60#issuecomment-449668251_

h1z1, 2020-01-03:
Was curious what this is since I'd never heard of it before.  The wiki link doesn't really have any details about it either. tl;dr - It's like xfr?

Only chips I could find with it are
 - Core i9-10900K
 - Core i9-9980HK 
 -  vPro Core i9-9880H 
 - Core i9-8950HK 
 - Xeon E-2286M 
 - Xeon E-2186M

cyring, 2020-01-03:
Hello, 
Do you have one of these processors to read the `0x150` MSR bits with ?

h1z1, 2020-01-06:
Didn't realize you replied to this, sorry for the late update.   I don't have any of those yet but figured it might be easier to list some in case someone else reading this wonders what on earth that is :)

cyring, 2020-01-06:
> Didn't realize you replied to this, sorry for the late update. I don't have any of those yet but figured it might be easier to list some in case someone else reading this wonders what on earth that is :)

Thanks for your help

May be the easiest should be that I go to any Walmart with my _CoreFreq_ live CD to dump those registers ;-)

h1z1, 2020-01-07:
See that isn't such a bad idea.  Was thinking about making a livecd or pxeboot yesterday.  Kind of a pita having to boot into full linux every time when overclocking/hardware tests.

cyring, 2020-01-07:
> See that isn't such a bad idea. Was thinking about making a livecd or pxeboot yesterday. Kind of a pita having to boot into full linux every time when overclocking/hardware tests.

This may help: 
[How to build the CoreFreq Live CD](http://corefreq.blogspot.com/2018/07/how-to-build-corefreq-live-cd.html)

cyring, 2020-03-18:
Unfortunately no hardware to test with.
Subject is transferred to the roadmap #169

closed 2020-03-18 by cyring