cyring/CoreFreq #124: Memory in GB

Tuquar, 2019-06-06:
Hi,

I stumbled upon this repo when looking for issues reading my Ryzen 2700X CPU temp. The app looks really great so I decided to give it a go. It does everything I need so far, so thankyou!

One thing I didn't like was the memory count in the bottom right, I thought KB is too small and I wanted to see what GB looks like. I'm a total C amateur though and only got as far as changing the label from "KB" to "GB" (as shown in the screenshot) :smile:

Could you point me to the right place for me to modify the actual numbers? I've had a hunt around and am struggling to trace them back to a function where I could do this.

[image]

Cheers

cyring, 2019-06-07:
Hello

# Code
## Function `PrintTaskMemory`
* It processes both tasks and memory usage.
https://github.com/cyring/CoreFreq/blob/1b2d21ffbd52703cca6f0e814a01bc69edae6de2/corefreq-cli.c#L6651
You will have to keep the same characters length when scaling numbers to GB

# Resources
Below are the labels printed in the footer.
Please keep the same length (or if a change is necessary, align the associated attributes `RSC_LAYOUT_FOOTER_SYSTEM_ATTR` in `corefreq-cli-rsc.h`)
## English
https://github.com/cyring/CoreFreq/blob/1b2d21ffbd52703cca6f0e814a01bc69edae6de2/corefreq-cli-rsc-en.h#L151
## French
https://github.com/cyring/CoreFreq/blob/1b2d21ffbd52703cca6f0e814a01bc69edae6de2/corefreq-cli-rsc-fr.h#L151

# Build
Fully rebuild the project before trying
`make clean all`

Cyril

cyring, 2019-06-08:
Feel free to push a PR to improve the units conversion; but keep in mind that `PrintTaskMemory()` is on the _hot path_ of the Client drawing call flow. 

The goal I'm pursuing is to reduce the CPU overhead as much as possible in favor of the Processor state counters. Thus, I'm trying to optimize any instructions that belongs to the UI loop where any instruction statement matters.

Tuquar, 2019-06-09:
Hey,

Thanks for all the info you posted, I've just got around to working on this. Had many dual boot issues this weekend.

So I managed to convert the values to doubles and pass them through, it does look much better.
I was having trouble getting Math in my scope, I was hoping to Math.round the values to 3dp. For now I've resorted to dividing the final number by 10000 and 1000, although this makes the value of the "free" memory 4dp.

I'm sure the PR will need some work so let me know what you think!

Tuquar, 2019-06-09:
I also wanted to query you on the actual value that comes out. When I look at the system monitor, I find that the value outputted by CoreFreq is a little under half that of the built in System Monitor, any ideas?    

System Monitors memory in the lower left, CoreFreq memory upper right:
[image]

cyring, 2019-06-09:
_CoreFreq_ collects the RAM usage from the Kernel.
It has been calibrated against these commands: `free` , `top`
```
$ corefreq-cli -k;echo;free -k;echo;top -E k -n 1
Linux:                                                                          
|- Release                                                  [5.1.7-arch1-1-ARCH]
|- Version                         [#1 SMP PREEMPT Sat Jun 8 12:37:33 CEST 2019]
|- Machine                                                              [x86_64]
Memory:                                                                         
|- Total RAM                                                         12285400 KB
|- Shared RAM                                                          151464 KB
|- Free RAM                                                           9607580 KB
|- Buffer RAM                                                          254680 KB
|- Total High                                                               0 KB
|- Free High                                                                0 KB
CPU-Freq driver                                               [  corefreqk-perf]
CPU-Idle driver                                               [  corefreqk-idle]
|- Idle Limit                                                          <     5> 
   |- State        POLL      C1     C1E      C3      C6  n/a  n/a  n/a  n/a  n/a
   |-           CPUIDLE  NHM-C1 NHM-C1E  NHM-C3  NHM-C6                         
   |- Power          -1       0       0       0       0                         
   |- Latency         0       3      10      20     200                         
   |- Residency       0       6      20      80     800                         

              total        used        free      shared  buff/cache   available
Mem:       12285400      965968     9607304      151464     1712128    10845920
Swap:             0           0           0

Tasks: 241 total,   1 running, 240 sleeping,   0 stopped,   0 zombie
%Cpu0  :   0.0/0.0     0[                                                     ]
%Cpu1  :   0.0/0.0     0[                                                     ]
%Cpu2  :   0.0/0.0     0[                                                     ]
%Cpu3  :   0.0/0.0     0[                                                     ]
%Cpu4  :   0.0/6.2     6[|||                                                  ]
%Cpu5  :   0.0/0.0     0[                                                     ]
%Cpu6  :   0.0/0.0     0[                                                     ]
%Cpu7  :   0.0/0.0     0[                                                     ]
%Cpu8  :   0.0/0.0     0[                                                     ]
%Cpu9  :   0.0/0.0     0[                                                     ]
%Cpu10 :   0.0/0.0     0[                                                     ]
%Cpu11 :   0.0/0.0     0[                                                     ]
KiB Mem : 11.7/12285400 [                                                     ]
KiB Swap:  0.0/0        [                                                     ]
```
Can you verify if "System Monitor" _equals_ to the `free` command (considering the same unit) ?

Tuquar, 2019-06-09:
Seems to be ~300mb more in CoreFreq than shown in the System Monitor now. 

[image]

```
Linux:                                                                          
|- Release                                                   [4.15.0-20-generic]
|- Version                         [#21-Ubuntu SMP Tue Apr 24 06:16:15 UTC 2018]
|- Machine                                                              [x86_64]
Memory:                                                                         
|- Total RAM                                                         32923132 KB
|- Shared RAM                                                          196968 KB
|- Free RAM                                                          27945216 KB
|- Buffer RAM                                                          146816 KB
|- Total High                                                               0 KB
|- Free High                                                                0 KB
CPU-Freq driver                                               [    acpi-cpufreq]
Governor                                                      [        ondemand]
CPU-Idle driver                                               [       acpi_idle]
|- Idle Limit                                                          [     3] 
   |- State        POLL      C1      C2  n/a  n/a  n/a  n/a  n/a  n/a  n/a      
   |-           CPUIDLE ACPI FF ACPI IO                                         
   |- Power          -1       0       0                                         
   |- Latency         0       1     400                                         
   |- Residency       0       2     800                                         

              total        used        free      shared  buff/cache   available
Mem:       32923132     1931692    27945180      196968     3046260    30352756
Swap:       2097148           0     2097148

top: unknown option 'E'
Usage:
  top -hv | -bcHiOSs -d secs -n max -u|U user -p pid(s) -o field -w [cols]

```

I'm getting an error at the end, but at least its spat out the RAM usage so I'm not too bothered.

Tuquar, 2019-06-10:
I'm still struggling to get the decimal point to the correct place. It seems that the free KB value is a factor of 10 bigger than it should be. If I do this:
```
void PrintTaskMemory(Layer *layer, CUINT row,
			int taskCount,
			unsigned long freeRAM,
			unsigned long totalRAM)
{
	double gigabyte_freeRAM = freeRAM/(1024^2);
	double gigabyte_totalRAM = totalRAM/(1024^2);
	
	gigabyte_freeRAM = gigabyte_totalRAM / 10000; // the extra 0 to make the value correct
	gigabyte_totalRAM = gigabyte_totalRAM / 1000;

	sprintf(buffer, "%6u" "%9g" "%-9g", taskCount, (gigabyte_freeRAM), (gigabyte_totalRAM));
```
I get roughly correct values (apart from free ram being 4dp rather than 3dp):
```
Mem [ 3.2088/32.088 GB]
```
If I change the code back to the original, both values have 8 digits. For example if I have 2gb used of 32gb, I get the following output, which I'll put above/below each other:
```
20000000
32923132
// Should be:
2000000
32923132
```
You can actually see this in my original screenshot, both values have the same number of digits!

I wanted to clean up my pull request as it didn't look like great code, but I am struggling with this particular issue. I had problems when trying to use the precision notation of .3. I might not have to do this though if the above free ram value can be corrected.
```
sprintf(buffer, "%6u" "%9g.3" "%g.3", taskCount, (freeRAM), (totalRAM));
```

cyring, 2019-06-11:
Perhaps you should switch to integer operations in the Q/R form using 64 bits variables & Euclidean division. 
The format will be some   `"%9u.%3u", q, r`

Tuquar, 2019-06-11:
I've pushed a change to make it 3dp for free and total memory now. 

I've done some googling about the difference between the system monitor and CoreFreq.
You can use commands like `sudo cat /proc/meminfo` and `free -m`. These show the same numbers as CoreFreq. So I'll be using this as my point of reference and not the System Monitor :smile:

Tuquar, 2019-06-19:
Right so this evening I've gone back to work on this some more. I've now realised I have been a total idiot.

The values which are output is `[free ram / total ram]`. I thought it was `[used ram / total ram]` - hence my confusion. 
I installed "stress" to inflate my ram usage and I was very confused when what I thought was `used ram` actually decreased. Obviously this was just my `free ram` which makes much more sense!

cyring, 2019-06-19:
If you are OK, I will provide new code for your need b/c it has major impact : not only the footer, which has to be dynamically resized based on the unit; but it also impacts other windows like "Kernel data" and "Memory Controller"

I was also asked to offer Fahrenheit conversion.

New menu items in Settings have to be programmed for units selection

So I will bundle all of these in one shot

Tuquar, 2019-06-19:
Yeah no problem at all. It was interesting to learn a bit of C code :)

I look forward to the changes!

cyring, 2019-06-29:
Last commit: use argument `-2` for Giga or `-1` for Mega or `-0` for Kilo (default), plut argument `-t` for UI

cyring, 2019-10-22:
Request implemented, closing the issue. Let me know for any other questions.
Regards

closed 2019-10-22 by cyring

cyring, 2020-07-06:
Hello,
Among the last changes in issue #195 , an I/O ASM access to the FCH have been added: can you please check if the last _CoreFreq_ version is starting fine ?
Thank you

mattlennon3, 2020-07-06:
Hey, 

Nice to see the project is still going strong!
Looks good to me:
[image]
Running latest commit: https://github.com/cyring/CoreFreq/commit/36a8fd6561b40afc9a52de70f1dcfcfae64d24ac

cyring, 2020-07-06:
Thank you for your confirmation.

At first, I saw a bug in `Mem [ 21/31 GB]` and asked myself what's wrong with RAM !
... then I remember I made this feature to your request.

Regards

mattlennon3, 2020-07-06:
No problem cyring :smiley: 

That is funny, I appreciate you adding the feature, it does make it easier for me to read the stats!

-Matt