cyring/CoreFreq #125: corefreq-cli -d receives SIGSEGV and SIGBUS

JohnAZoidberg, 2019-06-06:
When I run `corefreq-cli -d` it sometimes crashes before showing anything and sometimes it gets to show the output for a second and then crashes.

```
$ ./corefreq-cli -d
fish: “./corefreq-cli -d” terminated by signal SIGSEGV (Address boundary error)
```

Fish is the name of my shell.

```
$  gdb --args corefreq-cli -d
(gdb) run
Starting program: /home/zoid/media/clone/reference/CoreFreq/corefreq-cli -d
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libthread_db.so.1".

Program received signal SIGBUS, Bus error.
0x0000000000407628 in StateToSymbol ()
(gdb) where
#0  0x0000000000407628 in StateToSymbol ()
#1  0x0000000000407d5e in Draw_Card_Task ()
#2  0x0000000000402c39 in Draw_Dashboard ()
#3  0x000000000042686d in Top ()
#4  0x0000000000402a3f in main ()
```

Yes, I booted with `nmi_watchdog=0` on the kernel commandline.
I built from the latest master 1b2d21ffbd52703cca6f0e814a01bc69edae6de2.

cyring, 2019-06-07:
Thanks for your issues and PR
I will take time this WE to process them. 
Come back to you ASAP

JohnAZoidberg, 2019-06-07:
Is there any other information I can provide you with?

cyring, 2019-06-07:
Yes:
1. in [screenshots](https://gist.github.com/JohnAZoidberg/7af1d420062e6194d9e481ad66c58cd0#gistcomment-2935363) the Client background color. Is it a terminal choice ?
2. The one core Turbo Boost is supposed to go to 3.4 GHz. Are you stressing with my internal tool ?
3. According to the mwait sub-states, your Processor is down to C10 capable : I'm improving the idle sub-driver. Can you test this module option and check the resulting Core and Package C-States are well entered ?

I appreciate a lot your help. Thank you.

JohnAZoidberg, 2019-06-07:
1. Yes, it's this is the solarized light theme of my terminal - it's not white it's #fdf6e3
2. I'm stressing it with [stress](https://people.seas.harvard.edu/~apw/stress/) `stress -cpu 1`.
   How do I use the internal stress tester?
3. I should use `Register_CPU_Idle=1`? But for that I also need to blacklist Intel's idle driver?
   I did that:
   ```
   $ modprobe -c | grep corefreq
   options corefreqk Register_CPU_Idle=1
   $ journalctl -k | grep intel_cstate
   Jun 07 23:57:18 think-nix kernel: Kernel command line: initrd=\efi\nixos\initrd.efi loglevel=4 nmi_watchdog=0 modprobe.blacklist=intel_cstate idle=halt intel_idle.max_cstate=0 mitigations=off transparent_hugepage=never
   ```
   But `corefreq-cli` still only show up until C7
   [image]
    Is that what you meant?

I'm running Linux 5.1.5.

JohnAZoidberg, 2019-06-07:
I have to use `-D` to build the idle driver into the kernel module, right?
Now it shows:

```
$ cat /sys/devices/system/cpu/cpuidle/current_driver
corefreqk-idle
```

but `corefreq-cli` is still the same.

cyring, 2019-06-08:
Thanks for your returns.

3. `corefreqk-idle` : the sub-driver has been well registered. 

Press [k] (kernel data) or [s] (settings) to see confirmation.

C7 is so far the deepest _Core C-States_ implemented in the monitoring view.
However, press [g] to monitor the _Package C-States_ down to C10

You can also try to register the CPU frequency sub-driver. It probably needs to blacklist the other ACPI, pcc, p-state drivers to let CoreFreq takes the room.
When registered, you are in power to set the min, max, target HWP frequencies, press [p] ; but also the energy policy in the Power window.

Remark: when changing HWP policies, frequencies, watch the resulting Vcore and your Turbo highest performance frequency ratio. You may find for example the optimum processor settings for H24 use.

2. About the integrated stressing function : just press [F3] and choose among the algorithms.
A red dot appears beside the LCD when a stressing function is processing. 
Press [F10] to stop stressing at any time

1. The UI has been designed for a black background, but I had centralized the theme resources in header files to let users customize them.

If you are in need of a full "color" theme, I can refactor the code to instantiate a new set and provide a chooser option ?

JohnAZoidberg, 2019-06-08:
Now with the idle driver it crashes on the regular command:
```
$ gdb ./corefreq-cli
Reading symbols from ./corefreq-cli...done.
(gdb) run
Starting program: /home/zoid/media/clone/reference/CoreFreq/corefreq-cli
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libthread_db.so.1".
*** buffer overflow detected ***: /home/zoid/media/clone/reference/CoreFreq/corefreq-cli terminated

Program received signal SIGABRT, Aborted.
0x00007ffff7cafbe0 in raise () from /nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libc.so.6
(gdb) where
#0  0x00007ffff7cafbe0 in raise () from /nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libc.so.6
#1  0x00007ffff7cb0dc1 in abort () from /nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libc.so.6
#2  0x00007ffff7cf12ac in __libc_message () from /nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libc.so.6
#3  0x00007ffff7d7df68 in __fortify_fail_abort () from /nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libc.so.6
#4  0x00007ffff7d7df81 in __fortify_fail () from /nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libc.so.6
#5  0x00007ffff7d7c070 in __chk_fail () from /nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libc.so.6
#6  0x00007ffff7d7b5d9 in _IO_str_chk_overflow () from /nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libc.so.6
#7  0x00007ffff7cf51cb in _IO_default_xsputn () from /nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libc.so.6
#8  0x00007ffff7cc7d01 in vfprintf () from /nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libc.so.6
#9  0x00007ffff7d7b66c in __vsprintf_chk () from /nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libc.so.6
#10 0x00007ffff7d7b5bd in __sprintf_chk () from /nix/store/681354n3k44r8z90m35hm8945vsp95h1-glibc-2.27/lib/libc.so.6
#11 0x000000000042476b in Layout_Ruller_Load ()
#12 0x0000000000425589 in Layout_Header_DualView_Footer ()
#13 0x00000000004269a5 in Top ()
#14 0x0000000000402a4f in main ()
(gdb)
```

cyring, 2019-06-08:
* Running in a plain XTerm with fish (installed from this [package](https://www.archlinux.org/packages/community/x86_64/fish)): no crash in different load cases.
```
$ ./corefreq-cli -d
```
and
```
$ make clean all CC='cc -g' ; gdb corefreq-cli
run
```
[image]
[image]
[image]
[image]

* Despite you have disable the NMI from the Kernel command line, _CoreFreq_ may interact with other softwares which make use of the fixed counter registers: here's the [list](https://github.com/cyring/CoreFreq/wiki/Software-incompatibility).
When buffer overflows occur in the Client, it is a strong sign that the PMU counters are rewritten, to the detriment of the `corefreqk.ko` readings.

* I suggest to start the driver with no option until we solve the issue.

* [Here](https://git.archlinux.org/svntogit/packages.git/tree/trunk/config?h=packages/linux) is the Kernel config I'm running _CoreFreq_ oftenly with.

* My current setup:
ArchLinux Kernel 5.1.7
GNU Compiler (gcc) 8.3.0 , (gcc-libs) 8.3.0-1
GNU assembler (as) 2.32
GNU linker (ld) 2.32
GNU Make 4.2.1
GNU C Library (glibc) 2.29-1 , POSIX threads
GNU Debugger (gdb) 8.3-1
valgrind 3.14.0-1
binutils 2.32-1
xterm 346-1
```
$ gcc -E -v
Using built-in specs.
COLLECT_GCC=gcc
Target: x86_64-pc-linux-gnu
Configured with: /build/gcc/src/gcc/configure --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://bugs.archlinux.org/ --enable-languages=c,c++,ada,fortran,go,lto,objc,obj-c++ --enable-shared --enable-threads=posix --enable-libmpx --with-system-zlib --with-isl --enable-__cxa_atexit --disable-libunwind-exceptions --enable-clocale=gnu --disable-libstdcxx-pch --disable-libssp --enable-gnu-unique-object --enable-linker-build-id --enable-lto --enable-plugin --enable-install-libiberty --with-linker-hash-style=gnu --enable-gnu-indirect-function --enable-multilib --disable-werror --enable-checking=release --enable-default-pie --enable-default-ssp --enable-cet=auto
Thread model: posix
gcc version 8.3.0 (GCC)

$ grep FLAG /etc/makepkg.conf
# ARCHITECTURE, COMPILE FLAGS
CPPFLAGS="-D_FORTIFY_SOURCE=2"
CFLAGS="-march=westmere -mtune=westmere -O2 -pipe -fno-plt"
CXXFLAGS="-march=westmere -mtune=westmere -O2 -pipe -fno-plt"
LDFLAGS="-Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now"
MAKEFLAGS="-j16"
DEBUG_CFLAGS="-g -fvar-tracking-assignments"
DEBUG_CXXFLAGS="-g -fvar-tracking-assignments"
```

cyring, 2019-06-08:
Latest commit may fix the Idle States enumeration on your Processor architecture:
```
insmod corefreqk.ko Register_CPU_Idle=1
```
* If things OK, please show the " Performance Monitoring " and " Kernel " windows as below.
[image]

cyring, 2019-08-27:
Closing the issue if latest version is stable ?

closed 2019-08-27 by cyring